zigpy/zigpy #327: Can't reenable XBee switch or sensor if it was disabled before - sqlite3.OperationalError: database is locked

MaximumSU, 2020-03-21:
I have some XBee modules connected to HA. Each module have some switches and sensors, what i not need. I dasabled it in configs - no problems. But when i want to reenable it back and at that time all my ZHA XBee switches go to unavailable and all devices disappeared from HA. If HA restarted after that - all back and working. The log of problem:
```
2020-03-21 18:05:57 DEBUG (MainThread) [homeassistant.components.zha.core.gateway] Shutting down ZHA ControllerApplication
2020-03-21 18:05:57 DEBUG (MainThread) [zigpy_xbee.uart] Connection made
2020-03-21 18:06:02 ERROR (MainThread) [homeassistant.config_entries] Error setting up entry /dev/serial0 for zha
Traceback (most recent call last):
  File "/srv/homeassistant/lib/python3.7/site-packages/homeassistant/config_entries.py", line 215, in async_setup
    hass, self
  File "/srv/homeassistant/lib/python3.7/site-packages/homeassistant/components/zha/__init__.py", line 103, in async_setup_entry
    await zha_gateway.async_initialize()
  File "/srv/homeassistant/lib/python3.7/site-packages/homeassistant/components/zha/core/gateway.py", line 133, in async_initialize
    self.application_controller = radio_details[CONTROLLER](radio, database)
  File "/srv/homeassistant/lib/python3.7/site-packages/zigpy_xbee/zigbee/application.py", line 32, in __init__
    super().__init__(database_file=database_file)
  File "/srv/homeassistant/lib/python3.7/site-packages/zigpy/application.py", line 49, in __init__
    self._dblistener = zigpy.appdb.PersistingListener(database_file, self)
  File "/srv/homeassistant/lib/python3.7/site-packages/zigpy/appdb.py", line 38, in __init__
    self._create_table_devices()
  File "/srv/homeassistant/lib/python3.7/site-packages/zigpy/appdb.py", line 122, in _create_table_devices
    self._create_table("devices", "(ieee ieee, nwk, status)")
  File "/srv/homeassistant/lib/python3.7/site-packages/zigpy/appdb.py", line 113, in _create_table
    self.execute("PRAGMA user_version = %s" % (DB_VERSION,))
  File "/srv/homeassistant/lib/python3.7/site-packages/zigpy/appdb.py", line 51, in execute
    return self._cursor.execute(*args, **kwargs)
sqlite3.OperationalError: database is locked
2020-03-21 18:06:02 DEBUG (MainThread) [zigpy.ota] Initialize OTA providers
2020-03-21 18:06:02 DEBUG (MainThread) [zigpy.ota.provider] OTA image directory '/home/homeassistant/.homeassistant/zigpy_ota/' does not exist
```

ClayMav, 2021-06-30:
I also have a `database is locked` error.
Have not been able to get it to "unlock" no matter what.
Looking for help finding a cause for this and how to get this to go away.

```
Logger: homeassistant.config_entries
Source: components/zha/core/gateway.py:157
First occurred: 10:48:16 PM (1 occurrences)
Last logged: 10:48:16 PM

Error setting up entry HubZ Smart Home Controller - HubZ ZigBee Com Port, s/n: 315004C2 - Silicon Labs for zha
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 293, in async_setup
    result = await component.async_setup_entry(hass, self)  # type: ignore
  File "/usr/src/homeassistant/homeassistant/components/zha/__init__.py", line 102, in async_setup_entry
    await zha_gateway.async_initialize()
  File "/usr/src/homeassistant/homeassistant/components/zha/core/gateway.py", line 157, in async_initialize
    self.application_controller = await app_controller_cls.new(
  File "/usr/local/lib/python3.8/site-packages/zigpy/application.py", line 64, in new
    await app._load_db()
  File "/usr/local/lib/python3.8/site-packages/zigpy/application.py", line 53, in _load_db
    self._dblistener = await zigpy.appdb.PersistingListener.new(database_file, self)
  File "/usr/local/lib/python3.8/site-packages/zigpy/appdb.py", line 76, in new
    await listener.initialize_tables()
  File "/usr/local/lib/python3.8/site-packages/zigpy/appdb.py", line 55, in initialize_tables
    await self._create_table_devices()
  File "/usr/local/lib/python3.8/site-packages/zigpy/appdb.py", line 244, in _create_table_devices
    await self._create_table("devices", "(ieee ieee, nwk, status)")
  File "/usr/local/lib/python3.8/site-packages/zigpy/appdb.py", line 231, in _create_table
    await self.execute("CREATE TABLE IF NOT EXISTS %s %s" % (table_name, spec))
  File "/usr/local/lib/python3.8/site-packages/aiosqlite/core.py", line 184, in execute
    cursor = await self._execute(self._conn.execute, sql, parameters)
  File "/usr/local/lib/python3.8/site-packages/aiosqlite/core.py", line 129, in _execute
    return await future
  File "/usr/local/lib/python3.8/site-packages/aiosqlite/core.py", line 102, in run
    result = function()
sqlite3.OperationalError: database is locked
```

puddly, 2021-06-30:
The database is opened by zigpy exactly once so unless you somehow have two HA instances using the same `zigbee.db` file (or are editing it with an external program?), I'm having a hard time seeing how this can happen.

Have you tried rebooting? What does `sqlite3 /path/to/zigbee.db 'pragma integrity_check'` print? Make sure to run it with the correct path, by default it'll print `ok` even if the file doesn't exist.

closed 2022-01-03 by github-actions[bot]