zigpy/zigpy #328: Add support to Profalux (Covers)

Tobby-NZ, 2020-03-22:
Hello

In HA, Profalux covers appear as a switch and not as a cover.
I use Pi-Zigate, documentation and compatibility are describe here (but in french): 
 - Zigate : https://zigate.fr/
 -  Motor : https://zigate.fr/produits-profalux-compatibles-zigate/compatible/moteurvoletprofalux/
 - Remote : https://zigate.fr/produits-profalux-compatibles-zigate/compatible/telecommandesimpleprofalux/

How can I change type of this entity or anything else to manage level of covers ?
I don't know if this issue must be post here of in https://github.com/home-assistant/core/issues or in https://github.com/doudz/zigpy-zigate/issues

Some informations are missing :
- Basic (Endpoint id: 1, Id: 0x0000, Type: in)
  - Attributes : 
    - manufacturer (id: 0x0004) : None
    - model (id: 0x0005) : None
    - stack_version (id: 0x0002) : None
    - zcl_version (id: 0x0000) : 1
- ManufacturerSpecificCluster (Endpoint id: 1, Id: 0xfc20, Type: in)
  - No attribute
  - No commands
- ManufacturerSpecificCluster (Endpoint id: 1, Id: 0xfc21, Type: in)
  - No attribute
  - No commands
- ManufacturerSpecificCluster (Endpoint id: 1, Id: 0xfc20, Type: out)
  - No attribute
  - No commands

But LevelControl and other commands to manage cover are presents : 
- LevelControl (Endpoint id: 1, Id: 0x0008, Type: in)
  - Attributes : 
     - current_level (id: 0x0000)
     - default_move_rate (id: 0x0014)
     - off_transition_time (id: 0x0013)
     - on_level (id: 0x0011)
     - on_off_transition_time (id: 0x0010)
     - on_transition_time (id: 0x0012)
     - remaining_time (id: 0x0001)
  - Commands :
     - move (id: 0x0001)
     - move_to_level (id: 0x0000)
     - move_to_level_with_on_off (id: 0x0004)
     - move_with_on_off (id: 0x0005)
     - step (id: 0x0002)
     - step_with_on_off (id: 0x0006)
     - stop (id: 0x0003)
     - stop (id: 0x0007)

Hedda, 2020-03-30:
@Tobby-NZ For your information; these types of "quirk" issues can often (but not always) be resolved by creating a ZHA Device Handler, see https://github.com/dmulcahey/zha-device-handlers

pipiche38, 2020-03-30:
You can find some information there: https://github.com/pipiche38/Domoticz-Zigate-Wiki/blob/master/en-eng/Profalux-corner.md

Tobby-NZ, 2020-05-03:
Sorry for late, many thanks for your responses !

I will open a issue to zha-device-handlers. I tried to add my own quirck but the device is always see as Switch and not Cover but properties are ok. I think I forgot an item in quirck.

closed 2020-05-03 by Tobby-NZ