zigpy/zigpy #341: [REQUEST] ZGP (Zigbee Green Power) specification support

Hedda, 2020-03-30:
Please consider adding support for Zigbee Green Power (ZGP) specification, a.k.a. "Zigbee GreenPower"

- https://zigbeealliance.org/solution/green-power/

Summery; ZGP, a.k.a. "Zigbee GreenPower",  is a feature of Zigbee that allows for energy-harvesting technology to be used directly with the Zigbee stack. Using Zigbee Green Power, certain home automation devices may be powered by energy harvesting solutions, which when enabled introduces the potential of battery-less devices operating in a Zigbee network.  

- https://training.ti.com/zigbee-green-power-enable-battery-less-devices

Please see zigbee-herdsman PR 92 for a proof-of-concept implementation (to be used by Zigbee2mqtt):

- https://github.com/Koenkk/zigbee-herdsman/pull/92

PS: Zigbee Green Power (ZGP) is also included in the Zigbee 3.0 specification (Z3)

Hedda, 2020-04-15:
FYI, Green Power support has now been merged into zigbee-herdsman master via https://github.com/Koenkk/zigbee-herdsman/pull/92

dieneuser, 2020-09-06:
Is there any news here? I am also eagerly waiting for ZGP support in ZHA. Currently I use the Deconz/Phoscon integration only for my ZGP wall switches [module](https://www.enocean.com/en/products/enocean_modules_24ghz/ptm-216z/) from [switch/remote](https://zigbee.blakadder.com/Philips_9290011152X.html). I migrated all other Zigbee devices to ZHA and I am very happy with them.

Hedda, 2020-09-07:
@dieneuser I believe that perhaps this should maybe be separate requests for all the radio libraries and not common for all zigpy? At least @Adminiuga closed #360 as it request for zigpy to Zigbee 3.0 support because it is up to each radio library to support it.

As such I assume it will probably be the same for ZGP, or?

If so then you should first be asking the hardware manufacturer is the Zigbee adapter and its firmware that you are using support the ZGP (Zigbee Green Power) specification or not. If not then you would have to buy the other hardware adapter anyway.

PS: You mention Deconz and Phoscon, can I assume that you then have a ConBee or Raspbee 1 or 2 model? If so then you should be asking the dresden elektronik company if the firmware for the adapter you have support ZGP or not.

PPS: Is you device listed on the https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Supported-Devices list?

Hedda, 2020-09-07:
> Might it otherwise be that the specific devices that @dieneuser mention only require new "quirk" ZHA Device Handlers each?
> 
> https://github.com/zigpy/zha-device-handlers/blob/dev/README.md
> 
> https://github.com/zigpy/zha-device-handlers/issues

@dieneuser FYI, I created placeholders on your behalf in https://github.com/zigpy/zha-device-handlers/issues/477 and https://github.com/zigpy/zha-device-handlers/issues/478

Hedda, 2020-11-03:
FYI, it looks like @zoic21 has now begun working on ZGP implementation for zigpy in https://github.com/zigpy/zigpy/pull/519

Hedda, 2021-02-02:
> FYI, it looks like @zoic21 has now begun working on ZGP implementation for zigpy in #519

FYI, zoic2 has now submitted an updated pull request for "Zigbee Green Power" here -> https://github.com/zigpy/zigpy/pull/656

Hedda, 2021-04-13:
@zoic21 Just FYI, read that Zigbee EmberZNet 6.9.0.0 added a new feature to Silicon Labs EFR32 NCP (eg. [bellows = EZSP](https://github.com/zigpy/bellows)):

**_Green Power Bidirectional Messaging_**

_In this release, the proxy implementations now include a compile-time option of the bidirectional communication advanced feature. This feature allows the proxy to participate in bidirectional operational message transactions by processing the gp response to submit the outgoing GPDF to DGp stub._

_The DGp stub layer of the Green Power library implements multiple transmit queues, one per GPD, to hold the outgoing GPDFs when the proxy is elected as temp master. A set of new interface functions is available as access APIs for these APIs from SoC._

_For open issue related to BRD4161A board please refer to the Known Issues section._

**_emberAfPluginScenesServerCustomRecallSceneCallback()_**

_The Scenes cluster plugin now offers the callback function emberAfPluginScenesServerCustomRecallSceneCallback() to allow a devel-oper to implement custom transitional behavior for the RecallScene command. See the callback function description in the API documen-tation for detailed description. If this callback is not implemented, the plugin’s default behavior is applied (immediately set all attributes to the values specified in the selected scene, without any consideration for transition time / gradual transition)._

Reference:

https://www.silabs.com/documents/public/release-notes/emberznet-release-notes-6.9.0.0.pdf

https://www.silabs.com/documents/public/release-notes/emberznet-release-notes-6.9.1.0.pdf

https://www.silabs.com/documents/public/release-notes/emberznet-release-notes-6.9.2.0.pdf

Hedda, 2021-12-23:
Reopen and leave open for discussion even if currently no developers are actively working on ZGP for zigpy?

Grandma-Betty, 2022-05-01:
+1
I'm also waiting for the EnOcean Green Power PTM 215ZE module to be supported.

Friedi1970, 2022-06-07:
I am still waiting for support of Senic FoH. ZGP should be implemented.

ulope, 2022-07-24:
Those auto-close stale bots are just the worst.

This is not fixed. Please reopen.

Einhardj, 2022-08-06:
Hi, I'm new to ZHA and coming from deconz. I was very surprised that zha doesn't support scenic FOH switches / ZGP. Please consider support. Thanks

Hedda, 2022-08-06:
@puddly do you know if ZGP support is needed for Zigbee Alliance Certification / CSA-IOT Certification of an application like Home Assistant with ZHA? https://csa-iot.org/certification/why-certify/

APG-ZH, 2022-10-18:
Hi, I'm new to ZHA and coming from deconz. I was very surprised that zha doesn't support my Feller Friends of Hue Switch. Please consider support. Thanks

spectas, 2022-10-26:
+1
I think this would be great, as ZGA reduces not only power consumption but also reduces the need for all these batteries we currently use in so many smart home devices and need to throw them away after not much more than a year of using it. Thank you!

PS: is there any way for a non-programmer to support the implentation - for example by donation or something else? thx

MattWestb, 2022-10-26:
@spectas Im also no code worrier so cant contributing with deep coding.
I think ZHA / Zigpy have getting very much things fixed and its now easier / possible adding the support for most radios.

If some devs have the interest doing the work and is not having hardware for testing it can being good of some one can donating one or more device for making it possible getting working code being added.

Sending devices intercontinental is optimum but if some devs is near you its not so expensive sending by post office and also getting it back then its not  needed to you.

spectas, 2022-10-26:
Good point. Is there any experience here with crowdfunding hardware to send it to devs as a donation in the hope for further development? I'd be happy to spend some money and collect some money for some devices. Also it would be interesting if there's a dev interested in working with some of these devices. @puddly and @Adminiuga the two of you have been mentioned  - would it help you to have some of these devices? Of course with no obligation attached, as I know that there's never a guarantee for a succes plus it still is a hobby (I guess?) and should stay one. Also please tell me if I am mistaken and this is a complete wrong approach. Thanks and best regards, sp:ctas

puddly, 2022-10-27:
I've ordered a scene controller and will give this a try when I have some free time.

Hedda, 2022-10-29:
> @puddly do you know if ZGP support is needed for Zigbee 3.0 gateway certifications according to Zigbee Alliance Certification / CSA-IOT Certification of an application like Home Assistant with the ZHA integration want to become certified as a Zigbee 3.0 gateway? https://csa-iot.org/certification/why-certify/

Any throught if ZGP in relation to Zigbee 3.0 certificate requirements? I am specifically thinking if Nabu Casa would want to look into certifying a solution like the ZHA integration with specific hardware like the built in radio module inside ”Home Assistant Yellow” https://www.crowdsupply.com/nabu-casa/home-assistant-yellow and/or ”Home Assistant SkyConnect USB Stick” https://www.home-assistant.io/skyconnect/ (asking as Zigbee and Z-Wave certifications are something that Home Assistant founders have mentioned before in passing).

mpepping, 2022-11-05:
+1 for ZGP spec support in Zigpy

Hedda, 2023-01-09:
For reference, this pull request for ZHA documentation would at least add a mention of this limitation in an introduction for ZHA:

https://github.com/home-assistant/home-assistant.io/pull/24494

### _Limitations_

...

_ZHA does not yet support devices that can only use the ("Zigbee Green Power") profile which is used in a few batteryless self-powered or energy harvesting devices, (such as for example; Philips Hue Click, Philips Hue Tap, and some "Friends of Hue" partnership switches). The reason for this is that ZGP profile support has not yet been implemented in the [zigpy](https://github.com/zigpy/zigpy) library._

...

patpac9, 2023-01-14:
this has been open for almost 2 years now is it even likely that it will ever come? 
Or is that simply no longer considered important enough since matter is out?

slyglif, 2023-01-14:
I'd be more than happy to try and fix up / recreate the related PR, but so far I haven't seen any comments on what was actually bad back when it was created.  Without that I don't see how we can make one that will be acceptable.

MattWestb, 2023-01-14:
I think if some one like digging and trying getting it implanted today its easier then now most of all things that is needed is moved to zigpy from the radiolibs.
And is doing changes for getting it working is most made in zigpy and not in every radiolibs that is working in different ways under zigpy so it shall being easier implanting for all possible radios.

TheRealSimon42, 2023-01-27:
+1 for this, as it is preventing many of my subscribers to jump on ZHA (From Z2M)

b1ggi, 2023-02-03:
Would also be interested, my last device missing in zha

Utty88, 2023-02-23:
Would also be interested, my last device missing in zha

GelsemaP, 2023-02-24:
Interested as well. Willing to sponsor a foh switch.

kebian, 2023-02-26:
Also super interested.  Some of the ZGP switches are the best looking devices on the market (for example, Retrotouch) and it's really unfortunate that we can't use them.

xprezz, 2023-02-27:
Need this too and as others happy to donate a FoH switch for testing

chillkiller, 2023-03-07:
Most zigbee push button coupler I know (110V-230V powered) are ZGP so its really necessary for many if not the most users to have updated zigpy within ZGP

Please pull!!!

gianlucaleuthen, 2023-03-12:
Also interested in adding Senic FoH to my ZHA setup. Ist's the only reason I still need Hue bridge.

dusopn, 2023-03-19:
Hello, 

I am about to install multiple Niko FOH switches around the house, and would really like to switch from deconz to ZHA. Please integrate ZGP.

joelast, 2023-03-27:
I would be super over-excited if this ZGP was in ZHA. Please 🙏 integrate it

puddly, 2023-03-27:
There's no need to poke this issue with new +1 comments, ZGP is on the radar.

dusopn, 2023-03-29:
> There's no need to poke this issue with new +1 comments, ZGP is on the radar.

hello Puddly, sorry I have not realized this is already being worked on. Is there a specific thread somewhere where the status could be seen? I haven't found one so I figured like everyone else that there is none and so the issue could benefit from some community demand.

D

Mic1979, 2023-04-03:
I have about 32 FOH switches around the house, and switch from (sonoff zigbee 3.0 usb dongle plus p) - Z2M over to (SkyConnect) - ZHA. and now i can't use them anymore

Hedda, 2023-05-31:
Update: Based on PRs https://github.com/zigpy/zigpy/pull/1213 and https://github.com/zigpy/zha-device-handlers/pull/2414 it looks like @nworbneb is working on ZGP support for zigpy.

FYI, CSA also released Zigbee Protocol specification R23 (a.k.a. Zigbee PRO 2023) which might have added updates to ZGP/GPD?

https://csa-iot.org/wp-content/uploads/2023/04/05-3474-23-csg-zigbee-specification-compressed.pdf

Hedda, 2023-06-09:
@nworbneb what are the next steps for ZGP now that your https://github.com/zigpy/zigpy/pull/1213 and https://github.com/zigpy/zha-device-handlers/pull/2414 PRs have been merged?

zoic21, 2023-06-09:
Hello,

Sorry for the delay, jeedom will move de z2m it's too much work to maintain zigpy (due to python requierement and pip issue). Z2M is much easiers to install an maintain.

beginnersguide, 2023-06-14:
If this PR is added to HomeAssistant-Dev, are we able to add zgp devices?
https://github.com/home-assistant/core/pull/93989#issue-1738971156

Hedda, 2023-08-18:
Is it true that Zigbee Green Power endpoint devices can normally not be connected directly to the Zigbee Cooridnator NCP application and that they instead need a compatible ZGPP (Zigbee Green Power Proxy) device in range on the network to translate the ZGP messages? At least Zigbee2MQTT notes on the EnOcean PTM 215Z device seem to imply that:

https://www.zigbee2mqtt.io/devices/PTM_215Z.html

"**_This is a Zigbee Green Power device which allows it to be very energy efficient. Messages from Green Power devices cannot be "understood" by normal Zigbee devices, therefore they need to be "translated" first. Not all Zigbee devices can do this translation, currently the only devices known to do this are Philips Hue devices. This means that the Green Power device has to be in range of a Philips Hue device in order to use it._**"

However, Silabs has now released EmberZNet 7.3 which release notes seem to imply that it is the first Silicon Labs Zigbee firmware to be able to support ZGP endpoint on NCP application which suggests that ZGP endpoints without proxies could work with it, or?

https://www.silabs.com/documents/public/release-notes/emberznet-release-notes-7.3.1.0.pdf

"**_Green power endpoint support on NCP. This release updates green power cluster and framework components to enable the green power endpoint, cluster configuration and command processing to be added on an NCP application._**"

PS: By the way, also in Zigbee2MQTT notes; "**_Green Power devices don't support binding and are not included in network scans_**".

MattWestb, 2023-08-18:
ZGP NCP function is implanting and working for very long time in the NCP firmware.
Also Silabs have one NCP with extra ZGP functionality that i have using then its have some extras i like and its more stable then the normal EZSP NCP and some devic is running ZGP on Bellows in ther systems so its possible implanting but we need looking how implanting it in the future for getting good support for it also with RCP that is using EZSP in Zigbeed.

MattWestb, 2023-08-18:
## NCP UART Application with Multi-Rail
NCP UART Application with multi-RAIL library enabled for application-specific Green Power GPDF transmission scheduling.
This network co-processor (NCP) application is an extension to the standard ncp-uart-hw sample application with following changes.
Application configuration:
 - Multi-RAIL library enabled instead of single RAIL (in multi-RAIL one handle is used by the Zigbee stack whereas the other is used by the application)
 - multirail-demo plugin enabled (this initializes the additional RAIL handle)
 - GP library with sink and proxy table set to non zero.
This application implements a simple GP TX queue. Its size can be configured by defining EMBER_APPL_GP_BIDIRECTIONAL_TX_QUEUE_SIZE.
The queue works as follows. The host inititializes and submits the outgoing GPDF packets against a GPD address, and the queue holds it. When a GPDF is received (checked in emberPacketHandoffIncoming for MAC data type frames) from a GPD with the rxAfterTx bit set (in its GP NWK Ext FC) the queue is read and a transmission is scheduled using the additional RAIL handle for the rx offset time in the application GP_RX_OFFSET_USEC (i.e., 20000 micro seconds).
This application implements the following custom EZSP commands as the queue interface
EMBER_CUSTOM_EZSP_COMMAND_INIT_APP_GP_TX_QUEUE: Initializes and clears the application-specific GP outgoing tx queue.
EMBER_CUSTOM_EZSP_COMMAND_SET_APP_GP_TX_QUEUE: Sets (adds or overwrites) a GPDF frame in the queue for a given GPD.
EMBER_CUSTOM_EZSP_COMMAND_GET_APP_GP_TX_QUEUE: Gets (reads back) the content from the queue for a GPD. This is a test API for sending the raw command out using the additional RAIL handle.
EMBER_CUSTOM_EZSP_COMMAND_SEND_APP_GP_RAW: Sends a raw GP packet on a specific channel and time.

GhisHD, 2023-11-02:
> Is it still true that Zigbee Green Power endpoint devices can normally not be connected directly to the Zigbee Cooridnator NCP application and that they instead need a compatible ZGPP (Zigbee Green Power Proxy) device in range on the network to translate the ZGP messages? At least Zigbee2MQTT notes on EnOcean PTM 215Z, PTM 215ZE, and PTM 216Z devices seem to imply that:
> 
> * https://www.zigbee2mqtt.io/devices/PTM_215Z.html
> * https://www.zigbee2mqtt.io/devices/PTM_215ZE.html
> * https://www.zigbee2mqtt.io/devices/PTM_216Z.html
> 
> "**_This is a Zigbee Green Power device which allows it to be very energy efficient. Messages from Green Power devices cannot be "understood" by normal Zigbee devices, therefore they need to be "translated" first. Not all Zigbee devices can do this translation, currently the only devices known to do this are Philips Hue devices. This means that the Green Power device has to be in range of a Philips Hue device in order to use it._**"
> 
> However, Silabs has now released EmberZNet 7.3 which release notes seem to imply that it is the first Silicon Labs Zigbee firmware to be able to support ZGP endpoint on NCP application which suggests that ZGP endpoints without proxies could work with it, or?
> 
> * https://www.silabs.com/documents/public/release-notes/emberznet-release-notes-7.3.1.0.pdf
> 
> "**_Green power endpoint support on NCP. This release updates green power cluster and framework components to enable the green power endpoint, cluster configuration and command processing to be added on an NCP application._**"
> 
> PS: By the way, also in Zigbee2MQTT notes; "**_Green Power devices don't support binding and are not included in network scans_**" for EnOcean PTM 215Z and PTM 215ZE devices, however for EnOcean PTM 216Z it had this added comment; _**"Binding may work between this module and the actuator, depending if the actuator supports the Zigbee Green Power standard & if the actuators user interface supports "Finding & binding" or "EZ-mode". For more information, contact the supplier of the actuator"**_.

FYI: I have the Skyconnect + zigbee2mqtt (under home assistant), the PTM 216Z communicate directly to the Skyconnect (binding and receiving press and release button).

Hedda, 2023-11-03:
@GhisHD That is known and aldready mentioned above. This is not about Zigbee2MQTT, this is only about implementations that uses zigpy such as the ZHA integration in Home Assistant.

konistehrad, 2023-11-10:
Hi all,

I've started working in earnest on adding ZGP support to ZHA. My current branch is here: https://github.com/konistehrad/zigpy/tree/zgp

Goals for this fork are as follows:

- Implement the minimum functionality to conform to being a Green Power Target as per the Green Power Specification
- Create a GreenPowerController class that will do state tracking as per Figures 67-68 of the ZGP spec
- Create ZHA-compatible `Device`s that are transparently usable by consuming code

Specifically, this means we do not directly receive Green Power frames. We ask Green Power Proxy devices on the network to do that. The code intends to be flexible enough to allow for that in the future to become a Green Power Target+ device.

Currently I'm looking for feedback on the overall structure of the project, and also some help in figuring out the best way to parse incoming ZCL messages on the GreenPower clusters. Namely:

- What's the correct object to subscribe to to get this information?
- What method signatures will I need to consume this data?

Thanks,
Conrad

puddly, 2023-11-10:
Awesome! Could you create a draft pull request from your current branch? It will be easier to make comments from there.

konistehrad, 2023-11-10:
Done. Thanks in advance. :)

Hedda, 2023-12-04:
FYI, all of you are probably already aware of this, but for reference:

ConBee III (ConBee 3) Zigbee Coordinator adapter was released by about a month ago with ZGP-capable firmware.

https://phoscon.de/en/conbee3

As for the zigpy-deconz radio library it will need, puddly committed initial Conbee III support which is in version 0.22.0

https://github.com/zigpy/zigpy-deconz/pull/232

https://github.com/zigpy/zigpy-deconz/releases/tag/0.22.0

https://github.com/zigpy/zigpy-deconz/compare/0.21.1...0.22.0

The upcoming (not yet released) Home Assistant core 2023.12 looks to get bumped to this release for ZHA integration

https://github.com/home-assistant/core/pull/104335

https://rc.home-assistant.io/blog/2023/11/29/release-202312/

https://rc.home-assistant.io/changelogs/core-2023.12

Hedda, 2023-12-04:
> EZSP is also having support for GP endpoints but need one NCP firmware that supporting it

@MattWestb Do you mean that the out-of-the-box firmware shipping with most Silicon Labs based adapter will not support ZGP?

Do you mean they need to be flashed with an other firmware build where Green Power Sink (GPS) has specifically been enabled?

MattWestb, 2023-12-04:
Read the paper and you is knowing the different from standard NCP firmware.

NilkOne, 2024-02-18:
Hello, how to add it with Zigbee2MQTT ?
My model is [MOES ZT-LP-ZEU2S-WH-MS](https://moeshouse.com/products/zigbee-wireless-self-powered-scene-switch)

Hedda, 2024-02-19:
> Hello, how to add it with Zigbee2MQTT ?

@NilkOne That is way off-topic or you are confused. 

This zigpy libary has nothing to do with Zigbee2MQTT, (zigpy is instead only used by Home Assistant’s built-in Zigbee Home Automation integration that is known as ”ZHA”).

 Zigbee2MQTT is a third-party stand-alone Zigbee gateway application that has nothing to with zigpy nor with ZHA, and regardless, Zigbee2MQTT already has Zigbee Green Power support on its own.

Hedda, 2024-04-23:
FYI, Silicon Labs just announced a MG22E / EFR32MG22E which is an upcoming 802.15.4 based Zigbee Green Power (ZGP) and Bluetooth Low Energy (BLE) capable 2.4 GHz radio MCU/SoC that is targeting low power consumption for long-life battery-operation and energy harvesting for battery-free/batteryless IoT devices. 

https://news.silabs.com/2024-04-22-Silicon-Labs-Streamlines-Energy-Harvesting-Product-Development-for-Battery-Free-IoT

https://www.silabs.com/blog/building-a-more-sustainable-connected-world-with-xg22e

https://www.silabs.com/wireless/zigbee/efr32mg22-series-2-socs

https://community.silabs.com/s/article/what-is-zigbee-green-power-x

https://www.silabs.com/wireless/energy-harvesting

  "_**Where MG22 allows you to create energy-efficient applications, the MG22E – ‘E’ denoting Energy Conservation – takes this a step further by enhancing battery longevity and supporting designs that eliminate the need for batteries altogether. The MG22 SoCs’ combination of ultra-low transmit and receive power (8.2 mA TX at +6 dBm, 3.9 mA RX), 1.4 µA deep-sleep mode power and low-power peripherals delivers an industry-leading, energy-efficient solution for applications using the Zigbee protocol including Green Power.**_"

MG22E/EFR32MG22E (and the Bluetooth-only BG22E/EFR32BG22E) supposedly also support direction finding using Angle-of-Arrival (AoA) and Angle-of-Departure (AoD) estimation via Time-of-Arrival (ToA) algorithms that can be used for tracking and localization devices (i.e. key finder trackers or indoor positioning).

There is yet any new if and when they will release official EFR32MG22E Radio Board / Module Radio Boards (based on EFR32MG22E224F512IM40 or EFR32MG22E224F512IM32) that can be used with the EFR32xG22 Wireless Gecko Starter Kit for testing/development and prototyping.

https://www.silabs.com/wireless/zigbee/efr32mg22-series-2-socs/device.efr32mg22e224f512im40?tab=specs

https://www.silabs.com/wireless/zigbee/efr32mg22-series-2-socs/device.efr32mg22e224f512im32?tab=specs

Silabs have also announced that they will host an unboxing Tech Talk (called "Unboxing Silicon Labs' Latest Bluetooth SoC for Energy Harvesting") on the 9th of May 2024 to showcase Silicon Labs's new Bluetooth-only BG22E / EFR32BG22E energy-optimized xG22E Explorer Kit (which you can get energy harvesting shields for).

https://www.silabs.com/about-us/events/tech-talks-wireless-technology-training

During development of the Silicon Labs xG22E wireless microcontrollers they have apparently been tested with e-peas Power Managed Integrated Circuits (PMICs) designed for energy harvesting (which previously been used seen in the Conexio Stratus board with solar energy harvesting and WattUp 1W active energy harvesting developer kit.

[image]

News spotted on cnx-software.com:

https://www.cnx-software.com/2024/04/23/silicon-labs-bg22e-mg22e-fg22e-wireless-mcu-energy-harvesting-battery-free-iot-devices/

Hedda, 2024-05-08:
FYI, CSA has released updated Green Power Revision 1.1.2 specifications for Zigbee-based Energy-harvesting Technology:

https://csa-iot.org/newsroom/the-connectivity-standards-alliance-releases-green-power-1-1-2-for-zigbee-based-energy-harvesting-technology/

The latest revision of the Green Power standard, Revision 1.1.2 introduces additional features and scripts that simplify testing and allow Green Power developers to use the same unified test harness for Green Power receiver products as used for the traditional Zigbee protocol.

Zigbee PRO Green Power feature specification Basic functionality set version 1.1.2 can now be downloaded from 

https://csa-iot.org/developer-resource/specifications-download-request/

flying-sheep, 2024-11-04:
You have 23 open issues. Is the stale bot really a good idea?

Hedda, 2024-11-04:
@konistehrad Anything we not skilled enough can do to help with your "Zigbee Green Power Support" pull request? PR -> https://github.com/zigpy/zigpy/pull/1282

* https://github.com/zigpy/zigpy/pull/1282

konistehrad, 2024-11-05:
Sorry @Hedda I handed this off to @puddly as the remaining tasks were very specific to Zigpy (tests, architectural stuff, quirks refactor) and I was learning both ZGP and Zigbee on-the-fly. He’ll ultimately be responsible for finalizing the implementation or rejecting the patch. (I concede immediately that my approach wasn’t as clean or svelte as it could have been in retrospect.)

Akschirkan, 2025-04-10:
Is there any chance that ZGP will be implemented in Zigpy in the foreseeable future? Or is it better to switch to other stacks like Zigbee2MQTT if I want to use ZGP devices?

[9 more replies]

closed 2026-10-31 by puddly