zigpy/zigpy #351: [SUGGESTION] Move back to only using the PyPI-project "zigpy" and stop using the PyPI-project "zigpy-homeassistant"

Hedda, 2020-04-02:
Please consider moving back to only using the PyPI-project "zigpy" for the zigpy library package and stop using the PyPI-project "zigpy-homeassistant" to distribute the zigpy library package.

I understand that the active PyPI projects being named "zigpy-homeassistant" and "bellows-homeassistant" can be confusing to new developers, especially to those who wish to use zigpy and bellows without Home Assistant as they might think that Home Assistant is a requirement for using these libraries.

- https://pypi.org/project/zigpy/
- https://pypi.org/project/zigpy-homeassistant/

Note! This request is indirectly related to my suggestion to rename bellows -> https://github.com/zigpy/bellows/issues/215

PS: @Adminiuga & @damarco Regardless of above it is probably also a good idea if you could and would make a few more zigpy developers shared co-owner/co-author/co-maintainers of the "zigpy" and "zigpy-homeassistant" projects on PyPI:

- https://pypi.org/project/zigpy/
- https://pypi.org/project/zigpy-homeassistant/

Maybe you can consider making @dmulcahey and @puddly maintainers or those projects as well?

Hedda, 2020-04-15:
For reference, see related https://github.com/zigpy/zigpy/pull/363

Adminiuga, 2020-04-29:
Fixed in 0.20.0

closed 2020-04-29 by Adminiuga