zigpy/zigpy #356: Home Assistant's Device Type Categories and Device Classes missing from ZHA integration dependent on zigpy??

Gamester17, 2020-04-06:
This is just a question for starting a discussion about this so not really a request or bug-report for zigpy.

Maybe also more a question for Home Assistant ZHA and/or ZHA Device Handlers developers?

Noticed that the ZHA integration currently 'only' supports for the following device type categories within the [ZHA integration for Home Assistant](https://www.home-assistant.io/integrations/zha/)

- Binary Sensor
- Cover
- Fan
- Light
- Lock
- Sensor
- Switch

I believe that term in Home Assistant for those called device type categories are called "Templating"?

https://www.home-assistant.io/docs/configuration/templating/#home-assistant-template-extensions

I was just wondering what other device types of Zigbee based device categories are missing from that ZHA integration and whether or not those are dependent that zigpy add support for those first or not? Most importantly, what relatively common types of Zigbee products does today not have its device class/classes in the ZHA integration?

Example: 
- Alarm (alarm control panel, siren)
- Climate (for ex. HVAC, Thermostat, Temperature, Humidity, and vent control + keypad lock)? = https://www.home-assistant.io/integrations/climate
- DIY? (ex. https://www.zigbee2mqtt.io/information/supported_devices.html#custom-devices-diy and https://www.zigbee2mqtt.io/information/supported_devices.html#diyruz )
- Energy (energy measurement and energy metering over time)?
- Irrigation
- Presence Detection (I understand that the new Samsung SmartThings presence sensor is a Zigbee device?)

 Also, not sure in which category water cut-off/shut-off valve and gas shut-off valves call under but those are also relatively common Zigbee products?

Anyway, those Device Type Categories may also need additional attributes, such as for example; ArmDisarm, FanSpeed, HumiditySetting, TemperatureSetting, TemperatureControl, Modes, Toggles, RunCycle, Timer

PS: Slightly off-topic but Actions on Google now has  a very extensive conceptual list of Google Assistant Smart Home "Device types" and "Device traits":

- https://developers.google.com/assistant/smarthome/concepts/devices-traits
  - https://developers.google.com/assistant/smarthome/guides
  - https://developers.google.com/assistant/smarthome/traits

Would it be logical to match Zigbee device classes and attributes to Google Assistant's Device types and Device traits?

dmulcahey, 2020-04-06:
> I believe that term in Home Assistant for those called device type categories are called "Templating"?
> 
> https://www.home-assistant.io/docs/configuration/templating/#home-assistant-template-extensions

Nope, templating is different. These are the (components) domains that are currently supported. docs on architecture are here: https://developers.home-assistant.io/docs/architecture_components 

> I was just wondering what other device types of Zigbee based device categories are missing from that ZHA integration and whether or not those are dependent that zigpy add support for those first or [...]

As far as components go we're missing alarm_control_panel and climate. Both are currently in different stages of development.

DIY has its own integration or devices (like XBee) can implement the correct clusters to work with ZHA for whatever purpose suits folks. 

Metering and EM already exist. They are sensors. Additional sensors will be exposed in the future for historical consumption etc. We had to make ZHA changes to allow multiple sensors to leverage a single Zigbee cluster. This work is in flight. 

Presence detection is already supported. The SmartThings Tag is currently supported. If we get examples of more presence devices we can expand this functionality.

> Also, not sure in which category water cut-off/shut-off valve and gas shut-off valves call under but those are also relatively common Zigbee products?

These tend to be exposed as switches.

> PS: Slightly off-topic but Actions on Google now has a very extensive conceptual list of Google Assistant Smart Home "Device types" and "Device traits": [...]

This is a loaded question and I am going to refrain from answering ATM. I will say that several companies have their thoughts on how this should be modeled and they all have pros and cons. Apple, Google and Amazon all have differences though so I would be weary just leveraging one of them. RN we're actually using HA's model for the most part.

Gamester17, 2020-04-09:
That's cool! I just meant that if you would like to base all your smart home automation on Zigbee devices then integrations of devices types for categories like "Energy" and "Irrigation" really have to be their own separate components in Home Assistant because the 'smarts' in "Energy Management" and "Irrigation Management" is so much more complexed system than just energy/water-metering monitoring, timers, and keeping a history per device with actively managing energy/water consumption at the end-user level.

So I guess that for the Home Assistant to be able to offer "Energy Management" (via example an "Energy Control Panel" integration?) and "Irrigation Management" (via example an "Irrigation Control Panel" integration?) with entities from ZHA integration, or any other integrations, new such domains would first need to have their own internal control services integration component(s) designed in/for Home Assistant similar to but separate from that of the "alarm control panel" and "climate" integration components?

Thus it is not that "Energy" and "Irrigation" are only missing from ZHA but also still missing from Home Assistant because domain components for those have not yet been added in Home Assistant and by extension that is perhaps also why no one has yet created "Energy Control Panel" and "Irrigation Control Panel" integration for Home Assistant?

I guess my point was just that without such components/domains existing in Home Assistant then every end-user would have to design their own automation systems on their own from scratch for these instead of being able to simply use existing internal integration component in Home Assistant like you can do "alarm control panel" and "climate", right or?

Hedda, 2020-05-18:
FYI, Adminiuga has submitted a first pull request for "ZHA Climate" (beta) with thermostats support

https://github.com/home-assistant/core/pull/35682

Now we are cooking with radiators ;)

closed 2020-10-22 by Adminiuga