zigpy/zigpy #360: Zigbee 3.0 network with partial or full Zigbee 3.0 Protocol Stack?

Gamester17, 2020-04-09:
Some of the newer Zigbee coordinator hardware that zigpy now supports are fully compliant with Zigbee 3.0 so I was just wondering if anyone knows if anyone has thought or plans to allow zigpy to form a Zigbee 3.0 network and perhaps extend zigpy so support one or more additional Zigbee Application Profiles?

Maybe add support for Zigbee Light Link (ZLL) 1.0 or 2.0 application layer protocol profile or perhaps even extend zigpy to support the full Zigbee 3.0 stack, i.e. all existing Zigbee application layer protocol profiles with the exception of Zigbee Smart Energy (ZSE)?

- Zigbee Home Automation (ZHA)
- Zigbee Light Link (ZLL) 2.0
- Zigbee Remote Control (ZRC) 2.0
- Zigbee Building Automation (ZBA)
- Zigbee Retail Services (ZRS)
- Zigbee Health Care (ZHC)
- Zigbee Telecommunication services (ZTS)

By the way, these linked webpages below together offer quite good explanation meant for end-users to this dilemma of interoperability between different application layer protocols which cause headaches for end-users who don't know that there is a difference:

- https://www.unifore.net/home-alarm-system/introduction-to-zigbee-3-0-zigbee-ha-protocol.html
- https://www.rfwireless-world.com/Tutorials/Zigbee-3-tutorial.html

meremortals70, 2020-08-03:
Another vote for Zigbee 3.0

Hedda, 2020-08-10:
@Adminiuga and @puddly what is for reference missing from zigpy and its radio libraries to make them compatible with Zigbee 3.0?

That is, make it possible for zigpy to make the radio libraries form a Zigbee 3.0 network allowing newer devices to use a Zigbee 3.0 profile as an option instead of ZHA 1.2 profile in a Zigbee Home Automation 1.2 network if and when newer devices support Zigbee 3.0 profiles? 

If I understand it correctly, some of not all newer Zigbee Stacks support Zigbee network forming which allow multiple profiles at the same time, allowing Zigbee 3.0 devices to connect using a Zigbee 3.0 profile, Zigbee Light Link 1.x devices can connect using a ZLL 1.0 profile, and Zigbee Home Automation 1.2 devices can connect using a ZHA 1.2 profile, or?

I bellows radio library recently got support for the later EZSP API versions used Silicon Labs EmberZNet Zigbee Stack firmware that supports Zigbee 3.0 and the zigpy-znp radio library supports the newer Texas Instruments chips use Z-Stack 3.0 Zigbee Stack firmware which all do support Zigbee 3.0 standard. 

https://www.silabs.com/community/wireless/zigbee-and-thread/knowledge-base.entry.html/2018/04/24/how_to_form_zigbee3-zrzB

https://www.silabs.com/documents/public/application-notes/an1117-migrating-zigbee-profiles-to-z30.pdf

https://www.ti.com/lit/swra615

Previous versions of Silicon Labs firmware was also Zigbee PRO compatible, and the core of the Zigbee 3.0 standard is based on the Zigbee PRO 2015 specification, while the core of the Zigbee 3.1 standard will be based on the based on the Zigbee PRO 2017 specification.

Hedda, 2020-08-11:
FYI, most added Zigbee 3.0 support via OTA updates so you probably have many more Zigbee 3.0 capable devices than you know ;)

All new devices from a while back must be Zigbee 3.0 compatible to be added on Zigbee Alliance list of their certified devices:

https://zigbeealliance.org/zigbee_products/

Just some examples of very popular and commonly available devices include those from manufactures/brands such as;  GE, HEIMAN, IKEA, Innr, LEDVANCE, Leedarson, Legrand, Linkind, Müller-Licht, Namron, Paul Neuhaus, Paulmann, Philips (Hue), Schneider Electric, Sengled, Samsung (SmartThings), Sunricher, Sylvania, TaHoma, TuYa, Ubisys, and even Xiaomi (Aqara).

https://zigbeealliance.org/zigbee_products/?product_type=certified_product&se=Aqara
https://zigbeealliance.org/zigbee_products/?product_type=certified_product&se=GE
https://zigbeealliance.org/zigbee_products/?product_type=certified_product&se=HEIMAN
https://zigbeealliance.org/zigbee_products/?product_type=certified_product&se=hue
https://zigbeealliance.org/zigbee_products/?product_type=certified_product&se=IKEA
https://zigbeealliance.org/zigbee_products/?product_type=certified_product&se=Innr
https://zigbeealliance.org/zigbee_products/?product_type=certified_product&se=LEDVANCE
https://zigbeealliance.org/zigbee_products/?product_type=certified_product&se=Leedarson
https://zigbeealliance.org/zigbee_products/?product_type=certified_product&se=Legrand
https://zigbeealliance.org/zigbee_products/?product_type=certified_product&se=Linkind
https://zigbeealliance.org/zigbee_products/?product_type=certified_product&se=M%C3%BCller
https://zigbeealliance.org/zigbee_products/?product_type=certified_product&se=Namron
https://zigbeealliance.org/zigbee_products/?product_type=certified_product&se=Neuhaus
https://zigbeealliance.org/zigbee_products/?product_type=certified_product&se=Paulmann
https://zigbeealliance.org/zigbee_products/?product_type=certified_product&se=Schneider
https://zigbeealliance.org/zigbee_products/?product_type=certified_product&se=Sengled
https://zigbeealliance.org/zigbee_products/page/2/?product_type=certified_product&se=SmartThings
https://zigbeealliance.org/zigbee_products/?product_type=certified_product&se=Sunricher
https://zigbeealliance.org/zigbee_products/?product_type=certified_product&se=Sylvania
https://zigbeealliance.org/zigbee_products/?product_type=certified_product&se=TaHoma
https://zigbeealliance.org/zigbee_products/?product_type=certified_product&se=Ubisys

As you can see on that list many popular devices that have already been available to buy for a while are now Zigbee 3.0 certified.

Zigbee Alliance is pushing very hard for Zigbee 3.0 certification and devices need to be independently tested to be listed there.

PS: I also read that ITead is aiming to get its latest Sonoff branded Zigbee sensors (Silabs based) certified for Zigbee 3.0 now.

Adminiuga, 2020-09-04:
This is not a Zigpy requests. Zigbee 3 support depends on the coordinator firmware and just a little bit on the radio lib: Radio Lib needs to implement the method to support joining devices with an install code.

closed 2020-09-04 by Adminiuga

Hedda, 2020-09-07:
@Adminiuga Which radio libraries for zigpy does already support forming a Zigbee 3.0 network and joining Zigbee 3.0 devices as a true Zigbee 3.0 device in Zigbee 3.0 mode?

Adminiuga, 2020-09-07:
Bellows>=0.20.0, zigpy-znp
It is not as much depends on the library, as on firmware running on the stick. All libraries form the network same way.

Hedda, 2020-09-07:
zigpy-zigate too then since ZiGate has Zigbee 3.0 firmware?

I believe that the current deCONZ adapters, ConBee II and Raspbee II, even with latest firmware still only support Zigbee PRO, however not sure exactly which revision/version of Zigbee PRO, but newer revision/version of Zigbee PRO form the core Zigbee 3.0

Adminiuga, 2020-09-07:
On the flip side ConBee de report zgp endpoint so must be Zigbee 3? Need to check node descriptor of the ConBee

Adminiuga, 2020-09-08:
Zigate could be ZB3 firmware. I couldn't get their stick, as they won't ship to US :(

Hedda, 2020-09-08:
> On the flip side ConBee de report zgp endpoint so must be Zigbee 3? Need to check node descriptor of the ConBee

I believe that ZGP is part of newer revisions of Zigbee PRO and not specifically new for the first time in Zigbee 3.0

Hedda, 2020-09-08:
> Zigate could be ZB3 firmware. I couldn't get their stick, as they won't ship to US :(

Perhaps either @doudz or @pipiche38 has some connections at ZiGate that could ship you a ZiGate adapter/module for free? :)

fairecasoimeme, 2020-09-08:
Hello,
Zigate use a Zigbee 3.0 stack from NXP since the begining. For the moment, we didn't implement ZGP because the lack of memory but the issue will be resolved for the next ZiGate.
unusually, I can send a ZiGate to USA (with prototype status), you can send to me a mail to arrange this. contact[at]zigate.fr

Fred

pipiche38, 2020-09-08:
> > Zigate could be ZB3 firmware. I couldn't get their stick, as they won't ship to US :(
> 
> Perhaps either @doudz or @pipiche38 has some connections at ZiGate that could ship you a ZiGate adapter/module for free? :)

done ! and you got his answer above ;-)

Hedda, 2020-09-09:
@Adminiuga Please consider e-mail @fairecasoimeme to contact[at]zigate.fr for ZiGate sample. Maybe also @puddly is interested?