zigpy/zigpy #372: [REQUEST] Allow configuration of TX power for radios with built-in Power Amplifier (PA) frontend

Hedda, 2020-04-14:
Allow user to change/set "Frontend TX Power Amplification" if available option in the radio library?

This is a follow up to https://github.com/zigpy/zigpy-cc/issues/33 and https://github.com/zha-ng/zigpy-znp/issues/11

If and when radio libraries allow the setting TX power on modules with built-in Power Amplifier (PA) then Zigpy too needs to be able to configure the TX power for radios from software using zigpy like Home Assistant, or?

Adminiuga, 2020-04-14:
Unless this could be configured across all zigpy supported radios, I'd leave it to the radio lib implementation. Could be implemented with pending changes in zigpy. For example take a look at https://github.com/zigpy/bellows/commits/new-zigpy-init/dev  which implement EZSP config customization.

Hedda, 2020-04-15:
FYI, apparently @puddly has already enabled this in [3f6968d](https://github.com/zha-ng/zigpy-znp/commit/3f6968d7af53e15d893d92240136bf93ef42f7ed) for the zigpy-znp radio library.

Hedda, 2020-04-17:
As I understand this is related to https://github.com/zigpy/zigpy/issues/381 as "TX Power" is an extension that maybe not all radio libraries will use but zigpy / ZHA need to be able permit and config it.

closed 2020-05-10 by Adminiuga