zigpy/zigpy #385: [REQUEST] Use zigpy & bellows with Sonoff ZBBridge (Zigbee to WiFi Bridge/Gateway by Itead)

Hedda, 2020-04-20:
Please consider looking at using Sonoff ZBBridge as a networked Zigbee adapter for zigpy & bellows

- https://www.itead.cc/sonoff-zbbridge.html

Would be very cool if we could use this new Sonoff ZBBridge as a remote WiFi adapter for zigpy & bellows to connect directly to via IP if and when it is flashed with a third-party "OS" (firmware) for ESP8266. Perhaps as a proof-of-concept just using something simple like ser2net to allow ZHA to remote connect to it via serial over the network, like what Zigbee2mqtt does:

https://www.zigbee2mqtt.io/how_tos/how_to_connect_to_a_remote_adapter.html

Backstory bellow: 

Itead has just launched a new product called "Sonoff ZBBridge" which at less than $17 US-dollar is a very inexpensive Zigbee to Wi-Fi Bridge/Gateway which contain a very powerful Zigbee radio module

- https://www.cnx-software.com/2020/04/16/sonoff-zbbridge-wifi-to-zigbee-gateway/

According to the teardown on notenoughtech.com it sounds as if it contain a Silicon Labs EFR32MG21 (EFR32 Gecko Series 2) Zigbee 3.0 radio module support and ESP8266 for WiFi and bridge/gateway/controller software. Where ESP8266EX is basically an Arduino with integrated Wi-Fi.

- https://notenoughtech.com/home-automation/sonoff/sonoff-zigbee-bridge-preview/

zigpy devs already do something somewhat similar to connect remotely with ESP8266 based "ZiGate Pack WiFi"

- https://zigate.fr/produit/zigate-pack-wifi-v1-3/
- https://github.com/fairecasoimeme/ZiGate/
- https://github.com/zigpy/zigpy-zigate

A similar to Sonoff ZBBridge but non-Zigbee RF to WiFi device called "Sonoff RF Bridge" already existing and that device has been hacked to run third-party open-source ESP8266 firmware from ex. Tasmota, ESPurna, and OpenMQTTGateway. So instructions for different methods to flash firmware on that "Sonoff RF Bridge" device already exist.

- https://github.com/arendst/Tasmota
- https://github.com/xoseperez/espurna
- https://github.com/1technophile/OpenMQTTGateway

As you might already know, Tasmota also does something similar with its Zigbee2Tasmota (Z2T) subproject but by connecting an ESP8266 to a Texas Instrument CC2530 and acting as a MQTT gateway

- https://www.cnx-software.com/2019/12/23/tasmota-open-source-firmware-now-supports-zigbee-connectivity/#tasmota-zigbee-support
- https://tasmota.github.io/docs/Zigbee/

Hedda, 2020-04-20:
Maybe a good excuse to for some zigpy developers here to play with MicroPython on ESP8266?

- https://micropython.org
- https://en.wikipedia.org/wiki/MicroPython

I guess as a proof-of-concept you could use a popular ESP8266 development board like the Wemos D1 Mini or the ESP8266 12-E NodeMCU Kit and connect any already supported serial Zigbee module (like example CC2530) to its matching zigpy radio library by using ser2net or any other open-source serial server software that is compatible with ESP8266 / Arduino.

Hedda, 2020-04-20:
Do you think that you could use esp-link for this?

- https://github.com/jeelabs/esp-link

Hedda, 2020-04-20:
ESPEasy (ESP Easy) firmware for ESP8266 can otherwise be used as a generic Ser2Net gateway

https://www.letscontrolit.com/wiki/index.php/Ser2Net

https://github.com/letscontrolit/ESPEasy

Hedda, 2020-04-20:
@oxan might have a solution for the ESPHome firmware in https://github.com/esphome/feature-requests/issues/660

Hedda, 2020-04-21:
ESP8266 dev board like NodeMCU V2 with a module like Ebyte E180-ZG120B is about the equivalent.

E180-ZG120B (and the older E180-ZG120A) modules are sold on eBay and Aliexpress at low prices. 

- https://www.aliexpress.com/wholesale?catId=0&SearchText=%22E180-ZG120B%22

Example: 

- https://www.aliexpress.com/item/4000473346234.html

Ebyte is making an inexpensive development board called "E180-ZG120B-TB" made for testing this. 

- http://www.ebyte.com/en/product-view-news.aspx?id=896
  - http://www.ebyte.com/en/downpdf.aspx?id=896
- http://www.cdebyte.net/test-board/zigbee-3-0-module-usb-test-board-kits.html

This development is sold for less than $9 US-dollar on Aliexpress or about twice that on eBay UK:

- https://www.aliexpress.com/item/4000538003718.html
- https://www.ebay.co.uk/itm/EFR32-Zigbee-3-0-Test-Board-USB-Port-2-4GHZ-E180-ZG120B-TB-Test-Kit/233517910970

Note! Ebyte is now making two EFR32MG12 based modules called E180-ZG120A and E180-ZG120B

- http://www.ebyte.com/en/product-view-news.aspx?id=842
  - http://www.ebyte.com/en/downpdf.aspx?id=842
  - http://www.ebyte.com/en/pdf-down.aspx?id=161

Sonoff ZBBridge could be the perfect hardware if it was not for the case that it needs to be hacked.

Hedda, 2020-05-04:
FYI @SillyDay don't own a E180-ZG120B but has tried to build a firmware for it as per discussion here:

https://github.com/Koenkk/zigbee-herdsman/issues/168

SillyDay GitHub repo for EFR32 firmware:

https://github.com/SillyDay/EFR32

Hedda, 2020-05-05:
FYI, as per related request https://github.com/zigpy/zigpy/issues/405 I just learned that there is an existing ESP8266-based networked-attached Zigbee-adapter called "ZiGate Pack WiFi adapter" which has a new v2.0 firmware that archives this requested function of UART-to-TCP/IP (for Serial-port to WiFi-bridge function) using "ESP-LINK from Jeelab" software, in addition, using ESP-LINK also adds mDNS zeroconf to allow automatic network discovery and configuration:

- https://github.com/jeelabs/esp-link

Description of functions that using ESP-LINK will add to v2.0 firmware for ZiGate Pack WiFi adapter:

https://translate.google.com/translate?sl=fr&tl=en&u=https%3A%2F%2Fzigate.fr%2Fdocumentation%2Fdescription-du-firmware-v2-xx%2F

- ZiGate TCP 9999 to UART WiFi Bridge
- WiFi configuration in client and access point mode
- DHCP client or static IP
- Client syslog
- SNTP client
- mDNS
- MQTT client
- WiFi firmware update (OTA)
- Console for logging and validating the operation of the ZiGate

As I understand it, all ZiGate hardware look to be modular in design and the "ZiGate Pack WiFi adapter" is really just an optional ESP8266 based "dumb" UART-to-TCP/IP (for Serial-port to WiFi-bridge function) for the standard "ZiGate TTL adapter" that allows users to connect to it remotely using TCP/IP over your home LAN (Local Area Network) instead of plugging it directly to your computer via the optional USB adapter. 

- https://zigate.fr/produit/module-wifi-v1-3-compatible-zigate/

Specifically, please see the picture of "ZiGate Pack WiFi adapter" https://zigate.fr/produit/zigate-pack-wifi-v1-3/ compared to the picture of "ZiGate TTL USB adapter" https://zigate.fr/produit/zigate-ttl/

Thus "ZiGate Pack WiFi adapter" allows ZHA users to have a networked Zigbee adapter setup like this:

ZHA <–> zigpy/zigpy-zigate <–> TCP/IP over LAN <–> ZiGate-WiFi <–> UART <–> ZiGate Radio

Suggesting this now as I just learned from @doudz there that the new v2.0 version of the ZiGate Pack WiFi adapter firmware contains "ESP-LINK from Jeelab" software which among other things adds mDNS and  UART WiFi Bridge support over TCP. As I understand, version v1.x of the firmware for the ZiGate Pack WiFi adapter basically only contained a simple UART/serial-port server forwarding service ( serial server software that just acts as a dumb Zigbee to WiFi bridge for zigpy-zigate), while the new version v2.x also has more advanced features (which does not need to used) it still also contain a simple UART/serial-port server forwarding service, but now mDNS also makes it easier to discover the adapter on your local network.

Now it would be awesome if the ZHA integration component for Home Assistant from an end-user perspective supported just as an easy detection and configuration of network networked-attached Zigbee coordinator adapters, like the Sonoff ZBBridge and the ZiGate Pack WiFi adapter. 

I would therefore also suggest using some kind of [Zero-configuration networking (zeroconf)](https://en.wikipedia.org/wiki/Zero-configuration_networking) method, like for example [mDNS](https://en.wikipedia.org/wiki/Multicast_DNS), (as mDNS is already in use in Home Assistant Core), to make the ZHA integration component for Home Assistant automatically detect, connect, and configure compatible networked-attached Zigbee coordinator adapters like the "ZiGate Pack WiFi adapter" as that is otherwise already supported by the zigpy-zigate radio library for zigpy.

Hedda, 2020-05-13:
Home Assistant 0.110 & later should allow zeroconf instance sharing between different integrations:

https://github.com/home-assistant/core/pull/35484

This new approach allows the zeroconf instance to be shared between Home Assistant integrations.

Hedda, 2020-06-01:
FYI, @s-hadinger has now got one and started review https://github.com/arendst/Tasmota/issues/8583

First signs it is the EFR32 Zigbee module it has inside it has a EmberZNet / Ember based firmware on it.

Hedda, 2020-06-02:
There is now also a deep dive follow-up discussion here in parallel about hacking or flashing its EFR32:

- https://github.com/Koenkk/zigbee-herdsman/issues/168

Hedda, 2020-06-22:
FYI;  Tasmota developers has made great progress hacking Sonoff ZBBridge's EFR32 Zigbee module:

- https://github.com/arendst/Tasmota/issues/8583

@mtx512 also builds a custom ZNet EZSP NCP firmware for the EFR32 module inside Sonoff ZBBridge:

- https://github.com/mtx512/efr32/tree/master/SonoffZigbeeBridge

- https://github.com/mtx512/efr32

Interestingly @s-hadinger also added a TCP serial bridge to Tasmota firmware for remote access:

- https://github.com/arendst/Tasmota/pull/8702

> ## Description:
> Add a new feature: Serial to TCP bridge (similar to esp-link). This allows to remotely communicate to a MCU through ESP8266. Needs `#define USE_TCP_BRIDGE`
> 
> It adds 2 GPIO types: `TCP Tx (208)` and `TCP Rx (209)` and can work with hardware or software serial.
> 
> Commands:
> 
> * `TCPBaudRate <x>`: sets the baudrate for serial (only 8N1 mode), min `1200`, max `115200` by 1200 increments.
> * `TCPStart <port>`: listens to port `<port>`. This features supports 2 parallel TCP connexions, which can be useful if you need a terminal + a specific protocol (like XMODEM). The 3rd connection will disconnect an previous connection. The number of parallel connections is a compile-time option.
> * `TCPStart 0` or `TCPStart`: shuts down the TCP server and disconnects any existing connection.
> 
> For security reasons, the TCP bridge is not started at boot, and requires an explicit `TCPStart` command (can be automated with Rules).

Hedda, 2020-07-03:
FYI, @s-hadinger has after the last pull request posted saying that he now feels Tasmota ready for testing of EZSP v8 interface:

https://github.com/arendst/Tasmota/issues/8583

https://github.com/arendst/Tasmota/pull/8839

Not for everyone yet since still need to flash its EFR32 chip with some type of J-Link Debug Probe (might always be required?).

tube0013, 2020-07-21:
@Hedda If you follow the instructions in arendst/Tasmota#8583 to flash the bridge with tasmota (compile with the tcp-serial enabled), and flash the firmwares there. knowing that v8 ezp support is sort of a work in progress for bellows, you can connect to remote serial ports natively in HA by manually specifying the Port, picking the ezp type radio and using the socket://host:ip as the port.  https://youtu.be/RLIiAJXLU0A

Hedda, 2020-07-31:
Yes @digiblur posted step-by-step instruction here https://www.digiblur.com/2020/07/how-to-use-sonoff-zigbee-bridge-with.html

As you can read in the update there, flashing the EFR32 chip can now be done from the tasmota-zbbridge firmware once flashed.

Hedda, 2020-08-10:
Closing as this is now supported in bellows as long as flashed with Tasmota

https://www.digiblur.com/2020/07/how-to-use-sonoff-zigbee-bridge-with.html

https://www.youtube.com/watch?v=dC1AEyFhmnc

closed 2020-08-10 by Hedda

Hedda, 2021-03-11:
FYI, HW tip now is to instead use Tube's Zigbee Gateways open-source hardware by tube0013 (which is tested via ESPHome FW).

It is based on WT32-ETH01 ESP32 board from Wireless-Tag which also has wired Ethernet.

He designed two variants; one has a Silicon Labs EFR32 Series 2 module and one that has a Texas Instruments CC2652P module.

https://github.com/tube0013/tube_gateways

[https://github.com/zigpy/zigpy/discussions/584](https://github.com/zigpy/zigpy/discussions/584)

https://www.tubeszb.com/shop/coordinators/2

[image]

[image]