zigpy/zigpy #387: [REQUEST] Zigbee sniffer interface for Wireshark using zigpy to routing packets to Wireshark for display

Hedda, 2020-04-21:
Suggest porting or rewriting a Zigbee sniffer based on com.zsmartsystems.zigbee.sniffer concept.

- https://github.com/zsmartsystems/com.zsmartsystems.zigbee.sniffer

com.zsmartsystems.zigbee.sniffer (by @cdjackson for the @zsmartsystems project) to the zigpy project to let Wireshark Zigbee sniffer function utilize radio libraries from zigpy to sniff Zigbee network traffic. Could allow Wireshark to utilize all Zigbee radio that there are zigpy radio libraries for?

My thinking here is that having access to a Zigbee sniffer interface based on zigpy would help new and existing projects using zigpy and perhaps make more devs use and contribute to the zigpy project. This could also allow Wireshark to use different hardware than it is limited to today(?).

The [com.zsmartsystems.zigbee.sniffer](https://github.com/zsmartsystems/com.zsmartsystems.zigbee.sniffer) project act as a stand-alone example of how to features of an Ember dongle to provide a network sniffer to route frames to [Wireshark](https://www.wireshark.org/). See https://wiki.wireshark.org/IEEE_802.15.4

"This project uses the com.zsmartsystems.zigbee.dongle.ember driver to provide a ZigBee sniffer interface for Wireshark, and optionally write the data to a Silabs compatible event log. The software will connect to an Ember dongle using a serial interface, and send UDP packets on port 17754 which can be received and displayed by Wireshark."

"To use Wireshark, the loopback interface needs to be selected, and then a filter udp port 17754 is used to only display ZigBee packets."

- https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html
- https://www.cd-jackson.com/downloads/ZigBeeWiresharkSniffer.pdf

com.zsmartsystems.zigbee.sniffer as it is currently depends on the Ember (EZSP) driver from https://github.com/zsmartsystems/com.zsmartsystems.zigbee

Adminiuga, 2020-04-21:
Bellows does support dumping packets into a file, see `bellows -d /dev/ttyUSB1 dump -c 15 -w packet-capture-file.pcap`

The feature is radio dependent so won't be able to implement in `zigpy`. Remember, `zigpy` is common component between the radio lib because it implements functionality common  between all radios, particularly `ZCL` and `ZDO` (ZDP) parts of Zigbee specifications.

Adminiuga, 2020-04-29:
This would not be a zigpy function, as support between radios differs. Can be implemented for EZSP protocols and I think could be implemented for ZNP based radios. Others like XBee, Conbee do not provide interface for the raw MAC layer or require a special firmware image (Conbee).

closed 2021-09-07 by github-actions[bot]