zigpy/zigpy #394: [SUGGESTION] Use Nordic nRF52840 chip ...

captcha1, 2020-04-24:
The Nordic nRF52840 chip is inexpensive and has Zigbee ZBOSS support.

The [pca10059](https://www.nordicsemi.com/Software-and-tools/Development-Kits/nRF52840-Dongle)  is only $10.

Nordic has an SDK with different [Zigbee architectures (entire stack or just MAC layer)](https://infocenter.nordicsemi.com/topic/sdk_tz_v4.0.0/zigbee_architectures.html)

... and [example code](https://infocenter.nordicsemi.com/topic/sdk_tz_v4.0.0/zigbee_cli_wrapper.html)

... and [precompiled examples](https://infocenter.nordicsemi.com/topic/sdk_tz_v4.0.0/thread_zigbee__intro.html#thread_zigbee_precompiled_examples)

[MakerDairy sells similiar USB dongles in a nice case](https://store.makerdiary.com/collections/frontpage)

puddly, 2020-04-24:
I use the dongle as a sniffer and it seems to work fine. If you want to try writing a zigpy radio library for it, using their "CLI interface" might make this a relatively simple adapter to implement.

Gamester17, 2020-08-12:
Anyone working on a zigpy radio library for Nordic Semiconductor (Nordic Semi) nRF52833 and nRF52840 Zigbee hardware?

Their nRF5 SDK for Thread and Zigbee available without cost free from nordicsemi.com

https://www.nordicsemi.com/Software-and-tools/Software/Zigbee-software

https://www.nordicsemi.com/Software-and-tools/Software/nRF5-SDK-for-Thread-and-Zigbee

https://www.nordicsemi.com/Products/Low-power-short-range-wireless/Zigbee

It includes a pre-built Zigbee 3.0 stack (ZBOSS independent Zigbee stack by DSR)

http://www.dsr-zboss.com

As already mentioned its firmware application offer a CLI (Command line interface) 

https://infocenter.nordicsemi.com/index.jsp?topic=%2Fsdk_tz_v4.0.0%2Fzigbee_cli_wrapper.html

The zb-cli-wrapper is a Python package that includes example wrapper for Nordic Zigbee CLI Agent

https://pypi.org/project/zb-cli-wrapper/

Anyway, their nRF52840 Dongle for Zigbee 3.0 is cheap at about $9 and has broad support

https://www.nordicsemi.com/Software-and-tools/Development-Kits/nRF52840-Dongle

nRF52840 DK Development kit / development baord is also not expensive at around $49

https://www.nordicsemi.com/Software-and-tools/Development-Kits/nRF52840-DK

Gamester17, 2020-08-18:
Very powerful if the nRF52840 SoC is combined with an nRF21540 RF Font-End Module (FEM) chip that integrates a power amplifier (PA) and low noise amplifier (LNA) as on paper get a boost up to 16 times range increase and -100 dBm RX sensitivity.

https://www.nordicsemi.com/Products/Low-power-short-range-wireless/nRF21540

nRF21540 key features and specifications:

- Supported wireless standards
  -Bluetooth Low Energy (incl. Bluetooth mesh)
  -Thread and Zigbee (802.15.4)
  -Proprietary 2.4 GHz
-Output Power – Adjustable in small increments up to +21 dBm
-Receive Gain – +13 dB with 2.5 dB noise figure
-Antennas – 2x antenna ports for antenna diversity
-Host Interface – I/Os, SPI, or a combination
-Supply Voltage – 1.7 V to 3.6V
-Current consumption
  -Tx tuned to +20 dBm: 115 mA
  -Rx: 4.1 mA
  -Power down mode: 30 nA
-Temperature Range – -40°C to +105°C
-Package – QFN16  4×4 mm
-When nRF21540 is combined with an nRF52 or nRF53 SoCs, Bluetooth LE (1Mbps) gets up to 16 times range increase, and -100 dBm RX sensitivity.

nRF21540 development kit provides a full-featured development kit with the following features:

- WiSoC – Nordic Semi nRF52840 Arm Cortex-M4F SoC with Bluetooth LE, 802.15.4, 2.4 GHz protocol support
- Range Extender – nRF21540 RF front end module
- Antennas – 2x 2.4 GHz antennas for antenna diversity, NFC antenna connector
- USB – 1x Micro USB connected to nRF52840 SoC, 1x Micro USB for program/debug and power
- Expansion – Arduino Uno Rev3 compatible headers
- Debugging
  - Segger J-Link OB programming/debugging via Micro USB port
  - 2x SWF RF ports for direct RF measurements
- Misc – 4x user-programmable buttons and LEDs
- Power
- 5V via Micro USB “programming” port
  - Coin-cell battery holder
  - Pins for current consumption measurements

That board shares many of the same features as the original nRF52840 DK, meaning software support is basically identical.

Also, nRF21540 evaluation kit that is meant to connect the RF FEM to lab equipment or other radios via an SMA connector.

https://www.nordicsemi.com/Products/Low-power-short-range-wireless/nRF21540

Gamester17, 2020-08-18:
@NordicSemiconductor @coderbyheart @LuDuda @philips77 @roshanrajaratnam / @DSRCorporation @t7ko Does anyone at Nordic Semiconductor and/or DSR Corporation have any interest and resources that could develop a radio library for zigpy as POC?

ZBOSS and Nordic Semiconductor support is the last major Zigbee stack and Zigbee hardware that is missing zigpy support now.

coderbyheart, 2020-08-18:
Thank you for bringing this to our attention. I have forwarded this thread internally.
I think it's fair to say that interest for zigpy support is rather low, if you

coderbyheart, 2020-08-18:
Thank you for bringing this to my attention. I have forwarded this thread internally to colleagues who might have more input.

I think it's fair to say that interest for zigpy support is rather low ([DevZone search](https://devzone.nordicsemi.com/search?q=zigpy)) which is not meant to say that this project is not valuable. It's probably not on many peoples radar at Nordic.

Nordic provides a [Zigbee CLI wrapper for python](https://infocenter.nordicsemi.com/topic/sdk_tz_v4.1.0/zigbee_cli_wrapper.html) which you might be able to leverage for this effort, and [this is what we recommend to use when building solutions on Zigbee](https://devzone.nordicsemi.com/f/nordic-q-a/60497/hint-for-zigbee-coordinator-development/246415#246415).

Gamester17, 2020-08-18:
@coderbyheart That is probably because most people do not know that zigpy is the Zigbee library that Home Assistant depends on https://www.home-assistant.io/integrations/zha/

Home Assistant is otherwise very popular so I think it is hard to call that project not valuable.

coderbyheart, 2020-08-18:
Please double check my comment, I explicitly did write 

> which **is not meant** to say that this project is not valuable.

It's a double negative so I see it's easy to misread, sorry for that.

Gamester17, 2020-08-19:
> Please double check my comment, I explicitly did write
> 
> > which **is not meant** to say that this project is not valuable.
> 
> It's a double negative so I see it's easy to misread, sorry for that.

Sorry, yeah I am not a native English speaker, though I understand that they might also have problems with a double negative ;)

If look at it from a marketing value view you can see Nordic Semiconductor and ZBOSS support is missing from Home Assistant:

https://www.home-assistant.io/integrations/zha/

Home Assistant made it on the official GitHub Octoverse Top-10 list for most popular open source projects on GitHub this year:

https://octoverse.github.com/#trends

If there was support for Nordic Semiconductor in Home Assistant's ZHA component via zigpy then you get more devs and users.

https://github.com/zigpy/zigpy/blob/dev/README.md

I also read that a few  other open source project are looking into integrating zigpy because it has board hardware support.

MeisterBob, 2020-12-20:
I'm quite familiar with the nRF52 and IEEE802.15.4 but new to zigbee.

I wanted to write a zigpy radio library for the nRF52. I have read into the code of some of the other radio libraries and if I got it right, I need to implement all the function of the `ControllerApplication` Class that are marked as `@abc.abstractmethod` and those who raise an `NotImplementedError`.

Do you have an API description which is more complete than the one in the [contributing document](https://github.com/zigpy/zigpy/blob/dev/CONTRIBUTING.md#the-zigpy-api)?

And for testing: how can I add my new nRF52 lib to Homeassistant?

Hedda, 2020-12-20:
Not sure if any more help than looking at other radio libraries but there is a beginning of API documentation here:

https://github.com/zigpy/zigpy/pull/477

What/which is the Zigbee stack/firmware that you plan on those Nordic nRF52 chips?

Hedda, 2020-12-20:
Check out the zha custom radios custom component for testing a new radio library in Home Assistant:

https://github.com/zha-ng/zha-custom-radios

Archived as as not used any longer but you probably get the gist of it.

MeisterBob, 2020-12-20:
I plan on using the [zigbee cli](https://infocenter.nordicsemi.com/topic/sdk_tz_v4.1.0/zigbee_example_cli_reference.html) that nordic provides. Maybe I need to extend the firmware but it's a good starting point. It uses the zboss stack. There is also a [python zigbee cli wrapper](https://infocenter.nordicsemi.com/topic/sdk_tz_v4.1.0/zigbee_cli_wrapper.html) but I'm not sure if I will it or not.

Hedda, 2020-12-20:
Not sure if any more help than looking at all the other radio libraries but there is a beginning of API documentation in #477

LuDuda, 2020-12-20:
FYI @tomchy

Hedda, 2020-12-21:
> There is also a [python zigbee cli wrapper](https://infocenter.nordicsemi.com/topic/sdk_tz_v4.1.0/zigbee_cli_wrapper.html) but I'm not sure if I will it or not.

FYI, found this project by @hipiphock & @ninima0323 which looks to be using zb-cli-wrapper to command Zigbee devices:

https://github.com/hipiphock/Dongle-Handler/

https://github.com/ninima0323/TestGUI

If it existed then maybe they would be interested in using zigpy together with a zigpy-zboss radio library in such a project? ;)

tomchy, 2021-01-08:
@LuDuda: Thank you for bringing my attention to this conversation!

The first question that comes into my mind is: does the the zigpy project expects a well-defined device in a Zigbee network (e.g. the coordinator), able to send ZCL commands - or - a generic, universal Zigbee dongle, that (from Zigbee protocol perspective) is defined and configured through python classes?

If it is enough to pass the Zigbee CLI device definition through handlers and provide an API to send/receive ZCL commands, than the CLI example looks like a good choice to do the job. On the contrary, if it is expected to define a device through python (e.g. it is required to provide an API to add Zigbee endpoints, set simple descriptors, send/receive raw APS frames), I'd advise to look into the NCP solution, introduced in NCS 1.4.0.

MattWestb, 2021-01-09:
@tomchy @LuDuda and @MeisterBob  
I think im not the right person answer your question but some input i can giving.
The zigpy is trying doing one universal zigbee communication between the radio module and the host system and is using quirks for "compensating / translating" commands that is not compatible with zigbee cluster library = if one devices (example one On/Off plug) is 100% zigbee compatible zigpy is only letting the command thru to / from the host system (HA or others) and if not the quirks is "translating" them to true zigbee commands so the host system is understanding them.
One Xiaomi sensor is sending there attribute in property way is converted to real ZCL commands to the host system.

For TI coordinators is using the TIs Z-Stack Monitor and Test(MT) APIs for configuring and communicating with the devices in the mesh but its little high level and is limiting the functionality. 
Opposite is the Silabs EZSP is very low level making it not so easy implanting but is making it very flexible and powerful then being good implanted and is recommended way if its possible to being done. Silabs do also having one CLI to the zigbee stack but its not being users in zigpy (its being used in IKEA GW and so on) then its not so powerful as the native EZSP protocol for coordinators.

For the TI integration is @puddly that having the most experience and for the EZSP and others is @Adminiuga the mastermind and overall architect of zigpy.

I think implanting Nordic coordinators in Zigpy is one win win situation for both sides then its open more doors and making it more flexible and user friendly for the zigbee community but its not done over one night and need lot of work being done.

In lesser than one year zigpy devs have rewritten EZSP for support of more protocols versions (over 15K line codes changed) after implanting deCONZ radios and getting the all radios from alpha or beta to stable stage (exceptions is TI CC253X with old ZHA 1.2 stack) and its more or less only one large platform that is missing in the system, the Nordick one.

Hedda, 2021-01-09:
@doudz might also be able to answer some questions as he too has gone through this process with ZiGate (NXP Semi based)?

PS: FYI- There is now alsoa new discussion space available for zigpy here -> https://github.com/zigpy/zigpy/discussions as well as a brand new wiki here space for zigpy -> https://github.com/zigpy/zigpy/wiki (again it's brand new so not yet started to be populated).

argentumus, 2021-05-12:
I am also interested in using this chip in ZHA.  I want to connect  a few a few sensors and servos to HA and I am resisting to use wifi/ESPHome as much as I can. Zigbee has been great and I am forced to have wifi for some devices (when there is no option). I managed to stay away from Zwave, BLE, etc. and for that reason I prefer to have my custom projects connected to  my ZHA based Zigbee mesh.

tomchy, 2021-11-09:
I think it is a good idea to keep this suggestion :slightly_smiling_face: 
The CLI/Shell sample is now available on the NCS master branch:
- https://developer.nordicsemi.com/nRF_Connect_SDK/doc/latest/nrf/samples/zigbee/shell/README.html

And the NCP solution has reached the v1.0.0 milestone in the NCS v1.7.0:
- https://developer.nordicsemi.com/nRF_Connect_SDK/doc/latest/nrf/releases/release-notes-1.7.0.html#zigbee

Hedda, 2021-11-16:
@tomchy Is that still using the ZBOSS Zigbee stack?  .

FYI, Home Assistant has just posted some user statistics for the first time so maybe you could use that data as an incentive to convince ZBOSS and/or Nordic Semi bosses/management to contribute developer time to make a new radio library for zigpy:

https://www.home-assistant.io/blog/2021/11/12/100k-analytics/

As noted there, Home Assistant Analytics is opt-in only so most users have not enabled analytics and yet it shows that around 13% of all Home Assistant users who actively enabled opt-in for analytics  are using the Zigbee Home Automation (ZHA) integration:

https://analytics.home-assistant.io/#integrations

That ZHA) integration in turn depends on zigpy and its radio libraries:

https://www.home-assistant.io/integrations/zha

Home Assistant founders estimate that less than 20% of all users have enabled opt-in for analytics and if that is the case then there should be at least 50,000 Home Assistant installations with the ZHA integration installed, (though based on the community my guess would be way fewer users have opt-in for analytics and there and many more ZHA installations than so).

MattWestb, 2021-11-16:
More in the basket more you can do and is not locked so i think its great open more doors if possible.

The statics is interesting what deCOZ user is using there system stand alone or with other integrations then they is on place 34 and they is saying having over 100K installations (= radio adapter sold)(i think most in ZHA and Z2M and some other systems) but i think most is using then with deCONZ and HA as automation system for it.

Hedda, 2021-11-17:
I guess it is maybe important to add this note; while the [ZHA integration component](https://www.home-assistant.io/integrations/zha) is the only "native" Zigbee implementation in Home Assistant core, there are of course also many other integrations for Home Assistant that can indirectly support Zigbee devices. Other implementations that support Zigbee devices are however abstracted behind some kind of third-party hub/bridge/gateway hardware appliance or software application, such as deCONZ/Phoscon by dresden elektronik, Zigbee2MQTT, Tasmota (Zigbee2Tasmota), Philips Hue Bridge, IKEA TRÅDFRI Gateway, Samsung SmartThings, Xiaomi Gateway (or Aqara), and Tuya Smart Gateway, but all of those third-party hubs/bridges/gateways convert Zigbee into other protocols or APIs so as far as their Home Assistant integration integrations goes it is not really Zigbee support that has been implemented.

Hedda, 2021-12-16:
@tomchy @coderbyheart @LuDuda @philips77 @roshanrajaratnam @mjdietzx / @DSRCorporation @t7ko @MeisterBob

FYI, zigpy developer puddly have begun preliminary work on merging radio libraries respective command-line tools into zigpy-cli, so early on thus now would be great timíng for Nordic Semiconductor Zigbee developers to start giving your feedback and input on its design of this new unified Zigbee CLI for zigpy which is meant as common support for all radio manufacturers. See:

https://github.com/zigpy/zigpy-cli/

At the moment he is focusing on completely exposing the low-level network and node info, (for example energy scanning already works with TI ZNP and the deCONZ ConBee/RaspBee using the same code however the zigpy-znp command line tool just needs to be transplanted into zigpy-cli), so if you want a chance to influence what should go into that zigpy CLI versus what should go into the separate zigpy radio libraries for each manufacturer then now would be the best time to begin contributing. 

* This new CLI as a standard radio settings API is primarily being discussed here -> https://github.com/zigpy/zigpy/issues/842

* Development started here -> https://github.com/zigpy/zigpy/pull/848

* Cross-radio backup and restore -> https://github.com/zigpy/zigpy-cli/pull/2

* pully mentioned that he is eventually planning on trying to add a similar shell using [IPython](https://en.wikipedia.org/wiki/IPython) for sending commands interactively ("_**since it has autocompletion and is async-friendly**_").

* pully also mentioned that bootloader commands are low on the priority list, "_**since that code is often pretty complex and uses internal radio library code. A plugin system could work, but since the existing bootloader tools are synchronous, it'd take either a complete rewrite to work with zigpy-cli, or the tools would essentially be run externally in a separate thread, which is no easier than directly running the original tools.**_"

eexarevsky, 2022-01-31:
JFYI: DSR now open NCP protocol description.
See here: https://github.com/the-tcpdump-group/tcpdump-htdocs/blob/master/linktypes/LINKTYPE_ZBOSS_NCP.html
NCP protocol parser is now included into Wireshark.
Nordic has NCP support in their SDK: https://developer.nordicsemi.com/nRF_Connect_SDK/doc/latest/nrf/samples/zigbee/ncp/README.html
Speaking about Nordic SDK - you can now use it with Nordic HW only. Means, you can't do your own implementation of ZBOSS NCP protocol for Zigbee Platform on some another SoC (I am in doubt you may want to, but just in case).

DamKast, 2022-02-15:
Nice. Has anybody started to work on this? I can give a hand...

If not, I'll start to work on this locally. And I'll let you know if I go far enough.

Hedda, 2022-02-16:
@DamKast I do not think anyone has started so would be awesome if you could give it a go. FYI; @MeisterBob only wrote in 2020:

> I wanted to write a zigpy radio library for the nRF52. I have read into the code of some of the other radio libraries and if I got it right, I need to implement all the function of the `ControllerApplication` Class that are marked as `@abc.abstractmethod` and those who raise an `NotImplementedError`. Do you have an API description which is more complete than the one in the [contributing document](https://github.com/zigpy/zigpy/blob/dev/CONTRIBUTING.md#the-zigpy-api)? I plan on using the [zigbee cli](https://infocenter.nordicsemi.com/topic/sdk_tz_v4.1.0/zigbee_example_cli_reference.html) that nordic provides. Maybe I need to extend the firmware but it's a good starting point. It uses the zboss stack. There is also a [python zigbee cli wrapper](https://infocenter.nordicsemi.com/topic/sdk_tz_v4.1.0/zigbee_cli_wrapper.html) but I'm not sure if I will it or not.

Do you think you will make it from scratch based on their [Zigbee CLI Reference](https://infocenter.nordicsemi.com/topic/sdk_tz_v4.1.0/zigbee_example_cli_reference.html) or base it on Nordic Semi's [Zigbee CLI wrapper](https://infocenter.nordicsemi.com/index.jsp?topic=%2Fsdk_tz_v4.1.0%2Fzigbee_cli_wrapper.html)? I believe one of the problems with their [Zigbee CLI wrapper](https://infocenter.nordicsemi.com/index.jsp?topic=%2Fsdk_tz_v4.1.0%2Fzigbee_cli_wrapper.html) might be if source code licenses are not compatible with wanted license?

https://infocenter.nordicsemi.com/topic/sdk_tz_v4.1.0/zigbee_example_cli_reference.html

https://infocenter.nordicsemi.com/index.jsp?topic=%2Fsdk_tz_v4.1.0%2Fzigbee_cli_wrapper.html

https://infocenter.nordicsemi.com/topic/sdk_tz_v4.1.0/zigbee_example_cli_agent.html

Will you be using "nRF52840 Dongle" and/or "nRF52840 DK" be used as reference hardware (and recommend them for testers?)?

https://www.nordicsemi.com/Products/Development-hardware/nrf52840-dongle

https://www.nordicsemi.com/Products/Development-hardware/nrf52840-dk

While the new radio API for zigpy that @puddly been working on has not yet been merged it sounds as if it probably will sooner rather than later so might want to target that new radio API and use his zigpy-znp radio library branch for it as a reference(?):

* [[RFC] Standard radio settings API zigpy#842](https://github.com/zigpy/zigpy/issues/842)

* [New radio API zigpy#848](https://github.com/zigpy/zigpy/pull/848)
  * https://github.com/puddly/zigpy/tree/puddly/new-radio-settings-api
* [Implement new radio API zigpy-cli#2](https://github.com/zigpy/zigpy-cli/pull/2)
  * https://github.com/puddly/zigpy-cli/tree/puddly/zigpy-radio-api
* [zigpy-znp - Implement new zigpy radio API](https://github.com/zigpy/zigpy-znp/pull/97)
  * https://github.com/puddly/zigpy-znp/tree/puddly/new-radio-settings-api

Hedda, 2022-02-16:
> @DamKast I don't think anyone has started so would be awesome if you could give it a go. FYI; @MeisterBob only wrote in 2020:
> 
> > I wanted to write a zigpy radio library for the nRF52. I have read into the code of some of the other radio libraries and if I got it right, I need to implement all the function of the `ControllerApplication` Class that are marked as `@abc.abstractmethod` and those who raise an `NotImplementedError`. Do you have an API description which is more complete than the one in the [contributing document](https://github.com/zigpy/zigpy/blob/dev/CONTRIBUTING.md#the-zigpy-api)? I plan on using the [zigbee cli](https://infocenter.nordicsemi.com/topic/sdk_tz_v4.1.0/zigbee_example_cli_reference.html) that nordic provides. Maybe I need to extend the firmware but it's a good starting point. It uses the zboss stack. There is also a [python zigbee cli wrapper](https://infocenter.nordicsemi.com/topic/sdk_tz_v4.1.0/zigbee_cli_wrapper.html) but I'm not sure if I will it or not.

Ah, sorry, actually it looks like MeisterBob started on a zigpy radio for nrf52, see here:

https://github.com/MeisterBob/zigpy_nrf52

Again that will be before puddly started work on new radio API for zigpy.

tomchy, 2022-02-16:
Hi @Hedda !
I was also looking around to get an idea, what is the best protocol implementation to start with. @puddly advised me to look into the deconz protocol implementation. The minimal starting point that I came up with can be found here:
https://gist.github.com/tomchy/04ac4ff78d6e117d33ab92d9cc1de779

For the NCP protocol, I'd suggest to use the `zigpy-zbncp` name.

When it comes to the implementation, If I'm not mistaken, the protocol currently has 209 commands or indications. Not all of them are required though. I wanted to start with the bare initialization procedure implementation, so the controller application will configure the device, but do not start the Zigbee stack. That should allow to tackle all connectivity issues.

Going deeper, I see that by default the zigpy package implements the ZDO endpoint. Unfortunately, this approach is not compatible with NCP protocol as it defines ~22 commands or indication to do it on the connectivity side. This would mean that the `controller.zdo` has to be reattached to the custom class.
I see that most of the protocols face a very similar issue - with the APS groups & bindings management. I do not have a clear opinion on how that should be done, any suggestions are welcome.

tomchy, 2022-02-18:
Hi @DamKast !
I think it would be great to sync our thoughts, drafts and sketches.
Could you share some contact details?

DamKast, 2022-02-18:
@tomchy you can contact me at damien.kastner@kardia.no

Hedda, 2022-02-21:
> I was also looking around to get an idea, what is the best protocol implementation to start with. @puddly advised me to look into the deconz protocol implementation. The minimal starting point that I came up with can be found here:
> https://gist.github.com/tomchy/04ac4ff78d6e117d33ab92d9cc1de779

> zigpy-znp is going to be my reference.

@puddly what is your input on this and would your advice today be to look at [zigpy-znp](https://github.com/zigpy/zigpy-znp) and/or [zigpy-deconz](https://github.com/zigpy/zigpy-deconz) as a reference?

Little off-topic but FYI;  dresden-elektronik developers started a parallel development on a deCONZ RFC REST-API v2 back in 2020:

https://github.com/dresden-elektronik/deconz-rest-plugin-v2/

https://github.com/dresden-elektronik/deconz-rest-plugin-v2/issues/1

https://github.com/dresden-elektronik/deconz-rest-plugin-v2/wiki

https://github.com/dresden-elektronik/deconz-rest-plugin-v2/wiki/Device-Descriptions-Draft

https://github.com/dresden-elektronik/deconz-rest-plugin-v2/wiki/Fault-Tolerance

https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/RFC-REST-API-v2

https://github.com/dresden-elektronik/deconz-rest-plugin-v2

> ## Why
> I think it's time that we need to figure out a proper v2 REST-API to support a wider range of devices which are available today or might arrive in the future. I'm increasingly unhappy about the /lights hack which is currently used in some places to add devices which don't fit in the provided categories.
> 
> A first humble attempt to address more complex devices was the _/devices_ endpoint which allows querying all sub-devices behind one physical device referenced by its MAC address. [dresden-elektronik/deconz-rest-plugin@96daf83](https://github.com/dresden-elektronik/deconz-rest-plugin/commit/96daf83a87e38be60978283bb375f3365be4ef2a)
> 
> ## Request For Comments
> To proceed I invite everyone to share ideas and wishes how a new interface may look like. Beside this GH issue the [RFC REST-API v2](https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/RFC-REST-API-v2) wiki page should be used to develop the new API.
> 
> ### Initial thoughts
> * The current API v1 is used by several home automation projects, therefore v2 should be implemented as a parallel interface in order to not break any v1 client.
> * The v2 API should be considered completely independent from v1, it should be the best possible API and not a workaround or compatibility compromise.
> * A proper security scheme should be integrated. HTTPS can't be used due certificate complexity. However using widely available libsodium we can have end-to-end encryption, authentication and identity management without much hassle.
> * The internal `Resource` and `ResourceItem` C++ classes are a good common ground to build a generic API on top. These are already used in many places and simplify the implementation of the rules engine. For v2 they are likely to be extended to be more versatile.
> * deCONZ and the REST-API is now nearly 10 years old with some historically grown code. As a by-product of v2 new code must be designed and implemented with a proper architecture to escape spaghetti land. The code and many processes like joining new devices could be such much better than it is right now. By now we have so much knowledge about various devices, what can go wrong and how-to workaround issues manually, that's punishment of the user and should be handled by code. The related architecture and design decision should be thoughtful developed with RFC documents.
> * The implementation of various switches have shown that a general notion of capabilities might be desirable.
> * The v2 interface should be as high level as possible, clients shall not be required to understand any Zigbee specific knowledge. However an optional low-level v2 interface can be offered in parallel to provide full access to Zigbee world.
> * Currently Websockets are used to provide real time events, MQTT can be offered as well. Qt has native support for it so the implementation is simple (in fact there exists already a very basic prototype for it).

MeisterBob, 2022-02-23:
Yes, I started with a nRF52 zigpy implementation. On the hardware side it is based on the [Zigbee CLI Agent example](https://infocenter.nordicsemi.com/topic/sdk_tz_v4.1.0/zigbee_example_cli_agent.html) in the 'nRF5 SDK for Thread and Zigbee'. Because of the lack of Documentation and time I didn't continue this project.

I added the GPL-3 license to my zigpy-nrf52 repository so you could continue with what I started, but I would suggest to dump the firmware part, that is based on the 'nRF5 SDK for Thread and Zigbee', and start something new based on the 'nRF Connect SDK'.

tomchy, 2022-02-28:
@Hedda I received answers from @puddly - it seems it is not going to be a simple thing, but it will not be the first time to do ZDO wrapping in zigpy. The deconz seems to be the simplest one, the znp has some ZDO hooks, so both are a good start.

@eexarevsky : As spotted by @DamKast the document contains the following copyright notice:
```
Copyright @ 2020 DSR Corporation | Proprietary and Confidential 
```
which, if I am not mistaken, effectively prevents anyone without a written permission to use it to write a code based on it's contents. Could you verify if it is possible to change it into something, that can be used by software distributed under GPLv3?

Hedda, 2022-08-03:
> FYI, zigpy developer puddly have begun preliminary work on a new radio API for zigpy as well as on merging radio libraries respective high-level functions and command-line tools into zigpy-cli. Still veryearly on thus now would be great timíng for Nordic Semiconductor Zigbee developers to start giving your feedback and input on its design of this new unified Zigbee CLI for zigpy which is meant as common support for all radio manufacturers. See:
> 
> https://github.com/zigpy/zigpy-cli/ (and https://github.com/zigpy/zigpy-cli/issues plus there [zigpy/zigpy-cli#5](https://github.com/zigpy/zigpy-cli/issues/5))
> 
> At the moment he is focusing on completely exposing the low-level network and node info, (for example energy scanning already works with TI ZNP and the deCONZ ConBee/RaspBee using the same code however the zigpy-znp command line tool just needs to be transplanted into zigpy-cli), so if you want a chance to influence what should go into that zigpy CLI versus what should go into the separate zigpy radio libraries for each manufacturer then now would be the best time to begin contributing.
> 
> * This new CLI as a standard radio settings API is primarily being discussed here -> [[RFC] Standard radio settings API #842](https://github.com/zigpy/zigpy/issues/842)
> * Development started here -> [New radio API #848](https://github.com/zigpy/zigpy/pull/848)
> * Cross-radio backup and restore -> [ Implement new radio API zigpy-cli#2](https://github.com/zigpy/zigpy-cli/pull/2)
> * pully mentioned that he is eventually planning on trying to add a similar shell using [IPython](https://en.wikipedia.org/wiki/IPython) for sending commands interactively ("_**since it has autocompletion and is async-friendly**_").
> * pully also mentioned that bootloader commands are low on the priority list, "_**since that code is often pretty complex and uses internal radio library code. A plugin system could work, but since the existing bootloader tools are synchronous, it'd take either a complete rewrite to work with zigpy-cli, or the tools would essentially be run externally in a separate thread, which is no easier than directly running the original tools.**_"

FYI, puddly has now merged the new radio API so all all of zigpy radio libraries should now use the new unfied API.

rbultman, 2022-08-19:
Is the work on a nRF52840 stack dead?  I am interested in looking at this.

codebullfrog, 2022-08-21:
I am interested but my coding ability isnt that great.

I have had a look at MeisterBob's code (https://github.com/MeisterBob/zigpy_nrf52) and managed to get basic functionality running (creating a co-ordinator and adding a device).

I am falling short on what to to next and how to ensure it works with zigpy properly. My Goal is to get it working with Home Assistant.

codyc1515, 2023-02-17:
Still current

Hedda, 2023-04-14:
FYI, Nordic Semi announced a new nRF54 MCU family which includes new "H series" with nRF54H20 802.15.4 radio SoC to be first:

https://www.cnx-software.com/2023/04/13/nrf54h20-cortex-m33-risc-v-wireless-soc-bluetooth-5-4-le-audio-bluetooth-mesh-thread-matter/

nRF54H20 will when released according to Nordic Semi be best-is-class and feature state-of-the-art in all categories of functions:

https://www.youtube.com/watch?v=bPPm3fLiAfc&ab_channel=NordicSemiconductor

https://www.nordicsemi.com/News/2023/04/Nordic-Semiconductor-redefines-its-leadership-in-Bluetooth-Low-Energy-with-the-nRF54-Series

https://www.nordicsemi.com/Products/nRF54H20

PS: Bt the way, missed that Nordic Semi released nRF53 family, including nRF5340 SoC for Zigbee, Thread, and BLE back in 2019:

https://www.cnx-software.com/2019/11/17/nordic-semi-nrf5340-dual-arm-cortex-m33-soc-supports-bluetooth-5-1-mesh-nfc-thread-zigbee/

DamKast, 2023-05-10:
Check this out! [zigpy-zboss](https://github.com/kardia-as/zigpy-zboss)
I wrote the zigpy implementation for the nordic radio running the ZBOSS NCP sample (from the [nRF Connect SDK](https://developer.nordicsemi.com/nRF_Connect_SDK/doc/latest/nrf/index.html)). The package is called zigpy-zboss as you suggested, and it might be useful for esp-zigbee radio as well.
I based the implementation on the [zigpy-znp](https://github.com/zigpy/zigpy-znp) and reused a lot of stuff from there.

Let me know what you think, and feel free to test it out on HA

Hedda, 2023-05-10:
> Check this out! [zigpy-zboss](https://github.com/kardia-as/zigpy-zboss)
> I wrote the zigpy implementation for the nordic radio running the ZBOSS NCP sample (from the [nRF Connect [...]

That is awesome!

What firmware image do you recommend for the nRF52840 dongle and which tool to flash it?

Hedda, 2023-05-10:
> it might be useful for esp-zigbee radio as well.

I guess that esp-zigbee should really be discussed in its own seperate discussion(?) but it will be really interesting to hear more later once ESPHome adds support to the new ESP32-C6 SoC/modules which requires upcoming Espressif's ESP IDF 5.1-> https://github.com/zigpy/zigpy/discussions/783

ESP32-C6 Arduino support will require ESP IDF 5.1 framework which is still under development but the hardware is available:

https://github.com/espressif/arduino-esp32/issues/7713

https://github.com/espressif/arduino-esp32/issues/7852

https://github.com/espressif/esp-idf/issues/10423

esp-zigbee could be very interesting for ESP32-based Zigbee gateway / remote adapter products, like ex. Tube's Zigbee Gateways:

https://github.com/tube0013/tube_gateways (and https://tubeszb.com )

Espressif's "esp-zigbee" solution is also based on ZBOSS Open Initiative (ZOI) and 

https://dsr-zoi.com

Espressif already Zigbee certificated the ESP32-H2, so assume they will do the same for ESP32-C6 after ESP IDF 5.1 is released:

https://csa-iot.org/csa_product/esp32-h2/

https://docs.espressif.com/projects/esp-zigbee-sdk/en/latest/esp32/certification.html

https://api.knack.com/v1/applications/54e658034b4f44e42fb18201/download/asset/616f731cc3bc41001ec1b65f/docs080006070csgzigbee2015layerpicsandstackprofiles_esp32h2_oct_20.docx

### esp-zigbee:

https://github.com/espressif/esp-zigbee-sdk

https://github.com/espressif/esp-zigbee-sdk/tree/main/components/esp-zigbee-lib

https://github.com/espressif/esp-zboss-lib/

https://components.espressif.com/components/espressif/esp-zboss-lib

https://github.com/espressif/esp-idf/tree/master/examples/zigbee/

https://github.com/espressif/esp-idf/blob/master/examples/zigbee/esp_zigbee_rcp

https://github.com/espressif/esp-idf/tree/master/examples/zigbee/esp_zigbee_gateway

https://github.com/openthread/ot-br-posix/

https://openthread.io/platforms/co-processor

### esp-protocols:

esp-protocols including esp modem (for AT commands, etc.), esp websocket client, mDNS (Zeroconf and DNS-SD), Asio, and more:

https://espressif.github.io/esp-protocols/

https://github.com/espressif/esp-protocols

https://github.com/espressif/esp-protocols/tree/master/components/esp_modem

https://github.com/espressif/esp-protocols/tree/master/components/esp_websocket_client

https://github.com/espressif/esp-protocols/tree/master/components/mdns

https://github.com/espressif/esp-protocols/tree/master/components/asio

### Other indirectly related tools from Espressif:

https://github.com/espressif/esptool

https://github.com/espressif/openocd-esp32

https://github.com/espressif/esp-usb-bridge

DamKast, 2023-05-11:
> What firmware image/version do you recommend for Nordic Semi's [nRF52840 Dongle](https://www.nordicsemi.com/Products/Development-hardware/nrf52840-dongle) and are they any Python scripts to flash it?

The firmware is built from the open [nRF Connect SDK](https://developer.nordicsemi.com/nRF_Connect_SDK/doc/latest/nrf/index.html). I only changed memory configuration of the zboss stack in order to be able to build big zigbee networks (coordinator configuration). The SDK is built upon [Zephyr RTOS](https://www.zephyrproject.org/).

[Here](https://github.com/kardia-as/nrf-zboss-ncp) can you find a firmware for the dongle that was built from the nRF Connect SDK v2.2.0.
I don't have a python script to flash the dongle but I am using the nordic programmer tool that comes with [nRF Connect for Desktop](https://www.nordicsemi.com/Products/Development-tools/nrf-connect-for-desktop).

Hedda, 2023-05-16:
_Originally posted by @DamKast in https://github.com/kardia-as/zigpy-zboss/issues/1#issuecomment-1547426411_

> > _Does ZBOSS also successfully send any request to endpoint 0 and intercept the response? Or does it not allow the requests to be sent directly to endpoint 0 in the first place?_
> 
> _ZBOSS does actually send ZDO requests when using the APSDE data request command (I can see ZDO packets with the sniffer), but the response is intercepted and is not returned to zigpy. This is why we have to overwrite the Zdo class in order to redefine the request methods to use the specific ZBOSS commands._
> 
> > _**Also, do you happen to have a link to the serial protocol documentation? I'm unable to find any of the commands within the ZBOSS source code, nor in Nordic's documentation.**_
> 
> _I asked directly DSR and they sent me this documentation._

@DamKast Could you try to ask your contacts at DSR and/or Nordic Semiconductor if they are maybe willing to also add/share documentation on their public websites so that it could be added as a reference so all zigpy developers could reference that too?

https://github.com/kardia-as/zigpy-zboss/blob/main/README.md#external-links-documentation-and-other-development-references

https://github.com/zigpy/zigpy/discussions/595

eexarevsky, 2023-05-16:
Hi!

See there https://cloud.dsr-corporation.com/index.php/s/BAn4LtRWbJjFiAm

Yes, ZBOSS supposes that application does not send ZDO commands as just APS data. Use ZDO API instead.

There are full sources for the host side of ZBOSS NCP there https://developer.nordicsemi.com/Zigbee/ncp_sdk_for_host/ncp_host_v2.2.1.zip

And, remind, Wireshark can parse ZBOSS NCP. But I am in doubt Nordic uses that features.

Hedda, 2023-07-21:
FYI, @DamKast have committed a few fixes to https://github.com/kardia-as/zigpy-zboss and released zigpy-zboss v1.0.2

https://github.com/kardia-as/zigpy-zboss/releases

https://pypi.org/project/zigpy-zboss/

Hedda, 2024-01-17:
Can be closed if and when https://github.com/kardia-as/zigpy-zboss is moved to zigpy organization -> https://github.com/kardia-as/zigpy-zboss/issues/17

codyc1515, 2024-01-30:
Is there a willingness from the maintainers to add this?

Hedda, 2024-01-31:
@puddly would it be OK to move [zigpy-zboss ](https://github.com/kardia-as/zigpy-zboss) repository to zigpy organization if maintailed there by @DamKast as a member?

You did write a comment in https://github.com/kardia-as/zigpy-zboss/issues/2 that wanted to test it thoroughly first before including in the downstream ZHA component inside the Home Assistant core repo, but could it not be moved to the zigpy organization before that regardless?

It could be added as an "experimental" radio library in [zigpy ](https://github.com/zigpy/zigpy) only, long before adding to ZHA integration (also as "experimental").

Again, it is likely to be discovered and tested by more people if added to  zigpy organization and included in zigpy and on PyPI.

https://pypi.org/project/zigpy/

Hedda, 2024-11-04:
FYI, for reference, puddly has posted question to [DamKast](https://github.com/DamKast) if he would like to transfer [zigpy-zboss](https://github.com/kardia-as/zigpy-zboss) repo to the zigpy organization:

* https://github.com/kardia-as/zigpy-zboss/pull/58

> _**Do you want to transfer this package to the zigpy organization so we can keep this package updated? I'd eventually like to include it as an option in ZHA once we figure out a way to get firmware publicly built. If not directly for the nRF52840, I think we probably could do so for the Espressif Zigbee stack: https://github.com/andryblack/esp-coordinator**_

Hedda, 2025-05-24:
> For reference, [puddly](https://github.com/puddly) now posted question to [@DamKast](https://github.com/DamKast) if he would like to transfer [zigpy-zboss](https://github.com/kardia-as/zigpy-zboss) repo to the zigpy organization:
> 
> * [Update to support zigpy 0.70.0 kardia-as/zigpy-zboss#58](https://github.com/kardia-as/zigpy-zboss/pull/58)
> 
> > _**Do you want to transfer this package to the zigpy organization so we can keep this package updated? I'd eventually like to include it as an option in ZHA once we figure out a way to get firmware publicly built. If not directly for the nRF52840, I think we probably could do so for the Espressif Zigbee stack: https://github.com/andryblack/esp-coordinator**_

Not stale as @DamKast posted reply to @puddly here with wish to transfer repo to zigpy orginazation:

- https://github.com/kardia-as/zigpy-zboss/pull/58

Hedda, 2025-08-14:
_Originally posted by @DamKast in https://github.com/kardia-as/zigpy-zboss/issues/58#issuecomment-2876091284_

> > > @DamKast _Do you want to transfer this package to the zigpy organization so we can keep this package updated? I'd eventually like to include it as an option in ZHA once we figure out a way to get firmware publicly built._
> > 
> > @puddly _**Sure, I can transfer this package to the zigpy organization. What are the steps to do this?**_
>
> _I sent invites to the maintainers listed, but only @dmulcahey accepted for now_

@puddly Any progress on transferring [zigpy-zboss](https://github.com/kardia-as/zigpy-zboss/pull/zigpy-zboss) repository (and [its PyPI project](https://pypi.org/project/zigpy-zboss/)) to the [zigpy organization/admins](https://github.com/zigpy) on GitHub?

Hedda, 2026-10-01:
@DamKast has an open offer to transfer the repo, project and code to the zigpy organization (which is now governed by the Open Home Foundation)

* https://github.com/kardia-as/zigpy-zboss/

puddly, 2026-10-02:
I would direct any interested users to try https://github.com/zigpy/ziggurat/ with their nRF52840 radio flashed with OpenThread RCP firmware. It'll probably work fine, assuming Nordic's OpenThread firmware is feature complete.

We already own six radio libraries in zigpy and they all require maintenance, especially once they're available for users to install within Home Assistant. Adding more to the list requires time that we really don't have at the moment.