zigpy/zigpy #402: Cree light always returns Status.SUCCESS, even when powered off

quinnhosler, 2020-05-03:
I currently use ZHA through HomeAssistant. No matter the mains power state of my bulbs, the debug statement always returns ``Status.SUCCESS``. This only occurs with my Cree light bulbs, however, my Philips Hue bulbs respond appropriately when powered off with ``Status.Unavailable``.

My Cree bulbs also occasionally miss checkins and do not appear to respond to report their state through polling. Is it possible they don't have a mechanism to report their state at all?

Adminiuga, 2020-05-03:
What debug statement?

quinnhosler, 2020-05-03:
Though HomeAssistant; it always says the following regardless of mains power. From what I can gather this ``Status.SUCCESS`` is from the zigpy library and is being created there and passed to HomeAssistant

``[homeassistant.components.zha.entity] light.bathroom: turned on: {'on_off': [1, <Status.SUCCESS: 0>]}``

Adminiuga, 2020-05-03:
You would need to provide full debug log and some extra context, like what bulb you are trying to control. Bulb's entity  id and preferably network address if known.

If you are trying to control the bulb through a  group entity, then you always would get a Success, as there's no feedback mechanism for multicast requests.

enable debug logging at least for
- homeassistant.core
- homeassistant.components.zha
- zigpy
- bellows
- zigpy_deconz
- zigpy_cc
- zigpy_zigate
- zigpy_xbee

quinnhosler, 2020-05-04:
You're right @Adminiuga, it was a multicast request I was trying. It's weird that sometimes it is marked as ``SUCCESS`` and other times it's marked as ``undefined``. Good to know that the individual bulbs respond appropriately though. It'll help me work around the group changing state when the individual bulbs do not. 

Thanks!

closed 2020-05-04 by quinnhosler