zigpy/zigpy #405: [REQUEST] mDNS network detection of ZiGate Pack WiFi adapter and configuration flow for ZHA integration

Hedda, 2020-05-05:
@doudz @Adminiuga @dmulcahey not sure if any of you own a ZiGate Pack WiFi adapter but please consider discussing and brainstorming mDNS concepts for zigpy probe method + initialization routine for ZHA integration, not only for the ZiGate Pack WiFi adapter but also for similar none-ZiGate network-attached Zigbee-adapters that will probably come in the near future, (like the new [Sonoff ZBBridge](https://github.com/zigpy/zigpy/issues/385) which will surely get a hacked ESP8266 firmware with similar features, and probably sooner rather than later as it is inexpensive and looks to have a great Zigbee radio).

Anyway, this is a follow-up issue to mDNS question raised in https://github.com/home-assistant/core/pull/35161 where @Adminiuga implemented a configuration flow for ZHA integration component to make the probing type of to local connect adapters and initial initialization routine much more user-friendly for new installations in the ZHA integration component for Home Assistant.

Now it would be awesome if the ZHA integration component for Home Assistant from an end-user perspective supported just as an easy detection and configuration of network networked-attached Zigbee coordinator adapters, like the "ZiGate Pack WiFi adapter".

Therefore I suggest using some kind of [Zero-configuration networking (zeroconf)](https://en.wikipedia.org/wiki/Zero-configuration_networking) method, like for example [mDNS](https://en.wikipedia.org/wiki/Multicast_DNS), (as mDNS is already in use in Home Assistant Core), to automatically detect, connect, and configure compatible networked-attached Zigbee coordinator adapters like the "ZiGate Pack WiFi adapter" as that is otherwise already supported by the zigpy-zigate radio library for zigpy.

All ZiGate hardware look to be modular in design and the "ZiGate Pack WiFi adapter" is what I understand really just an optional ESP8266 based "dumb" UART/Serial-to-WiFi bridge for the standard "ZiGate TTL adapter" that allows users to connect to it remotely using TCP/IP over your home LAN (Local Area Network) instead of plugging it directly to your computer via the optional USB adapter. 

- https://zigate.fr/produit/module-wifi-v1-3-compatible-zigate/

Specifically, please see the picture of "ZiGate Pack WiFi adapter" https://zigate.fr/produit/zigate-pack-wifi-v1-3/ compared to the picture of "ZiGate TTL USB adapter" https://zigate.fr/produit/zigate-ttl/

Thus "ZiGate Pack WiFi adapter" allows ZHA users to have a networked Zigbee adapter setup like this:

ZHA <–> zigpy/zigpy-zigate <–> TCP/IP over LAN <–> ZiGate-WiFi <–> UART <–> ZiGate Radio

Suggesting this now as I just learned from @doudz there that the new v2.0 version of the ZiGate Pack WiFi adapter firmware contains "ESP-LINK from Jeelab" software which among other things adds mDNS and  UART WiFi Bridge support over TCP. As I understand, version v1.x of the firmware for the ZiGate Pack WiFi adapter basically only contained a simple UART/serial-port server forwarding service ( serial server software that just acts as a dumb Zigbee to WiFi bridge for zigpy-zigate), while the new version v2.x also has more advanced features (which does not need to used) it still also contain a simple UART/serial-port server forwarding service, but now mDNS also makes it easier to discover the adapter on your local network.

- https://zigate.fr/documentation/description-du-firmware-v2-xx/

- https://github.com/jeelabs/esp-link

Here is the list of features of this new v2.xx firmware for the ZiGate Pack WiFi adapter:

- ZiGate TCP 9999 to UART WiFi Bridge
- WiFi configuration in client and access point mode
- DHCP client or static IP
- Client syslog
- SNTP client
- mDNS
- MQTT client
- WiFi firmware update (OTA)
- Console for logging and validating the operation of the ZiGate

PS: @doudz Do you maybe happen to have any connections or contact with the manufacturers of ZiGate and could possibly get them to sponsor each of you guys (yourself plus @Adminiuga and @dmulcahey ) with both the ZiGate Pack WiFi adapter as well as the standard ZiGate TTL USB adapter for ZHA, zigpy and zigpy-zigate development purposes for free under the premises of no promises and no-strings-attached so that there is no pressure for you to produce as this is still just a hobby project?

Hedda, 2020-05-05:
FYI, posted a copy of request in https://github.com/fairecasoimeme/ZiGate-WiFi/issues/3 on ZiGate-WiFi repo that redirect back here.

doudz, 2020-05-05:
For information, I implemented it in zigate lib https://github.com/doudz/zigate/blob/master/zigate/transport.py#L362

Adminiuga, 2020-05-05:
zeroconf discovery is outside zigpy scope.

Hedda, 2020-05-06:
@doudz @dmulcahey Do you think some type zeroconf discovery of networked Zigbee adapters like ZiGate WiFi (with 2.0 firmware) belongs in the ZHA integration component for Home Assistant?  

Again, the idea was a similar configuration flow as https://github.com/home-assistant/core/pull/35161 to make initial setup of networked Zigbee radio adapter in ZHA a more user-friendly experience.

Adminiuga, 2020-05-06:
>  to make initial discovery + setup of networked Zigbee radio adapters in ZHA easier

Exactly. And I think we've discussed this already, but Zigpy is just a library for handling ZCL/ZDO requests. It does not manage the radio. it is unaware of multiple radio lib existing.

Actual radio management, configuration, detection is done by the respective  radio lib.

Hedda, 2020-05-07:
@Adminiuga OK, it is then possible for you or @doudz to move this issue to the zigpy-zigate repo? 

...or should I copy the text and repost this there? -> https://github.com/zigpy/zigpy-zigate/issues