zigpy/zigpy #406: Implement ZCL Default Response for Attribute Reports

oscar-b, 2020-05-10:
The Hue Motion (SML001) blinks red on motion with ZHA, from Z2M issue tracker it seems the coordinator needs to ack the presence message for the sensor to not indicate a problem with the connection. It also seems to trigger the motion event slower than with deCONZ (when the red LED goes off again).

https://github.com/Koenkk/zigbee2mqtt/issues/897#issuecomment-536764802

https://github.com/dresden-elektronik/deconz-rest-plugin/issues/771

https://community.hubitat.com/t/zigbee-zcl-default-response-0x0b/11151

Adminiuga, 2020-05-10:
Should be able to fix for "attribute report" received command, but this raise an interesting issue with  `zigpy`: 

- When receiving cluster commands (either General or cluster specific), there should be a "response", unless "disable default response" bit is set. 
- if we delegate the "response" generation to cluster listeners, then:  1)  we need to pass the header  in 2) in case of multiple listeners, which one's response should we send out

closed 2020-05-11 by Adminiuga

oscar-b, 2020-05-15:
@Adminiuga This seems to work properly, as our Hue Motion sensors no longer flash red. Thanks!

michalmo, 2020-12-08:
Hi @Adminiuga, i came across this issue while working on getting my [Saswell SEA802-Zigbee](https://www.saswell.com/etrv-smart-radiator-thermostat-tuya-zigbee-thermostatic-radiator-valve-sea802_p107.html) TRV to work.

This is a Tuya device and requires the coordinator to respond to received commands from the Tuya manufacturer specific Cluster (0xEF00).

I can get it to work by overriding `handle_message` but that requires duplicating the condition there. Based on your comment above it seems that it would be better to have support in `zigpy`.

Adminiuga, 2020-12-08:
question is, does Tuya look for the ZCL "default response" command, or does it want to receive a Tuya cluster command response with the status?

michalmo, 2020-12-08:
The ZCL default response. the same call added in the PR works:

```python
self.create_catching_task(
    self.general_command(
        foundation.Command.Default_Response,
        hdr.command_id,
        foundation.Status.SUCCESS,
        tsn=hdr.tsn,
    )
)
```

It also sets `disable_default_response=False` in the ZCLHeader for these commands:
```
2020-12-08 01:07:59 DEBUG (MainThread) [zigpy.zcl] [0x44bf:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=2 command_id=1>
2020-12-08 01:07:59 DEBUG (MainThread) [zigpy.zcl] [0x44bf:1:0xef00] ZCL request 0x0001: [Command(status=0, tsn=1, command_id=357, function=0, data=[1, 0])]
```

PR #408 (2020-05-11): Send default response for received Report_Attributes command
Send default response for received Report_Attributes command, unless `disable_default_response` is set.

Fixes #406