zigpy/zigpy #419: [REQUEST] Zigbee "Alarm Control Panel" support (ZHA Alarm)?

Hedda, 2020-05-18:
Home Assistant has a generic component that has an interface which is similar to a classic alarm system

- https://www.home-assistant.io/integrations/alarm_control_panel/

- https://www.home-assistant.io/integrations/alarm_control_panel.template/

Could zigpy be made to support different Zigbee Alarms (Control Panels for Zigbee alarm systems)?

Also known as Premises Control Unit (PCU), Alarm Control Panel (ACP), Security Panel, or Alarm Panel

As an example implementation please see the MQTT Alarm Control Panel integration component:

- https://www.home-assistant.io/integrations/alarm_control_panel.mqtt/

MQTT Alarm Control Panel integration will accept the following states from your Alarm Panel:

disarmed

armed_home

armed_away

armed_night

armed_custom_bypass

pending

triggered

arming

disarming

The integration can control your Alarm Panel by publishing to the command_topic when a user interacts with the Home Assistant frontend.

In addition, might also want to check out the IFTTT Alarm Control Panel integration implementation:

- https://www.home-assistant.io/integrations/alarm_control_panel.ifttt/

Hedda, 2020-05-18:
Example of what seems to be relativly common Zigbee alarm control "panel" or "keypad" products:

CentraLite Security Keypad (by Ezlo Innovation LLC)
- https://centralite.com/products/security-keypad
Centralite 3-Series Security Keypad is compatible with Zigbee HA 1.2
  - https://zigbeealliance.org/zigbee_products/3-series-security-keypad/
It is also sold under other brands, Ecolink by niversal Electronics,, Xfinity by Comcast & iris by Lowes
- https://zigbeealliance.org/zigbee_products/ecolink-wireless-keypad/
- https://zigbeealliance.org/zigbee_products/comcast-xfinity-home-classic-keypad/

Linkind Keypad Control Panel from "Linkind Smart Home Security System Starter Kit"
- https://www.amazon.com/Linkind-Security-Starter-Control-Protection/dp/B07W49TS34
Not sure if Linkind Keypad Control Panel is backwards compatible with ZHA 1.2 or not?
  - https://zigbeealliance.org/zigbee_products/linkind-keypad/

NookBox Remote Keypad with Display
- https://getnookbox.com/remote-keypad/

NookBox Remote Keypad (without display)
- https://getnookbox.com/keypad/

Hedda, 2020-05-18:
Could this about "Brand Specific Arming Sequences" be relative this "Alarm Control Panel" feature?

- https://github.com/home-assistant/core/pull/35761

That PR is otherwise for AlarmDecoder integration which also has concepts which can be borrowed?

- https://www.home-assistant.io/integrations/alarmdecoder/

Hedda, 2020-05-18:
Looks like @dmulcahey has been working on alarm control panel in one in his IAS ACE branch

- https://github.com/dmulcahey/home-assistant/tree/dm/zha-ias-ace-support
  - https://github.com/dmulcahey/home-assistant/commit/46cf61867eff78733419a46ab4000b86f241c022
    - https://github.com/home-assistant/core/compare/dev...dmulcahey:dm/zha-ias-ace-support

dmulcahey, 2020-05-18:
Yep. I’ll get back to this soon hopefully

Hedda, 2020-05-18:
> Yep. I’ll get back to this soon hopefully

Nice! 

I'm still only trying to grasp the concept, though TI.com wiki does has a pretty good description here:

- https://processors.wiki.ti.com/index.php?title=Fire_Alarm_System

I understand this is mandatory and optional supported the Zigbee Home Automation (HA) Profile:
 
Mandatory = IAS CIE, IAS ACE, IAS WD, IAS Zone, and Identify(?)

Optional = HA Generic Devices, Group, (and Scenes which probably do not apply to ZHA?)(?)

- Intruder Alarm System (IAS) Devices
  - IAS Control and Indicating Equipment (CIE) = Device ID 0x0400 (e.i. Central Control Unit, i.e. ZHA)
  - IAS Ancillary Control Equipment (ACE) = 0x0401 (i.e. Remote Control Units like Keypad/Panel)
  - IAS Warning Device (WD) = 0x0403 (Warning Devices, e.i. Inddor/Outdoor Sirens, Strobes, Sprinklers?, etc. )
  - IAS Zone = 0x0402 (Zones, i.e. Areas which contain Zone Devices like Panic Button, Smoke Detectors, Heat/Fire Detectors, Motion Sensors, Door/Window Sensors, Vibration Alarms, Sprinklers?, etc. ).

The Control and Indicating Equipment (CIE) device provides the functionality for the
central control node of an IAS. As such, this device communicates with all the other
IAS devices (ACE, Zone and WD).

[image]

Above is from NXP Laboratories documentation and according to it, Zigbee IAS Devices are not exclusively concerned with intruders but also cover other emergencies, such as fire, as depicted in the above example system.
- https://www.nxp.com/docs/en/user-guide/JN-UG-3076.pdf

Hedda, 2020-08-10:
> Yep. I’ll get back to this soon hopefully

@dmulcahey Any updates or further thoughts on your "IAS ACE" (Intruder Alarm System / Alarm Control Panel) project for ZHA?

https://github.com/dmulcahey/home-assistant/tree/dm/zha-ias-ace-support

aquatix, 2021-01-27:
I would love to know too, just received my Linkind keypad, which only shows up with two entities: IAS zone, and battery level. Going to listen to events now.

Hedda, 2021-01-28:
@sebres is this maybe what you were thinking about when you submitted your https://github.com/home-assistant/core/pull/45473 pull request for ZHA?

If so then maybe you would be interested in digging into dmulcahey's "IAS ACE" branch for ZHA which can be found here:

- https://github.com/dmulcahey/home-assistant/tree/dm/zha-ias-ace-support
  - https://github.com/dmulcahey/home-assistant/commit/46cf61867eff78733419a46ab4000b86f241c022
    - https://github.com/home-assistant/core/compare/dev...dmulcahey:dm/zha-ias-ace-support

There are many parts in Zigbee IAS and that is only a start however it might be easier to build on that and not start from scratch?

Scags104, 2021-04-23:
i see configs for a ton of models but not the Centralite 3405-L.  Have it connected to ZHA with only temp, IAS Zone and battery. I can also see the cluster attribute for tamper is working. 

any plans to get this model working? Im not sure where to go from here.

Hedda, 2021-04-26:
Yes it looks like dmulcahey is working on alarm control panel support to ZHA for Home Assistant core -> https://github.com/home-assistant/core/pull/49080

thecobra666, 2021-05-26:
Is there a way to enable a delay? Now the alarm activates instantly.

Hedda, 2021-05-27:
> Is there a way to enable a delay? Now the alarm activates instantly.

Feature request for ZHA as a 10-second delay before arming is probably wanted by default on all alarm control panels as standard?

I understand that the industry standard on the commercial security alarm system is to beep for 10-seconds to warn before arming?

https://github.com/home-assistant/architecture/discussions

https://community.home-assistant.io/c/feature-requests/

aquatix, 2021-11-23:
My Linkind keypad at least works pretty much as expected now, with a delay on arming when setting the 'arm away'.

tomplast, 2022-01-04:
> My Linkind keypad at least works pretty much as expected now, with a delay on arming when setting the 'arm away'.

Would you happen to know how to change the pin code for the keypad?

tomplast, 2022-01-14:
> > Would you happen to know how to change the pin code for the keypad?
> 
> See picture of Alarm Control Panel Options in [home-assistant/core#49080](https://github.com/home-assistant/core/pull/49080)

Thank you, that worked. Another question, is it possible to add multiple pin codes to be able to known who interacted with the panel?

Finally I have a lot of sensors and I want to setup a burglar alarm. I have created  a manual alarm in Homeassistant but now I have two alarms, both the manual one and another for my keypad. Is there some smooth way of making them work together without setting up automation?

Thank you for your time.

Hedda, 2022-01-14:
> Another question, is it possible to add multiple pin codes to be able to known who interacted with the panel?

I do not think so yet(?). That would probably need to be best submitted as a new feature request(s) per implementation, see ex:

* https://community.home-assistant.io/c/feature-requests/13

> Finally I have a lot of sensors and I want to setup a burglar alarm. I have created a manual alarm in Homeassistant but now I have two alarms, both the manual one and another for my keypad. Is there some smooth way of making them work together without setting up automation?

Good question. Quick search now I found "Alarmo" which is an unofficial alarm system integration for Home Assistant:

* https://github.com/nielsfaber/alarmo
* https://community.home-assistant.io/t/custom-component-alarmo-browser-managed-alarm-system/238910

I think otherwise `manual` could be used to make more easily reusable/sharable alarm system Blueprints for Home Assistant?

* https://www.home-assistant.io/integrations/manual/
* https://community.home-assistant.io/t/synchronize-ring-alarm-keypad-v2-with-alarmo/349461
* https://community.home-assistant.io/t/example-of-the-manual-alarm/14658/
* https://community.home-assistant.io/t/a-different-take-on-an-alarm-system/7809
* https://community.home-assistant.io/t/first-alert-smoke-detector-and-carbon-monoxide-alarm-setup/66761
* https://community.home-assistant.io/t/my-combination-manual-alarm-yaml-and-node-red-alarm-example-and-looking-for-constructive-criticism/135018

Perhaps Blueprints could be used to extend [manual alarm](https://www.home-assistant.io/integrations/manual) in order to make it simpler to configure an alarm system in UI?

* https://www.home-assistant.io/blog/2020/12/13/release-202012/#blueprints
* https://community.home-assistant.io/c/configuration/blueprints/54
* https://community.home-assistant.io/t/blueprint-turn-off-lights-when-the-alarm-is-armed/251914
* https://community.home-assistant.io/t/blueprint-turn-on-lights-when-the-alarm-is-disarmed/251917
* https://community.home-assistant.io/t/do-stuff-before-and-after-an-alarm/305624
* https://community.home-assistant.io/t/alarmsystem-with-appdaemon/31312
* https://community.home-assistant.io/t/blueprint-for-alarm-alert-based-on-binary-triggers/305091
* https://community.home-assistant.io/t/how-to-arm-disarm-several-alarm-control-panel-at-the-same-time/186061/

PS: Tip is to also check out this somewhat related "Template Alarm Control Panel" and the "Alarm Panel Card":

* https://www.home-assistant.io/integrations/alarm_control_panel.template/
* https://www.home-assistant.io/lovelace/alarm-panel/

Scags104, 2022-01-14:
I was able to accomplish multiple codes by checking what was entered in the zha-event using node red and parsing that out to preform different activities. 

Example:
0000 is master key on the config page
If I type 1111 I can open my garage
2222 can unlock front door
3333 could send me a notification for fake user
And so on.

closed 2022-07-26 by github-actions[bot]

masi, 2024-03-09:
Is the whole idea of implementing Zigbee IAS fuctionallity abandoned?