zigpy/zigpy #423: OTA fails with Tradfri outlet

roger-, 2020-06-11:
I'm using HA 0.111.0 with a CC2531 dongle and trying to update an IKEA Tradfri outlet (firmware 2.0.019) following [this guide](https://www.reddit.com/r/homeassistant/comments/fak430/how_to_update_your_ikea_or_ledevance_firmware/) but I get the following when I issue the `image_notify` command:

```
2020-06-11 09:36:25 DEBUG (MainThread) [zigpy.zcl] [0x3040:1:0x0019] Schema and args lengths do not match in reply
2020-06-11 09:36:25 DEBUG (MainThread) [zigpy_cc.zigbee.application] request (0x3040, 260, 25, 1, 1, 79, b'\x19O\x00', False, False)
2020-06-11 09:36:25 DEBUG (MainThread) [zigpy_cc.api] --> SREQ AF dataRequest tsn: 79 {'dstaddr': 12352, 'destendpoint': 1, 'srcendpoint': 1, 'clusterid': 25, 'transid': 80, 'options': 0, 'radius': 30, 'len': 3, 'data': b'\x19O\x00'}
2020-06-11 09:36:25 DEBUG (MainThread) [zigpy_cc.uart] Send: b'\xfe\r$\x01@0\x01\x01\x19\x00P\x00\x1e\x03\x19O\x00Z'
2020-06-11 09:36:25 DEBUG (MainThread) [zigpy_cc.uart] Frame received: <UnpiFrame type=CommandType.SRSP subsystem=Subsystem.AF command_id=1 data=b'\x00' length=1 fcs=100>
2020-06-11 09:36:25 DEBUG (MainThread) [zigpy_cc.api] <-- SRSP AF dataRequest tsn: None {'status': 0}
2020-06-11 09:36:25 DEBUG (MainThread) [zigpy_cc.uart] Frame received: <UnpiFrame type=CommandType.AREQ subsystem=Subsystem.AF command_id=128 data=b'\x00\x01P' length=3 fcs=150>
2020-06-11 09:36:25 DEBUG (MainThread) [zigpy_cc.api] <-- AREQ AF dataConfirm tsn: None {'status': 0, 'endpoint': 1, 'transid': 80}
2020-06-11 09:36:25 DEBUG (MainThread) [zigpy_cc.zigbee.application] message ignored: dataConfirm
```

Nothing else seems to happen.

Same issue was reported [here](https://community.home-assistant.io/t/ikea-bulb-firmware-upgrade-fails/185578) with an IKEA bulb.

Adminiuga, 2020-06-11:
Power cycle the outlet and wait 10min. Do you see firmware request from the outlet?

Enable debug for homeassistant.core and post debug of the service call for notify command and a screenshot shot of how you are invoking it

roger-, 2020-06-11:
Power cycling gives the following log, don't really see any firmware requests (probably other traffic mixed in too):

```
2020-06-11 10:20:03 DEBUG (MainThread) [zigpy_cc.uart] Frame received: <UnpiFrame type=CommandType.AREQ subsystem=Subsystem.ZDO command_id=196 data=b'\xa6|\x01@0' length=5 fcs=47>
2020-06-11 10:20:03 DEBUG (MainThread) [zigpy_cc.api] <-- AREQ ZDO srcRtgInd tsn: None {'dstaddr': 0x7ca6, 'relaycount': 1, 'relaylist': [12352]}
2020-06-11 10:20:03 DEBUG (MainThread) [zigpy_cc.zigbee.application] message ignored: srcRtgInd
2020-06-11 10:20:03 DEBUG (MainThread) [zigpy_cc.uart] Frame received: <UnpiFrame type=CommandType.AREQ subsystem=Subsystem.AF command_id=129 data=b'\x00\x00\x00\x00\xa6|\x01\x01\x00,\x00xu<\x00\x00*\x1c_\x11\xbd\n\x01\xffB"\x01!\xe5\x0b\x03(\x13\x04!\xa8\x13\x05!H\x00\x06$\x01\x00\x00\x00\x00\x08!\x04\x02\n!@0d\x10\x00@0\x1c' length=62 fcs=38>
2020-06-11 10:20:03 DEBUG (MainThread) [zigpy_cc.api] <-- AREQ AF incomingMsg tsn: None {'groupid': 0, 'clusterid': 0, 'srcaddr': 0x7ca6, 'srcendpoint': 1, 'dstendpoint': 1, 'wasbroadcast': 0, 'linkquality': 44, 'securityuse': 0, 'timestamp': 3962232, 'transseqnumber': 0, 'len': 42, 'data': b'\x1c_\x11\xbd\n\x01\xffB"\x01!\xe5\x0b\x03(\x13\x04!\xa8\x13\x05!H\x00\x06$\x01\x00\x00\x00\x00\x08!\x04\x02\n!@0d\x10\x00'}
2020-06-11 10:20:03 INFO (MainThread) [zigpy_cc.zigbee.application] handle_message incomingMsg
2020-06-11 10:20:03 DEBUG (MainThread) [zigpy.zcl] [0x7ca6:1:0x0000] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=True is_reply=False disable_default_response=True> manufacturer=4447 tsn=189 command_id=Command.Report_Attributes>
2020-06-11 10:20:03 DEBUG (MainThread) [zigpy.zcl] [0x7ca6:1:0x0000] ValueError exception for: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=True is_reply=True disable_default_response=True> manufacturer=4447 tsn=189 command_id=Command.Report_Attributes> payload: b'01ff42220121e50b0328130421a8130521480006240100000000082104020a214030641000'
2020-06-11 10:20:03 DEBUG (MainThread) [zigpy.zcl] [0x7ca6:1:0x0000] new data: b'1c5f11bd0a01ff42210121e50b0328130421a8130521480006240100000000082104020a214030641000'
2020-06-11 10:20:03 DEBUG (MainThread) [zigpy.zcl] [0x7ca6:1:0x0000] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=True is_reply=False disable_default_response=True> manufacturer=4447 tsn=189 command_id=Command.Report_Attributes>
2020-06-11 10:20:03 DEBUG (MainThread) [zigpy.zcl] [0x7ca6:1:0x0000] ZCL request 0x000a: [[<Attribute attrid=65281 value=<TypeValue type=CharacterString, value=!ï¿½
(!ï¿½!H>>]]
2020-06-11 10:20:03 DEBUG (MainThread) [zigpy.zcl] [0x7ca6:1:0x0000] Attribute report received: 65281=!ï¿½
(!ï¿½!H
2020-06-11 10:20:30 DEBUG (MainThread) [zigpy_cc.uart] Frame received: <UnpiFrame type=CommandType.AREQ subsystem=Subsystem.ZDO command_id=196 data=b'@0\x00' length=3 fcs=242>
2020-06-11 10:20:30 DEBUG (MainThread) [zigpy_cc.api] <-- AREQ ZDO srcRtgInd tsn: None {'dstaddr': 0x3040, 'relaycount': 0, 'relaylist': []}
2020-06-11 10:20:30 DEBUG (MainThread) [zigpy_cc.zigbee.application] message ignored: srcRtgInd
2020-06-11 10:20:30 DEBUG (MainThread) [zigpy_cc.uart] Frame received: <UnpiFrame type=CommandType.AREQ subsystem=Subsystem.AF command_id=129 data=b'\x00\x00\x06\x00@0\x01\x01\x00/\x00}\xbe=\x00\x00\x07\x08\x04\n\x00\x00\x10\x00@0\x1d' length=27 fcs=5>
2020-06-11 10:20:30 DEBUG (MainThread) [zigpy_cc.api] <-- AREQ AF incomingMsg tsn: None {'groupid': 0, 'clusterid': 6, 'srcaddr': 0x3040, 'srcendpoint': 1, 'dstendpoint': 1, 'wasbroadcast': 0, 'linkquality': 47, 'securityuse': 0, 'timestamp': 4046461, 'transseqnumber': 0, 'len': 7, 'data': b'\x08\x04\n\x00\x00\x10\x00'}
2020-06-11 10:20:30 INFO (MainThread) [zigpy_cc.zigbee.application] handle_message incomingMsg
2020-06-11 10:20:30 DEBUG (MainThread) [zigpy.zcl] [0x3040:1:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=4 command_id=Command.Report_Attributes>
2020-06-11 10:20:30 DEBUG (MainThread) [zigpy.zcl] [0x3040:1:0x0006] ZCL request 0x000a: [[<Attribute attrid=0 value=<TypeValue type=Bool, value=Bool.false>>]]
2020-06-11 10:20:30 DEBUG (MainThread) [zigpy.zcl] [0x3040:1:0x0006] Attribute report received: on_off=0
2020-06-11 10:20:30 DEBUG (MainThread) [zigpy_cc.zigbee.application] request (0x3040, 260, 6, 1, 1, 4, b'\x18\x04\x0b\n\x00', False, False)
2020-06-11 10:20:30 DEBUG (MainThread) [zigpy_cc.api] --> SREQ AF dataRequest tsn: 4 {'dstaddr': 12352, 'destendpoint': 1, 'srcendpoint': 1, 'clusterid': 6, 'transid': 87, 'options': 0, 'radius': 30, 'len': 5, 'data': b'\x18\x04\x0b\n\x00'}
2020-06-11 10:20:30 DEBUG (MainThread) [zigpy_cc.uart] Send: b'\xfe\x0f$\x01@0\x01\x01\x06\x00W\x00\x1e\x05\x18\x04\x0b\n\x00\r'
2020-06-11 10:20:30 DEBUG (MainThread) [zigpy_cc.uart] Frame received: <UnpiFrame type=CommandType.SRSP subsystem=Subsystem.AF command_id=1 data=b'\x00' length=1 fcs=100>
2020-06-11 10:20:30 DEBUG (MainThread) [zigpy_cc.api] <-- SRSP AF dataRequest tsn: None {'status': 0}
2020-06-11 10:20:30 DEBUG (MainThread) [zigpy_cc.uart] Frame received: <UnpiFrame type=CommandType.AREQ subsystem=Subsystem.AF command_id=128 data=b'\x00\x01W' length=3 fcs=145>
[... cut ...]
2020-06-11 10:22:18 DEBUG (MainThread) [zigpy_cc.api] <-- AREQ AF incomingMsg tsn: None {'groupid': 0, 'clusterid': 1027, 'srcaddr': 0xff92, 'srcendpoint': 1, 'dstendpoint': 1, 'wasbroadcast': 0, 'linkquality': 55, 'securityuse': 0, 'timestamp': 4383581, 'transseqnumber': 0, 'len': 17, 'data': b'\x18\x96\n\x00\x00)\xdf\x03\x14\x00(\xff\x10\x00)\xbe&'}
2020-06-11 10:22:18 INFO (MainThread) [zigpy_cc.zigbee.application] handle_message incomingMsg
2020-06-11 10:22:18 DEBUG (MainThread) [zigpy.device] Ignoring message (b'18960a000029df03140028ff100029be26') on cluster 1027: unknown endpoint or cluster id: 1
```

Let me enable core logging...

Adminiuga, 2020-06-11:
it's not imediate, but after a power cycle IKEA devices usually check for firmware update in a few minutes on `clusterid: 25` see if you can find those

roger-, 2020-06-11:
Not seeing any additional `'clusterid': 25` messages after rebooting (only the first one I posted above).

puddly, 2020-06-11:
This is a problem with the radio library. I checked my logs for the past six days and I do not see a single OTA image request.

This issue should be moved to zigpy-cc and is simple to fix. I've added an endpoint with the required input cluster to zigpy-znp (https://github.com/zha-ng/zigpy-znp/commit/d84a21b6fb4b33514aad665808b185595c98e613) and have just successfully upgraded two Trådfri bulbs.

roger-, 2020-06-11:
Great.

Can you or someone move it or should I create a new issue there?

Adminiuga, 2020-06-11:
@puddly I thought zigpy-cc already was doing that in https://github.com/zigpy/zigpy-cc/blob/23e5cc937fd95dbf276c09b43a7cc8b6750a59e0/zigpy_cc/zigbee/start_znp.py#L41-L47

Question: does TI really need that many endpoints or could they be combined? at least endpoints with the same appprofileid?

puddly, 2020-06-11:
You're right, it was updated last month to include that endpoint. There is a typo in it, however, which might be a problem (`appnuminclusters=1` but `appoutclusterlist` is provided?).

As for the endpoints, I'm not 100% sure, they seem to have been added to zigbee-herdsman for various reasons (https://github.com/Koenkk/zigbee-herdsman/blame/9299a5ffed13baa0007866ad10af8ef0d1bfb63d/src/adapter/z-stack/adapter/startZnp.ts).

Adminiuga, 2020-06-11:
> which might be a problem (appnuminclusters=1 but appoutclusterlist is provided?).
Hrm, I think you are right.

> As for the endpoints, I'm not 100% sure, they seem to have been added to zigbee-herdsman for various reasons

ConBee would not pass traffic, if the `appProfileId` of the message doesn't match the profile id of the endpoint. Probably same with ZNP. Maybe needs a similar patch which was done for multicast messages, and don't match app profile id to the endpoint :shrug: