zigpy/zigpy #428: [REQUEST] PowerOn Attribute for Power Cycle Initial State

kurtwarwick-new, 2020-06-17:
Hi

Could the PowerOn attribute be added to the On/Off Cluster to allow Power Cycle Initial State changes?

I have been attempting to contribute to the code base but the CI build kept failing for a reason I was not able to figure out. 

To help out, you can check my already tested commit made to my fork [here](https://github.com/kurtwarwick-new/zigpy/commit/80e695166fd5b558585f2e22520a71f3446d7933).

Hope to see this added soon.

Thanks!

Kurt

Adminiuga, 2020-06-17:
Hi Kurt,

The `0x4003` attribute of the On/Off cluster is not in the standard [Zigbee Cluster Library](https://zigbeealliance.org/wp-content/uploads/2019/12/07-5123-06-zigbee-cluster-library-specification.pdf) so it must be vendor specific. Currently, for vendor specific attributes we just create a custom cluster/device quirk in https://github.com/zigpy/zha-device-handlers

To do so, open the issue and submit the device signature retrieved from ZHA config panel for your device, so the custom cluster applies only to those devices.

closed 2020-07-03 by Adminiuga

kurtwarwick-new, 2020-07-06:
Hi @Adminiuga 

Thanks for the response. Did you mean I should re-open this issue? If so, here is the signature:

`{
  "node_descriptor": "<NodeDescriptor byte1=1 byte2=64 mac_capability_flags=142 manufacturer_code=4107 maximum_buffer_size=71 maximum_incoming_transfer_size=45 server_mask=0 maximum_outgoing_transfer_size=45 descriptor_capability_field=0>",
  "endpoints": {
    "11": {
      "profile_id": 49246,
      "device_type": "0x0210",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0008",
        "0x0300",
        "0x1000",
        "0xfc01"
      ],
      "out_clusters": [
        "0x0019"
      ]
    },
    "242": {
      "profile_id": 41440,
      "device_type": "0x0061",
      "in_clusters": [
        "0x0021"
      ],
      "out_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "Philips",
  "model": "LCT015",
  "class": "zigpy.device.Device"
}`

The vendor attribute is `0x4003` with a description of `power_on_state`. The possible values (enum8) are 

`Off = 0x00
On = 0x01
Previous = 0xff`

I hope that's what you needed. Please let me know if there anything I should do.

Thanks.

Kurt

Adminiuga, 2020-07-06:
I meant this issue belongs to zigpy/zha-device-handlers repository.

Check if your light model is covered by https://github.com/zigpy/zha-device-handlers/pull/396

kurtwarwick-new, 2020-07-07:
Ah!

Got it. Thanks! 😄 

K