zigpy/zigpy #435: Duplicate transaction sequence numbers with Centralite 3130

sam-ward, 2020-07-05:
Hi,

I have been having a problem with Centralite 3130 (Sylvana lightify diming) switches and my Home assistant setup.  After chatting in the HA discord channel, it was suggested to post here.

Background: 
I am running Home Assistant on a RPi with ZHA and a conbee II stick (connected via a 30cm USB extension).  I use some centralite 3130 switches (the 2 button ones, not the 4 button) to control some hue lights (via home assistant automations). Currently I have automations setup to trigger on a zha_event, filtering by the device_ieee and command.  

Problem:
Sometimes the button press will work fine, and then other times the zha event seems to fire 2 or 3 times, causing the light to go on, off, on again (you get the idea).  

Looking through the debug logs, you can see where the button push message first appears, and then re-appears about 11 seconds later: https://pastebin.com/e8TGQHmU

Similarly you can see where what looks to be the battery status get reported multiple times: https://pastebin.com/gRAHVpgT earlier in the day.

In both cases it appears to be using the same transaction sequence number.

I have 2 of these 3130's active at the moment, and they both exhibit the same behaviour.  If I shutdown and restart the whole PI, it will usually come good for a bit, but then the repeated event behaviour will slowley creep back in.  Hard to say if its releated, but when I used to use these 3130 switches with smartthings, they did not exhibit this behaviour. 

Any help or suggestions of where else I can check would be greatly appreciated.  

cheers
Sam

Adminiuga, 2020-07-05:
It is received on uart so the message does come in twice. As for why, I have no ideas. Check if more than 1 Zigbee router thinks it's being a parent for this child device.

sam-ward, 2020-07-11:
There is only a single zigbee gateway on the network (the conbee stick) and no other routers (everything else is battery powered).

It might well the be the device itself, From what I have been able to tell its only those 2 3130's that are showing the error/retry behavior.  

I might raise an issue against ZHA to look at how it handles this scenario (either don't pass the message until its ACK'd, or ignore subsequent messages with duplicate sequence numbers).

dmulcahey, 2020-07-11:
> I might raise an issue against ZHA to look at how it handles this scenario (either don't pass the message until its ACK'd, or ignore subsequent messages with duplicate sequence numbers).

Please don’t, this isn’t at ZHA level

sam-ward, 2020-07-11:
My comment about zha was in respect to how the receiving of multiple duplicate messages is handled. Maybe zha is not the correct context / layer to have this discussion.

Currently the switch is sending multiple messages with the same tsn, possibly due to the switch not receiving an ack from the conbee/zigpy/ zha / home assistant (I’m still not clear on which layer is responsible for what). The result is that within home assistant it is raising multiple events. 

My suggestion is that at some layer, either the message should not result in an event till the ack is received (which I don’t think is practical / doable), or subsequent messages that are received with the same tsn should be ignored /not result in an event being raised.

But as I said above, I’m not clear on the responsibilities of the layers and therefore where the appropriate place to have this discussion is.

Adminiuga, 2020-07-11:
```
2020-07-05 01:26:30 DEBUG (MainThread) [zigpy_deconz.zigbee.application] Error while sending 27 req id frame: TXStatus.MAC_NO_ACK
```

There's a problem with ConBee where it doesn't remember the children after a power cycle. Can you remove and reinsert the battery to force device to rejoin ConBee and see if it fixes the issue?

Adminiuga, 2020-07-11:
> Currently the switch is sending multiple messages with the same tsn, possibly due to the switch not receiving an ack from the conbee/zigpy/ zha / home assistant (I’m still not clear on which layer is responsible for what). The result is that within home assistant it is raising multiple events.

There are two acks, one is sent by the network/coordinator, when a requests from the sensor successfully reaches the coordinator. and one at the application level, ZCL default response or a response to the command. 99.9% devices care about the 1st one. And very few care about ZCL default response command. ATM I'm only aware about HUE motion sensor. Since centralite doesn't always send the duplicates, most likely  it doesn't really care about the application level default response.

sam-ward, 2020-07-14:
> There's a problem with ConBee where it doesn't remember the children after a power cycle. Can you remove and reinsert the battery to force device to rejoin ConBee and see if it fixes the issue?

Thanks @Adminiuga.  I have removed replaced the battery, and so far it is working as expected.  Now that I know what to check for, I will monitor how the switch behaves in association with HA restarts. 

I'm still of the opinion that HA shouldnt be raising multiple events for duplicate TSNs, but that is a seperate discussion / github repo :-) 

Cheers
Sam

closed 2020-09-09 by Adminiuga