zigpy/zigpy #437: How to read zigpy DEBUG 'Send' messages

LionelMe, 2020-07-08:
Hi

I have included a new -unknown- Zigbee device in ZHA that I try to communicate with, that fails. My issue is that I cannot decode the messages built by zigpy.

Here is an example that works with an IKEA tradfri wallplug:
2020-07-08 08:50:41 DEBUG (MainThread) [zigpy_cc.zigbee.application] request (0x75cc, 260, 6, 1, 1, 112, b'\x01p\x00', True, False)
2020-07-08 08:50:41 DEBUG (MainThread) [zigpy_cc.api] --> SREQ AF dataRequest tsn: 112 {'dstaddr': 30156, 'destendpoint': 1, 'srcendpoint': 1, 'clusterid': 6, 'transid': 113, 'options': 0, 'radius': 30, 'len': 3, 'data': b'\x01p\x00'}
2020-07-08 08:50:41 DEBUG (MainThread) [zigpy_cc.uart] Send: b'\xfe\r$\x01\xccu\x01\x01\x06\x00q\x00\x1e\x03\x01p\x00\x8a'

Even in there I cannot translate from the intent to the 'Send' message. Anyone has a pointer, a doc, a dictionary how to read this 'Send' message ?

Sorry if I am not in the right place, I have tried in HA groups without success

Adminiuga, 2020-07-08:
This is more specific to zigpy-cc.
In zigpy-cc.application and zigpy-cc.api you are going to see more higher level messages. You would need to look at the code to see what each parameter means exactly, but at high level you are interested at the device nwk address, destination cluster, destination endpoint and the payload.  
Payload is going to be a bytes data of a ZDO or ZCL payload depending on the destination endpoint id ( endpoint id of 0 is a ZDO message) the format of the individual zcl  message depends on the cluster id in use.
To understand it, you need to know zcl frame format. Google for Zigbee cluster library specification, it covers all this minutiae

Adminiuga, 2020-07-08:
In your case, it is cluster id 6 -- on off cluster. The 1st byte of payload is zcl frame control, saying whether it is a cluster specific or ZCL general command, like read_attributes. The 2nd byte in this case is a transaction sequence number -- to match request to replies (in some cases depending on frame control, there could be manufacturer id code) and third byte is command is. My guess it is a cluster specific command id==0 either on or off command.

LionelMe, 2020-07-13:
Hi again
has anyone done something ? It now works -almost- like a charm.
With no intervention from my side, I tried today to click on UP or DOWN arrow, and the shutter moved accordingly to my command. Even the position slider works, not perfectly calibrated thu.
I like the idea it works now, but being an engineer I desperately need to understand how it started working.

closed 2020-08-20 by Adminiuga