zigpy/zigpy #452: [SUGESTION] - Documentation on how to use the library

pipiche38, 2020-08-10:
As per definition the zigpy is implemented as Library for Zigbee devices.
Whoever all documentation I found is very focussed on HA.

I was under the impression that I could use zigpy as a pure python Library , maybe I'm wrong !

puddly, 2020-08-10:
Zigpy doesn't depend on HA in any way (other than HA being the biggest project directly utilizing it) so you can definitely use it standalone with the appropriate radio library. Here's an (untested) example:

```Python
import asyncio

# There are many different radio libraries but they all have the same API
from zigpy_znp.zigbee.application import ControllerApplication

class MainListener:
    """
    Contains callbacks that zigpy will call whenever something happens.
    Look for `listener_event` in the Zigpy source or just look at the logged warnings.
    """

    def __init__(self, application):
        self.application = application

    def device_joined(self, device):
        print(f"Device joined: {device}")

    def attribute_updated(self, device, cluster, attribute_id, value):
        print(f"Received an attribute update {attribute_id}={value}"
              f" on cluster {cluster} from device {device}")

async def main():
    app = ControllerApplication(ControllerApplication.SCHEMA({
        "database_path": "/path/to/zigbee.db",
        "device": {
            "path": "/dev/serial/by-id/your-serial-port",
        }
    }))

    listener = MainListener(app)
    app.add_listener(listener)

    await app.startup(auto_form=True)

    # Permit joins for a minute
    await app.permit(60)
    await asyncio.sleep(60)

    # Just run forever
    await asyncio.get_running_loop().create_future()

if __name__ == "__main__":
    await asyncio.run(main())
```

pipiche38, 2020-08-10:
thanks that will be very useful. What about sending a command to  a device  ?

Like On/Off on Cluster 0x0006 or a LevelCOntrol on Cluster 0x0008 N

puddly, 2020-08-10:
You would do something like `await app.get_device(nwk=0x1234).endpoints[ep_id].on_off.on()`. Some places to look:

 - https://github.com/zigpy/zigpy/blob/371f1198dad5a3f0587907d01169b6656e5262b3/zigpy/application.py#L332-L341
 - https://github.com/zigpy/zigpy/blob/41986610e84a51cd89ff09701ce1a055d7c9a6a6/zigpy/zcl/clusters/general.py#L383-L404

pipiche38, 2020-08-11:
@puddly thanks for all those informations. I'm making progress, as I have been able to get connected to the HW ( RPI Zigate ), and get devices paired.
However I have few issues/questions 

I have those error messages which seems coming when the device is sending messages ( Xiaomi Motion Sensor )
```Error calling listener.attribute_updated: 'NoneType' object has no attribute 'attribute_updated'```

As the Database is concerned, do I have to create it first ? Or will the library will create at the first occasion ? After pairing devices, I still didn't get it !

I also got that message:
```
[0xae81:1] Failed ZDO request during endpoint initialization
Traceback (most recent call last):
  File "/var/lib/domoticz/plugins/Domoticz-Zigpy/zigpy/endpoint.py", line 53, in initialize
    self._device.nwk, self._endpoint_id, tries=3, delay=2
  File "/var/lib/domoticz/plugins/Domoticz-Zigpy/zigpy/util.py", line 109, in retry
    r = await func()
  File "/var/lib/domoticz/plugins/Domoticz-Zigpy/zigpy/device.py", line 205, in request
    self.nwk, dst_ep, cluster
zigpy.exceptions.DeliveryError: [0xae81:0:0x0004]: Message send failure
Failed ZDO request during device initialization
Traceback (most recent call last):
  File "/var/lib/domoticz/plugins/Domoticz-Zigpy/zigpy/device.py", line 107, in _initialize
    epr = await self.zdo.Active_EP_req(self.nwk, tries=3, delay=2)
  File "/var/lib/domoticz/plugins/Domoticz-Zigpy/zigpy/util.py", line 109, in retry
    r = await func()
  File "/var/lib/domoticz/plugins/Domoticz-Zigpy/zigpy/device.py", line 205, in request
    self.nwk, dst_ep, cluster
zigpy.exceptions.DeliveryError: [0x5616:0:0x0005]: Message send failure
```

pipiche38, 2020-08-11:
Ok going from errors to errors ....

Here is my setup:

python3.7.7
HW is a USB Zigate (I'm using zigpy-zigate) libs

When switching to dev branch of zigpy-zigate  and zigpy on master branche I'm getting 

```Error on asyncio.run: Can't instantiate abstract class ControllerApplication with abstract methods probe```

When switching to dev branch of zigpy-zigate  and zigpy on dev branche I'm getting  an other error
```
  File "/var/lib/domoticz/plugins/Domoticz-Zigpy/zigpy_zigate/types.py", line 132, in <module>
    class NWK(zigpy.types.HexRepr, uint16_t):
AttributeError: module 'zigpy.types' has no attribute 'HexRepr'
```

pipiche38, 2020-08-11:
One more comment, I wanted to use the Quirks library but this looks ZHA specific, isn't ?

For instance, trying to pair a Legrand Dimer  (here are the logs ), I beleive the quirks for the Device was'nt found !

``
Aug 11 13:23:14 rasp domoticz[18070]: DEBUG:zigpy.endpoint:[0xbcac:1] Manufacturer:  Legrand
Aug 11 13:23:14 rasp domoticz[18070]: DEBUG:zigpy.endpoint:[0xbcac:1] Model:  Dimmer switch w/o neutral
Aug 11 13:23:14 rasp domoticz[18070]: DEBUG:zigpy.quirks.registry:Checking quirks for  Legrand  Dimmer switch w/o neutral (00:04:74:00:00:82:a5:4f)
Aug 11 13:23:14 rasp domoticz[18070]: DEBUG:zigpy_zigate.uart:Frame received: 800200596b0001040000010102bcac020000180801040000421f204c656772616e64000000
```

Adminiuga, 2020-08-11:
It's not. Depending on what are you trying to do you may not need it.

pipiche38, 2020-08-11:
> It's not. Depending on what are you trying to do you may not need it.

Where , how should I install it ?

Adminiuga, 2020-08-11:
https://pypi.org/project/zha-quirks/

pipiche38, 2020-08-11:
> https://pypi.org/project/zha-quirks/

Thanks I have it, but still I do not see the library taking it , this is why I was asking if there are any specific config/setup to be done when using it outside of ZHA!

Adminiuga, 2020-08-11:
you need to install it and you need to `import zhaquirks` prior Zigpy restores its application state.

Hedda, 2020-08-11:
As @sanyatuning already mentioned to @iftahgi in https://github.com/zigpy/zigpy-cc/issues/61 Be sure to check out the ZHA component code in Home Assistant at https://github.com/home-assistant/core/tree/dev/homeassistant/components/zha as its code is currently the best reference application for how-to utilize zigpy and its radio libraries inside a other projects/application.

@sanyatuning also has a tip to checkout https://github.com/zigpy/zigpy-cc/blob/master/test.py as he uses that to test zigpy-cc without Home Assistant (check git history for this test.py code file in the zigpy-cc project).

Also, if I understand it correctly, the zigpy library was originally a refactored fork from the [bellows](https://github.com/zigpy/bellows) library as it was split into two to separate duties, however even though the bellows was refactored to use zigpy, the parts moved to zigpy is still left in bellows as a legacy.

Therefore a tip can be to in addition look for non-HA standalone application projects are still using bellows without zigpy? As unlike the other radio libraries for zigpy, bellows can still (or at least) could be used without the zigpy library. As well as look for non-HA standalone application projects using zigpy of course.

I believe that those non-HA standalone application projects include but are might not be limited to [ZigCoHTTP ](https://github.com/daniel17903/ZigCoHTTP), [qzig](https://github.com/Seluxit/qzig), and the [hue-thief](https://github.com/vanviegen/hue-thief). 

Again there might also be other non-HA standalone application projects out there as well that use zigpy or bellows as a library. Publicly we can only see in the "used by" dependency graphs those public project that is listed on GitHub as using zigpy or bellows as dependencies: 

https://github.com/zigpy/bellows/network/dependents?package_id=UGFja2FnZS01MDI3Mjc2Mg%3D%3D

https://github.com/zigpy/zigpy/network/dependents?package_id=UGFja2FnZS01MjcyNjUwNg%3D%3D

Hedda, 2020-08-11:
> HW is a USB Zigate (I'm using zigpy-zigate) libs

@pipiche38 Tip on the hardware-side is that you might want to consider buying a Silicon Labs Zigbee adapter like example the latest [Elelabs Zigbee USB adapter](https://elelabs.com/products/elelabs-usb-adapter.html) or probably better today is the new [Sonoff ZBBridge with Tasmota](https://www.digiblur.com/2020/07/how-to-use-sonoff-zigbee-bridge-with.html) (for use with bellows) as an addition to also buying a newer Texas Instruments like the [LAUNCHXL-CC26X2R1](https://www.ti.com/tool/LAUNCHXL-CC26X2R1) dev board or the [slaesh CC2652RB stick](https://slae.sh/projects/cc2652/) adapters. 

Reason for this is that I know ZiGate support is experimental and has only ever had one developer working on it, while there have over the years been more developers working on the [bellows](https://github.com/zigpy/bellows) library for Silabs support and [zigpy-cc](https://github.com/zigpy/zigpy-cc) + [zigpy-znp](https://github.com/zha-ng/zigpy-znp) libraries for TI support. Currently, there might in release be more than one developer each working actively on those as well but there are at least many more developers out there with those two types of hardware than there are those with ZiGate hardware.

I think that it awesome that zigpy support multiple adapters from different manufacturers, but if you are going develop a brand new standalone implementation of zigpy then you probably want to be using the same type of hardware as most active developers are.

pipiche38, 2020-08-11:
> @pipiche38 Tip on the hardware-side is that you might want to consider buying a Silicon Labs Zigbee adapter like example the latest [Elelabs Zigbee USB [...]

For now I'm the developper of the Domoticz plugin for Zigate. I do consider having a zigpy plugin for Domoticz.
I known pretty well the ZiGate and the Zigbee 3.0 stack available there (which I know is not used by zigpy). 
So I'm not right now ready to invest in other materials

puddly, 2020-08-11:
@pipiche38 Here's an example that actually works with a Xiaomi motion sensor. Make sure to poke the pairing button every second or so to keep the sensor awake while it is joining.

```Python
import asyncio
import logging

import coloredlogs
coloredlogs.install(milliseconds=True, level=logging.DEBUG)

import zhaquirks

# There are many different radio libraries but they all have the same API
from zigpy_znp.zigbee.application import ControllerApplication

LOGGER = logging.getLogger(__name__)

class MainListener:
    """
    Contains callbacks that zigpy will call whenever something happens.
    Look for `listener_event` in the Zigpy source or just look at the logged warnings.
    """

    def __init__(self, application):
        self.application = application

    def device_initialized(self, device, *, new=True):
        """
        Called at runtime after a device's information has been queried.
        I also call it on startup to load existing devices from the DB.
        """
        LOGGER.info("Device is ready: new=%s, device=%s", new, device)

        for ep_id, endpoint in device.endpoints.items():
            # Ignore ZDO
            if ep_id == 0:
                continue

            # You need to attach a listener to every cluster to receive events
            for cluster in endpoint.in_clusters.values():
                # The context listener passes its own object as the first argument
                # to the callback
                cluster.add_context_listener(self)

    def attribute_updated(self, cluster, attribute_id, value):
        # Each object is linked to its parent (i.e. app > device > endpoint > cluster)
        device = cluster.endpoint.device

        LOGGER.info("Received an attribute update %s=%s on cluster %s from device %s",
            attribute_id, value, cluster, device)

async def main():
    app = await ControllerApplication.new(
        config=ControllerApplication.SCHEMA({
            "database_path": "/tmp/zigbee.db",
            "device": {
                "path": "/dev/cu.usbmodemL1100H861",
            }
        }),
        auto_form=True,
        start_radio=True,
    )

    listener = MainListener(app)
    app.add_listener(listener)

    # Have every device in the database fire the same event so you can attach listeners
    for device in app.devices.values():
        listener.device_initialized(device, new=False)

    # Permit joins for a minute
    await app.permit(60)
    await asyncio.sleep(60)

    # Run forever
    await asyncio.get_running_loop().create_future()

if __name__ == "__main__":
    asyncio.run(main())
```

@Adminiuga Do you think there's a benefit to hard-coding the node descriptor responses into quirks based solely on the device model name, like the Xiaomi gateway does? It'd make the joining process very fast for battery-powered Xiaomi devices that are known to have only one firmware version available.

Adminiuga, 2020-08-11:
hrm, this would require overriding `initialize()` method for each Xiaomi custom device. Just complicates things in the long run if we ever decide to refactor device joining.

Gamester17, 2020-08-12:
Can I suggested to all (zigpy) developers to also start using some kind of inline code documentation in the source code files?

That way the bulk of any code documentation does not have to be maintained separately (which would normally quickly be outdated) and can instead be generated live at any time from the latest code three using documentation generator tools. 

You would really only have to agree to follow a standard for inline code documentation., which should not be too hard as Doxygen is the de facto standard and do support Python

https://www.doxygen.nl/index.html

CodeDocs will then generate and publish documentation from Doxygen for you, (free to setup with public GitHub repositories).

https://codedocs.xyz/

For example, you can check out the Kodi code documentation generated on codedocs.xyz

https://codedocs.xyz/AlwinEsch/kodi/

https://codedocs.xyz/AlwinEsch/kodi/_doxygen__on__addon.html

There are of course many different standards for inline code documentation and also many documentation generation tools, ex:

https://wiki.python.org/moin/DocumentationTools

pipiche38, 2020-08-13:
I had a look to the various events which can be trigger and while I found device_initialized, device_joined, group_added ..., I do not see any cluster command feedbacks like Configure Reporting Response, Write Attribute Response ...

How those are communicated ?

Hedda, 2020-08-13:
> @pipiche38 Here's an example that actually works with a Xiaomi motion sensor. Make sure to poke the pairing button every second or so to keep the sensor awake while it is joining.
>
> ```python
> import [...]

Maybe scripts like that should be [added to the zigpy repo as a test script](https://github.com/zigpy/zigpy/tree/dev/tests) as examples for independent tests without Home Assistant?

pipiche38, 2020-08-13:
Making progress, but now facing abig issue with Domoticz integration. Getting a Segmentation fault 
on zigpy/appdb.py", line 52 in execute

puddly, 2020-08-13:
What platform and Python distribution are you running? You shouldn't be getting segfaults in a pure Python project, especially in the sqlite module.

pipiche38, 2020-08-13:
> What platform and Python distribution are you running? You shouldn't be getting segfaults in a pure Python project, especially in the sqlite module.

I don't expect that the issue is on the zigpy / sqlite3 side.

Domoticz is not as HA. 
Domoticz is C program, and we benefits from the Python facilities to call python from a C program. In such there is a Framework embedded into Domoticz to launch python applications.

In that context and to coexist with the embedded python framework, I have to do the zigpy part in a dedicated thread (which is not a problem)

From my current investigation the segmentation fault occurred on the execute() method and only when doing the SELECT * from devices statements and other tables as well. But the Table creation and Index creation works without any issues.

To answer your question on platform:
- RPI3B+
- Fedora 31
- python3 .7.8

pipiche38, 2020-08-16:
@puddly , I'm making progress. I have been able to duplicate the Domoticz / plugin issue with the appdb part, by extracting only the sqlite3 related class/method.

If Domoticz guru's troubleshoot and fix the issue, I think that I'm on the good path to develop a Domoticz plugin while using zigpy library.

puddly, 2020-08-17:
That's excellent news. I don't have much experience with embedded Python but one possible thing to check is making sure the version of SQLite that Python might be dynamically linked against is compatible with what you have installed.

pipiche38, 2020-08-17:
> That's excellent news. I don't have much experience with embedded Python but one possible thing to check is making sure the version of SQLite that Python might be dynamically linked against is compatible with what you have installed.

Found the zigpy statement causing the segmentation violation.

Line 118 in appdb.py
`self.execute("PRAGMA user_version = %s" % (DB_VERSION,))`

I'll be doing more investigations during the day,

Hedda, 2020-08-18:
> One more comment, I wanted to use the Quirks library but this looks ZHA specific, isn't ?

@dmulcahey I think this is related to https://github.com/zigpy/zha-device-handlers/issues/327 which is a suggestion rename the zha-device-handlers library and repository to "zigpy-device-handlers" for clarity and consistency as it is not dependant on ZHA so it does not make sense that its current library/repo name seem to reflect that

pipiche38, 2020-08-18:
> > One more comment, I wanted to use the Quirks library but this looks ZHA specific, isn't ?
> 
> @dmulcahey I think this is related to [zigpy/zha-device-handlers#327](https://github.com/zigpy/zha-device-handlers/issues/327) which is a suggestion rename the zha-device-handlers library and repository to "zigpy-device-handlers" for clarity and consistency as it is not dependant on ZHA so it does not make sense that its current library/repo name seem to reflect that

Yep, that was my worries as prefixed with zha. So It think it would be much better to rename  it device-handler without zha

Hedda, 2020-08-18:
> Yep, that was my worries as prefixed with zha. So It think it would be much better to rename it device-handler without zha

My personal suggestion in https://github.com/zigpy/zha-device-handlers/issues/327 is to specifically rename it to "zigpy-device-handlers" instead.

There are now already all these libraries for zigpy where only "zha-device-handlers" and "bellows" do not follow the name-standard for the library and repository names (or 'brand' if you will):

- zigpy https://github.com/zigpy/zigpy/
  - **bellows https://github.com/zigpy/bellows**
  - zigpy-cc https://github.com/zigpy/zigpy-cc
  - zigpy-deconz https://github.com/zigpy/zigpy-deconz
  - zigpy-xbee https://github.com/zigpy/zigpy-xbee
  - zigpy-zigate https://github.com/zigpy/zigpy-zigate
  - zigpy-znp https://github.com/zha-ng/zigpy-znp
  - zigpy-deconz-parser https://github.com/zha-ng/zigpy-deconz-parser
  - **zha-device-handlers https://github.com/zigpy/zha-device-handlers**

IMHO, best would be if both zha-device-handlers, as well as the bellows, has their libraries/repos renamed with "zigpy-*." prefix.

PS: If a library and its repo is renamed then also its PyPI project cand and probably should be renamed as well for consistency.

Gamester17, 2020-08-19:
Not a developer myself but from other project I've seen I believe these are good examples of basic documentation for integrators:

https://github.com/NordicSemiconductor/pynrfjprog/blob/master/README.md

https://github.com/NordicSemiconductor/pc-ble-driver-py/blob/master/README.md

That is, just some very basic information for what the libraries are intended to be used for and minimum step-by-step documentation on how-to used started with the development of a brand new integration to a generic application when starting from scratch :)

Gamester17, 2020-08-20:
> Not a developer myself but some other projects I've seen I believe these are good examples of basic documentation for integrators:
> 
> https://github.com/NordicSemiconductor/pynrfjprog/blob/master/README.md
> 
> https://github.com/NordicSemiconductor/pc-ble-driver-py/blob/master/README.md
> 
> That is, just some very basic information for what the libraries are intended to be used for and minimum step-by-step documentation on how-to used started with the development of a brand new integration to a generic application when starting from scratch :)
> 
> The competing ZigBee Cluster Library by Z-Smart Systems also has more information about packages included in its framework:
> 
> https://github.com/zsmartsystems/com.zsmartsystems.zigbee/blob/master/README.md

zha-quirks "CONTRIBUTING.md" is also good as provides a how-to guide with information and examples for getting started:

https://github.com/zigpy/zha-device-handlers/blob/dev/CONTRIBUTING.md

Maybe the introduction information could be copied from there as it summarizes some relative parts of the Zigbee specification?

Gamester17, 2020-08-21:
@pipiche38 FYI, see https://github.com/zigpy/zigpy/pull/459 as a start.

pipiche38, 2020-08-21:
> @pipiche38 FYI, see #459 as a start.

Yep, I saw it and that is great move.

Gamester17, 2020-08-21:
@pipiche38 Also checkout new PR https://github.com/zigpy/zigpy/pull/460 by @puddly

pipiche38, 2020-08-24:
Just a short update, for now I'm stuck as the appdb.py implementation hangs with the embedded python framework in Domoticz. The funny things is that it creates the Database (all CREATE, INDEXES and PRAGMA statements are correctly executed), but at the first SELECT statement it crashes in the sqlite3.so with a segmentation violation.

I suspect some conflict between the embedded Python and the Domoticz which are both doing SQLITE.

I'm thinking to migrate the zigpy/appdb.py from sqlite3 to dict with json. If by chance anyone has already done some work on that matter I would be interested.

puddly, 2020-08-24:
@pipiche38 it's not feature-complete but it works enough for most things:

```Python
import json
import pathlib

import zigpy.types
from zigpy.zdo import types as zdo_t

class JSONPersistingListener:
    def __init__(self, database_file, application):
        self._database_file = pathlib.Path(database_file)
        self._application = application

        self._db = {'devices': {}}

    def device_joined(self, device):
        self.raw_device_initialized(device)

    def device_initialized(self, device):
        # This is passed in a quirks device
        pass

    def device_left(self, device):
        self._db['devices'].pop(str(device.ieee))
        self._dump()

    def raw_device_initialized(self, device):
        self._db['devices'][str(device.ieee)] = self._serialize_device(device)
        self._dump()

    def device_removed(self, device):
        self._db['devices'].pop(str(device.ieee))
        self._dump()

    def attribute_updated(self, cluster, attrid, value):
        self._dump()

    def _serialize_device(self, device):
        obj = {
            'ieee': str(device.ieee),
            'nwk': device.nwk,
            'status': device.status,
            'node_descriptor': None if not device.node_desc.is_valid else list(device.node_desc.serialize()),
            'endpoints': [],
        }

        for endpoint_id, endpoint in device.endpoints.items():
            if endpoint_id == 0:
                continue  # Skip zdo

            endpoint_obj = {}
            endpoint_obj['id'] = endpoint_id
            endpoint_obj['status'] = endpoint.status
            endpoint_obj['device_type'] = getattr(endpoint, 'device_type', None)
            endpoint_obj['profile_id'] = getattr(endpoint, 'profile_id', None)
            endpoint_obj['output_clusters'] = [cluster.cluster_id for cluster in endpoint.out_clusters.values()]
            endpoint_obj['input_clusters'] = [cluster.cluster_id for cluster in endpoint.in_clusters.values()]

            obj['endpoints'].append(endpoint_obj)

        return obj
    
    def _dump(self):
        devices = []

        for device in self._application.devices.values():
            devices.append(self._serialize_device(device))

        existing = self._database_file.read_text()
        new = json.dumps({'devices': devices}, separators=(', ', ': '), indent=4)

        # Don't waste writes
        if existing == new:
            return

        self._database_file.write_text(new)

    def load(self):
        try:
            state_obj = json.loads(self._database_file.read_text())
        except FileNotFoundError:
            logger.warning('Database is empty! Creating an empty one...')
            self._database_file.write_text('')

            state_obj = {'devices': []}

        for obj in state_obj['devices']:
            ieee = zigpy.types.named.EUI64([zigpy.types.uint8_t(int(c, 16)) for c in obj['ieee'].split(':')][::-1])

            assert obj['ieee'] in repr(ieee)

            device = self._application.add_device(ieee, obj['nwk'])
            device.status = zigpy.device.Status(obj['status'])

            if 'node_descriptor' in obj and obj['node_descriptor'] is not None:
                device.node_desc = zdo_t.NodeDescriptor.deserialize(bytearray(obj['node_descriptor']))[0]

            for endpoint_obj in obj['endpoints']:
                endpoint = device.add_endpoint(endpoint_obj['id'])
                endpoint.profile_id = endpoint_obj['profile_id']
                device_type = endpoint_obj['device_type']

                try:
                    if endpoint.profile_id == 260:
                        device_type = zigpy.profiles.zha.DeviceType(device_type)
                    elif endpoint.profile_id == 49246:
                        device_type = zigpy.profiles.zll.DeviceType(device_type)
                except ValueError:
                    pass

                endpoint.device_type = device_type
                endpoint.status = zigpy.endpoint.Status(endpoint_obj['status'])

                for output_cluster in endpoint_obj['output_clusters']:
                    endpoint.add_output_cluster(output_cluster)

                for input_cluster in endpoint_obj['input_clusters']:
                    cluster = endpoint.add_input_cluster(input_cluster)

class JSONControllerApplication(ControllerApplication):
    def __init__(self, config):
        super().__init__(self.SCHEMA(config))

        self._dblistener = JSONPersistingListener(self.config['json_database_path'], self)
        self.add_listener(self._dblistener)
        self._dblistener.load()
        self._dblistener._dump()

```

Example JSON database:

```json
{
    "devices": [
        {
            "ieee": "00:15:8d:00:01:b7:b0:42", 
            "nwk": 22045, 
            "status": 2, 
            "node_descriptor": [
                2, 
                64, 
                128, 
                55, 
                16, 
                127, 
                100, 
                0, 
                0, 
                0, 
                100, 
                0, 
                0
            ], 
            "endpoints": [
                {
                    "id": 1, 
                    "status": 1, 
                    "device_type": 24321, 
                    "profile_id": 260, 
                    "output_clusters": [
                        0, 
                        4, 
                        6
                    ], 
                    "input_clusters": [
                        0, 
                        3
                    ]
                }
            ]
        }
    ]
}
```

pipiche38, 2020-08-25:
@puddly, you save me a lot of time and you made my day

Device is paired under Domoticz via ... 

It so confirmed : 
- SQLITE3 was the issue with  Domoticz.
- zigpy-zigate seems to work, and I assumed would be enought for developing the first version of the plugin for Domoticz

This sound pretty great. Questions:

1/ Can we have the JSONPersistingListener supported in the zigpy stack. We can imagine that at a point of time we can select JSON or SQLITE3 as the persistent storage mode ?

2/ This question is more related to Zigbee devices. After almost 3 years in the Zigbee world developing the Zigate plugin for Domoticz, I saw quiet a large number of devices not fully standard. Even if they are ZB3.0 compliant they do not behave as the norm expect. Xiaomi  is one brand, Legrand , Livolo, Konke, Tuya are other examples do you plan to support and manage all specifics behavours ? Where to so see the border between what is done on the zigpy stack and the quirk libraries and what is expected to be done on the above layer (like a gateway application would do ) ?

The first device paired with Domoticz <-> zigpy <-> zigpy-zigate <-> zigate
Aqara Opple Switch

```
{
    "devices": [
        {
            "ieee": "00:15:8d:00:01:c6:1b:bb",
            "nwk": 0,
            "status": 0,
            "node_descriptor": null,
            "endpoints": []
        },
        {
            "ieee": "04:cf:8c:df:3c:79:4d:cb",
            "nwk": 4476,
            "status": 2,
            "node_descriptor": [
                2,
                64,
                132,
                95,
                17,
                127,
                100,
                0,
                0,
                44,
                100,
                0,
                0
            ],
            "endpoints": [
                {
                    "id": 1,
                    "status": 1,
                    "device_type": 261,
                    "profile_id": 260,
                    "output_clusters": [
                        3,
                        6,
                        8,
                        768
                    ],
                    "input_clusters": [
                        0,
                        3,
                        1
                    ]
                },
                {
                    "id": 2,
                    "status": 3,
                    "device_type": null,
                    "profile_id": null,
                    "output_clusters": [],
                    "input_clusters": []
                },
                {
                    "id": 3,
                    "status": 3,
                    "device_type": null,
                    "profile_id": null,
                    "output_clusters": [],
                    "input_clusters": []
                },
                {
                    "id": 4,
                    "status": 3,
                    "device_type": null,
                    "profile_id": null,
                    "output_clusters": [],
                    "input_clusters": []
                },
                {
                    "id": 5,
                    "status": 3,
                    "device_type": null,
                    "profile_id": null,
                    "output_clusters": [],
                    "input_clusters": []
                },
                {
                    "id": 6,
                    "status": 3,
                    "device_type": null,
                    "profile_id": null,
                    "output_clusters": [],
                    "input_clusters": []
                }
            ]
        }
    ]
```

Hedda, 2020-08-25:
> It so confirmed :
>
> * SQLITE3 was the issue with  Domoticz.
> * zigpy-zigate seems to work, and I assumed would be enought for developing the first version of the plugin for Domoticz
>
> This sound pretty [...]

@pipiche38 Do you think that a Domoticz developer can now be convinced to fix SQLite issues inside Domoticz so it can be used?

> 2/ This question is more related to Zigbee devices. After almost 3 years in the Zigbee world developing the Zigate plugin for Domoticz, I saw quiet a large number of devices not fully standard. Even [...]

@dmulcahey I understand this is what quirks / device-handlers is for or are there also workarounds in HA's ZHA as well?

https://github.com/zigpy/zha-device-handlers/blob/dev/CONTRIBUTING.md

https://github.com/zigpy/zha-device-handlers/blob/dev/README.md

pipiche38, 2020-08-25:
@puddly sorry to come with basic question, but as now, I have paired and receiving messages, I'm not sure on how to process  such messages

provided by
```
def attribute_updated(self, cluster, attribute_id, value):
    device = cluster.endpoint.device
    Domoticz.Log("Received an attribute update %s=%s on cluster %s from device %s/%s" %( attribute_id, value, cluster, device, device._ieee) )
```

`Received an attribute update 0=bitmap8.1 on cluster <zhaquirks.xiaomi.OccupancyCluster object at 0xab1938e0> from device <zhaquirks.xiaomi.aqara.motion_aq2.MotionAQ2 object at 0xac2af1c0>`

In other terms how can I decode : 
`bitmap8.1 `into the real value ?
`cluster <zhaquirks.xiaomi.OccupancyCluster `into the real cluster number which is 0x0406 ?

puddly, 2020-08-25:
1. > Can we have the JSONPersistingListener supported in the zigpy stack. We can imagine that at a point of time we can select JSON or SQLITE3 as the persistent storage mode ?

    A configurable database class would be relatively simple to implement but I don't think adding this specific class to zigpy would be all that helpful (or reduce the maintenance burden). I only wrote that database handler so that I could quickly edit device's clusters/endpoints without having to use a SQLite GUI or write queries. I don't think more than a couple of users really care what storage format the device database uses (and I believe a few external tools directly query the SQLite database) so for now, copy/pasting it into your library seems like a workable solution.

2. Zigpy itself handles low-level spec deviations (e.g. https://github.com/zigpy/zigpy/pull/457#issuecomment-673758818) but the interface for handling superficial translation, like standardizing Xiaomi's proprietary attribute reports into "virtual" clusters, is done through https://github.com/zigpy/zha-device-handlers/ (which has no dependencies on ZHA). I believe the intended effect is that whatever library is using the zigpy stack can just assume every device is standards compliant once it has been initialized and the appropriate quirk in place.

3. [According to the spec](https://zigbeealliance.org/wp-content/uploads/2019/12/07-5123-06-zigbee-cluster-library-specification.pdf#page=296), the occupancy cluster's `Occupancy` attribute (`0x0000`) has a bitmap value:

   > ### 4.8.2.2.1.1 *Occupancy* Attribute
   > The *Occupancy* attribute is a bitmap. Bit 0 specifies the sensed occupancy as follows: 1 =    occupied, 0 = unoccupied. All other bits are reserved.

   `t.bitmap8` is an integer type so you'll just have to check if bit 0 is set:

  ```Python
   In [5]: bool(t.bitmap8(1) & 0b00000001)  # you can also use `& 1`, I'm just being verbose
   Out[5]: True

   In [5]: bool(t.bitmap8(1) & 0b00000010)
   Out[5]: False
   ```

    Ideally we would name the bitmap's bits within zigpy so the first bit would be called `Occupancy` or something.

4. Look at `cluster.cluster_id`.

pipiche38, 2020-08-25:
Thanks. on how to manipulate the bitmap that is ok. 

My question was more does that mean, that I need at the upper level to know what is the Attribute type in order to manipulate it ? 

Now , I understand the issue that zigpy have with the Konke switch where on 0x0006/0x0000 it reports something else than a Bool (the zigpy stack expect a Bool from the standard, but Konke decided otherwise).

thanks for the cluster info

puddly, 2020-08-25:
If I'm understanding your question correctly, zigpy doesn't do any high-level translation like pulling apart a bitmap into individual bits (a lot of sensors are more complicated than they seem at first glance, like the [IAS `ZoneStatus` attribute](https://zigbeealliance.org/wp-content/uploads/2019/12/07-5123-06-zigbee-cluster-library-specification.pdf#page=458)). ZHA's approach is to support individual clusters, which are extracted from a device's endpoints once it is initialized.

However, naming the bitmap fields is something I'd like to do in the future to make them easier to work with:

```Python
In [1]: import zigpy.types as t

In [2]: class OccupancyAttribute(t.bitmap8):
  ...     Occupancy = 0x01
  ...     Reserved2 = 0x02
  ...     Reserved3 = 0x04
  ...     Reserved4 = 0x08
  ...     Reserved5 = 0x10
  ...     Reserved6 = 0x20
  ...     Reserved7 = 0x40
  ...     Reserved8 = 0x80
  ...

In [3]: attr = OccupancyAttribute(0b10000001)

In [4]: attr
Out[4]: <OccupancyAttribute.Reserved8|Occupancy: 129>

In [5]: attr & OccupancyAttribute.Occupancy
Out[5]: <OccupancyAttribute.Occupancy: 1>

In [6]: attr & OccupancyAttribute.Reserved2
Out[6]: <OccupancyAttribute.0: 0>

In [7]: OccupancyAttribute._member_map_
Out[7]:
{'Occupancy': <OccupancyAttribute.Occupancy: 1>,
 'Reserved2': <OccupancyAttribute.Reserved2: 2>,
 'Reserved3': <OccupancyAttribute.Reserved3: 4>,
 'Reserved4': <OccupancyAttribute.Reserved4: 8>,
 'Reserved5': <OccupancyAttribute.Reserved5: 16>,
 'Reserved6': <OccupancyAttribute.Reserved6: 32>,
 'Reserved7': <OccupancyAttribute.Reserved7: 64>,
 'Reserved8': <OccupancyAttribute.Reserved8: 128>}
```

This might make it possible for you to not have to do *all* of the parsing within your application code.

pipiche38, 2020-09-01:
Just a quick update on my end.

The good news.

Based on the PersistentDD provided by @puddly , I have now a "working" POC based on a xiaomi Motion sensor

When the plugin is started from Domoticz, it then create the corresponding widget in Domoticz, and then will update it when there is a motion detected and/or a Lux variation is detected

[image]
[image]

The so not good news, nobody at Domoticz side is interested so far to dive into the SQLITE3 problem.

Here is the thread on the Domoticz forum:
https://www.domoticz.com/forum/viewtopic.php?f=65&t=33642

I have open an issue on Domoticz side:
https://github.com/domoticz/domoticz/issues/4312

pipiche38, 2020-09-01:
In regards to #459 I wonder if there is something special can be made for the zigpy-hardware layer.
For instance if I well understood each zigpy-harwarde layer will provide an implementation of the ControllerApplication in regards to the use of the SCHEMA.

In such it would be effecient that for each hw layer we have:

- What are the parameters supported
- For each parameter what are the possible values
- For each parameter what are the impact//effects

pipiche38, 2020-09-01:
@puddly , @Hedda , @Gamester17 

For your information, I have found the root cause of the sqlite3 issue. This is Domoticz which is compiling and linking with its own sqlite3 implementation ( don't ask me why). I have rebuild domoticz and link it with sqlite3 standard shared lib and all works fine.

Today was the day !

In order to move forward, what would you prefer ?
- I create a new Issue in order to track the progress and ask for support time to time
- continue using that one ?

Last but not least, I'll not make big progress during the next 3 weeks, as I'll be out . but for sure we have now make a big step!

closed 2020-09-02 by pipiche38

Gamester17, 2020-09-02:
> Can I suggested to all (zigpy) developers to also start using some kind of inline code documentation in the source code files?
> 
> That way the bulk of any code documentation does not have to be maintained separately (which would normally quickly be outdated) and can instead be generated live at any time from the latest code three using documentation generator tools.
> 
> You would really only have to agree to follow a standard for inline code documentation., which should not be too hard as Doxygen is the de facto standard and do support Python. Doxygen, is a documentation system that supports most programming languages today. It can generate html docs documenting a projects source code, by either extracting special tags from the source code (put there by people wanting to make use of doxygen), or doxygen will otherwise attempt to build the documentation from the existing source code.
> 
> https://www.doxygen.nl/index.html
> 
> CodeDocs will then generate and publish documentation from Doxygen for you, (free to setup with public GitHub repositories).
> 
> https://codedocs.xyz/
> 
> For example, you can check out the Kodi code documentation generated on codedocs.xyz
> 
> https://codedocs.xyz/AlwinEsch/kodi/
> 
> There are of course many different standards for inline code documentation and also many documentation generation tools, ex:
> 
> https://wiki.python.org/moin/DocumentationTools

Copied this suggestion to its own issue here -> https://github.com/zigpy/zigpy/issues/469