zigpy/zigpy #453: AttributeError: module 'zigpy.types' has no attribute 'HexRepr' with dev branch

pipiche38, 2020-08-12:
Looks like HexRepr class definition has been removed from zigpy.types in dev branch.

Does that mean it has to be defined localy where we need to use , or an other class must be used ?

```
class HexRepr:
    def __repr__(self):
        return ("0x{:0" + str(self._size * 2) + "x}").format(self)

    def __str__(self):
        return ("0x{:0" + str(self._size * 2) + "x}").format(self)
```

pipiche38, 2020-08-12:
Ok. I'm lost I'll wait that the guy in charge of the Zigpy_Zigate to update its code!

closed 2020-08-12 by pipiche38

Adminiuga, 2020-08-12:
@pipiche38 Don't wait on him. If you know how to fix it, then submit a PR. Otherwise use pinned versions for zigpy and zigpy-zigate in your code.