zigpy/zigpy #455: Database is not created and all devices are lost after restart

pipiche38, 2020-08-12:
RPI3B+
OS: Fedora
Zigbee HW: Zigate (using zigpy-zigate)

I'm using a simple code (see attached )
[testzigpy.py.txt](https://github.com/zigpy/zigpy/files/5063190/testzigpy.py.txt)

From the logs, pairing is happening, but nothing is created in the folder specified as config.CONF_DATABASE: "/var/lib/domoticz/plugins/Domoticz-Zigpy/Datas/zigpy.db"
The folder exist, but the database not . I guess I don't have to create it and it will be created automatically by the library

Ref: https://github.com/zigpy/zigpy-zigate/issues/41

puddly, 2020-08-12:
Take a look at [my example application in the other issue](https://github.com/zigpy/zigpy/issues/452#issuecomment-672026121). You have to create an instance of `ControllerApplication` by calling its `new` async method:

```Python
async def main( ):

    import zhaquirks  # noqa: F401

    zigpyApp = await ControllerApplication.new(APP_CONFIG, auto_form=True)

    listener = MainListener(zigpyApp)
    zigpyApp.add_listener(listener)

    # Permit joins for a minute
    await zigpyApp.permit(240)
    await asyncio.sleep(240)

    # Just run forever
    await asyncio.get_running_loop().create_future()
```

Also, [`form_network` in zigpy-deconz](https://github.com/zigpy/zigpy-deconz/blob/e9496cf3d3fe21968c36b9b40fb3cf7235682957/zigpy_deconz/zigbee/application.py#L92-L103) doesn't actually utilize any of the passed parameters so there's little point in calling it manually.

pipiche38, 2020-08-12:
This doesn’t work ! Maybe there is an issue with the zigpy-zigate instance !

This is based on master branch of zippy and zippy-zigate

```
python3 testzigpy.py
DEBUG:asyncio:Using selector: EpollSelector
Traceback (most recent call last):
  File "testzigpy.py", line 109, in <module>
    asyncio.run(main())
  File "/usr/lib/python3.7/asyncio/runners.py", line 43, in run
    return loop.run_until_complete(main)
  File "/usr/lib/python3.7/asyncio/base_events.py", line 587, in run_until_complete
    return future.result()
  File "testzigpy.py", line 71, in main
    zigpyApp = await ControllerApplication(APP_CONFIG)
TypeError: object ControllerApplication can't be used in 'await’ expression
```

Here is the code used:

```
async def main( ):

    import zhaquirks  # noqa: F401

    zigpyApp = await ControllerApplication(APP_CONFIG)

    listener = MainListener(zigpyApp)
    zigpyApp.add_listener(listener)

    await zigpyApp.startup(auto_form=False)
    await zigpyApp.form_network( channel=11 )

    # Permit joins for a minute
    await zigpyApp.permit(240)
    await asyncio.sleep(240)

    # Just run forever
    await asyncio.get_running_loop().create_future()
```

puddly, 2020-08-12:
You need to call the `new` classmethod:

```Python
zigpyApp = await ControllerApplication.new(APP_CONFIG, auto_form=True)
                                      ^^^^
```

pipiche38, 2020-08-13:
Much better. Thanks

closed 2020-08-13 by pipiche38