zigpy/zigpy #471: [SUGGESTION] Zigpy developer guide

pipiche38, 2020-09-03:
Here is the summary of a refection I had after trying to build a poc for Domoticz. This is not about documenting the code itself, but more how to start.
Maybe it is just about creating the file, and then having everyone contributing to it.

I would be more than happy to handle and challenge the content.

As a library you need to provide some guide on how-to use such library. I tried with #470, but I don't think we understand each other.

Just have a look to https://www.domoticz.com/wiki/Developing_a_Python_plugin which describe the steps on how to develop a plugin in Domoticz.

What I would expect from a Zigpy Developer Guide are:

1- Concepts
- What are the zigpy principle. What does it do (from pairing a device, to handling of messages).
- The usage of asyncio and the development consequences
- how to receive messages from devices (flow)
   - receiving attributes values
   - receiving commands
- how to send command to devices (flow)
- the concept of listner
- How to deal with the Zigbee radio sub libraries
- how to use quirks

2- My first App using zigpy

- create the PersistentDB and open the zigbee radio (BTW, I don't understand what the 2 are linked together; I would expect (i) PersistentDB (ii) Zigbee radio communication)
- Loading the DB and adding listener
- setting the call back methods

3- API list

4- How to deal with errors

Gamester17, 2020-09-03:
@dmulcahey also already written a very nice guide on how to build a quirk for ZHA Device Handlers which has good info too: https://github.com/zigpy/zha-device-handlers/blob/dev/CONTRIBUTING.md

In addition for reference, comparing zigpy to OpenZWave, which even though is written in C++ and is for Z-Wave instead Zigbee looks in an overview on paper to have the same concept as zigpy, it too has a "Getting Started" introduction which includes both online Developers documentation (generated from source comments by Doxygen) as well as an example application called "MinOZW" (short for 'Minimum OpenZWave') which is based on the library, and the idea is that they use this example application in their guides.

https://github.com/OpenZWave/open-zwave/blob/master/README.md

https://github.com/OpenZWave/open-zwave/blob/master/README.md

## _Getting Started_

_There is an example application based on the library, it is called MinOZW and it is part of the [open-zwave repository](https://github.com/OpenZWave/open-zwave). This currently offers a minimal example. There are also plenty of other examples on the Internet that you can research._

_On Linux, macOS, BSD run "make" to build the library and MinOZW. Optionally, run "make install" to install the library, header and config files system-wide. Run "BUILD=DEBUG make" to make a debug build._

_Online [Developers documentation](http://www.openzwave.com/dev/) is generated from source comments by Doxygen. If you do "make install" and have Doxygen on your system, a description of the api will be in open-zwave/docs/api/html/index.html_ 

_On Windows, install "Visual Studio" then open "open-zwave/cpp/build/windows/vs2010/OpenZWave.sln" to build the library or "open-zwave/cpp/examples/windows/MinOZW/vs2010/MinOZW.sln" to build both library and sample application._

pipiche38, 2020-09-03:
I might be wrong, but from what I read from the zha-device-handers is much more oriented on developing a quirk rather than an application using the full stack.

Again, pointing to existing application is good, but this doesn't give the perspective and the general design, and required to understand and take time to read.

I'm convinced that all the knowledge is there, the point is just to extract it and put it in a user guide.

And this is not about automatic documentation as here I agree reading the code is the best. But you need to get the main principles , the general design and architecture to be able to navigate to the code and the all library

Gamester17, 2020-09-03:
FYI, showcasing the importance of getting started guidance see the introduction on [the homepage of OpenZWave](http://www.openzwave.net) listing of features:

* **_Fully Documented Code Base_**
* **_Includes simple examples to demonstrate how to interface to the code_**

http://www.openzwave.net

With the exception of those missing two points they list [the homepage of OpenZWave](http://www.openzwave.net) only basically the same features zigpy have:

  **_Introduction_**

_Our goal is to create free software library that interfaces with selected Z-Wave PC controllers, allowing anyone to create applications that manipulate and respond to devices on a Z-Wave network, without requiring in-depth knowledge of the Z-Wave protocol._

_This software is currently aimed at Application Developers who wish to incorporate Z-Wave functionality into their applications._

_If you are a looking for applications to control your Z-Wave devices, please consult this [page](http://code.google.com/p/open-zwave/wiki/ProjectsUsingOZW) which lists applications that utilize OpenZWave._

  **_Features_**
* _Developed in C++_
* _Dotnet wrapper for Windows_
* _Portable to most Linux, Mac OSX and Windows Platforms_
* _Works with the multiple ZWave Controllers_
* _ZWave Protocol is abstracted away hiding a lot of the complexity_
* _Works with most switches, dimmers, thermostats, energy monitors, motion sensors, appliance modules,key fobs, door/window * sensors etc (see here for a complete list)_
* S_ecurity related devices such as Locks are Supported_
* _Reverse Engineered the ZWave Protocol_
* _Easy to add support for new devices_
* **_Fully Documented Code Base_**
* **_Includes simple examples to demonstrate how to interface to the code_**

pipiche38, 2020-09-03:
Here is a starting file
https://github.com/pipiche38/Domoticz-Zigpy/blob/master/zigpy_api.md

feel free to add/comment

Hedda, 2020-09-09:
> Here is a starting file
> https://github.com/pipiche38/Domoticz-Zigpy/blob/master/zigpy_api.md
> 
> feel free to add/comment

See that submitted a PR with this file to zigpy now in https://github.com/zigpy/zigpy/pull/477 so have something all can build on. Thank you @pipiche38 That is great!

Hedda, 2020-10-14:
@pipiche38 please also see related replies from @puddly to @Mopele in https://github.com/zha-ng/zigpy-znp/issues/25

Hedda, 2021-09-06:
Perhaps best to keep this open? And instead close https://github.com/zigpy/zigpy/issues/715 which looks to maybe be a duplicate?

closed 2021-09-06 by pipiche38