zigpy/zigpy #475: Migrate asynctest to stdlib's AsyncMock

fabaff, 2020-09-08:
Asynctest doesn't work with Python 3.8 and above.

Asynctest upstream is aware of this breakage and recommends using the built-in AsyncMock instead.

See also Martiusweb/asynctest#144 and Martiusweb/asynctest#126

puddly, 2020-09-08:
What's the recommended approach for mocking coroutines on Python 3.7? Is there any backport of `AsyncMock` other than the one in `asynctest`?

```Python
try:
    # Python 3.8 already has this
    from mock import AsyncMock as CoroutineMock
except ImportError:
    from asynctest import CoroutineMock
```

Adminiuga, 2020-09-09:
we could take HA's core approach and just import `CoroutineMock` and/or `mock` from the local `asyncmock` module
https://github.com/home-assistant/core/blob/dev/tests/async_mock.py

closed 2020-09-16 by Adminiuga