zigpy/zigpy #481: Issue with some device (Sonoff)

zoic21, 2020-09-10:
Hello,

I begin developpement of zigbee for french home automation application. I can make a deamon which send data to my application. I test with Xiaomi aquara and it's perfectly work, but I have an issue with all sonoff device I test (swith, motion, temperature...). For all I can include it but after I don't have any usefull cluster (only Basic).

Since I'm rather new on zigpy, I don't really know where to start to find the problem.

Here the log after including device : 

```
[2020-09-10 20:10:25][DEBUG] : Application frame 36 (trustCenterJoinHandler) received: b'656bfe14d61c004b120001000000'
[2020-09-10 20:10:25][DEBUG] : Received trustCenterJoinHandler frame with [0x6b65, 00:12:4b:00:1c:d6:14:fe, <EmberDeviceUpdate.STANDARD_SECURITY_UNSECURED_JOIN: 1>, <EmberJoinDecision.USE_PRECONFIGURED_KEY: 0>, 0x0000]
[2020-09-10 20:10:25][INFO] : Device 0x6b65 (00:12:4b:00:1c:d6:14:fe) joined the network
[2020-09-10 20:10:25][INFO] : ************************* Device joined : <zigpy.device.Device object at 0x7f76065f90b8>
[2020-09-10 20:10:25][DEBUG] : Send command findKeyTableEntry: (00:12:4b:00:1c:d6:14:fe, True)
[2020-09-10 20:10:25][INFO] : [0x6b65] Requesting 'Node Descriptor'
[2020-09-10 20:10:25][DEBUG] : Tries remaining: 2
[2020-09-10 20:10:25][DEBUG] : [0x6b65] Extending timeout for 0x03 request
[2020-09-10 20:10:25][DEBUG] : Extending timeout for 00:12:4b:00:1c:d6:14:fe/0x6b65
[2020-09-10 20:10:25][DEBUG] : Send command setExtendedTimeout: (00:12:4b:00:1c:d6:14:fe, True)
[2020-09-10 20:10:25][DEBUG] : Sending: b'467921a9212aeba68f884a6eb855933df87e'
[2020-09-10 20:10:25][DEBUG] : Data frame: b'6579a1a9212aeaab867e'
[2020-09-10 20:10:25][DEBUG] : Sending: b'87009f7e'
[2020-09-10 20:10:25][DEBUG] : Application frame 117 (findKeyTableEntry) received: b'ff'
[2020-09-10 20:10:25][DEBUG] : Sending: b'577d5e21a92a2aeba68f884a6eb8559377fc7e'
[2020-09-10 20:10:25][DEBUG] : Data frame: b'767ea1a92a2ac7897e'
[2020-09-10 20:10:25][DEBUG] : Sending: b'8070787e'
[2020-09-10 20:10:25][DEBUG] : Application frame 126 (setExtendedTimeout) received: b''
[2020-09-10 20:10:25][DEBUG] : Send command sendUnicast: (<EmberOutgoingMessageType.OUTGOING_DIRECT: 0>, 0x6B65, EmberApsFrame(profileId=0, clusterId=2, sourceEndpoint=0, destinationEndpoint=0, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY|APS_OPTION_RETRY: 320>, groupId=0, sequence=3), 4, b'\x03ek')
[2020-09-10 20:10:25][DEBUG] : Sending: b'607f21a9602a15d732944a27aa5592099d4e27a8e9cd64ee9604b67e'
[2020-09-10 20:10:25][DEBUG] : Data frame: b'077fa1a9602a15cd328c7e'
[2020-09-10 20:10:25][DEBUG] : Sending: b'8160597e'
[2020-09-10 20:10:25][DEBUG] : Application frame 52 (sendUnicast) received: b'007f'
[2020-09-10 20:10:26][DEBUG] : Data frame: b'177fb1a9112a11b259874a25aa5592499c4eaf1588a59874f1c606e2026ae9bbeb86cc6f0ffd63167e'
[2020-09-10 20:10:26][DEBUG] : Sending: b'82503a7e'
[2020-09-10 20:10:26][DEBUG] : Application frame 69 (incomingMessageHandler) received: b'04000013000000000000000088be656bffff0c00656bfe14d61c004b12008002'
[2020-09-10 20:10:26][DEBUG] : Received incomingMessageHandler frame with [<EmberIncomingMessageType.INCOMING_BROADCAST: 4>, EmberApsFrame(profileId=0, clusterId=19, sourceEndpoint=0, destinationEndpoint=0, options=<EmberApsOption.APS_OPTION_NONE: 0>, groupId=0, sequence=0), 136, -66, 0x6b65, 255, 255, b'\x00ek\xfe\x14\xd6\x1c\x00K\x12\x00\x80']
[2020-09-10 20:10:26][INFO] : ZDO Device announce: 0x6b65, 00:12:4b:00:1c:d6:14:fe
[2020-09-10 20:10:26][INFO] : Device 0x6b65 (00:12:4b:00:1c:d6:14:fe) joined the network
[2020-09-10 20:10:26][DEBUG] : Skip initialization for existing device 00:12:4b:00:1c:d6:14:fe
[2020-09-10 20:10:26][DEBUG] : [0x6b65:zdo] ZDO request ZDOCmd.Device_annce: [0x6B65, 00:12:4b:00:1c:d6:14:fe, 128]
[2020-09-10 20:10:26][DEBUG] : Data frame: b'277fb1a96b2a134fa6944a36aa5592499c4e27abedce67f7b57e'
[2020-09-10 20:10:26][DEBUG] : Sending: b'83401b7e'
[2020-09-10 20:10:26][DEBUG] : Application frame 63 (messageSentHandler) received: b'06fdff0000130000000000000000000000'
[2020-09-10 20:10:26][DEBUG] : Received messageSentHandler frame with [<EmberOutgoingMessageType.OUTGOING_BROADCAST: 6>, 65533, EmberApsFrame(profileId=0, clusterId=19, sourceEndpoint=0, destinationEndpoint=0, options=<EmberApsOption.APS_OPTION_NONE: 0>, groupId=0, sequence=0), 0, <EmberStatus.SUCCESS: 0>, b'']
[2020-09-10 20:10:26][DEBUG] : Unexpected message send notification tag: 0
[2020-09-10 20:10:27][DEBUG] : Data frame: b'377fb1a96b2a15d732944a27aa5592099d4e27d4e9ce672e947e'
[2020-09-10 20:10:27][DEBUG] : Sending: b'8430fc7e'
[2020-09-10 20:10:27][DEBUG] : Application frame 63 (messageSentHandler) received: b'00656b000002000000400100007f040000'
[2020-09-10 20:10:27][DEBUG] : Received messageSentHandler frame with [<EmberOutgoingMessageType.OUTGOING_DIRECT: 0>, 27493, EmberApsFrame(profileId=0, clusterId=2, sourceEndpoint=0, destinationEndpoint=0, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY|APS_OPTION_RETRY: 320>, groupId=0, sequence=127), 4, <EmberStatus.SUCCESS: 0>, b'']
[2020-09-10 20:10:27][DEBUG] : Data frame: b'477fb1a9112a15b25996ca25aa5593499c4fa31688a59874ecc563ec977c7f27ebcd8ecf8fffc77bd5d26bdf407e'
[2020-09-10 20:10:27][DEBUG] : Sending: b'8520dd7e'
[2020-09-10 20:10:27][DEBUG] : Application frame 69 (incomingMessageHandler) received: b'00000002800000000100000184bd656bffff110300656b024080000050a0000000a0000002'
[2020-09-10 20:10:27][DEBUG] : Received incomingMessageHandler frame with [<EmberIncomingMessageType.INCOMING_UNICAST: 0>, EmberApsFrame(profileId=0, clusterId=32770, sourceEndpoint=0, destinationEndpoint=0, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY: 256>, groupId=0, sequence=1), 132, -67, 0x6b65, 255, 255, b'\x03\x00ek\x02@\x80\x00\x00P\xa0\x00\x00\x00\xa0\x00\x00']
[2020-09-10 20:10:27][INFO] : [0x6b65] Node Descriptor: NodeDescriptor(byte1=2, byte2=64, mac_capability_flags=128, manufacturer_code=0, maximum_buffer_size=80, maximum_incoming_transfer_size=160, server_mask=0, maximum_outgoing_transfer_size=160, descriptor_capability_field=0)
[2020-09-10 20:10:27][INFO] : [0x6b65] Discovering endpoints
[2020-09-10 20:10:27][DEBUG] : Tries remaining: 3
[2020-09-10 20:10:27][DEBUG] : [0x6b65] Extending timeout for 0x05 request
[2020-09-10 20:10:27][DEBUG] : Extending timeout for 00:12:4b:00:1c:d6:14:fe/0x6b65
[2020-09-10 20:10:27][DEBUG] : Send command setExtendedTimeout: (00:12:4b:00:1c:d6:14:fe, True)
[2020-09-10 20:10:27][DEBUG] : Sending: b'757c21a92a2aeba68f884a6eb8559351f37e'
[2020-09-10 20:10:27][DEBUG] : Data frame: b'507ca1a92a2a3be37e'
[2020-09-10 20:10:27][DEBUG] : Sending: b'8610be7e'
[2020-09-10 20:10:27][DEBUG] : Application frame 126 (setExtendedTimeout) received: b''
[2020-09-10 20:10:27][DEBUG] : Send command sendUnicast: (<EmberOutgoingMessageType.OUTGOING_DIRECT: 0>, 0x6B65, EmberApsFrame(profileId=0, clusterId=5, sourceEndpoint=0, destinationEndpoint=0, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY|APS_OPTION_RETRY: 320>, groupId=0, sequence=5), 6, b'\x05ek')
[2020-09-10 20:10:27][DEBUG] : Sending: b'067d5d21a9602a15d732944a20aa5592099d4e27aeebcd62ee96b4757e'
[2020-09-10 20:10:27][DEBUG] : Data frame: b'617da1a9602a153241ec7e'
[2020-09-10 20:10:27][DEBUG] : Sending: b'87009f7e'
[2020-09-10 20:10:27][DEBUG] : Application frame 52 (sendUnicast) received: b'0080'
[2020-09-10 20:10:27][DEBUG] : Data frame: b'717db1a96b2a15d732944a20aa5592099d4e272bebce67576f7e'
[2020-09-10 20:10:27][DEBUG] : Sending: b'8070787e'
[... 58 lines cut ...]
[2020-09-10 20:10:28][DEBUG] : Considering <class 'zhaquirks.netvox.z308e3ed.Z308E3ED'>
[2020-09-10 20:10:28][DEBUG] : Fail because device_type mismatch on at least one endpoint
[2020-09-10 20:10:28][DEBUG] : Considering <class 'zhaquirks.osram.a19twhite.A19TunableWhite'>
[2020-09-10 20:10:28][DEBUG] : Fail because endpoint list mismatch: {3} {1}
[2020-09-10 20:10:28][DEBUG] : Considering <class 'zhaquirks.philips.rom001.PhilipsROM001'>
[2020-09-10 20:10:28][DEBUG] : Fail because device_type mismatch on at least one endpoint
[2020-09-10 20:10:28][DEBUG] : Considering <class 'zhaquirks.philips.rwl020.PhilipsRWL020'>
[2020-09-10 20:10:28][DEBUG] : Fail because endpoint list mismatch: {1, 2} {1}
[2020-09-10 20:10:28][DEBUG] : Considering <class 'zhaquirks.philips.rwl021.PhilipsRWL021'>
[2020-09-10 20:10:28][DEBUG] : Fail because endpoint list mismatch: {1, 2} {1}
[2020-09-10 20:10:28][DEBUG] : Considering <class 'zhaquirks.smartthings.multi.SmartthingsMultiPurposeSensor'>
[... cut ...]
[2020-09-10 20:10:28][DEBUG] : Considering <class 'zhaquirks.xbee.xbee_io.XBeeSensor'>
[2020-09-10 20:10:28][DEBUG] : Fail because endpoint list mismatch: {232, 230} {1}
[2020-09-10 20:10:28][DEBUG] : Considering <class 'zhaquirks.xiaomi.mija.smoke.MijiaHoneywellSmokeDetectorSensor'>
[2020-09-10 20:10:28][DEBUG] : Fail because device_type mismatch on at least one endpoint
[2020-09-10 20:10:28][DEBUG] : Considering <class 'bellows.zigbee.application.EZSPCoordinator'>
[2020-09-10 20:10:28][DEBUG] : Fail because device_type mismatch on at least one endpoint
```

Thank in advance.

puddly, 2020-09-10:
Your device reports that it has more than just the Basic cluster:

```
[2020-09-10 20:10:28][INFO] : [0x6b65:1] Discovered endpoint information: SizePrefixedSimpleDescriptor(endpoint=1, profile=260, device_type=0, device_version=0, input_clusters=[0, 3, 1], output_clusters=[6, 3])
```

Specifically, you should be able to use the commands in [the OnOff cluster](https://github.com/zigpy/zigpy/blob/bc1040f7957cc7f14542ade0bdfa0725f3d3bb64/zigpy/zcl/clusters/general.py#L383-L404) (6), like:

```Python
await device.endpoints[1].on_off.toggle()
```

zoic21, 2020-09-10:
Ok but this a Switch (info only no action) ans when I clic on it not new log line on zigpy...

puddly, 2020-09-10:
Oh, my mistake. XIaomi devices automatically send attribute reports, which isn't standard behavior. You probably need to bind to the `on_off` cluster and possibly configure attribute reporting when the device first joins:

```Python
await device.endpoints[1].on_off.bind()
```

Take a look at how the ZHA component of Home Assistant does it: https://github.com/home-assistant/core/blob/fd8a4182d980d5cdcbc23d39049d76fd6a7b4b6e/homeassistant/components/zha/core/channels/base.py#L135-L182

zoic21, 2020-09-11:
Ok perfect I make it work now (like temperature and occupancy).

I want to say zigpy is very great project, I love it.

Thank a lot

closed 2020-09-11 by zoic21