zigpy/zigpy #487: Strange data on neighbours

zoic21, 2020-09-15:
Hello,

I have a small question about neighbours, when I launch neighbours request `status, val = await self.device.zdo.request(zdo_t.ZDOCmd.Mgmt_Lqi_req, idx, tries=3, delay=1)` I got some device which is not in my zigbee network... How it's possible ?

Second question : after restart when I launch a neighbours scan control does not have any neighbours...

My code is base on zha-map, here debug log : 

```
[2020-09-15 10:10:33][DEBUG] : Building topology starting from coordinator
[2020-09-15 10:10:33][DEBUG] : Tries remaining: 3
[2020-09-15 10:10:33][DEBUG] : Send command sendUnicast: (<EmberOutgoingMessageType.OUTGOING_DIRECT: 0>, 0x0000, EmberApsFrame(profileId=0, clusterId=49, sourceEndpoint=0, destinationEndpoint=0, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY|APS_OPTION_RETRY: 320>, groupId=0, sequence=17), 18, b'\x11\x00')
[2020-09-15 10:10:33][DEBUG] : Sending: b'302021a9602a15b259944a14aa5592099d4e27baffcc768bb6f57e'
[2020-09-15 10:10:33][DEBUG] : Data frame: b'0420a1a9602a150f63897e'
[2020-09-15 10:10:33][DEBUG] : Sending: b'8160597e'
[2020-09-15 10:10:33][DEBUG] : Application frame 52 (sendUnicast) received: b'00bd'
[2020-09-15 10:10:33][DEBUG] : Data frame: b'1420b1a9112a15b259a54a25aa1593499cf3d8abedce9874ffd76352257e'
[2020-09-15 10:10:33][DEBUG] : Sending: b'82503a7e'
[2020-09-15 10:10:33][DEBUG] : Application frame 69 (incomingMessageHandler) received: b'0000003100000040010000bdff000000ffff021100'
[2020-09-15 10:10:33][DEBUG] : Received incomingMessageHandler frame with [<EmberIncomingMessageType.INCOMING_UNICAST: 0>, EmberApsFrame(profileId=0, clusterId=49, sourceEndpoint=0, destinationEndpoint=0, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY|APS_OPTION_RETRY: 320>, groupId=0, sequence=189), 255, 0, 0x0000, 255, 255, b'\x11\x00']
[2020-09-15 10:10:33][DEBUG] : [0x0000:zdo] ZDO request ZDOCmd.Mgmt_Lqi_req: [0]
[2020-09-15 10:10:33][DEBUG] : [0x0000:zdo] Unsupported ZDO request:ZDOCmd.Mgmt_Lqi_req
[2020-09-15 10:10:33][DEBUG] : Data frame: b'2420b1a9112a15b259a5ca25aa1593499cf0d8abedce9874bad7638ffc7d9fea4499621fa0a1230b7ad069015323a41c643ba4ead5cfeeccf0563cefa27cee000e4ca9e55b8f98459b1a3e0230a37fa9fb3464bf298b486f0009d8663dafee44c9517e'
[2020-09-15 10:10:33][DEBUG] : Sending: b'83401b7e'
[2020-09-15 10:10:33][DEBUG] : Data frame: b'3420b1a96b2a15b259944a14aa5592099d4e2716ffce67c4c87e'
[2020-09-15 10:10:33][DEBUG] : Application frame 69 (incomingMessageHandler) received: b'0000003180000040010000beff000000ffff471100060003a04daf54bc702f5ee4d0af02008d15000df012000100a04daf54bc702f5e420cd61c004b120091ea12000157a04daf54bc702f5e511d781b004b120064381200018b'
[2020-09-15 10:10:33][DEBUG] : Sending: b'8430fc7e'
[2020-09-15 10:10:33][DEBUG] : Received incomingMessageHandler frame with [<EmberIncomingMessageType.INCOMING_UNICAST: 0>, EmberApsFrame(profileId=0, clusterId=32817, sourceEndpoint=0, destinationEndpoint=0, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY|APS_OPTION_RETRY: 320>, groupId=0, sequence=190), 255, 0, 0x0000, 255, 255, b'\x11\x00\x06\x00\x03\xa0M\xafT\xbcp/^\xe4\xd0\xaf\x02\x00\x8d\x15\x00\r\xf0\x12\x00\x01\x00\xa0M\xafT\xbcp/^B\x0c\xd6\x1c\x00K\x12\x00\x91\xea\x12\x00\x01W\xa0M\xafT\xbcp/^Q\x1dx\x1b\x00K\x12\x00d8\x12\x00\x01\x8b']
[2020-09-15 10:10:33][DEBUG] : Application frame 63 (messageSentHandler) received: b'00000000003100000040010000bd120000'
[2020-09-15 10:10:33][DEBUG] : Received messageSentHandler frame with [<EmberOutgoingMessageType.OUTGOING_DIRECT: 0>, 0, EmberApsFrame(profileId=0, clusterId=49, sourceEndpoint=0, destinationEndpoint=0, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY|APS_OPTION_RETRY: 320>, groupId=0, sequence=189), 18, <EmberStatus.SUCCESS: 0>, b'']
[2020-09-15 10:10:33][DEBUG] : neighbor request Status: Status.SUCCESS. Response: Neighbors(Entries=6, StartIndex=0, NeighborTableList=[Neighbor(PanId=5e:2f:70:bc:54:af:4d:a0, IEEEAddr=00:15:8d:00:02:af:d0:e4, NWKAddr=0xF00D, NeighborType=18, PermitJoining=0, Depth=1, LQI=0), Neighbor(PanId=5e:2f:70:bc:54:af:4d:a0, IEEEAddr=00:12:4b:00:1c:d6:0c:42, NWKAddr=0xEA91, NeighborType=18, PermitJoining=0, Depth=1, LQI=87), Neighbor(PanId=5e:2f:70:bc:54:af:4d:a0, IEEEAddr=00:12:4b:00:1b:78:1d:51, NWKAddr=0x3864, NeighborType=18, PermitJoining=0, Depth=1, LQI=139)])
[2020-09-15 10:10:33][DEBUG] : ******************Neighbors of 84:2e:14:ff:fe:90:6a:fe => [Neighbor(PanId=5e:2f:70:bc:54:af:4d:a0, IEEEAddr=00:15:8d:00:02:af:d0:e4, NWKAddr=0xF00D, NeighborType=18, PermitJoining=0, Depth=1, LQI=0), Neighbor(PanId=5e:2f:70:bc:54:af:4d:a0, IEEEAddr=00:12:4b:00:1c:d6:0c:42, NWKAddr=0xEA91, NeighborType=18, PermitJoining=0, Depth=1, LQI=87), Neighbor(PanId=5e:2f:70:bc:54:af:4d:a0, IEEEAddr=00:12:4b:00:1b:78:1d:51, NWKAddr=0x3864, NeighborType=18, PermitJoining=0, Depth=1, LQI=139)]
[2020-09-15 10:10:34][DEBUG] : Querying next starting at 3
[2020-09-15 10:10:34][DEBUG] : Tries remaining: 3
[2020-09-15 10:10:34][DEBUG] : Send command sendUnicast: (<EmberOutgoingMessageType.OUTGOING_DIRECT: 0>, 0x0000, EmberApsFrame(profileId=0, clusterId=49, sourceEndpoint=0, destinationEndpoint=0, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY|APS_OPTION_RETRY: 320>, groupId=0, sequence=19), 20, b'\x13\x03')
[2020-09-15 10:10:34][DEBUG] : Sending: b'442121a9602a15b259944a14aa5592099d4e27b8f9cc7488eb717e'
[2020-09-15 10:10:34][DEBUG] : Data frame: b'4521a1a9602a150d4aa97e'
[2020-09-15 10:10:34][DEBUG] : Sending: b'8520dd7e'
[2020-09-15 10:10:34][DEBUG] : Application frame 52 (sendUnicast) received: b'00bf'
[2020-09-15 10:10:34][DEBUG] : Data frame: b'5521b1a9112a15b259a54a25aa1593499cf1d8abedce9874ffd5606a8c7e'
[2020-09-15 10:10:34][DEBUG] : Sending: b'8610be7e'
[2020-09-15 10:10:34][DEBUG] : Application frame 69 (incomingMessageHandler) received: b'0000003100000040010000bfff000000ffff021303'
[2020-09-15 10:10:35][DEBUG] : Received incomingMessageHandler frame with [<EmberIncomingMessageType.INCOMING_UNICAST: 0>, EmberApsFrame(profileId=0, clusterId=49, sourceEndpoint=0, destinationEndpoint=0, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY|APS_OPTION_RETRY: 320>, groupId=0, sequence=191), 255, 0, 0x0000, 255, 255, b'\x13\x03']
[2020-09-15 10:10:35][DEBUG] : [0x0000:zdo] ZDO request ZDOCmd.Mgmt_Lqi_req: [3]
[2020-09-15 10:10:35][DEBUG] : [0x0000:zdo] Unsupported ZDO request:ZDOCmd.Mgmt_Lqi_req
[2020-09-15 10:10:35][DEBUG] : Data frame: b'6521b1a9112a15b259a5ca25aa1593499c8ed8abedce9874bad5638fff7d9fea4499621fa0a173b81ed069015323662a643ba4ead5cfeeccf0563cef84f353e2f1baa00950a198459b4d3e0230a37fa9fb34b470cd6eb74b1f09a96b3dafee532b0b7e'
[2020-09-15 10:10:35][DEBUG] : Sending: b'87009f7e'
[2020-09-15 10:10:35][DEBUG] : Application frame 69 (incomingMessageHandler) received: b'0000003180000040010000c0ff000000ffff471300060303a04daf54bc702f5eb463cb02008d1500cfc612000100a04daf54bc702f5e64836bfeffbd1bec9ac412000100a04daf54bc702f5e81d29cfeff6f0d0015351200019c'
[2020-09-15 10:10:35][DEBUG] : Received incomingMessageHandler frame with [<EmberIncomingMessageType.INCOMING_UNICAST: 0>, EmberApsFrame(profileId=0, clusterId=32817, sourceEndpoint=0, destinationEndpoint=0, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY|APS_OPTION_RETRY: 320>, groupId=0, sequence=192), 255, 0, 0x0000, 255, 255, b'\x13\x00\x06\x03\x03\xa0M\xafT\xbcp/^\xb4c\xcb\x02\x00\x8d\x15\x00\xcf\xc6\x12\x00\x01\x00\xa0M\xafT\xbcp/^d\x83k\xfe\xff\xbd\x1b\xec\x9a\xc4\x12\x00\x01\x00\xa0M\xafT\xbcp/^\x81\xd2\x9c\xfe\xffo\r\x00\x155\x12\x00\x01\x9c']
[2020-09-15 10:10:35][DEBUG] : Data frame: b'7521b1a96b2a15b259944a14aa5592099d4e2714f9ce67520b7e'
[2020-09-15 10:10:35][DEBUG] : Sending: b'8070787e'
[2020-09-15 10:10:35][DEBUG] : Application frame 63 (messageSentHandler) received: b'00000000003100000040010000bf140000'
[2020-09-15 10:10:35][DEBUG] : Received messageSentHandler frame with [<EmberOutgoingMessageType.OUTGOING_DIRECT: 0>, 0, EmberApsFrame(profileId=0, clusterId=49, sourceEndpoint=0, destinationEndpoint=0, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY|APS_OPTION_RETRY: 320>, groupId=0, sequence=191), 20, <EmberStatus.SUCCESS: 0>, b'']
[2020-09-15 10:10:35][DEBUG] : neighbor request Status: Status.SUCCESS. Response: Neighbors(Entries=6, StartIndex=3, NeighborTableList=[Neighbor(PanId=5e:2f:70:bc:54:af:4d:a0, IEEEAddr=00:15:8d:00:02:cb:63:b4, NWKAddr=0xC6CF, NeighborType=18, PermitJoining=0, Depth=1, LQI=0), Neighbor(PanId=5e:2f:70:bc:54:af:4d:a0, IEEEAddr=ec:1b:bd:ff:fe:6b:83:64, NWKAddr=0xC49A, NeighborType=18, PermitJoining=0, Dept [... cut ...]
[2020-09-15 10:10:35][DEBUG] : 00:12:4b:00:1c:d6:14:fe (eWeLink WB01) was not found in the neighbours tables
[2020-09-15 10:10:35][DEBUG] : 08:6b:d7:ff:fe:20:23:fd (IKEA of Sweden TRADFRI on/off switch) was not found in the neighbours tables
[2020-09-15 10:10:35][DEBUG] : 00:04:74:00:00:92:70:70 ( Legrand  Mobile outlet) was not found in the neighbours tables
```

I suppose I don't understand somethink (maybe philips hue light can be a neighbours even if is not on zigbee network ?)

Thank in advance

Adminiuga, 2020-09-15:
> I got some device which is not in my zigbee network... How it's possible

The device reporting that neighbor thinks is has that neighbor.  Depending on device_type -- Router or End Device, could be a child device which wasn't properly aged out. If it is a router, then it could be that `zigpy` is not aware of the device.

Does that neighbour has a proper EUI64 address? Are you sure you never had that device connected and it is not on your network?

Adminiuga, 2020-09-15:
> Second question : after restart when I launch a neighbours scan control does not have any neighbours...

zha_map scans are not persistent across reboots.

zoic21, 2020-09-16:
Yes device have propre EUI64 adress. It can be old device inclusion maybe... How can I force remove those devices ?

Adminiuga, 2020-09-16:
what is the device type?

zoic21, 2020-09-17:
It's end device, to remove device I use `application.remove(device.ieee)`.

I also noticed that when there are router nodes, the list of neighboring nodes is much more accurate. To test I start without a router node, I have no neighboring node with the controller (however I receive the end device information well), I plug in a router node and everything is fine

zoic21, 2020-09-28:
After update zigpy and all lib (like zha-map) issue gone...

closed 2020-09-28 by zoic21