zigpy/zigpy #495: [REQUEST] Common zigpy radio library API for energy detection scan

Hedda, 2020-09-21:
Good or bad idea to share a common zigpy radio library API / command for energy detection scan and access via zigpy?

Some unified channel energy scanning interface via some common energy scanner tool that could work for all radio libraries?

Related to radio library requests:
* https://github.com/zigpy/bellows/issues/352
* https://github.com/zigpy/zigpy-xbee/issues/112
* https://github.com/zigpy/zigpy-zigate/issues/54
* https://github.com/zigpy/zigpy-deconz/issues/130
* https://github.com/zigpy/zigpy-cc/issues/74

Requesting as @puddly  implementing "energy detection scan" feature in zigpy-znp but other libraries do not yet have feature parity.

https://github.com/zha-ng/zigpy-znp/blob/dev/README.md

# Energy scan
Perform an energy scan to find a quiet Zigbee channel:

```console
$ python -m zigpy_znp.tools.energy_scan /dev/cu.usbmodem14101
Channel energy (mean of 1 / 5):
------------------------------------------------
 + Lower energy is better
 + Active Zigbee networks on a channel may still cause congestion
 + Using 26 in the USA may have lower TX power due to FCC regulations
 + Zigbee channels 15, 20, 25 fall between WiFi channels 1, 6, 11
 + Some Zigbee devices only join networks on channels 15, 20, and 25
------------------------------------------------
 - 11    61.57%  #############################################################
 - 12    60.78%  ############################################################
 - 13    12.16%  ############
 - 14    58.43%  ##########################################################
 - 15    57.65%  #########################################################
 - 16    29.80%  #############################
 - 17    38.82%  ######################################
 - 18    47.06%  ###############################################
 - 19    36.86%  ####################################
 - 20    10.98%  ##########
 - 21    16.47%  ################
 - 22    33.73%  #################################
 - 23    30.59%  ##############################
 - 24    20.39%  ####################
 - 25     5.88%  #####
 - 26*   20.39%  ####################
```

puddly, 2020-09-21:
Bellows does have the command, it just isn't documented: https://github.com/zigpy/bellows/blob/88b9fe9f3a6563808b715b356f7bbb2bfeb7c090/bellows/cli/network.py#L129-L157

A unified `zigpy` command line interface, perhaps as a separate package, would be useful. Even a simple one that calls `form_network`.

pipiche38, 2020-09-21:
THis sound great. I don't for other gateways but as the NXP chipset the ZiGate allows to do an Energy Scan on the gateway, but also on any routers, so it might be good that the API allows to give the NwkId of the device to scan (either 0x0000 as the gateway, or a router)

Adminiuga, 2020-09-21:
Zigpy does energy scan via ZDO command, see `Mgmt_NWK_Update_req` command.
Energy Scan via radio specific calls can be implemented in the radio lib itself. 

A unified command line interface is needed. Maybe split the one from `bellows` into a separate package and allow extension by each individual radio lib, so radio specific commands could be exposed as well.

Adminiuga, 2020-09-21:
[image]

Hedda, 2020-11-30:
> A unified command line interface is needed. Maybe split the one from `bellows` into a separate package and allow extension by each individual radio lib, so radio specific commands could be exposed as well.

For reference, this idea is related to https://github.com/zigpy/zigpy/issues/557

closed 2021-09-07 by github-actions[bot]