zigpy/zigpy #496: Be aware if command is take by device (not in queue)

zoic21, 2020-09-23:
Hello,

Is it possible to know if an attribute is pending writing on device ?
I have a problem with the thermostats where we must give a setpoint delta (setpoint_raise_lower). The user gives the desired setpoint, so I calculate the delta between the current setpoint and the desired one. The concern is if the user sends 2 commands. 
Example:
- thermostat at 18
- user asks for 21
- send a delta to 3
- the user asks again 21
- the write_attributes function has finished well (await) but the thermostat has not yet taken it I still have a setpoint at 18
- return of a delta to 3
- themrostat take +3 and +3 so setpoint is now 24 inplace of 21

I found a temporary fix which instead of taking the real value of the thermostat I take the last setpoint if it is less than 10s but I don't find that great. So I would need a status to know if the thermostat has taken the last delta or not.

Thank in advance

pipiche38, 2020-09-23:
shouldn't you to control if the setpoint has been taken in consideration by the Thermostat. 
Either by doing a Read Attribute and get the Setpoint value to cross-check that it matches what was set.

zoic21, 2020-09-23:
Yes but I don't find a good solution, before send command I to read_attributes (with allow_cache=False) to get current setpoint than calcul delta. Issue is due to command in queue no taken by thermostat so read_attribut is nearly false... So solution is to wait fix time (10s) or check is there is command in queue for device.

Adminiuga, 2020-09-24:
Does writing the desired setpoint work?

zoic21, 2020-09-24:
Yes it's work, I can use that to handle my issue !!!! Thank a lot

closed 2020-09-24 by zoic21