zigpy/zigpy #497: OTA fails and loops, size mismatch?

grischard, 2020-09-23:
I'm getting symptoms very similar to the ones described on https://github.com/Koenkk/zigbee2mqtt/issues/4361.

**How to reproduce:**

1. Purchase, borrow or steal a TRADFRI bulb E27 WS opal 1000lm running firmware 0x20022623
2. Add it to Home Assistant with the ZHA integration
3. Wait

**Expected results:**

The firmware updates to 2.0.029 and there is a blinding light appearing in the sky and there is much rejoicing

**Actual results:**

Firmware update gets attempted every couple of minutes

**Relevant logs**

```log
2020-09-23 13:15:01 DEBUG (MainThread) [zigpy_cc.zigbee.application] Config: OrderedDict([('source_routing', True), ('ezsp_config', OrderedDict([('CONFIG_MAX_END_DEVICE_CHILDREN', 5)])), ('ota', OrderedDict([('ikea_provider', True), ('ledvance_provider', True), ('otau_directory', '/etc/homeassistant/ota')])), ('database_path', '/etc/homeassistant/zigbee.db'), ('device', {'baudrate': 115200, 'flow_control': None, 'path': '/dev/serial/by-id/usb-Texas_Instruments_TI_CC2531_USB_CDC___0X00124B001949BE20-if00'}), ('topology_scan_period', 240), ('network', {'key': None, 'update_id': 0, 'tc_link_key': [(snip)], 'extended_pan_id': None, 'tc_address': None, 'channels': <Channels.CHANNEL_25|CHANNEL_20|CHANNEL_15: 34635776>, 'key_sequence_number': 0, 'channel': 15, 'pan_id': None})])
2020-09-23 13:15:14 DEBUG (MainThread) [zigpy.ota.provider] Downloading http://fw.ota.homesmart.ikea.net/Tradfri_OTA_release_signed_2020_09_08_141305/bin/10038562-2.1-TRADFRI-sy5882-bulb-ws-2.0.029.ota.ota.signed for ImageKey(manufacturer_id=4476, image_type=16900)
2020-09-23 13:15:15 DEBUG (MainThread) [zigpy.ota.provider] Finished downloading 209342 bytes from http://fw.ota.homesmart.ikea.net/Tradfri_OTA_release_signed_2020_09_08_141305/bin/10038562-2.1-TRADFRI-sy5882-bulb-ws-2.0.029.ota.ota.signed for ImageKey(manufacturer_id=4476, image_type=16900) ver None

2020-09-23 13:48:09 DEBUG (MainThread) [zigpy.zcl] [0x15da:1:0x0019] OTA upgrade progress: 100.0
2020-09-23 13:48:09 DEBUG (MainThread) [zigpy.zcl] [0x15da:1:0x0019] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=97 command_id=6>
2020-09-23 13:48:09 DEBUG (MainThread) [zigpy.zcl] [0x15da:1:0x0019] ZCL request 0x0006: [<Status.SUCCESS: 0>, 4476, 16900, 537040419]
2020-09-23 13:48:09 DEBUG (MainThread) [zigpy.zcl] [0x15da:1:0x0019] OTA upgrade_end handler for 'IKEA of Sweden TRADFRI bulb E27 WS opal 1000lm': status=Status.SUCCESS, manufacturer_id=4476, image_type=16900, file_version=537040419
2020-09-23 13:49:42 DEBUG (MainThread) [zigpy.zcl] [0x15da:1:0x0019] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=5 command_id=1>
2020-09-23 13:49:42 DEBUG (MainThread) [zigpy.zcl] [0x15da:1:0x0019] ZCL request 0x0001: [1, 4476, 16900, 537011747, 1]
2020-09-23 13:49:42 DEBUG (MainThread) [zigpy.zcl] [0x15da:1:0x0019] OTA query_next_image handler for 'IKEA of Sweden TRADFRI bulb E27 WS opal 1000lm': field_control=1, manufacture_id=4476, image_type=16900, current_file_version=537011747, hardware_version=1
2020-09-23 13:49:42 DEBUG (MainThread) [zigpy.zcl] [0x15da:1:0x0019] OTA image version: 537040419, size: 208242. Update needed: True
2020-09-23 13:49:42 INFO (MainThread) [zigpy.zcl] [0x15da:1:0x0019] Updating: IKEA of Sweden TRADFRI bulb E27 WS opal 1000lm
```

puddly, 2020-09-23:
Have you tried rebooting the light?

grischard, 2020-09-23:
> Have you tried rebooting the light?

Yes.

Adminiuga, 2020-09-23:
Include the log from zigpy-cc or better switch to zigpy-znp.
Include initial query from the device about the firmware.

What's the current version on the bulb?

puddly, 2020-09-23:
Here's a relevant section of the log (https://gist.github.com/grischard/726db1e594c1834b7e6358b95f2b500e):
```
# Bulb successfully finishing previous upgrade
2020-09-23 15:53:23 INFO (MainThread) [zigpy_cc.zigbee.application] handle_message incomingMsg
2020-09-23 15:53:23 DEBUG (MainThread) [zigpy.zcl] [0x15da:1:0x0019] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=12 command_id=3>
2020-09-23 15:53:23 DEBUG (MainThread) [zigpy.zcl] [0x15da:1:0x0019] ZCL request 0x0003: [0, 4476, 16900, 537040419, 208214, 63, None, None]
2020-09-23 15:53:23 DEBUG (MainThread) [zigpy.zcl] [0x15da:1:0x0019] OTA image_block handler for 'IKEA of Sweden TRADFRI bulb E27 WS opal 1000lm': field_control=0, manufacturer_id=4476, image_type=16900, file_version=537040419, file_offset=208214, max_data_size=63, request_node_addr=Noneblock_request_delay=None
2020-09-23 15:53:23 DEBUG (MainThread) [zigpy.zcl] [0x15da:1:0x0019] OTA upgrade progress: 100.0

2020-09-23 15:53:23 DEBUG (MainThread) [zigpy_cc.zigbee.application] request (0x15DA, 260, 25, 1, 1, 12, b'\x19\x0c\x05\x00|\x11\x04B#\x96\x02 V-\x03\x00\x1c<\x0b\x00 \\\x0b\x00 |\x0b\x00 \x84\x0c\x00 \xcc\x0c\x00 \xbc\r\x00 \x18\x0e\x00 ', False, False)
2020-09-23 15:53:23 DEBUG (MainThread) [zigpy_cc.api] --> SREQ AF dataRequest tsn: 12 {'dstaddr': 5594, 'destendpoint': 1, 'srcendpoint': 1, 'clusterid': 25, 'transid': 12, 'options': 0, 'radius': 30, 'len': 45, 'data': b'\x19\x0c\x05\x00|\x11\x04B#\x96\x02 V-\x03\x00\x1c<\x0b\x00 \\\x0b\x00 |\x0b\x00 \x84\x0c\x00 \xcc\x0c\x00 \xbc\r\x00 \x18\x0e\x00 '}
2020-09-23 15:53:23 DEBUG (MainThread) [zigpy_cc.uart] Send: b'\xfe7$\x01\xda\x15\x01\x01\x19\x00\x0c\x00\x1e-\x19\x0c\x05\x00|\x11\x04B#\x96\x02 V-\x03\x00\x1c<\x0b\x00 \\\x0b\x00 |\x0b\x00 \x84\x0c\x00 \xcc\x0c\x00 \xbc\r\x00 \x18\x0e\x00 \xeb'
2020-09-23 15:53:23 DEBUG (MainThread) [zigpy_cc.uart] Frame received: <UnpiFrame command_type=CommandType.SRSP subsystem=Subsystem.AF command_id=1 data=b'\x00' length=1 fcs=100>
2020-09-23 15:53:23 DEBUG (MainThread) [zigpy_cc.api] <-- SRSP AF dataRequest tsn: None {'status': 0}
2020-09-23 15:53:23 DEBUG (MainThread) [zigpy_cc.uart] Frame received: <UnpiFrame command_type=CommandType.AREQ subsystem=Subsystem.AF command_id=128 data=b'\x00\x01\x0c' length=3 fcs=202>
2020-09-23 15:53:23 DEBUG (MainThread) [zigpy_cc.api] <-- AREQ AF dataConfirm tsn: None {'status': 0, 'endpoint': 1, 'transid': 12}
2020-09-23 15:53:23 DEBUG (MainThread) [zigpy_cc.uart] Frame received: <UnpiFrame command_type=CommandType.AREQ subsystem=Subsystem.ZDO command_id=196 data=b'\xda\x15\x00' length=3 fcs=77>
2020-09-23 15:53:23 DEBUG (MainThread) [zigpy_cc.api] <-- AREQ ZDO srcRtgInd tsn: None {'dstaddr': 0x15DA, 'relaycount': 0, 'relaylist': []}

2020-09-23 15:53:23 DEBUG (MainThread) [zigpy_cc.uart] Frame received: <UnpiFrame command_type=CommandType.AREQ subsystem=Subsystem.AF command_id=129 data=b'\x00\x00\x19\x00\xda\x15\x01\r\x00,\x00\xa30\x99\x00\x00\x0c\x01\r\x06\x00|\x11\x04B#\x96\x02 \xda\x15\x1d' length=32 fcs=113>
2020-09-23 15:53:23 DEBUG (MainThread) [zigpy_cc.api] <-- AREQ AF incomingMsg tsn: None {'groupid': 0, 'clusterid': 25, 'srcaddr': 0x15DA, 'srcendpoint': 1, 'dstendpoint': 13, 'wasbroadcast': 0, 'linkquality': 44, 'securityuse': 0, 'timestamp': 10039459, 'transseqnumber': 0, 'len': 12, 'data': b'\x01\r\x06\x00|\x11\x04B#\x96\x02 '}
2020-09-23 15:53:23 INFO (MainThread) [zigpy_cc.zigbee.application] handle_message incomingMsg
2020-09-23 15:53:23 DEBUG (MainThread) [zigpy.zcl] [0x15da:1:0x0019] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=13 command_id=6>
2020-09-23 15:53:23 DEBUG (MainThread) [zigpy.zcl] [0x15da:1:0x0019] ZCL request 0x0006: [<Status.SUCCESS: 0>, 4476, 16900, 537040419]
2020-09-23 15:53:23 DEBUG (MainThread) [zigpy.zcl] [0x15da:1:0x0019] OTA upgrade_end handler for 'IKEA of Sweden TRADFRI bulb E27 WS opal 1000lm': status=Status.SUCCESS, manufacturer_id=4476, image_type=16900, file_version=537040419

2020-09-23 15:53:23 DEBUG (MainThread) [zigpy_cc.zigbee.application] request (0x15DA, 260, 25, 1, 1, 13, b'\x19\r\x07|\x11\x04B#\x96\x02 \x00\x00\x00\x00\x00\x00\x00\x00', False, False)
2020-09-23 15:53:23 DEBUG (MainThread) [zigpy_cc.api] --> SREQ AF dataRequest tsn: 13 {'dstaddr': 5594, 'destendpoint': 1, 'srcendpoint': 1, 'clusterid': 25, 'transid': 13, 'options': 0, 'radius': 30, 'len': 19, 'data': b'\x19\r\x07|\x11\x04B#\x96\x02 \x00\x00\x00\x00\x00\x00\x00\x00'}
2020-09-23 15:53:23 DEBUG (MainThread) [zigpy_cc.uart] Send: b'\xfe\x1d$\x01\xda\x15\x01\x01\x19\x00\r\x00\x1e\x13\x19\r\x07|\x11\x04B#\x96\x02 \x00\x00\x00\x00\x00\x00\x00\x00A'
2020-09-23 15:53:23 DEBUG (MainThread) [zigpy_cc.uart] Frame received: <UnpiFrame command_type=CommandType.SRSP subsystem=Subsystem.AF command_id=1 data=b'\x00' length=1 fcs=100>
2020-09-23 15:53:23 DEBUG (MainThread) [zigpy_cc.api] <-- SRSP AF dataRequest tsn: None {'status': 0}
2020-09-23 15:53:23 DEBUG (MainThread) [zigpy_cc.uart] Frame received: <UnpiFrame command_type=CommandType.AREQ subsystem=Subsystem.AF command_id=128 data=b'\x00\x01\r' length=3 fcs=203>
2020-09-23 15:53:23 DEBUG (MainThread) [zigpy_cc.api] <-- AREQ AF dataConfirm tsn: None {'status': 0, 'endpoint': 1, 'transid': 13}

[... cut ...]
2020-09-23 15:57:38 DEBUG (MainThread) [zigpy.zcl] [0x15da:1:0x0019] ZCL request 0x0003: [0, 4476, 16900, 537040419, 80, 63, None, None]
2020-09-23 15:57:38 DEBUG (MainThread) [zigpy.zcl] [0x15da:1:0x0019] OTA image_block handler for 'IKEA of Sweden TRADFRI bulb E27 WS opal 1000lm': field_control=0, manufacturer_id=4476, image_type=16900, file_version=537040419, file_offset=80, max_data_size=63, request_node_addr=Noneblock_request_delay=None
2020-09-23 15:57:38 DEBUG (MainThread) [zigpy.zcl] [0x15da:1:0x0019] OTA upgrade progress: 0.0
```

Adminiuga, 2020-09-23:
@puddly actually I can confirm the same with 'IKEA of Sweden TRADFRI bulb E26 WS clear 806lm' on 0.115.2 on zigpy-deconz
It finishes the update, sends End update, bulb resets and rejoins and starts again
```
2020-09-23 11:42:47 DEBUG (MainThread) [zigpy.zcl] [0x6e57:1:0x0019] ZCL deserialize: <ZCLHeader fram
e_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=False disable
_default_response=False> manufacturer=None tsn=89 command_id=3>
2020-09-23 11:42:47 DEBUG (MainThread) [zigpy.zcl] [0x6e57:1:0x0019] ZCL request 0x0003: [0, 4476, 16
901, 587531825, 207400, 63, None, None]
2020-09-23 11:42:47 DEBUG (MainThread) [zigpy.zcl] [0x6e57:1:0x0019] OTA image_block handler for 'IKE
A of Sweden TRADFRI bulb E26 WS clear 806lm': field_control=0, manufacturer_id=4476, image_type=16901
, file_version=587531825, file_offset=207400, max_data_size=63, request_node_addr=Noneblock_request_d
elay=None
2020-09-23 11:42:47 DEBUG (MainThread) [zigpy.zcl] [0x6e57:1:0x0019] OTA upgrade progress: 100.0
2020-09-23 11:42:47 DEBUG (MainThread) [zigpy_deconz.zigbee.application] Sending Zigbee request with 
tsn 89 under 59 request id, data: b'195905007c11054231060523282a0300265500fffffffff00f040188640300000
0000000c003000ac00300fffffffff00f0401b0640300'
2020-09-23 11:42:47 DEBUG (MainThread) [zigpy_deconz.api] 'aps_data_indication' response from <Deconz
Address address_mode=ADDRESS_MODE.NWK_AND_IEEE address=0x6e57>, ep: 1, profile: 0x0104, cluster_id: 0
x0019, data: b'015903007c11054231060523282a03003f'
2020-09-23 11:42:47 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_request (70, 59, 0
, <DeconzAddressEndpoint address_mode=2 address=0x6E57 endpoint=1>, 260, 25, 1, b'\x19Y\x05\x00|\x11\
x05B1\x06\x05#(*\x03\x00&U\x00\xff\xff\xff\xff\xf0\x0f\x04\x01\x88d\x03\x00\x00\x00\x00\x00\x00\xc0\x
03\x00\n\xc0\x03\x00\xff\xff\xff\xff\xf0\x0f\x04\x01\xb0d\x03\x00', 2, 0)

...

2020-09-23 11:42:47 DEBUG (MainThread) [zigpy_deconz.api] APS data indication response: [43, <DeviceS
tate.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|2: 34>, <DeconzAddress address_mode=ADDRESS_MODE.NWK address=
0x0000>, 1, <DeconzAddress address_mode=ADDRESS_MODE.NWK_AND_IEEE address=0x6e57>, 1, 260, 25, b'\x01
Z\x06\x00|\x11\x05B1\x06\x05#', 0, 175, 255, 163, 0, 1, 2, -45]
2020-09-23 11:42:47 DEBUG (MainThread) [zigpy.zcl] [0x6e57:1:0x0019] ZCL deserialize: <ZCLHeader fram
e_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=False disable
_default_response=False> manufacturer=None tsn=90 command_id=6>
2020-09-23 11:42:47 DEBUG (MainThread) [zigpy.zcl] [0x6e57:1:0x0019] ZCL request 0x0006: [<Status.SUC
CESS: 0>, 4476, 16901, 587531825]
2020-09-23 11:42:47 DEBUG (MainThread) [zigpy.zcl] [0x6e57:1:0x0019] OTA upgrade_end handler for 'IKE
A of Sweden TRADFRI bulb E26 WS clear 806lm': status=Status.SUCCESS, manufacturer_id=4476, image_type
=16901, file_version=587531825
2020-09-23 11:42:47 DEBUG (MainThread) [zigpy_deconz.zigbee.application] Sending Zigbee request with 
tsn 90 under 60 request id, data: b'195a077c110542310605230000000000000000'

...
Bulb turns on, the only reason i've noticed

2020-09-23 11:42:55 DEBUG (MainThread) [zigpy_deconz.api] APS data indication response: [43, <DeviceS
tate.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|2: 34>, <DeconzAddress address_mode=ADDRESS_MODE.NWK address=
0xfffd>, 0, <DeconzAddress address_mode=ADDRESS_MODE.NWK_AND_IEEE address=0x6e57>, 0, 0, 19, b'\x81Wn
\xf1J\x95\xfe\xffW\xb4\x14\x8e', 0, 175, 255, 163, 0, 1, 2, -45]
2020-09-23 11:42:55 INFO (MainThread) [zigpy_deconz.zigbee.application] New device joined: 0x6e57, 14
:b4:57:ff:fe:95:4a:f1
2020-09-23 11:42:55 INFO (MainThread) [zigpy.application] Device 0x6e57 (14:b4:57:ff:fe:95:4a:f1) joi
ned the network
2020-09-23 11:42:55 DEBUG (MainThread) [zigpy.application] Skip initialization for existing device 14
:b4:57:ff:fe:95:4a:f1
2020-09-23 11:42:55 DEBUG (MainThread) [zigpy.zdo] [0x6e57:zdo] ZDO request ZDOCmd.Device_annce: [0x6
E57, 14:b4:57:ff:fe:95:4a:f1, 142]
2020-09-23 11:42:55 DEBUG (MainThread) [zigpy_deconz.zigbee.application] Sending Zigbee request with 
tsn 61 under 62 request id, data: b'013d0200'

...

Starts upgrade again:
2020-09-23 11:45:23 DEBUG (MainThread) [zigpy_deconz.api] APS data indication response: [45, <DeviceS
tate.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|2: 34>, <DeconzAddress address_mode=ADDRESS_MODE.NWK address=
0x0000>, 1, <DeconzAddress address_mode=ADDRESS_MODE.NWK_AND_IEEE address=0x6e57>, 1, 260, 25, b'\x01
\x05\x01\x01|\x11\x05B1&\x02!\x01\x00', 0, 175, 255, 163, 0, 1, 4, -43]
2020-09-23 11:45:23 DEBUG (MainThread) [zigpy.zcl] [0x6e57:1:0x0019] ZCL deserialize: <ZCLHeader fram
e_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=False disable
_default_response=False> manufacturer=None tsn=5 command_id=1>
2020-09-23 11:45:23 DEBUG (MainThread) [zigpy.zcl] [0x6e57:1:0x0019] ZCL request 0x0001: [1, 4476, 16
901, 553788977, 1]
2020-09-23 11:45:23 DEBUG (MainThread) [zigpy.zcl] [0x6e57:1:0x0019] OTA query_next_image handler for 'IKEA of Sweden TRADFRI bulb E26 WS clear 806lm': field_control=1, manufacture_id=4476, image_type=16901, current_file_version=553788977, hardware_version=1
2020-09-23 11:45:23 DEBUG (MainThread) [zigpy.zcl] [0x6e57:1:0x0019] OTA image version: 587531825, size: 207438. Update needed: True
2020-09-23 11:45:23 INFO (MainThread) [zigpy.zcl] [0x6e57:1:0x0019] Updating: IKEA of Sweden TRADFRI bulb E26 WS clear 806lm
2020-09-23 11:45:23 DEBUG (MainThread) [zigpy_deconz.zigbee.application] Sending Zigbee request with tsn 5 under 68 request id, data: b'190502007c110542310605234e2a0300'

```

puddly, 2020-09-23:
Is it stuck on the same firmware version? Maybe there's some intermediate firmware we have to upgrade to first?

Adminiuga, 2020-09-23:
and this is what triggered it
```
2020-09-23 11:28:43 DEBUG (MainThread) [zigpy.zcl] [0x6e57:1:0x0019] OTA query_next_image handler for
 'IKEA of Sweden TRADFRI bulb E26 WS clear 806lm': field_control=1, manufacture_id=4476, image_type=1
6901, current_file_version=553788977, hardware_version=1
2020-09-23 11:28:43 DEBUG (MainThread) [zigpy_deconz.api] 'aps_data_indication' response from <Deconz
Address address_mode=ADDRESS_MODE.NWK_AND_IEEE address=0x6e57>, ep: 1, profile: 0x0104, cluster_id: 0
x0019, data: b'011301017c110542312602210100'
2020-09-23 11:28:43 DEBUG (MainThread) [zigpy.ota.provider] Downloading http://fw.ota.homesmart.ikea.
net/Tradfri_OTA_release_signed_2020_09_08_141305/bin/10040611-3.2-TRADFRI-sy5882-unified-2.3.050.ota.
ota.signed for ImageKey(manufacturer_id=4476, image_type=16901)
2020-09-23 11:28:43 DEBUG (MainThread) [zigpy.ota.provider] Finished downloading 208574 bytes from ht
tp://fw.ota.homesmart.ikea.net/Tradfri_OTA_release_signed_2020_09_08_141305/bin/10040611-3.2-TRADFRI-
sy5882-unified-2.3.050.ota.ota.signed for ImageKey(manufacturer_id=4476, image_type=16901) ver None
2020-09-23 11:28:43 DEBUG (MainThread) [zigpy.zcl] [0x6e57:1:0x0019] OTA image version: 587531825, si
ze: 207438. Update needed: True
2020-09-23 11:28:43 INFO (MainThread) [zigpy.zcl] [0x6e57:1:0x0019] Updating: IKEA of Sweden TRADFRI 
bulb E26 WS clear 806lm
```

BTW, my IKEA 5 button remote reports
```
2020-09-23 11:18:58 DEBUG (MainThread) [zigpy.zcl] [0x5b75:1:0x0019] OTA query_next_image handler for
 'IKEA of Sweden TRADFRI remote control': field_control=1, manufacture_id=4476, image_type=4545, curr
ent_file_version=587286065, hardware_version=1
2020-09-23 11:18:58 DEBUG (MainThread) [zigpy.zcl] [0x5b75:1:0x0019] OTA image version: 587286065, si
ze: 182590. Update needed: False
```

And doesn't OTA loop.

I'm not sure what intermediate updates are needed. Unless IKEA screwed up and published wrong firmware. Have you tried firmwares from the test URL?

PS: should we print firmware versions in hex?

grischard, 2020-09-23:
The test URL seems to have the same firmware: http://fw.ota.homesmart.ikea.net/Tradfri_OTA_release_signed_2020_09_08_141305/bin/10038562-2.1-TRADFRI-sy5882-bulb-ws-2.0.029.ota.ota.signed vs http://fw.test.ota.homesmart.ikea.net/global/GW1.0/01.11.051/bin/10038562-2.1-TRADFRI-sy5882-bulb-ws-2.0.029.ota.ota.signed

puddly, 2020-09-23:
Odd. The image headers are identical but the actual firmwares within the images differ.
```python
import requests

from zigpy.ota.image import OTAImage

def get_fw(url):
    data = requests.get(url).content

    assert len(data) > 24
    offset = int.from_bytes(data[16:20], "little")
    size = int.from_bytes(data[20:24], "little")
    assert len(data) > offset + size

    ota_image, _ = OTAImage.deserialize(data[offset : offset + size])

    return ota_image

prod_fw = get_fw('http://fw.test.ota.homesmart.ikea.net/global/GW1.0/01.11.051/bin/10038562-2.1-TRADFRI-sy5882-bulb-ws-2.0.029.ota.ota.signed')
test_fw = get_fw('http://fw.ota.homesmart.ikea.net/Tradfri_OTA_release_signed_2020_09_08_141305/bin/10038562-2.1-TRADFRI-sy5882-bulb-ws-2.0.029.ota.ota.signed')

assert test_fw.header == prod_fw.header
assert test_fw.subelements[0] == prod_fw.subelements[0]
```

Adminiuga, 2020-09-23:
Hrm, maybe should try test firmware???

Adminiuga, 2020-09-23:
I only have these in my prod, but they are older HW versions

```
020-09-23 00:51:36 DEBUG (MainThread) [zigpy.zcl] [0x14d4:1:0x0019] OTA query_next_image handler for 'IKEA of Sweden TRADFRI bulb E26 CWS opal 600lm': field_control=1, manufacture_id=4476, image_type=10241, current_file_version=318846322, hardware_version=1
2020-09-23 07:27:56 DEBUG (MainThread) [zigpy.zcl] [0x4b28:1:0x0019] OTA query_next_image handler for 'IKEA of Sweden TRADFRI bulb E26 W opal 1000lm': field_control=1, manufacture_id=4476, image_type=8449, current_file_version=304170354, hardware_version=1
```

grischard, 2020-09-23:
I'll try the test firmware.

Adminiuga, 2020-09-23:
from the newer hardware i've go TRADFRI bulb E26 WW 806lm by IKEA of Sweden and it also OTA loops. 

But from the looks of it, it seems to be a normal OTA flow, just for some reason the lights reject the firmware

puddly, 2020-09-23:
It seems like the test firmware worked.

@Adminiuga to prevent something like this from continuing to happen, perhaps we should blacklist the affected firmware URLs/hashes? Or maybe keep track of upgrade loops within zigpy? As it stands, people who have these bulbs could have been silently experiencing hundreds of these upgrade loops for the past two weeks.

Adminiuga, 2020-09-23:
i'm open to suggestions. 
don't know if we reliably can track all affected hashes. Probably should track OTA attempt counts.
or have a config section within `ota` with IEEE addresses and pinned firmware version number?

puddly, 2020-09-23:
I'm working on a pull request to keep track of upgrade loops at runtime. The info isn't persisted in the DB but it should at least prevent this from happening more than once during a run.

Adminiuga, 2020-09-23:
I think that would be acceptable. 
Are we going to count only those which go through `OTA upgrade_end handler` ? and how many attempts should we allow?

puddly, 2020-09-23:
Yeah, I'm thinking of disallowing more than one successful upgrade per file. Should it keep track of failures too?

Adminiuga, 2020-09-23:
you are more strict than me. I was thinking about allowing three successful as in got to "OTA upgrade_end_handler" cases. 
Those counters should self reset in some period of time, 24hours?

puddly, 2020-09-23:
I don't think it's necessary to add expiration. If a device "upgraded" with a specific image and then asks to do that exact same upgrade again, why would that second one succeed when the first one failed?

Adminiuga, 2020-09-23:
what if Ikea pushes new fixes image with the same version 🤷

puddly, 2020-09-23:
I'm using the image object itself, not just the header, so that should still work.

pipiche38, 2020-09-23:
> I don't think it's necessary to add expiration. If a device "upgraded" with a specific image and then asks to do that exact same upgrade again, why would that second one succeed when the first one failed?

From my understanding (and I have already seen it) for various reason the device itself, could consider the Load of the new image incorrect (while the image is correct on tghe gateway side).
Several reason I could think of:
- Data corruption during the transfer
- Lost of packets during the transfer

So from my end yes, and 2nd or a 3rd try can be successful!

puddly, 2020-09-23:
Hmm. How about allowing a single image to be sent to a device three times, without expiration? I've never seen a failed OTA upgrade but I think the chances of three happening during the transfer of the exact same image would rule out the possibility of a fourth one ever working.

Adminiuga, 2020-09-23:
device itself requests the packet, so if the packet is lost the device is just going to request it again. It is the device pulling the update, not the coordinator pushing it out, so I don't see lost packet.
Data corruption could happen, although I don't know how likely, cause then it probably would break the packet sig.

Adminiuga, 2020-09-23:
> ould rule out the possibility of a fourth one ever working

I'm fine with that. The only reason i wanted expiration, just in case IKEA publishes the fixed image under the same version number which was failing previously. But i think till that time i'm gonna reboot my instance anyway

pipiche38, 2020-09-23:
Why a device for different reason could consider stopping the upgrade process, in order to restart it later ?
I did see it during some of my test, and didn't find it abnormal. the gateway is only there to serve firmware. The end device is making the request. 

As preventing sending firmware when a successful upgrade was done previously is ok, but I think that should be configurable in the way, that we can still authorise it.

MattWestb, 2020-09-24:
IKEA have signed the firmware but have not encrypting it as most of the vendors is doing. I think its open up for storage and transmission corruption can happens and its only then the device have downloading it in the external flash and checking the crc and then its wrong the device is throwing it.
Very likely its the firmware corrupted on the IKEA server (The updating log / readmy was down for one week ago).
One good thing its only cutting the header of the signing and flashing the rest if need one type of firmware to one device but it can being that the userdata is not being OK and need one dumped from one real device for getting working.

puddly, 2020-09-24:
Is the container format documented anywhere? I was able pick out only the bare minimum fields to grab the size and offset of the contained image but being able to verify image integrity would be very useful.

MattWestb, 2020-09-24:
Stripping the first part i have reading but it was old and the offset is not longer constant "[Firmware images start at offset 0x1f6 and are loaded at 0x4000](https://trmm.net/Ikea)" but its easy finding the start and then flashing it and its working (have comparing one dump after OTA updating and the OTA file). Flashing must being made in raw format (with GDB = efl file = bin file),
The [UG266: Silicon Labs Gecko Bootloader User’s Guide](https://www.silabs.com/documents/public/user-guides/ug266-gecko-bootloader-user-guide.pdf) is the file formal described and [UG162: Simplicity Commander Reference Guide](https://www.silabs.com/documents/public/user-guides/ug162-simplicity-commander-reference-guide.pdf) is how to signing and extracting the GBL files.
I think you can "extracting" the raw firmware with this information then the IKEA have only signing the firmware and not encrypted it. So it souls being one "normal" GBL file that is wrapped with the ikea signing.

puddly, 2020-09-25:
Here's one example:

 - **GOOD OLD VERSION**: http://fw.ota.homesmart.ikea.net/Tradfri_OTA_release_signed_2020_03_18_121911/bin/10046695-1.1-TRADFRI-light-unified-w-2.1.022.ota.ota.signed
 - **BAD NEW VERSION**: http://fw.ota.homesmart.ikea.net/Tradfri_OTA_release_signed_2020_09_08_141305/bin/10046695-1.1-TRADFRI-light-unified-w-2.3.050.ota.ota.signed
 - **GOOD NEW VERSION**: "http://fw.test.ota.homesmart.ikea.net/global/GW1.0/01.11.051/bin/10046695-1.1-TRADFRI-light-unified-w-2.3.050.ota.ota.signed

You can extract the image by stripping both the IKEA OTA container and then the Zigbee OTA container:

```Python
import pathlib

from zigpy.ota.image import OTAImage

firmwares = [
    "last-good/10046695-1.1-TRADFRI-light-unified-w-2.1.022.ota.ota.signed",
    "bad/10046695-1.1-TRADFRI-light-unified-w-2.3.050.ota.ota.signed",
    "good/10046695-1.1-TRADFRI-light-unified-w-2.3.050.ota.ota.signed",
]

for filename in firmwares:
    path = (pathlib.Path('~/zigpy/ikea-firmwares/') / filename).expanduser()
    ikea_container = path.read_bytes()

    ota_offset = int.from_bytes(ikea_container[16:20], "little")
    ota_size = int.from_bytes(ikea_container[20:24], "little")

    image, _ = OTAImage.deserialize(ikea_container[ota_offset:ota_offset + ota_size])

    assert len(image.subelements) == 1
    path.with_suffix('.fw.bin').write_bytes(image.subelements[0])
```

I'm not familiar with the SiLabs ecosystem but Simplicity Commander prints some potentially useful info about each one:

```console
$ commander ebl print last-good/10046695-1.1-TRADFRI-light-unified-w-2.1.022.ota.ota.fw.bin
QObject::connect: No such signal EblHandler::warning(QString)
Found EBL Tag = 0x0000, length  140, [EBL Header]
  Version:     0x0201
  Signature:   0xE350 (Correct)
  Flash Addr:  0x00004000
  AAT CRC:     0x3D437516
  AAT Size:    128 bytes
    HalAppBaseAddressTableType
      Top of Stack:       0x20004F28
      Reset Vector:       0x00031FCD
      Hard Fault Handler: 0x00031E21
      Type:               0x0AA7
      Version:            0x010A
      HalVectorTable:     0x00004200
    Full AAT Size:    172
    Ember Version:    6.3.1.1
    Ember Build:      245
    Timestamp:        0x5D4D140E (Fri Aug 09, 2019 06:34:54 UTC [-0500])
    Image Info String:''
    Image CRC:        0x554E1AD4
    Customer Version: 0x21022631
    Image Stamp:      0x3E9922FE9C1EB46D

Found EBL Tag = 0xFD03, length 1924, [Erase then Program Data]
  Flash Addr: 0x00004080
Found EBL Tag = 0xFD03, length 2052, [Erase then Program Data]
  Flash Addr: 0x00004800

(96 additional tags of the same type and length.)

Found EBL Tag = 0xFD03, length 1288, [Erase then Program Data]
  Flash Addr: 0x00035000
Found EBL Tag = 0xFC04, length    4, [EBL End Tag]
  CRC: 0xFDB46EBA

The CRC of this EBL file is valid (0xdebb20e3)

File has 12 bytes of end padding.
Calculated image stamp matches value found in AAT.
DONE

$ commander ebl print bad/10046695-1.1-TRADFRI-light-unified-w-2.3.050.ota.ota.fw.bin
QObject::connect: No such signal EblHandler::warning(QString)
Found EBL Tag = 0x0000, length  140, [EBL Header]
  Version:     0x0201
  Signature:   0xE350 (Correct)
  Flash Addr:  0x00004000
  AAT CRC:     0xED616DC7
  AAT Size:    128 bytes
    HalAppBaseAddressTableType
      Top of Stack:       0x20004980
      Reset Vector:       0x00032541
      Hard Fault Handler: 0x00032395
      Type:               0x0AA7
      Version:            0x010A
      HalVectorTable:     0x00004200
    Full AAT Size:    172
    Ember Version:    6.3.1.1
    Ember Build:      245
    Timestamp:        0x5F45F4A5 (Wed Aug 26, 2020 05:35:33 UTC [-0500])
    Image Info String:''
    Image CRC:        0x85B68D62
    Customer Version: 0x23050631
    Image Stamp:      0x5032C9232859F9AD

Found EBL Tag = 0xFD03, length 1924, [Erase then Program Data]
  Flash Addr: 0x00004080
Found EBL Tag = 0xFD03, length 2052, [Erase then Program Data]
  Flash Addr: 0x00004800
ERROR: Could not read tag 0xFD03 with length 2052 bytes. EBL invalid (tag offset: 0x00031320)
ERROR: Image stamp calculated in file does not match calculated value.
ERROR:   Calculated: 1EF5DEE18B229EA7
ERROR:      In file: 5032C9232859F9AD
  DONE

$ commander ebl print good/10046695-1.1-TRADFRI-light-unified-w-2.3.050.ota.ota.fw.bin
QObject::connect: No such signal EblHandler::warning(QString)
ERROR: Unknown tag ID 0xEB17.  Can't validate length.
DONE
```

I believe the last one doesn't work because it's in the GBL format. Both EBL and GBL are a `<tag id> <tag length> <payload>` style format so they're trivial to parse and therefore superficially validate.

puddly, 2020-09-25:
Only these EBL/GBL files from the previous, current, and test IKEA releases fail to parse, all of them from the current release:

 - `bad/10005777-6.1-TRADFRI-control-outlet-2.0.024.ota.ota.signed`
 - `bad/10043101-3.1-TRADFRI-dimmer-2.1.024.ota.ota.signed`
 - `bad/10005778-10.1-TRADFRI-onoff-shortcut-control-2.2.010.ota.ota.signed`
 - `bad/10040611-3.2-TRADFRI-sy5882-unified-2.3.050.ota.ota.signed`
 - `bad/10047227-1.2-TRADFRI-cv-cct-unified-2.3.050.ota.ota.signed`
 - `bad/10038562-2.1-TRADFRI-sy5882-bulb-ws-2.0.029.ota.ota.signed`
 - `bad/191100-4.1-TRADFRI-sy5882-driver-ws-2.0.029.ota.ota.signed`
 - `bad/10046695-1.1-TRADFRI-light-unified-w-2.3.050.ota.ota.signed`

Here's the code I'm using:

```Python
def parse_ebl(data):
    assert len(data) % 64 == 0
    orig_data = data

    while data:
        assert len(data) >= 4
        tag = data[:2]
        length = int.from_bytes(data[2:4], "big")
        value = data[:length]

        assert len(value) == length
        data = data[4 + length:]
        yield tag, value

        if tag == b'\xFC\x04':
            # The EBL file should have nothing but padding at the end
            assert not data.strip(b"\xFF")
            break

def parse_gbl(data):
    assert len(data) % 64 == 0

    while data:
        assert len(data) >= 8
        tag = data[:4]
        length = int.from_bytes(data[4:8], "little")
        value = data[:length]

        assert len(value) == length
        data = data[8 + length:]
        yield tag, value
```

MattWestb, 2020-09-25:
I was trying simplicity commander and parsing the firmware from the spring for my RGBW that wos working in deCONZ (copy it from deCONZ download) with 
````
PS C:\Users\Mattias\Documents\ICCA1\Simplicity Commander>  .\commander.exe gbl parse .\10035515-TRADFRI-bulb-cws-1.3.013.ota.ota.signed  --app  1111.gbl -d EFR32MG1P132F256GM32 --force
QObject::connect: No such signal EblHandler::warning(QString)
Parse .ebl format for flash
ERROR: Processing of ebl file .\10035515-TRADFRI-bulb-cws-1.3.013.ota.ota.signed failed (error 0x46)
DONE
````
I getting the same error for all "new" ota files i have downloaded from IKEAs servers so i think SC is complaining about the certificate is not present so this way is not possible verifying the firmware (its only signed mot encrypted). That was sad because if it have working the CRC check of the GBL file have showing if its was corrupt.
I dont have any control outlet in deCONZ (all is on the IKEA GW for christmas lights) so cant easy trying updating the firmware by hand one more time that sould being possible in deCONZ.
Attaching the firmware from deCONZ downloaded 2020-03-31 for comparing.
[410005777-6.1-TRADFRI-control-outlet-2.0.024.ota.ota.zip](https://github.com/zigpy/zigpy/files/5284315/410005777-6.1-TRADFRI-control-outlet-2.0.024.ota.ota.zip)
I think (without knowing as normal) that IKEA was having problem with there back end was down for some weeks and the https was offline and have getting the files corrupted.

Adminiuga, 2020-09-25:
the `10046695-1.1-TRADFRI-light-unified-w-2.3.050.ota.ota.fw.bin` I've downloaded on Jun 30th from Ikea passes your check. Which tells me the corruption is on ikea side or CDN

puddly, 2020-09-25:
What does `commander ebl print` on that file show? The name is identical but it could be that they didn't increment the version numbers for more than a couple of releases.

Adminiuga, 2020-09-25:
the good one
```
063e545e88c3cf1cbc5bde7c31bb95c0e28ce432017b5889949921495bccc68f  10046695-1.1-TRADFRI-light-unified-w-2.3.050.ota.ota.signed
```

The bad one I've just downloaded:
```
4d019f039e4e6ef64f6f896a3a2738894c354060005dd4827062cab1040f4ef8  tests/10046695-1.1-TRADFRI-light-unified-w-2.3.050.ota.ota.signed
```

Both have same file size.
Here's same file from test OTA
```
5a6c844a7c1d048b02a85c8906cd4074b42b5f5a1a5151aa7f131d070f7578b0  10046695-1.1-TRADFRI-light-unified-w-2.3.050.ota.ota.test.signed
```

Lemme run it through the commander

Adminiuga, 2020-09-25:
btw, zigpy won't run on py3.8.6 on windows. giving me same error about `TypeError: object.__new__(Channels) is not safe, use int.__new__()`

Adminiuga, 2020-09-25:
all fail with `ERROR: Unknown tag ID 0x4E47.  Can't validate length.` aren't those GBLs?

puddly, 2020-09-25:
Make sure you extract the firmware subelement from the Zigbee OTA container. I couldn't get `commander` to accept anything else.

Adminiuga, 2020-09-25:
Doh

Adminiuga, 2020-09-25:
Old 2.3.050 good
```
QObject::connect: No such signal EblHandler::warning(QString)
Found EBL Tag = 0x0000, length  140, [EBL Header]
  Version:     0x0201
  Signature:   0xE350 (Correct)
  Flash Addr:  0x00004000
  AAT CRC:     0x007D8AED
  AAT Size:    128 bytes
    HalAppBaseAddressTableType
      Top of Stack:       0x20004980
      Reset Vector:       0x00032541
      Hard Fault Handler: 0x00032395
      Type:               0x0AA7
      Version:            0x010A
      HalVectorTable:     0x00004200
    Full AAT Size:    172
    Ember Version:    6.3.1.1
    Ember Build:      245
    Timestamp:        0x5EE30572 (Fri Jun 12, 2020 04:32:50 Eastern Standard Time [Eastern Standard Time])
    Image Info String:''
    Image CRC:        0x85B68D62
    Customer Version: 0x23050631
    Image Stamp:      0x4828FDBCAEE27761

Found EBL Tag = 0xFD03, length 1924, [Erase then Program Data]
  Flash Addr: 0x00004080
Found EBL Tag = 0xFD03, length 2052, [Erase then Program Data]
  Flash Addr: 0x00004800

(97 additional tags of the same type and length.)

Found EBL Tag = 0xFD03, length  820, [Erase then Program Data]
  Flash Addr: 0x00035800
Found EBL Tag = 0xFC04, length    4, [EBL End Tag]
  CRC: 0x3470B9E6

The CRC of this EBL file is valid (0xdebb20e3)

File has 24 bytes of end padding.
Calculated image stamp matches value found in AAT.
DONE
```

the test one fails with
```
QObject::connect: No such signal EblHandler::warning(QString)
ERROR: Unknown tag ID 0xEB17.  Can't validate length.
DONE
```

and the bad one which was just downloaded:
```
QObject::connect: No such signal EblHandler::warning(QString)
Found EBL Tag = 0x0000, length  140, [EBL Header]
  Version:     0x0201
  Signature:   0xE350 (Correct)
  Flash Addr:  0x00004000
  AAT CRC:     0xED616DC7
  AAT Size:    128 bytes
    HalAppBaseAddressTableType
      Top of Stack:       0x20004980
      Reset Vector:       0x00032541
      Hard Fault Handler: 0x00032395
      Type:               0x0AA7
      Version:            0x010A
      HalVectorTable:     0x00004200
    Full AAT Size:    172
    Ember Version:    6.3.1.1
    Ember Build:      245
    Timestamp:        0x5F45F4A5 (Wed Aug 26, 2020 05:35:33 Eastern Standard Time [Eastern Standard Time])
    Image Info String:''
    Image CRC:        0x85B68D62
    Customer Version: 0x23050631
    Image Stamp:      0x5032C9232859F9AD

Found EBL Tag = 0xFD03, length 1924, [Erase then Program Data]
  Flash Addr: 0x00004080
Found EBL Tag = 0xFD03, length 2052, [Erase then Program Data]
  Flash Addr: 0x00004800
ERROR: Could not read tag 0xFD03 with length 2052 bytes. EBL invalid (tag offset: 0x00031320)
ERROR: Image stamp calculated in file does not match calculated value.
ERROR:   Calculated: 1EF5DEE18B229EA7
ERROR:      In file: 5032C9232859F9AD
DONE
```

Adminiuga, 2020-09-25:
the timestamps are diferent though, but customer version is the same. Why?

maybe hackers are trying to replace OTA images...

puddly, 2020-09-25:
Yeah, they're doing a great job by uploading broken images and making our lives more difficult.

Any ideas for how to get `commander` to print out something similar for GBL files? All of the images from the `test` subdomain parse fine for me but that doesn't mean they're not broken in some other way.

MattWestb, 2020-09-25:
In deCONZ the OTA files is copied to one new file with mfg and device id and version and build number from the metadata in the header, The test version is normally that same a the later released only changed name of the file.
The signing is added in the beginning of the GBL file and also some at the end but the "IKEA" container is the bin file. So if having same build number it sold being the same "bin-file" (I have dumped one flashed OTA file and its 100% flashed from the "ikea container" in the OTA file).

@Adminiuga I think the commander dont like the file then you not have provided the IKEA cert for signing and then getting the error message. If stripping the signing from the OTA file then you have one pure bin file of the firmware (flashing it at 0x4000 and its working) then the firmware is not encrypted.
Is the time stamp in the signing pat and the buld and version is the same its little strange can being hardware corruption or hacked / virus infected. Its from the http or the https servers ?

Adminiuga, 2020-09-25:
http, don't think https works anymore

puddly, 2020-09-25:
HTTPS works for retrieving the individual images but not the main `/feed/` URL. The two images are identical.

MattWestb, 2020-09-25:
In both case hardware or "external" corruption its must going pretty fast for IKEA fixing it buy deleting all files and setting up it with new intact files if not having losing control of there certificate for signing the OTA files. If not possible checking CRC of the firmware i think its best disabling IKEA OTA updates until its being fixed on the server side.
Perhaps IKEA have changing servers for there own GW ?
One more strange thing its the latest release notes its some old fw on the list as updated but with the old firmware version (the control outlet is from first Q 2020).
Perhaps they is very well hacked ??

Adminiuga, 2020-09-25:
I couldn't find anything comparable for gbl command and it is not possible to convert a gbl to ebl

MattWestb, 2020-09-25:
I think you need the certificate that the signed file was signed with for doing it.
I was trying parsing then but was not working only error 0x46 also with one file that have working.
Stripping the signing away as puddly is doing but is not helping verifying the "bin file" in the container if not can checking CRC.

puddly, 2020-09-25:
I'll try to add checksumming support then, at least for EBL files because I know where to find that info in the header. GBL should be similar. Forging a CRC-32 isn't difficult at all but unless someone wants to break open the TRÅDFRI gateway and extract some public key, I don't think it's going to be easy to validate. The GBL/EBL images are all unencrypted and contain no special sections for anything but erasing and writing flash.

I wonder if the signature is valid for the IKEA container (the header is `NGIS`, I've never seen it)? That could indicate whether or not the firmware was damaged before it was signed.

MattWestb, 2020-09-25:
The mfg token its not so easy getting then its in the normally read protected lockbit aira and its the same as in the ICC-1-A moule as in the GW or  its not working.
[LED1836G9.zip](https://github.com/zigpy/zigpy/files/5285493/LED1836G9.zip)
One LED1836G9  with internal flash, user data and lockits plus memory dumped and some photos of "disassembling".

MattWestb, 2020-09-25:
Its one OTA from 2020.06.29 with the same version as in the dumped file.
The internal flash has first the bootloaders at 0x0, 0x800 and the "app" = firmware at 0X4000.

[10046695-1.1-TRADFRI-light-unified-w-2.1.022.ota.ota (1).zip](https://github.com/zigpy/zigpy/files/5285557/10046695-1.1-TRADFRI-light-unified-w-2.1.022.ota.ota.1.zip)

prilly-dev, 2020-10-20:
Seems that there has been some OTA related changes: https://ww8.ikea.com/ikeahomesmart/releasenotes/releasenotes.html

MattWestb, 2020-10-20:
I was applying the GW update earlier today.
I have all the devices updated so no new updates of devices is made (all was updated before the OTA problems on the servers).

Some on reddit have complaining of that the GW is eating bandwidth all the time for some weeks (1.11.47) so i think the IKEA GW is also having problems with broken files on the server.
Other users have problems that some lights is rebooting but dont knowing its failing OTA updates that is doing it.

Was looking on Zigbee alliance and no updated devices is certified only some new and the released GW 1.11.51.

MattWestb, 2020-11-21:
New status of IKEA OTA file download service:
Normal http is working with the "new" updated images for some device that is broken and not new only different packaged.
Normal htpps is down from the end of september.

As of yesterday http://fw.test.ota.homesmart.ikea.net/feed/version_info.json is getting errors.
ZHA log:
```
Logger: zigpy.ota.provider
Source: /usr/local/lib/python3.8/site-packages/zigpy/ota/provider.py:164
First occurred: 9:13:35 AM (1 occurrences)
Last logged: 9:13:35 AM

Trådfri: Couldn't download 'http://fw.test.ota.homesmart.ikea.net/feed/version_info.json': 503/Service Temporarily Unavailable
```
Web Viewer: 
```
503 ERROR
The request could not be satisfied.
The Lambda function associated with the CloudFront distribution is invalid or doesn't have the required permissions. We can't connect to the server for this app or website at this time. There might be too much traffic or a configuration error. Try again later, or contact the app or website owner.
If you provide content to customers through CloudFront, you can find steps to troubleshoot and help prevent this error by reviewing the CloudFront documentation.
Generated by cloudfront (CloudFront)
Request ID: noEt4QhCQWuwPZvy3Ziet5UAWVhSDcxb1KB2lfiPUTgfKdAsk-Dhgw==
```
Cand being one temporery thing with the servers.

If need the files then use this json from 2020.06.29 for downloading the OTA files (looks like all links is working but not all is tasted to day 5 month later) and putting them in your local OTA folder and its working. [version_info.json.txt](https://github.com/zigpy/zigpy/files/5577318/version_info.json.txt) (txt extension added for possible github upload).

Newer file on the HTTP feed is the NCP (hopefully you dont upgrading it with ZHA) nd the new shortcut button that is not released in all markets yet.

If this feed is stop working I have many old to the current version in my deCONZ OTA folder as backup for any reasons.

Interesting wath its going on with our mainstream suppliers OTA updates.
Perhaps is they trying closing the updating from outside there GW then LIDL and company is starting rolling out tuya devices all over europe.

closed 2021-09-07 by github-actions[bot]

Adminiuga, 2021-11-12:
My Ikea 2button remote loops through OTA upgrade again.

MattWestb, 2021-11-12:
What version is it having ?
If running 2.3.075 then you need slashing it  with SWD then its cant flashing the 2.3.079 and rolling back the old bad version.

Adminiuga, 2021-11-12:
yes it is 2.3.075

Adminiuga, 2021-11-12:
is it because it was test firmware? It won't accept the old firmware?

MattWestb, 2021-11-12:
From the bad test feed :-((
All IKE controllers is blocked for downgrading if reading certificate and i have trying 3 typed and is not working.

Last working firmware that puddly have helping extracting and SWD flashing it.
https://github.com/zigpy/zigpy/discussions/660#discussioncomment-1565541
https://github.com/zigpy/zigpy/discussions/660#discussioncomment-1563280

20 minute is fixed !!

2 Z2M user have testing it  and was working for them ;-)

[7 more replies]