zigpy/zigpy #513: Understand IAS Zone (through SNZB-03)

zoic21, 2020-10-02:
Hello,

I try to understand how IAS Zone work through Sonoff SNZB-03 module. I use zigpy with deconz in standalone (no HA) and I can't understand this module. But I think I don't really understand IAS Zone.

This what I do on inclusion : 

- bind IAS Zone cluster ``cluster.bind()``
- send cie_addr ``cluster.write_attributes({"cie_addr": ieee})``
- Test each command to this cluster to see if it's an enroll requested, if it I reply ``cluster.enroll_response(0, 0)``

But with this module I never get enroll requested it's normal ? Module don't work well. First minute after zigpy start it's work but after that module send ZoneStatus.0 then ZoneStatus.Alarm_1 every minute and after 20minutes stop all send.

Are there any commands to send to the module after receiving a message from it? Or maybe my module have an issue (it's a sample module) ? I test also xioami aquara motion and no problem with it.

Here log : 

`````
[2020-10-02 14:03:43][DEBUG] : Frame received: 0x0e43000700aa00
[2020-10-02 14:03:43][DEBUG] : Device state changed response: [<DeviceState.128|APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_INDICATION|2: 170>, 0]
[2020-10-02 14:03:43][DEBUG] : Command Command.aps_data_indication (1, 1)
[2020-10-02 14:03:43][DEBUG] : Send: 0x1736000800010001
[2020-10-02 14:03:43][DEBUG] : Frame received: 0x1c44000c000500028255ffc6
[2020-10-02 14:03:43][DEBUG] : Frame received: 0x1736002700200022020000010210ef01040100050900190f0000000000000000af5f5b335000ae
[2020-10-02 14:03:43][DEBUG] : APS data indication response: [32, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|2: 34>, <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x0000>, 1, <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0xef10>, 1, 260, 1280, b'\x19\x0f\x00\x00\x00\x00\x00\x00\x00', 0, 175, 95, 91, 51, 80, 0, -82]
[2020-10-02 14:03:43][DEBUG] : [0xef10:1:0x0500] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=15 command_id=0>
[2020-10-02 14:03:43][DEBUG] : [0xef10:1:0x0500] ZCL request 0x0000: [<ZoneStatus.0: 0>, <bitmap8.0: 0>, 0, 0]
[2020-10-02 14:03:44][DEBUG] : Frame received: 0x0e38000700aa00
[2020-10-02 14:03:44][DEBUG] : Device state changed response: [<DeviceState.128|APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_INDICATION|2: 170>, 0]
[2020-10-02 14:03:44][DEBUG] : Command Command.aps_data_indication (1, 1)
[2020-10-02 14:03:44][DEBUG] : Send: 0x1737000800010001
[2020-10-02 14:03:44][DEBUG] : Frame received: 0x1737002700200022020000010210ef0104010005090019100001000000000000af575b335000ad
[2020-10-02 14:03:44][DEBUG] : APS data indication response: [32, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|2: 34>, <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x0000>, 1, <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0xef10>, 1, 260, 1280, b'\x19\x10\x00\x01\x00\x00\x00\x00\x00', 0, 175, 87, 91, 51, 80, 0, -83]
[2020-10-02 14:03:44][DEBUG] : [0xef10:1:0x0500] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=16 command_id=0>
[2020-10-02 14:03:44][DEBUG] : [0xef10:1:0x0500] ZCL request 0x0000: [<ZoneStatus.Alarm_1: 1>, <bitmap8.0: 0>, 0, 0]
````

Thank in advance

Adminiuga, 2020-10-02:
See https://zigbeealliance.org/wp-content/uploads/2019/12/07-5123-06-zigbee-cluster-library-specification.pdf Section `8.2.2.2.3`

I'm not 100% positive binding on IAS cluster is needed, that came from the original implementation. an IAS is enrolled either via enroll_request/enroll_response https://github.com/home-assistant/core/blob/576fbbe0be3b684618e453fc0bce1809673663cd/homeassistant/components/zha/core/channels/security.py#L129-L132 or by writting the `cieee_address` attribute

But per your log, since you are receiving cluster command_id == 0x0000, that means it is enrolled correctly. as to why it triggers every second :shrug: try to blank it so it doesn't get any movement?

zoic21, 2020-10-02:
Ok thank I will read documentation. 

I try to hide but same issue : 
```
[2020-10-02 15:22:26][DEBUG] : {"devices":{"00:12:4b:00:1c:d6:0c:42":{"1":{"1280":{"cmd":[{"value":"0","cluster_name":"IAS Zone"},{"value":"ZoneStatus.0","cluster_name":"IAS Zone"}]}}}}}
[2020-10-02 15:22:26][DEBUG] : {"devices":{"00:12:4b:00:1c:d6:0c:42":{"1":{"1280":{"cmd":[{"value":"0","cluster_name":"IAS Zone"},{"value":"ZoneStatus.Alarm_1","cluster_name":"IAS Zone"}]}}}}}
[2020-10-02 15:29:11][DEBUG] : {"devices":{"00:12:4b:00:1c:d6:0c:42":{"1":{"1280":{"cmd":[{"value":"0","cluster_name":"IAS Zone"},{"value":"ZoneStatus.0","cluster_name":"IAS Zone"}]}}}}}
[2020-10-02 15:29:11][DEBUG] : {"devices":{"00:12:4b:00:1c:d6:0c:42":{"1":{"1280":{"cmd":[{"value":"0","cluster_name":"IAS Zone"},{"value":"ZoneStatus.Alarm_1","cluster_name":"IAS Zone"}]}}}}}
[2020-10-02 15:30:17][DEBUG] : {"devices":{"00:12:4b:00:1c:d6:0c:42":{"1":{"1280":{"cmd":[{"value":"0","cluster_name":"IAS Zone"},{"value":"ZoneStatus.0","cluster_name":"IAS Zone"}]}}}}}
[2020-10-02 15:30:17][DEBUG] : {"devices":{"00:12:4b:00:1c:d6:0c:42":{"1":{"1280":{"cmd":[{"value":"0","cluster_name":"IAS Zone"},{"value":"ZoneStatus.Alarm_1","cluster_name":"IAS Zone"}]}}}}}
[2020-10-02 15:31:17][DEBUG] : {"devices":{"00:12:4b:00:1c:d6:0c:42":{"1":{"1280":{"cmd":[{"value":"0","cluster_name":"IAS Zone"},{"value":"ZoneStatus.0","cluster_name":"IAS Zone"}]}}}}}
[2020-10-02 15:31:18][DEBUG] : {"devices":{"00:12:4b:00:1c:d6:0c:42":{"1":{"1280":{"cmd":[{"value":"0","cluster_name":"IAS Zone"},{"value":"ZoneStatus.Alarm_1","cluster_name":"IAS Zone"}]}}}}}
[2020-10-02 15:32:18][DEBUG] : {"devices":{"00:12:4b:00:1c:d6:0c:42":{"1":{"1280":{"cmd":[{"value":"0","cluster_name":"IAS Zone"},{"value":"ZoneStatus.0","cluster_name":"IAS Zone"}]}}}}}
[2020-10-02 15:32:19][DEBUG] : {"devices":{"00:12:4b:00:1c:d6:0c:42":{"1":{"1280":{"cmd":[{"value":"0","cluster_name":"IAS Zone"},{"value":"ZoneStatus.Alarm_1","cluster_name":"IAS Zone"}]}}}}}
[2020-10-02 15:33:19][DEBUG] : {"devices":{"00:12:4b:00:1c:d6:0c:42":{"1":{"1280":{"cmd":[{"value":"0","cluster_name":"IAS Zone"},{"value":"ZoneStatus.0","cluster_name":"IAS Zone"}]}}}}}
[2020-10-02 15:33:19][DEBUG] : {"devices":{"00:12:4b:00:1c:d6:0c:42":{"1":{"1280":{"cmd":[{"value":"0","cluster_name":"IAS Zone"},{"value":"ZoneStatus.Alarm_1","cluster_name":"IAS Zone"}]}}}}}
[2020-10-02 15:34:19][DEBUG] : {"devices":{"00:12:4b:00:1c:d6:0c:42":{"1":{"1280":{"cmd":[{"value":"0","cluster_name":"IAS Zone"},{"value":"ZoneStatus.0","cluster_name":"IAS Zone"}]}}}}}
[2020-10-02 15:34:20][DEBUG] : {"devices":{"00:12:4b:00:1c:d6:0c:42":{"1":{"1280":{"cmd":[{"value":"0","cluster_name":"IAS Zone"},{"value":"ZoneStatus.Alarm_1","cluster_name":"IAS Zone"}]}}}}}
[2020-10-02 15:35:20][DEBUG] : {"devices":{"00:12:4b:00:1c:d6:0c:42":{"1":{"1280":{"cmd":[{"value":"0","cluster_name":"IAS Zone"},{"value":"ZoneStatus.0","cluster_name":"IAS Zone"}]}}}}}
[2020-10-02 15:35:21][DEBUG] : {"devices":{"00:12:4b:00:1c:d6:0c:42":{"1":{"1280":{"cmd":[{"value":"0","cluster_name":"IAS Zone"},{"value":"ZoneStatus.Alarm_1","cluster_name":"IAS Zone"}]}}}}}
[2020-10-02 15:36:21][DEBUG] : {"devices":{"00:12:4b:00:1c:d6:0c:42":{"1":{"1280":{"cmd":[{"value":"0","cluster_name":"IAS Zone"},{"value":"ZoneStatus.0","cluster_name":"IAS Zone"}]}}}}}
[2020-10-02 15:36:21][DEBUG] : {"devices":{"00:12:4b:00:1c:d6:0c:42":{"1":{"1280":{"cmd":[{"value":"0","cluster_name":"IAS Zone"},{"value":"ZoneStatus.Alarm_1","cluster_name":"IAS Zone"}]}}}}}
[2020-10-02 15:37:21][DEBUG] : {"devices":{"00:12:4b:00:1c:d6:0c:42":{"1":{"1280":{"cmd":[{"value":"0","cluster_name":"IAS Zone"},{"value":"ZoneStatus.0","cluster_name":"IAS Zone"}]}}}}}
[2020-10-02 15:37:22][DEBUG] : {"devices":{"00:12:4b:00:1c:d6:0c:42":{"1":{"1280":{"cmd":[{"value":"0","cluster_name":"IAS Zone"},{"value":"ZoneStatus.Alarm_1","cluster_name":"IAS Zone"}]}}}}}
[2020-10-02 15:38:22][DEBUG] : {"devices":{"00:12:4b:00:1c:d6:0c:42":{"1":{"1280":{"cmd":[{"value":"0","cluster_name":"IAS Zone"},{"value":"ZoneStatus.0","cluster_name":"IAS Zone"}]}}}}}
[2020-10-02 15:38:23][DEBUG] : {"devices":{"00:12:4b:00:1c:d6:0c:42":{"1":{"1280":{"cmd":[{"value":"0","cluster_name":"IAS Zone"},{"value":"ZoneStatus.Alarm_1","cluster_name":"IAS Zone"}]}}}}}
[2020-10-02 15:39:23][DEBUG] : {"devices":{"00:12:4b:00:1c:d6:0c:42":{"1":{"1280":{"cmd":[{"value":"0","cluster_name":"IAS Zone"},{"value":"ZoneStatus.0","cluster_name":"IAS Zone"}]}}}}}
[2020-10-02 15:39:23][DEBUG] : {"devices":{"00:12:4b:00:1c:d6:0c:42":{"1":{"1280":{"cmd":[{"value":"0","cluster_name":"IAS Zone"},{"value":"ZoneStatus.Alarm_1","cluster_name":"IAS Zone"}]}}}}}
[2020-10-02 15:40:23][DEBUG] : {"devices":{"00:12:4b:00:1c:d6:0c:42":{"1":{"1280":{"cmd":[{"value":"0","cluster_name":"IAS Zone"},{"value":"ZoneStatus.0","cluster_name":"IAS Zone"}]}}}}}
[2020-10-02 15:40:24][DEBUG] : {"devices":{"00:12:4b:00:1c:d6:0c:42":{"1":{"1280":{"cmd":[{"value":"0","cluster_name":"IAS Zone"},{"value":"ZoneStatus.Alarm_1","cluster_name":"IAS Zone"}]}}}}}
[2020-10-02 15:41:24][DEBUG] : {"devices":{"00:12:4b:00:1c:d6:0c:42":{"1":{"1280":{"cmd":[{"value":"0","cluster_name":"IAS Zone"},{"value":"ZoneStatus.0","cluster_name":"IAS Zone"}]}}}}}
[2020-10-02 15:41:25][DEBUG] : {"devices":{"00:12:4b:00:1c:d6:0c:42":{"1":{"1280":{"cmd":[{"value":"0","cluster_name":"IAS Zone"},{"value":"ZoneStatus.Alarm_1","cluster_name":"IAS Zone"}]}}}}}
[... cut ...]
[2020-10-02 16:06:01][DEBUG] : {"devices":{"00:12:4b:00:1c:d6:0c:42":{"1":{"1280":{"cmd":[{"value":"0","cluster_name":"IAS Zone"},{"value":"ZoneStatus.Alarm_1","cluster_name":"IAS Zone"}]}}}}}
[2020-10-02 16:07:01][DEBUG] : {"devices":{"00:12:4b:00:1c:d6:0c:42":{"1":{"1280":{"cmd":[{"value":"0","cluster_name":"IAS Zone"},{"value":"ZoneStatus.0","cluster_name":"IAS Zone"}]}}}}}
[2020-10-02 16:07:01][DEBUG] : {"devices":{"00:12:4b:00:1c:d6:0c:42":{"1":{"1280":{"cmd":[{"value":"0","cluster_name":"IAS Zone"},{"value":"ZoneStatus.Alarm_1","cluster_name":"IAS Zone"}]}}}}}
```

Here the log, I hide it at 15:45.

Maybe there is an issue on my device. I will ask to have another sample.

puddly, 2020-10-02:
What device is this? The Bosch ISW-ZPR1-WP13 goes into a "test mode" state for ~15 minutes after it enrolls but then resumes normal attribute reports.

zoic21, 2020-10-02:
It Sonoff SNZB-03 in commercial world, on realty it's an eWeLink MS01

Maybe it's due like Bosch ISW-ZPR1-WP13 but after 15min no more word for me, since 16:07 no more message.

I will received next week a new sample to see if it's normal or not

pipiche38, 2020-10-02:
> See https://zigbeealliance.org/wp-content/uploads/2019/12/07-5123-06-zigbee-cluster-library-specification.pdf Section `8.2.2.2.3`
> 
> I'm not 100% positive binding on IAS cluster is needed, that came from the original implementation. an IAS is enrolled either via enroll_request/enroll_response https://github.com/home-assistant/core/blob/576fbbe0be3b684618e453fc0bce1809673663cd/homeassistant/components/zha/core/channels/security.py#L129-L132 or by writting the `cieee_address` attribute
> 
> But per your log, since you are receiving cluster command_id == 0x0000, that means it is enrolled correctly. as to why it triggers every second 🤷 try to blank it so it doesn't get any movement?

Agree with you on the binding. What I have observed is for the Konke Door sensor you must avoid doing the binding, while for Heiman the binding is required .. As usall Manufacturers do what they want ;-)

Adminiuga, 2020-10-02:
heiman required binding on IAS zone?

Adminiuga, 2020-10-02:
cause I was thinking about making binding mandatory only on clusters we configure attribute reporting. Still have to test if we really need a binding on OTA cluster?

zoic21, 2020-10-05:
Hello,

While waiting for the arrival of a new module I reincluded the module without the binding on the IAS zone (I test by only binding the clusters with attributes reports). This time I got enroll request (it's a progress). I will do some tests and I will tell you if it corrects the problem or not.

zoic21, 2020-10-05:
Unfortunatly same issue : 

````
[2020-10-05 10:09:03][DEBUG] : Event for 00:12:4b:00:1c:d6:0c:42 logicalId : 1::1280::cmd::1 => ZoneStatus.0
[2020-10-05 10:09:05][DEBUG] : Event for 00:12:4b:00:1c:d6:0c:42 logicalId : 1::1280::cmd::1 => ZoneStatus.Alarm_1
[2020-10-05 10:11:51][DEBUG] : Event for 00:12:4b:00:1c:d6:0c:42 logicalId : 1::1280::cmd::1 => ZoneStatus.Alarm_1
[2020-10-05 10:11:51][DEBUG] : Event for 00:12:4b:00:1c:d6:0c:42 logicalId : 1::1280::cmd::1 => ZoneType.Motion_Sensor
[2020-10-05 10:14:12][DEBUG] : Event for 00:12:4b:00:1c:d6:0c:42 logicalId : 1::1280::cmd::1 => ZoneStatus.0
[2020-10-05 10:16:07][DEBUG] : Event for 00:12:4b:00:1c:d6:0c:42 logicalId : 1::1280::cmd::1 => ZoneStatus.Alarm_1
[2020-10-05 10:17:07][DEBUG] : Event for 00:12:4b:00:1c:d6:0c:42 logicalId : 1::1280::cmd::1 => ZoneStatus.0
[2020-10-05 10:17:08][DEBUG] : Event for 00:12:4b:00:1c:d6:0c:42 logicalId : 1::1280::cmd::1 => ZoneStatus.Alarm_1
[2020-10-05 10:18:08][DEBUG] : Event for 00:12:4b:00:1c:d6:0c:42 logicalId : 1::1280::cmd::1 => ZoneStatus.0
[2020-10-05 10:19:08][DEBUG] : Event for 00:12:4b:00:1c:d6:0c:42 logicalId : 1::1280::cmd::1 => ZoneStatus.Alarm_1
[2020-10-05 10:20:08][DEBUG] : Event for 00:12:4b:00:1c:d6:0c:42 logicalId : 1::1280::cmd::1 => ZoneStatus.0
[2020-10-05 10:24:21][DEBUG] : Event for 00:12:4b:00:1c:d6:0c:42 logicalId : 1::1280::cmd::1 => ZoneStatus.Alarm_1
[2020-10-05 10:27:32][DEBUG] : Event for 00:12:4b:00:1c:d6:0c:42 logicalId : 1::1280::cmd::1 => ZoneStatus.0
[2020-10-05 10:27:48][DEBUG] : Event for 00:12:4b:00:1c:d6:0c:42 logicalId : 1::1280::cmd::1 => ZoneStatus.Alarm_1
[2020-10-05 10:52:16][DEBUG] : Event for 00:12:4b:00:1c:d6:0c:42 logicalId : 1::1280::cmd::1 => ZoneStatus.0
[2020-10-05 10:52:17][DEBUG] : Event for 00:12:4b:00:1c:d6:0c:42 logicalId : 1::1280::cmd::1 => ZoneStatus.Alarm_1
[2020-10-05 10:53:17][DEBUG] : Event for 00:12:4b:00:1c:d6:0c:42 logicalId : 1::1280::cmd::1 => ZoneStatus.0
[2020-10-05 10:53:18][DEBUG] : Event for 00:12:4b:00:1c:d6:0c:42 logicalId : 1::1280::cmd::1 => ZoneStatus.Alarm_1
[2020-10-05 10:54:18][DEBUG] : Event for 00:12:4b:00:1c:d6:0c:42 logicalId : 1::1280::cmd::1 => ZoneStatus.0
[2020-10-05 10:54:19][DEBUG] : Event for 00:12:4b:00:1c:d6:0c:42 logicalId : 1::1280::cmd::1 => ZoneStatus.Alarm_1
[2020-10-05 10:55:19][DEBUG] : Event for 00:12:4b:00:1c:d6:0c:42 logicalId : 1::1280::cmd::1 => ZoneStatus.0
[2020-10-05 10:55:19][DEBUG] : Event for 00:12:4b:00:1c:d6:0c:42 logicalId : 1::1280::cmd::1 => ZoneStatus.Alarm_1
[2020-10-05 10:57:11][DEBUG] : Event for 00:12:4b:00:1c:d6:0c:42 logicalId : 1::1280::cmd::1 => ZoneStatus.0
[2020-10-05 10:57:12][DEBUG] : Event for 00:12:4b:00:1c:d6:0c:42 logicalId : 1::1280::cmd::1 => ZoneStatus.Alarm_1
[2020-10-05 10:58:12][DEBUG] : Event for 00:12:4b:00:1c:d6:0c:42 logicalId : 1::1280::cmd::1 => ZoneStatus.0
[2020-10-05 10:58:13][DEBUG] : Event for 00:12:4b:00:1c:d6:0c:42 logicalId : 1::1280::cmd::1 => ZoneStatus.Alarm_1
[2020-10-05 10:59:13][DEBUG] : Event for 00:12:4b:00:1c:d6:0c:42 logicalId : 1::1280::cmd::1 => ZoneStatus.0
[2020-10-05 10:59:13][DEBUG] : Event for 00:12:4b:00:1c:d6:0c:42 logicalId : 1::1280::cmd::1 => ZoneStatus.Alarm_1
[2020-10-05 11:00:13][DEBUG] : Event for 00:12:4b:00:1c:d6:0c:42 logicalId : 1::1280::cmd::1 => ZoneStatus.0
[2020-10-05 11:00:14][DEBUG] : Event for 00:12:4b:00:1c:d6:0c:42 logicalId : 1::1280::cmd::1 => ZoneStatus.Alarm_1
[2020-10-05 11:01:14][DEBUG] : Event for 00:12:4b:00:1c:d6:0c:42 logicalId : 1::1280::cmd::1 => ZoneStatus.0
[2020-10-05 11:01:15][DEBUG] : Event for 00:12:4b:00:1c:d6:0c:42 logicalId : 1::1280::cmd::1 => ZoneStatus.Alarm_1
````

It's work correctly from 10h09 to 10h16 than send event 0/1 in loop....

zoic21, 2020-10-13:
I have advanced a bit I clearly think that the operation of the module is weird.

- I had a battery problem hence the fact that it no longer worked after a few hours after inclusion (new battery included with the module, less than a month)
- then when the module sends a 1 (movement) it is obliged to send a 0 if the previous state was 1. Basically it cannot do 1 1 it will necessarily do 1 0 1

closed 2020-10-13 by zoic21