zigpy/zigpy #514: Gledopto GL-C-007 2ID - issue with endpoint 15

jedi7, 2020-10-05:
**How to reproduce:**

* power up Gledopto GL-C-007 2ID (two devices in one)
* Add it to Home Assistant with the ZHA integration
* Device GL-C-007 with 2 entities should appear
* Try turn on/off the entities

**Expected results:**

Entity A (with endpoint 11, RGB/color) will on/off, etc
Entity B (with endpoint 15, CCT/white) will on/off, etc

**Actual results:**

Entity A (with endpoint 11, RGB/color) will on/off, etc
Entity B (with endpoint 15, CCT/white) does not respond (see logs)

**Relevant logs**
https://pastebin.com/5UgBGLn7

device description:
https://pastebin.com/LwyfS8mL

**Problem research**
When I tried the zigbee2mqtt, then it worked properly. And the difference is, the zigbee2mqtt uses always source endpoint 1.
So it looks like the firmware in Gledopto have issue when the source endpoint 15 is used.

When I patched this zigpy code and forced source endpoint to 1, then I was able to change visually both entities (A,B). 
I.E. I was able to make send proper command, but response was not matched in code. And I'm not sure where to change it.

If you gave me some hints, I can try to change the code and made PR.

Adminiuga, 2020-10-05:
zigpy-cc should pick the proper source endpoint, when communicating with the device. See issues for zigpy-cc as it have a few opened with the same root cause. 
As a fix, upgrade to 0.116.b4 and use zigpy-znp radio type for zha instead.

jedi7, 2020-10-08:
Thanks for info. I was directed to quirks, where is possible to solve it.

FYI: my dirty initial qurik: https://pastebin.com/8q2NYfNW

closed 2020-10-08 by jedi7