zigpy/zigpy #516: How to write specific attribute

zoic21, 2020-10-06:
Hello,

I need to write manufacturer specific attribute on some device (specific manufacturer configuration). For example on Danfoss Ally I need to write endpoint 1, cluster 0x0204, attribute 0x4000 to 0x01, to invert display with manufacturer specific option. But I got : 

````
[2020-10-06 18:13:33][ERROR] : [0x9214:1:0x0204] 16384 is not a valid attribute id
````

I understand zigpy don't know this attribute but how can I force write ?

Thank in advance

Adminiuga, 2020-10-06:
if `0x4000` is not in the ZCL, then you need to create a quirk for the device. For example this https://github.com/zigpy/zha-device-handlers/blob/427761d2b6e7577fe944986908b946e768195a34/zhaquirks/sinope/thermostat.py#L37-L46 adds two new manufacturer specific attributes.

See https://github.com/zigpy/zha-device-handlers/blob/dev/CONTRIBUTING.md on how to create a quirk

zoic21, 2020-10-07:
Thank for your reply.

Here my PR https://github.com/zigpy/zha-device-handlers/pull/517, I hope it's OK, it's my first...

closed 2020-10-08 by zoic21