zigpy/zigpy #522: OTA support for Securifi Peanut Plug

Madseason41, 2020-10-23:
Is there any way to utilize the OTA ability to perform a firmware update on the Securifi Peanut Plugs?

Currently there is a hackish way to update the firmware in order to enable power reporting on these plugs. It would be great to be able to utilize the OTA ability of Home Assistant.

https://community.hubitat.com/t/peanut-plug-firmware-ota-update/30508

The request to do this thru deconz is linked below. The firmware file is in the link as well. Looks like it fell dead there.

https://github.com/dresden-elektronik/deconz-rest-plugin/issues/419

Also, when adding the Peanut Plugs the device comes up as unknown model and manufacturer. Why is this? The switching works fine and the power monitoring entity comes up (albeit it doesn't work, which is supposed to be fixed by the FW upgrade).

If this has been addressed or is in the wrong place, I apologize. I am starting anew with zigbee stuff. I searched but didn't find anything pertaining to the peanut plug in this repo.

I run Hassio via docker on Ubuntu Server 18.04 with a HUSBZB-1 with the latest firmware if it is pertinent.

R,

puddly, 2020-10-23:
Since there is no HTTP endpoint to retrieve these images from, the only way to do this is to enable [OTA updates](https://old.reddit.com/r/tradfri/comments/famdi7/for_the_home_assistant_users_out_there/) and specify an `otau_directory`. Unzipping [`ZPS_CS5490_039.ota`](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/419#issuecomment-667109912) and moving it into the above directory will let zigpy utilize that OTA file to upgrade your plug.

In case it doesn't work, enable https://www.home-assistant.io/integrations/zha/#debug-logging and record the whole upgrade process.

Madseason41, 2020-10-23:
I setup the directory and uploaded the file. I am familiar with this method as I used it yesterday to update Ikea firmware from the directory.

I moved the peanut firmware to the directory and then went to clusters (OTA) and clusters commands (Image_notify) and clicked issue command.

Below is the log, I hope.

2020-10-22 22:19:00 DEBUG (MainThread) [zigpy.zcl] [0x240a:1:0x0019] Schema and args lengths do not match in reply
2020-10-22 22:19:00 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0x240A](unk_model): Issued cluster command: cluster_id: 25 command: 0 command_type: client args: () cluster_id: out manufacturer: None endpoint_id: 1
2020-10-22 22:19:00 DEBUG (MainThread) [homeassistant.components.zha.api] Issued command for: cluster_id: [25] cluster_type: [out] endpoint_id: [1] command: [0] command_type: [client] args: [] manufacturer: [None] response: EmberStatus.SUCCESS
2020-10-22 22:19:14 DEBUG (MainThread) [zigpy.zcl] [0x117f:1:0x0000] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=True is_reply=False disable_default_response=True> manufacturer=4447 tsn=6 command_id=Command.Report_Attributes>
2020-10-22 22:19:14 DEBUG (MainThread) [zigpy.zcl] [0x117f:1:0x0000] ZCL request 0x000a: [[Attribute(attrid=65281, value=<TypeValue type=CharacterString, value=!�(!�!�$>)]]
2020-10-22 22:19:14 DEBUG (MainThread) [zigpy.zcl] [0x117f:1:0x0000] Attribute report received: 65281=!�(!�!�$
2020-10-22 22:19:25 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x240A:1:0x0b04]: async_update
2020-10-22 22:19:25 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xB560:1:0x0b04]: async_update
2020-10-22 22:19:25 DEBUG (MainThread) [zigpy.zcl] [0x240a:1:0x0b04] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=145 command_id=Command.Read_Attributes_rsp>
2020-10-22 22:19:25 DEBUG (MainThread) [zigpy.zcl] [0xb560:1:0x0b04] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=147 command_id=Command.Read_Attributes_rsp>

puddly, 2020-10-23:
If nothing seems to happen soon, try physically unplugging the device and plugging it back in. That usually causes it to ask for a new OTA image within a few minutes.

Madseason41, 2020-10-23:
So far no luck. I have restarted a few times and unplugged and plugged it back in. Nothing showing up in the logs yet.

MattWestb, 2020-10-23:
@puddly I think (= not knowing) its not working because the device is unknown and the endpoints and clusters is not mapped at all > Zingpy is not understanding the OTA requests from the device.

Adminiuga, 2020-10-23:
@dmulcahey were you able to update your penaut plug?

If the plug is not looking for ota server after a power up ,  ota command or in 24 hours, then nothing would work

dmulcahey, 2020-10-23:
Nope, had to buy a 15$ router and flash it with almond firmware to do it

Madseason41, 2020-10-23:
Buying the router was what I was trying to avoid but may go that route just to have something to tinker with.

puddly, 2020-10-23:
It'd be interesting to get an over-the-air capture of the update process to see what is exchanged between the router and the plug.

Madseason41, 2020-10-23:
How would I do that? I am not even sure that these have the old firmware on them. I do know that the sensor for power consumption does appear but doesn't work. 

Also, how do I fix the unk_model?

I have ordered the $15 dollar router.

Adminiuga, 2020-10-23:
you can't fix. The real fix: vote with your money and return the product. Buy one which follows the specs.

puddly, 2020-10-23:
`unk_model` means that the device didn't respond when zigpy asked it for its model name. Not much you can do about that.

You can use your HUSBZB-1 as a sniffer to capture the packets. With HA not running, something like this might work:

 - Repeatedly run `bellows -d /dev/serial/by-id/your-husbzb1 -b 115200 scan` to find the specific Zigbee channel the $15 router is running on (i.e. see what new network shows up when it's plugged in). Don't join the plug to the router yet, as you will need the sniffer running when it first joins to capture the network encryption key.

  - Once you've found the Zigbee channel (should be between 11 and 26, most likely one of 11, 15, 20, 25), run `bellows -d /dev/serial/by-id/your-husbzb1 -b 115200 dump -c your_channel -w capture.pcap`, join the plug to the $15 router's network, perform the OTA update, and stop the sniffer. That capture should contain everything.

tube0013, 2020-10-24:
I have the plug and router incoming I. The next few days and can do this.

Madseason41, 2020-10-25:
It looks like there are different (custom) firmware files referenced in the following link.

https://community.hubitat.com/t/almond-routers-to-update-the-firmware-on-securifi-peanut-plugs/17439

tube0013, 2020-10-26:
here is the capture, it took me a few tries to join the peanut as it kept joining my HA network... any way, joined it, checked for update, pushed the update. then I toggled the "retain state" option - I assume it's an option for it to retain it's last state if it loses power... 

[capture.pcap.zip](https://github.com/zigpy/zigpy/files/5440575/capture.pcap.zip)

puddly, 2020-10-26:
It appears the hub does the following to get the plug to ask for a new image:

```Python
success, failure = await device.endpoints[0xFF].ota.read_attributes(["upgrade_server_id"])
# success["upgrade_server_id"] should look like "27 00 00 00 00 00 00 00"

await device.endpoints[0xFF].scenes.write_attributes({"current_group": 0xC15E})
```

Endpoint `0xFF` is not advertised by the device in its simple descriptor response.

The plug itself asks for an image a second later.

Unfortunately three 80-byte image blocks (out of 3169!) were not captured so the image itself can't be fully reconstructed but its header is `OTAImageHeader(upgrade_file_id=200208670, header_version=256, header_length=56, field_control=<FieldControl.0: 0>, manufacturer_id=4098, image_type=24256, file_version=57, stack_version=2, header_string='securifi Power switch', image_size=155902)` and it's a Silicon Labs EBL firmware image with a CRC-32 of b"\xEF\x2A\x34\xCA"`.

dmulcahey, 2020-10-26:
I have the OTA file they sent it to me

Adminiuga, 2020-10-26:
endpoint 0xFF is a broadcast endpoint, essentially it addresses all endpoints on the device. My expectation writing to the actual endpoint should be equivalent to addressing the broad cast endpoint, cause ATM i don't think you can do that with zigpy, not unless you build the request manually.

tube0013, 2020-10-27:
I have another plug coming. with some guidance can experiment with it.

tube0013, 2020-10-29:
okay so with @walthowd 's help we got this to work. 🎉🎉🎉🎉

 I put the OTA file linked here:

https://github.com/dresden-elektronik/deconz-rest-plugin/issues/419#issuecomment-667109912

into my zigpy_ota folder - restarted HA. FW was parsed okay.

then went to developer tools > Services and called zha.set_zigbee_cluster_attribute with this:

```
ieee: '00:0d:6f:00:05:3a:bb:a5'
endpoint_id: 1
cluster_id: 5
attribute: 2
cluster_type: out
value: 49502
```
log out put:

https://paste.ubuntu.com/p/WwxwgV6qZW/

```
2020-10-29 10:52:54 DEBUG (MainThread) [zigpy.zcl] [0x53e7:1:0x0019] ZCL request 0x0006: [<Status.SUCCESS: 0>, 4098, 24256, 57]
2020-10-29 10:52:54 DEBUG (MainThread) [zigpy.zcl] [0x53e7:1:0x0019] OTA upgrade_end handler for 'Securifi Ltd. None': status=Status.SUCCESS, manufacturer_id=4098, image_type=24256, file_version=57
```
[image]

It's above my head, but maybe a quirk can be created to set this up a command for the scene cluster so it can be called via the config panel cluster commands - but this would like be confusing as it would be under the Scenes. so this maybe just best in documentation --- idk  🤷‍♂️🤷‍♂️

puddly, 2020-10-29:
Awesome! Maybe we can make a quirk for the OTA cluster that first calls:

```Python
await self.endpoint.scenes.write_attributes({"current_group": 0xC15E})
```

when the OTA cluster is sent a unicast `image_notify`? That should at least keep the OTA functionality in one place.

tube0013, 2020-10-29:
I haven't seen it ask for firmware again itself, if I re-write that attribute it asks for firmware right away.

you can also get it to query for an image by issuing this command (From @walthowd 's guide without the args):
```
ieee: '00:0d:6f:00:05:3a:bb:a5'
endpoint_id: 1
cluster_type: out
command_type: client
cluster_id: 25
command: 0
```

puddly, 2020-10-29:
Does that work with the old firmware as well? Or just on the updated one?

tube0013, 2020-10-29:
Need someone with the older FW to test that now.

Madseason41, 2020-10-30:
I have one with the original firmware. What do i do?

walthowd, 2020-10-30:
@Madseason41 Find the IEEE of your Peanut (Config > Integrations > ZHA > Devices > Peanut) Then go to Developer Tools -> Services and choose `zha.issue_zigbee_cluster_command` -- Enter the following service data:
```
ieee: 'YOURPEANUTIEEE'
endpoint_id: 1
cluster_type: out
command_type: client
cluster_id: 25
command: 0
```

Then watch your `home-assistant.log` (with all zigpy/bellows debugs on) and see if you see any lines like this:

`2020-10-29 22:25:44 DEBUG (MainThread) [zigpy.zcl] [0xf5fa:1:0x0019] OTA query_next_image handler for 'sengled E11-G13': field_control=0, manufacture_id=4448, image_type=3, current_file_version=9, hardware_version=None`

Or any other lines matching zigpy and OTA.

Madseason41, 2020-10-30:
I don't think it worked or I didn't do something right.

OTA directory is enabled and it seemed to recognize the FW file on start.

Issued the command at 0745.00

2020-10-30 07:45:01 DEBUG (MainThread) [zigpy.zcl] [0x231a:1:0x0019] Schema and args lengths do not match in reply
2020-10-30 07:45:18 DEBUG (MainThread) [zigpy.zcl] [0x3d9b:1:0x0b04] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=51 command_id=Command.Read_Attributes_rsp>
2020-10-30 07:45:18 DEBUG (MainThread) [zigpy.zcl] [0x231a:1:0x0b04] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=53 command_id=Command.Read_Attributes_rsp>

puddly, 2020-10-31:
Have you also tried adding the optional arguments to that command?

```yaml
args:
  - 0
  - 100
```

Madseason41, 2020-11-01:
Issued the following cluster command:
```

ieee: '00:0d:6f:00:0a:79:17:86'
endpoint_id: 1
cluster_type: out
command_type: client
cluster_id: 25
command: 0
args:
  - 0
  - 100
```

Nothing shows up in the logs for OTA.

puddly, 2020-11-01:
@Madseason41 Thanks for testing this out. It sounds like writing `0xC15E` as the current group is necessary.

Madseason41, 2020-11-01:
Yeah, as soon as I set the attribute like below, it worked.

```
ieee: '00:0d:6f:00:0a:79:17:86'
endpoint_id: 1
cluster_id: 5
attribute: 2
cluster_type: out
value: 49502
```

FidgetyRat, 2020-11-16:
I just wanted to chime in real fast and thank all of you for your work. I was able to use this approach and update the firmware on my peanuts through HomeAssistant enabling the wattage output. 

Thanks a lot!

Hedda, 2021-04-30:
> It'd be interesting to get an over-the-air capture of the update process to see what is exchanged between the router and the plug.

FYI pipiche38 has begun an interesting project that is trying to get OTA firmware URL via info get from Wireshark Zigbee sniffing:

https://github.com/pipiche38/Capture-OTA-from-Wireshark

https://github.com/pipiche38/Domoticz-Zigate-Wiki/blob/master/en-eng/Corner_Retreiving-Legrand-Firmware.md

closed 2021-11-03 by github-actions[bot]

bvlaicu, 2022-12-12:
Just a note that for me the process worked as described here, except that I had to set `cluster_type: in` instead of `out` when setting the cluster attribute.
To summarize, the service I issued was:
```
service: zha.set_zigbee_cluster_attribute
data:
  ieee: '00:0d:**:**:**:**:58:02'
  endpoint_id: 1
  cluster_id: 5
  attribute: 2
  cluster_type: in
  value: 49502
```
I didn't have to issue the cluster command, I just set the attribute above.