zigpy/zigpy #523: Lightlink send reset

zoic21, 2020-10-23:
Hello,

I wonder if it's possible to send lightlink reset (for Hue by example). I see a server command "reset_to_factory_new" but how to use it ? I don't have lightlink cluster on my deconz 2 key it's normal ? It's a zigpy-deconz issue ?

Thank in advance

MattWestb, 2020-10-23:
Drestnen have not implanting the touch link functionality  in the second gen NCPs firmware.
I have one RaspBee first gen and the touch link scan and reset is working great in deCONZ (old web app) but its not doing for user with second gen RaspBee / CornBee.
The first and second gen sharing the same serial protocol so the command is there but not working. If I was you i would sending one request to DE for implanting the funktion in the firmware and hoping they is doing it then enough users is asking for it.

I have asking for implanting touch link scan and reset in Ziggpy then at least TI, EZSP and deCONZ 1 gen have the function in firmware but have not getting any positive response.

For EZSP device its gets one python application (HUE thief) but its broken then python have changing the commands and i cant fixing it.

One more thing is that i have expired more times that IKEA bulbs is not responding on hardware reset (6X repower) after being deleted in ZHA and is in "limbo" until being TL resetted.

zoic21, 2020-10-23:
Thank for your reply, it's much clearer now

I open an issue https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3485, 
I don't think it changes anything but you never know

closed 2020-10-23 by zoic21

MattWestb, 2020-10-23:
I going there and supporting you little but the moderators there dont like my so much ;-((