zigpy/zigpy #524: Issue after restart and "New device joined"

zoic21, 2020-10-23:
Hello,

I have a little issue after zigpy, I'am in standalone use not in HA. Sometimes (not always) after a zigpy restart when a remote (in my case it's an ikea remote controller) send I order (click) it's make a "New device joined" but not handle command (click), I need do redo I new one to make it's work.

Here the log : 

```
[2020-10-23 14:27:42][DEBUG] : Device state changed response: [<DeviceState.128|APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_INDICATION|2: 170>, 0]
[2020-10-23 14:27:42][DEBUG] : Command Command.aps_data_indication (1, 1)
[2020-10-23 14:27:42][DEBUG] : Send: 0x17cc000800010001
[2020-10-23 14:27:42][DEBUG] : Frame received: 0x17cc002900220022020000010297ae01040101000b0008160a200020192100201000af87760e5400b3
[2020-10-23 14:27:42][DEBUG] : APS data indication response: [34, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|2: 34>, <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x0000>, 1, <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0xae97>, 1, 260, 1, b'\x08\x16
\x00 \x19!\x00 \x10', 0, 175, 135, 118, 14, 84, 0, -77]
[2020-10-23 14:27:42][DEBUG] : [0xae97:1:0x0001] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=22 command_id=Command.Report_Attributes>
[2020-10-23 14:27:42][DEBUG] : [0xae97:1:0x0001] ZCL request 0x000a: [[Attribute(attrid=32, value=<TypeValue type=uint8_t, value=25>), Attribute(attrid=33, value=<TypeValue type=uint8_t, value=16>)]]
[2020-10-23 14:27:42][INFO] : [08:6b:d7:ff:fe:20:23:fd][listener.general_command] Cluster: <zhaquirks.DoublingPowerConfigurationCluster object at 0x7f1286220048> ClusterId: 0x0001 command_id: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=22 command_id=Command.Report_Attributes> args: ([[Attribute(attrid=32, value=<TypeValue type=uint8_t, value=25>), Attribute(attrid=33, value=<TypeValue type=uint8_t, value=16>)]],)
[2020-10-23 14:27:42][DEBUG] : [0xae97:1:0x0001] Attribute report received: battery_voltage=25, battery_percentage_remaining=16
[2020-10-23 14:27:42][INFO] : [08:6b:d7:ff:fe:20:23:fd][listener.attribute_updated] Received an attribute update 32=25 on cluster 1
[2020-10-23 14:27:42][INFO] : [08:6b:d7:ff:fe:20:23:fd][listener.attribute_updated] Received an attribute update 33=32 on cluster 1
[2020-10-23 14:27:42][DEBUG] : Sending Zigbee request with tsn 22 under 194 request id, data: b'18160b0a00'
[2020-10-23 14:27:42][DEBUG] : 'aps_data_indication' response from <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0xae97>, ep: 1, profile: 0x0104, cluster_id: 0x0001, data: b'08160a2000201921002010'
[2020-10-23 14:27:42][DEBUG] : Command Command.aps_data_request (20, 194, 0, <DeconzAddressEndpoint address_mode=2 address=0xAE97 endpoint=1>, 260, 1, 1, b'\x18\x16\x0b
\x00', 2, 0)
[2020-10-23 14:27:42][DEBUG] : Send: 0x12cd001b001400c2000297ae010401010001050018160b0a000200
[2020-10-23 14:27:42][DEBUG] : Frame received: 0x12cd000900020022c2
[2020-10-23 14:27:42][DEBUG] : APS data request response: [2, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|2: 34>, 194]
[2020-10-23 14:27:42][DEBUG] : Frame received: 0x1cce000c000500028255ffc5
[2020-10-23 14:27:43][DEBUG] : Frame received: 0x1ccf000c000500028255ffc5
[2020-10-23 14:27:44][DEBUG] : Frame received: 0x1cd0000c000500028255ffc5
[2020-10-23 14:27:45][DEBUG] : Frame received: 0x1cd1000c000500026486dfbe
[2020-10-23 14:27:45][DEBUG] : Frame received: 0x0ed2000700a600
[2020-10-23 14:27:45][DEBUG] : Device state changed response: [<DeviceState.128|APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_CONFIRM|2: 166>, 0]
[2020-10-23 14:27:45][DEBUG] : Frame received: 0x1cd3000c0005000297ae7fb2
[2020-10-23 14:27:45][DEBUG] : Command Command.aps_data_confirm (0,)
[2020-10-23 14:27:45][DEBUG] : Send: 0x04ce0007000000
[2020-10-23 14:27:45][DEBUG] : Frame received: 0x04ce0013000c0022c20297ae01010000000000
[2020-10-23 14:27:45][DEBUG] : APS data confirm response for request with id 194: 00
[2020-10-23 14:27:45][DEBUG] : Request id: 0xc2 'aps_data_confirm' for <DeconzAddressEndpoint address_mode=ADDRESS_MODE.NWK address=0xae97 endpoint=1>, status: 0x00
[2020-10-23 14:27:45][DEBUG] : Frame received: 0x1ccf000c0005000297ae77b1
[2020-10-23 14:27:45][DEBUG] : Frame received: 0x0ed0000700aa00
[2020-10-23 14:27:45][DEBUG] : Device state changed response: [<DeviceState.128|APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_INDICATION|2: 170>, 0]
[2020-10-23 14:27:45][DEBUG] : Command Command.aps_data_indication (1, 1)
[2020-10-23 14:27:45][DEBUG] : Send: 0x17cf000800010001
[2020-10-23 14:27:45][DEBUG] : Frame received: 0x1cd1000c0005000297ae7fb2
[2020-10-23 14:27:45][DEBUG] : Frame received: 0x17cf002a0023002202fdff000297ae00000013000c008297aefd2320feffd76b088000af77760e5400b1
[2020-10-23 14:27:45][DEBUG] : APS data indication response: [35, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|2: 34>, <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0xfffd>, 0, <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0xae97>, 0, 0, 19, b'\x82\x97\xae\xfd# \xfe\xff\xd7k\x08\x80', 0, 175, 119, 118, 14, 84, 0, -79]
[2020-10-23 14:27:45][INFO] : New device joined: 0xae97, 08:6b:d7:ff:fe:20:23:fd
[2020-10-23 14:27:45][INFO] : Device 0xae97 (08:6b:d7:ff:fe:20:23:fd) joined the network
[2020-10-23 14:27:45][DEBUG] : Skip initialization for existing device 08:6b:d7:ff:fe:20:23:fd
[2020-10-23 14:27:45][DEBUG] : [0xae97:zdo] ZDO request ZDOCmd.Device_annce: [0xAE97, 08:6b:d7:ff:fe:20:23:fd, 128]
[2020-10-23 14:27:45][DEBUG] : 'aps_data_indication' response from <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0xae97>, ep: 0, profile: 0x0000, cluster_id: 0x0013, data: b'8297aefd2320feffd76b0880'
[2020-10-23 14:27:45][DEBUG] : Frame received: 0x1cd0000c0005000297ae77b1
[2020-10-23 14:27:45][DEBUG] : Frame received: 0x1cd1000c00050002825
```

Maybe it's normal (we don't restart zigbee network often), or maybe I miss something on initialisation of zigpy.

Thank in advance.

Adminiuga, 2020-10-23:
This works as expected. Device sent zdo device announcement message which triggers handling of the device joining.
Since it an existing device, it doesn't do much

closed 2020-10-23 by Adminiuga

zoic21, 2020-10-24:
Ok I understand that but bouton click is not send or not handle ? In this case I press on bouton, device only send zdo ? I need to click twice on "on" bouton (same for off) to received attribyte update.

Adminiuga, 2020-10-24:
Can't tell. Depends on how buttons n was configured if at all when it joined the 1st time.