zigpy/zigpy #525: Philips Hue OTAs padding causes "Image contains trailing data" error

TheJulianJES, 2020-10-25:
Apparently some of the Philips Hue OTAs (possibly others too) have a padding at the end of each file.
With the following file: https://github.com/Koenkk/zigbee-OTA/blob/a3c4aad6ea0dc04b6afcbc92a2dc762cca9726e5/index.json#L147 (url: http://fds.dc1.philips.com/firmware/ZGB_100B_0111/16782848/100B-0111-01001600-ConfLight-ModuLum-EFR32MG13.zigbee", imageType: 273, manufacturerCode: 4107, fileSize: 441566 and fileVersion: 16782848, I get the following error when the device (finally) queries (note: it doesn't seem to support 'image_notify):
```
2020-10-25 07:40:46 DEBUG (MainThread) [zigpy.zcl] [0xcfa7:11:0x0019] OTA query_next_image handler for 'Philips LCT026': field_control=0, manufacture_id=4107, image_type=273, current_file_version=16782850, hardware_version=None
2020-10-25 07:40:46 WARNING (MainThread) [zigpy.ota.validators] Image OTAImageHeader(upgrade_file_id=200208670, header_version=256, header_length=56, field_control=<FieldControl.0: 0>, manufacturer_id=4107, image_type=273, file_version=16782848, stack_version=2, header_string='', image_size=441566) is invalid: Image contains trailing data
```

zigbee2mqtt also experiences this problem, but it seems like the padding at the end of the file is "ok" and shouldn't cause the image to be invalidated: https://github.com/Koenkk/zigbee2mqtt/issues/2921#issuecomment-716163402
The following PR was also submitted: https://github.com/Koenkk/zigbee-herdsman-converters/pull/1704
It seems like the 0x0 bytes at the end of the file are just removed before the file is validated(?)

(Currently using the latest Home Assistant 0.117.0b3) which should include the latest version of zigpy.)

puddly, 2020-10-25:
Ah, I tested against every image in that repository before implementing validation but the `index.json` file also links to external images.

TheJulianJES, 2020-10-25:
Just applied your fix manually and it seems to work perfectly!
Thanks already.

commit b8ba1459 (2020-10-26): Allow null padding bytes in GBL files (fixes #525) (#526)
closed 2020-10-26 by Adminiuga