zigpy/zigpy #533: Request for Workaround for IKEAs 5 button remotes sterling CT and RGBW light scenes.

MattWestb, 2020-10-29:
IKEA CT and RGBW bulbs is loading 3 default scenes (0,1,2) (scenes clusters attribute count (0x0000) = 3)  with groupe ID (Current_groue(0x0005) 0xff09 =) 65289 then upgrading the firmware thru OTA and is working with one group bunded remote.
I think its because not losing the function then updating the firmware and not need repairing / configuring the bulb after updating is made (custom made scenes is being lost but can being reconfigured in the APP).
[image]
(deCONZ screenshot)
The problem is that then doing one reset the "default" scenes is also deleted and not being created the doing one hardware or TL reset.
[image]
(deCONZ screenshot)
Then pairing with  IKEAs GW and also the remote (doing LL pairing without GW) the controller device (GW or remote) is configuring / loading the "default scenes" to the bulb with IKEA property commands (not verified but no one have catching how is made) or normal scenes.

If doing one force downgrade of the firmware in deCONZ  (around 55 minutes) and then letting it update to current version (55 more minutes) then the "HW scenes" is created and working until the bulb is reseted and loosing the funktion.
 If paring one downgraded bulb in ZHA and paring it adding it to one light groupe and ZHA is upgrading to current version its working great until resetting the bulb but is is not losing the "HW scenes" then sending reconfig to one bulb.

Its little frustrating deleting bulb in ZHA, paring with deCONZ and force firmware downgrade, deleting in deCONZ, pairing in ZHA and configuring group and then letting it being upgraded agen for recreating the "HW scenes" but its working well (have doing it 3 times) and more then 2 hours later.

ZHA can reading and sending attributes and cluster commands but have not implanting all the logic for configuring scenes so its not possible getting it working if the "HW scenes" is lost.

Z2H have thousands of posts then users is missing the funktionality and cant getting it back then not knowing the way to do it and can't recreating the "HW scenes" there.

ZHA can doing firmware upgrade from local files but is blocking forcing loading older versions.

By implanting "force" OTA file load from local file with older version, user can doing one manual downgrade and then letting ZHA upgrading the bulb and getting the "HW scenes" until resetting the device. 
How to doing one good / nice implementing i dont knowing if only override if local file exist with right manufacture and model ID and ignoring the one from server or more advance with possible configuring / picking file (ala deCONZ).

One comment to the screenshots: The Scene valid is true then having changing scene with the remote. Then changing some parameters like brightness the value is set fale in the bulb (= the scenes is selected but modified).

If ZHA have getting the more advanced cluster manipulation in the GU is caan being possible to recreating the "HW scenes" but for the moment its not possible (for my) to do as one normal user.

Thanks in advance and for great work done with ZHA and keep coding strong !!!

Attached bulb-ws-2.0.019 OTA file for downgrading test.
[10038562-TRADFRI-sy5882-bulb-ws-2.0.019.1.ota.ota.signed.zip](https://github.com/zigpy/zigpy/files/5460092/10038562-TRADFRI-sy5882-bulb-ws-2.0.019.1.ota.ota.signed.zip)

puddly, 2020-10-29:
Hmm. Perhaps we can make a `pinned/` folder for images or something that is checked for compatible images before every other provider?

MattWestb, 2020-10-29:
Thats making it very easy to "administrating" for the user and doing the "down/upgrade" its only renaming the directory (Xpinned/) and zigpy is using the OTA from the server for the next OTA request.
Most imported possible overriding the devices FW version but good implanting is great.
The name is not importer but OTA_pinned, OTA_override or something like that.
If you is making it then Z2M must rebuilding their OTA handling for solving the "HW scenes" for matching ZHA or do more serious IKEA sniffing  :-)))

closed 2021-09-07 by github-actions[bot]