zigpy/zigpy #540: Network topology scan issue with Legrand Netatmo and Sonoff Zigbee devices

smmoroz, 2020-11-03:
Hi, 

I'm not sure if this is the right place to post this issue. If it is not, please advise the right place.

I have several Zigbee devices in my network:

[image]

I have an issue with Legrand Netatmo devices, for example with the one highlighted in red above with `nwk=0x83d7`. At least two Sonoff Zigbee devices are connected to it -  the devices with `nwk=0x4241` and `nwk=0x9368`. Those Sonoff devices are online and work correctly. 

But as one can see from the log below, the network scan ignores them because they have invalid `ieee=00:00:00:00:00:00:00:00`. 

How to fix this problem?

```
2020-11-03 14:11:13 DEBUG (MainThread) [zigpy.neighbor] [0x83d7] request status: Status.SUCCESS. response: Neighbors(entries=6, start_index=2, neighbor_table_list=[Neighbor(extended_pan_id=ed:cd:9d:b7:26:e2:ed:d7, ieee=00:0b:57:ff:fe:de:e9:02, nwk=0x0000, packed=36, permit_joining=<PermitJoins.Accepting: 1>, depth=0, lqi=119), Neighbor(extended_pan_id=ed:cd:9d:b7:26:e2:ed:d7, ieee=00:00:00:00:00:00:00:00, nwk=0x4241, packed=37, permit_joining=<PermitJoins.Accepting: 1>, depth=3, lqi=191)])
2020-11-03 14:11:13 DEBUG (MainThread) [zigpy.neighbor] [0x83d7] ignoring invalid neighbor: 00:00:00:00:00:00:00:00
2020-11-03 14:11:15 DEBUG (MainThread) [zigpy.neighbor] [0x83d7] Querying next starting at 4
2020-11-03 14:11:15 DEBUG (MainThread) [zigpy.util] Tries remaining: 3
2020-11-03 14:11:15 DEBUG (MainThread) [bellows.ezsp.protocol] Send command sendUnicast: (<EmberOutgoingMessageType.OUTGOING_DIRECT: 0>, 0x83D7, EmberApsFrame(profileId=0, clusterId=49, sourceEndpoint=0, destinationEndpoint=0, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY|APS_OPTION_RETRY: 320>, groupId=0, sequence=114), 115, b'r\x04')
2020-11-03 14:11:15 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'50be2157541e1565da944a14aa5592099d4e27d99ecc158fdede7e'
2020-11-03 14:11:15 DEBUG (bellows.thread_0) [bellows.uart] Data frame: b'06bea157541e15aee86d7e'
2020-11-03 14:11:15 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'8160597e'
2020-11-03 14:11:15 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame 52 (sendUnicast) received: b'001c'
2020-11-03 14:11:15 DEBUG (bellows.thread_0) [bellows.uart] Data frame: b'16beb15754151565da944a14aa5592099d4e27b79ece6795fd7e'
2020-11-03 14:11:15 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'82503a7e'
2020-11-03 14:11:15 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame 63 (messageSentHandler) received: b'00d783000031000000400100001c730000'
2020-11-03 14:11:15 DEBUG (MainThread) [bellows.zigbee.application] Received messageSentHandler frame with [<EmberOutgoingMessageType.OUTGOING_DIRECT: 0>, 33751, EmberApsFrame(profileId=0, clusterId=49, sourceEndpoint=0, destinationEndpoint=0, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY|APS_OPTION_RETRY: 320>, groupId=0, sequence=28), 115, <EmberStatus.SUCCESS: 0>, b'']
2020-11-03 14:11:15 DEBUG (bellows.thread_0) [bellows.uart] Data frame: b'26beb157546f15b259a5ca25aa1593499c507b183a4d9874ccb4638ff87ce84a09eb69f242127398d02c965b2d2b083c533aa666a26fa3befbbbde5ce070381c0e07bbe5a2f6af4499f2d2247e'
2020-11-03 14:11:15 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'83401b7e'
2020-11-03 14:11:15 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame 69 (incomingMessageHandler) received: b'00000031800000400100001e5cb3d783ffff317200060402d7ede226b79dcdedb44305feffd76b08a1d02501038cd7ede226b79dcded00000000000000006893250103bf'
2020-11-03 14:11:15 DEBUG (MainThread) [bellows.zigbee.application] Received incomingMessageHandler frame with [<EmberIncomingMessageType.INCOMING_UNICAST: 0>, EmberApsFrame(profileId=0, clusterId=32817, sourceEndpoint=0, destinationEndpoint=0, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY|APS_OPTION_RETRY: 320>, groupId=0, sequence=30), 92, -77, 0x83d7, 255, 255, b'r\x00\x06\x04\x02\xd7\xed\xe2&\xb7\x9d\xcd\xed\xb4C\x05\xfe\xff\xd7k\x08\xa1\xd0%\x01\x03\x8c\xd7\xed\xe2&\xb7\x9d\xcd\xed\x00\x00\x00\x00\x00\x00\x00\x00h\x93%\x01\x03\xbf']
2020-11-03 14:11:15 DEBUG (MainThread) [zigpy.neighbor] [0x83d7] request status: Status.SUCCESS. response: Neighbors(entries=6, start_index=4, neighbor_table_list=[Neighbor(extended_pan_id=ed:cd:9d:b7:26:e2:ed:d7, ieee=08:6b:d7:ff:fe:05:43:b4, nwk=0xD0A1, packed=37, permit_joining=<PermitJoins.Accepting: 1>, depth=3, lqi=140), Neighbor(extended_pan_id=ed:cd:9d:b7:26:e2:ed:d7, ieee=00:00:00:00:00:00:00:00, nwk=0x9368, packed=37, permit_joining=<PermitJoins.Accepting: 1>, depth=3, lqi=191)])
2020-11-03 14:11:15 DEBUG (MainThread) [zigpy.neighbor] [0x83d7] ignoring invalid neighbor: 00:00:00:00:00:00:00:00
2020-11-03 14:11:15 DEBUG (MainThread) [zigpy.neighbor] [0x83d7] Done scanning. Total 4 neighbours
```

Adminiuga, 2020-11-03:
Contact legrand and ask them to fix the firmware? Legrand is reporting a neighbor incorrectly, as I'm pretty sure the sonoffs don't have ieee of all zeroes.

smmoroz, 2020-11-05:
Just to be sure, that I understand you correctly. Does it mean that Legrand does not implement `Mgmt_Lqi_rsp` correctly? But why in this case for some devices `ieee` is correct, and for some others `ieee=00:00:00:00:00:00:00:00`. For the latter `nwk` is returned correctly. Can't we use `nwk` to reconstruct `ieee`?  At least at [this](https://github.com/zigpy/zigpy/blob/e83733847e848c9a52ef76e4a3026add640c72ec/zigpy/topology.py#L67) level, we have `nwk --> ieee` mapping.

Adminiuga, 2020-11-05:
Yes - Legrand does not implement Mgmt_Lqi_rsp correctly. In retrospect that is not the only thing Legrand implemented in their specific way for some reason :shrug:
I don't know why it shows `ieee=00:00:00:00:00:00:00:00` for some, but not the others. May have to do that those are Sonoffs???  Some one else reported similar issue long time ago with zha_map. 
We could use NWK to do lookup, but depends. ieee should be unique. NWK may change.

pipiche38, 2020-11-05:
Could you kindly be more precised on the Legrand issue ?

I never saw any issue on Legrand devices badly reporting neighbor when using the NXP Zigbee 3.0 stack .

Last if you would like me to relay some issues to Legrand please feel free.

Adminiuga, 2020-11-05:
question is, why for some reported neighbors the IEEE address is all zeroes
```
2020-11-03 14:11:15 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame 69 (incomingMessageHandler) received: b'00000031800000400100001e5cb3d783ffff317200060402d7ede226b79dcdedb44305feffd76b08a1d02501038cd7ede226b79dcded00000000000000006893250103bf'
2020-11-03 14:11:15 DEBUG (MainThread) [bellows.zigbee.application] Received incomingMessageHandler frame with [<EmberIncomingMessageType.INCOMING_UNICAST: 0>, EmberApsFrame(profileId=0, clusterId=32817, sourceEndpoint=0, destinationEndpoint=0, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY|APS_OPTION_RETRY: 320>, groupId=0, sequence=30), 92, -77, 0x83d7, 255, 255, b'r\x00\x06\x04\x02\xd7\xed\xe2&\xb7\x9d\xcd\xed\xb4C\x05\xfe\xff\xd7k\x08\xa1\xd0%\x01\x03\x8c\xd7\xed\xe2&\xb7\x9d\xcd\xed\x00\x00\x00\x00\x00\x00\x00\x00h\x93%\x01\x03\xbf']
2020-11-03 14:11:15 DEBUG (MainThread) [zigpy.neighbor] [0x83d7] request status: Status.SUCCESS. response: Neighbors(entries=6, start_index=4, neighbor_table_list=[Neighbor(extended_pan_id=ed:cd:9d:b7:26:e2:ed:d7, ieee=08:6b:d7:ff:fe:05:43:b4, nwk=0xD0A1, packed=37, permit_joining=<PermitJoins.Accepting: 1>, depth=3, lqi=140), Neighbor(extended_pan_id=ed:cd:9d:b7:26:e2:ed:d7, ieee=00:00:00:00:00:00:00:00, nwk=0x9368, packed=37, permit_joining=<PermitJoins.Accepting: 1>, depth=3, lqi=191)])
```

Neighbor(extended_pan_id=ed:cd:9d:b7:26:e2:ed:d7, ieee=00:00:00:00:00:00:00:00, nwk=0x9368, packed=37, permit_joining=<PermitJoins.Accepting: 1>, depth=3, lqi=191)

Maybe it has something to do with the neighbor being a Sonoff. I never saw this personally.

pipiche38, 2020-11-05:
Find here after a Neightbor table build for a Netatmo-Legrand device

[image]

smmoroz, 2020-11-05:
@pipiche38 , here is a list of my devices.

[image]

I highlighted in red devices located in one room (in kitchen).
```
0x0dc4 - central light (Legrand Dimmer)
0x5c41 - main light (Legrand Dimmer) 
0x4af7 - motion sensor kitchen (Sonoff)
0x54f6 - magnet door (Sonoff)
```

I paired devices `0x4af7` and `0x54f6` through either `0x0dc4` or 0c5c41`. 

Here is the debug output of `zha_map.scan_now` command:

```
2020-11-05 20:15:51 DEBUG (MainThread) [zigpy.topology] Scanning neighbors of 00:0b:57:ff:fe:de:e9:02/0x0000 device
2020-11-05 20:15:51 DEBUG (MainThread) [zigpy.neighbor] [0x0000] request status: Status.SUCCESS. response: Neighbors(entries=8, start_index=0, neighbor_table_list=[Neighbor(extended_pan_id=ed:cd:9d:b7:26:e2:ed:d7, ieee=00:04:74:00:00:0b:ce:db, nwk=0x4003, packed=37, permit_joining=<PermitJoins.Unknown: 2>, depth=15, lqi=49), Neighbor(extended_pan_id=ed:cd:9d:b7:26:e2:ed:d7, ieee=00:04:74:00:00:8b:77:6e, nwk=0x5C41, packed=37, permit_joining=<PermitJoins.Unknown: 2>, depth=15, lqi=54), Neighbor(extended_pan_id=ed:cd:9d:b7:26:e2:ed:d7, ieee=00:04:74:00:00:8b:77:5a, nwk=0x83D7, packed=37, permit_joining=<PermitJoins.Unknown: 2>, depth=15, lqi=82)])
2020-11-05 20:15:52 DEBUG (MainThread) [zigpy.neighbor] [0x0000] Querying next starting at 3
2020-11-05 20:15:52 DEBUG (MainThread) [zigpy.neighbor] [0x0000] request status: Status.SUCCESS. response: Neighbors(entries=8, start_index=3, neighbor_table_list=[Neighbor(extended_pan_id=ed:cd:9d:b7:26:e2:ed:d7, ieee=00:04:74:00:00:0b:0d:68, nwk=0xBD12, packed=37, permit_joining=<PermitJoins.Unknown: 2>, depth=15, lqi=74), Neighbor(extended_pan_id=ed:cd:9d:b7:26:e2:ed:d7, ieee=08:6b:d7:ff:fe:05:43:b4, nwk=0xD0A1, packed=37, permit_joining=<PermitJoins.Unknown: 2>, depth=15, lqi=148), Neighbor(extended_pan_id=ed:cd:9d:b7:26:e2:ed:d7, ieee=00:15:8d:00:02:00:99:f3, nwk=0xABB9, packed=18, permit_joining=<PermitJoins.NotAccepting: 0>, depth=1, lqi=210)])
2020-11-05 20:15:53 DEBUG (MainThread) [zigpy.neighbor] [0x0000] Querying next starting at 6
2020-11-05 20:15:54 DEBUG (MainThread) [zigpy.neighbor] [0x0000] request status: Status.SUCCESS. response: Neighbors(entries=8, start_index=6, neighbor_table_list=[Neighbor(extended_pan_id=ed:cd:9d:b7:26:e2:ed:d7, ieee=00:15:8d:00:04:83:ae:59, nwk=0xAF63, packed=18, permit_joining=<PermitJoins.NotAccepting: 0>, depth=1, lqi=110), Neighbor(extended_pan_id=ed:cd:9d:b7:26:e2:ed:d7, ieee=00:15:8d:00:04:2c:ca:1f, nwk=0x582B, packed=18, permit_joining=<PermitJoins.NotAccepting: 0>, depth=1, lqi=176)])
2020-11-05 20:15:54 DEBUG (MainThread) [zigpy.neighbor] [0x0000] Done scanning. Total 8 neighbours
2020-11-05 20:15:57 DEBUG (MainThread) [zigpy.topology] Scanning neighbors of 08:6b:d7:ff:fe:05:43:b4/0xd0a1 device
2020-11-05 20:17:59 DEBUG (MainThread) [zigpy.topology] Scanning neighbors of 00:04:74:00:00:0b:0d:68/0xbd12 device
2020-11-05 20:18:00 DEBUG (MainThread) [zigpy.neighbor] [0xbd12] request status: Status.SUCCESS. response: Neighbors(entries=6, start_index=0, neighbor_table_list=[Neighbor(extended_pan_id=ed:cd:9d:b7:26:e2:ed:d7, ieee=00:04:74:00:00:0b:ce:db, nwk=0x4003, packed=37, permit_joining=<PermitJoins.Accepting: 1>, depth=1, lqi=135), Neighbor(extended_pan_id=ed:cd:9d:b7:26:e2:ed:d7, ieee=08:6b:d7:ff:fe:05:43:b4, nwk=0xD0A1, packed=37, permit_joining=<PermitJoins.Accepting: 1>, depth=1, lqi=7)])
2020-11-05 20:18:01 DEBUG (MainThread) [zigpy.neighbor] [0xbd12] Querying next starting at 2
2020-11-05 20:18:01 DEBUG (MainThread) [zigpy.neighbor] [0xbd12] request status: Status.SUCCESS. response: Neighbors(entries=6, start_index=2, neighbor_table_list=[Neighbor(extended_pan_id=ed:cd:9d:b7:26:e2:ed:d7, ieee=00:04:74:00:00:8b:7a:12, nwk=0x0DC4, packed=37, permit_joining=<PermitJoins.Accepting: 1>, depth=1, lqi=5), Neighbor(extended_pan_id=ed:cd:9d:b7:26:e2:ed:d7, ieee=00:0b:57:ff:fe:de:e9:02, nwk=0x0000, packed=36, permit_joining=<PermitJoins.Accepting: 1>, depth=0, lqi=69)])
2020-11-05 20:18:07 DEBUG (MainThread) [zigpy.neighbor] [0xbd12] Querying next starting at 4
2020-11-05 20:18:07 DEBUG (MainThread) [zigpy.neighbor] [0xbd12] request status: Status.SUCCESS. response: Neighbors(entries=6, start_index=4, neighbor_table_list=[Neighbor(extended_pan_id=ed:cd:9d:b7:26:e2:ed:d7, ieee=00:04:74:00:00:8b:77:6e, nwk=0x5C41, packed=37, permit_joining=<PermitJoins.Accepting: 1>, depth=1, lqi=5), Neighbor(extended_pan_id=ed:cd:9d:b7:26:e2:ed:d7, ieee=00:04:74:00:00:8b:77:5a, nwk=0x83D7, packed=37, permit_joining=<PermitJoins.Accepting: 1>, depth=1, lqi=3)])
2020-11-05 20:18:07 DEBUG (MainThread) [zigpy.neighbor] [0xbd12] Done scanning. Total 6 neighbours
2020-11-05 20:18:10 DEBUG (MainThread) [zigpy.topology] Scanning neighbors of 00:04:74:00:00:0b:ce:db/0x4003 device
2020-11-05 20:18:12 DEBUG (MainThread) [zigpy.neighbor] [0x4003] request status: Status.SUCCESS. response: Neighbors(entries=5, start_index=0, neighbor_table_list=[Neighbor(extended_pan_id=ed:cd:9d:b7:26:e2:ed:d7, ieee=00:04:74:00:00:0b:0d:68, nwk=0xBD12, packed=37, permit_joining=<PermitJoins.Accepting: 1>, depth=2, lqi=137), Neighbor(extended_pan_id=ed:cd:9d:b7:26:e2:ed:d7, ieee=00:0b:57:ff:fe:de:e9:02, nwk=0x0000, packed=36, permit_joining=<PermitJoins.Accepting: 1>, depth=0, lqi=25)])
2020-11-05 20:18:13 DEBUG (MainThread) [zigpy.neighbor] [0x4003] Querying next starting at 2
[... cut ...] d=ed:cd:9d:b7:26:e2:ed:d7, ieee=00:00:00:00:00:00:00:00, nwk=0x4AF7, packed=37, permit_joining=<PermitJoins.Accepting: 1>, depth=4, lqi=111)])
2020-11-05 20:48:48 DEBUG (MainThread) [zigpy.neighbor] [0x5c41] ignoring invalid neighbor: 00:00:00:00:00:00:00:00
2020-11-05 20:48:48 DEBUG (MainThread) [zigpy.neighbor] [0x5c41] Done scanning. Total 7 neighbours
```

As we can see, `ieee=00:15:8d:00:04:21:df:ca, nwk=0xD561` is properly captured. 

So, what is the reason of the issue? Legrand or Sonoff?

Adminiuga, 2020-11-05:
- do you have non legrand routers on the network?
- do those non legrand routers report Sonoff neighbours with a valid IEEE?

if answer to both is yes --> legrand problem. 
E.g. Does the ikea driver report any Sonoff neighbours?

smmoroz, 2021-08-18:
Hi, guys. I'm sorry for not being able to answer before. I just tested Sonoff HT sensor (SNZB-02) paired through IKEA router. And it works well. IKEA correctly shows it as neighbor. 

[image]

So, I conclude that the issue is with Legrand router. Does anyone know how to report the issue to Legrand? Or maybe it is already fixed in a more recent router firmware?

smmoroz, 2021-10-31:
I have updated one Legrand router firmware to version 57 (0x0039). I repaired some Sonoff devices to this router. And finally the routing table works!

[image]

So, it was definitely Legrand router issue. I close the issue now.

smmoroz, 2021-11-01:
I was too fast. Just after repairing the end devices to Legrand router, network topology was correctly shown. But after about 80 mins all end devices have disappeared from the topology. In particular, all "RxOnWhenIdle.Off" & "RelationShip.Child" devices have disappeared, see below:

[image]

At the same time, although these end devices are not shown as children of any device, they are online, they still work, they send messages, etc.

As I'm not a Zigbee expert, can anyone help me to understand, is this a normal behavior?

Adminiuga, 2021-11-01:
Again: zigpy builds this based on what devices are reporting. If Legrand does not report these children in it's neighbors table, then zigpy wouldn't know.

smmoroz, 2021-11-01:
I understand. I have just send a message to Legrand [here](https://developer.legrand.com/forums/topic/zigbee-neighbor-table-issue/)

smmoroz, 2021-11-07:
Legrand doesn't want to investigate with third-party devices. I close the issue.

closed 2021-11-07 by smmoroz