zigpy/zigpy #541: ZHA is unstable

Jens-Wymeersch, 2020-11-03:
Dear,

I've installed the Sonoff bridge a few weeks ago. Although the start was smoothly, I'm always scared that ZHA won't work. 
Sometimes the devices aren't accessible and a few minutes/hours later there back on. I don't know what to do to make it stabler nor that I know how to get it back up when it's down.
Here are a subset of the log messages I get:

`Logger: zigpy.zcl
Source: /usr/local/lib/python3.8/site-packages/zigpy/zcl/__init__.py:110
First occurred: 13:28:50 (1172 occurrences)
Last logged: 20:50:38

[0xe6c7:1:0x0000] Unknown cluster-specific command 10
[0x8789:1:0x0000] Unknown cluster-specific command 10
[0x0974:1:0x0000] Unknown cluster-specific command 10
[0x1ff3:1:0x0000] Unknown cluster-specific command 10
[0x3caf:1:0x0000] Unknown cluster-specific command 10`

`Logger: bellows.zigbee.application
Source: /usr/local/lib/python3.8/site-packages/bellows/zigbee/application.py:642
First occurred: 15:48:22 (17 occurrences)
Last logged: 19:09:23

Watchdog heartbeat timeout:`

`Logger: homeassistant
Source: /usr/src/homeassistant/homeassistant/runner.py:115
First occurred: 18:20:26 (1 occurrences)
Last logged: 18:20:26

Error doing job: Task exception was never retrieved
Traceback (most recent call last):
  File "/usr/local/lib/python3.8/site-packages/zigpy/device.py", line 201, in request
    raise zigpy.exceptions.DeliveryError(
zigpy.exceptions.DeliveryError: [0x41ba:1:0x0500]: Message send failure`

Can someone help ?

Regards,
Jens Wymeersch

Adminiuga, 2020-11-03:
> Unknown cluster-specific command 10

That's normal. Ignore it. That command is not defined in ZCL specs. 

```
Logger: bellows.zigbee.application
Source: /usr/local/lib/python3.8/site-packages/bellows/zigbee/application.py:642
First occurred: 15:48:22 (17 occurrences)
Last logged: 19:09:23

Watchdog heartbeat timeout:
```

🤷 not enough details. But since you are connecting to the gateway through WiFi network and nobody knows how good it is then maybe it's not bad. And officially serial over TCP not really supported/approved. Exactly of the issues like this, because bad network translates to bad zigpy/zha experience

```
File "/usr/local/lib/python3.8/site-packages/zigpy/device.py", line 201, in request
raise zigpy.exceptions.DeliveryError(
zigpy.exceptions.DeliveryError: [0x41ba:1:0x0500]: Message send failure`
```

Device is not responding. Normal part of operation. 

This issue really does not belong here, as it is not a support forum. I suggest checking with the #zigbee channel on HA discord server for support.

closed 2020-11-03 by Adminiuga

Jens-Wymeersch, 2020-11-04:
@Adminiuga 

Today again, my zigbee devices aren't reachable and then suddenly without any interaction of myself they start again working. Can you please help me figure out what's going on and how I can avoid this ?
Thank you so much,
Jens

Adminiuga, 2020-11-04:
1. Stop tagging people
2. enable debug logging https://www.home-assistant.io/integrations/zha/#debug-logging
3. Analyze the logs for clues
4. check with the #zigbee channel on HA discord server

Make the WiFi network stable or get an elelabs stick. "Watchdog heartbeat timeout" means there's no connection to Sonoff bridge and that's why it is unsupported.

Jens-Wymeersch, 2020-11-04:
I'll follow your advice and see where i get