zigpy/zigpy #542: Travis CI for zigpy not active?

Hedda, 2020-11-06:
Travis CI for zigpy not currently active?

https://travis-ci.org/github/zigpy/zigpy

Used by bellows, zigpy-znp, zigpy-deconz, zigpy-xbee, zigpy-cc, and /zha-device-handlers

https://github.com/zigpy/bellows -> https://travis-ci.org/github/zigpy/

https://github.com/zigpy/zigpy-znp -> https://travis-ci.org/github/zigpy/zigpy-znp

https://github.com/zigpy/zigpy-deconz -> https://travis-ci.org/github/zigpy/zigpy-deconz

https://github.com/zigpy/zigpy-xbee -> https://travis-ci.org/github/zigpy/zigpy-xbee

https://github.com/zigpy/zigpy-cc -> https://travis-ci.org/github/zigpy/zigpy-cc

https://github.com/zigpy/zigpy-zigate -> https://travis-ci.org/github/zigpy/zigpy-zigate

https://github.com/zigpy/zha-device-handlers -> https://travis-ci.org/github/zigpy/zha-device-handlers

puddly, 2020-11-06:
Travis CI's new pricing structure and small allotment of build minutes per project will last zigpy only one month. We are migrating to GitHub Actions in the meantime.

closed 2020-11-06 by Adminiuga

Adminiuga, 2020-11-06:
I meant to say no more travis. That's intended. `travis.yaml` is still there so you are free to run it on your repo.