zigpy/zigpy #544: Allow connecting to existing networks

dajudge, 2020-11-06:
I have two conbee2 USB sticks (connected to two different PCs).

One forms a network w/ several devices already joined.

For debugging/development purposes I want the other stick to join the existing network.

Is this already possible w/ zigpy? If so there seems to be little documentation / sample code for this use case - maybe you could give me some pointers here?

MattWestb, 2020-11-08:
In the zigbee world its only allowed /possible to have one trustcenter that is manage the security off the network. Therefore its is NOT possible for one second coordinator joining one existing network.
You can joining it a normal router without any function and can forwarding /  routing traffic in the network.

If you have one good firmware with one "controller bridge" it can receiving and sending commands to all devices but i have not seen that for Rasp/CornBee devices.

Adminiuga, 2020-11-08:
What do you want to debug? For dev purposes just form another network and have test devices join in. For Zigbee sniffer just install the sniffer firmware.
But zigpy/radio libs currently do not support joining networks.

dajudge, 2020-11-08:
Thanks for your replies! I am using a conbee 2 on my homeserver to control my zigbee network and I was hoping I could develop/debug w/ the same network using a 2nd conbee 2. But if that's not possible a 2nd network for dev purposes seems to be the way to go :-)

Hedda, 2020-11-30:
You can use the second ConBee as a Zigbee network packet sniffer with Wireshark using Phoscon ZSHARK extension:

https://phoscon.de/en/conbee/software

"ZSHARK extension for network analysis sniffer software Wireshark".

Though I believe most common is to use a Texas Instruments as sniffers because of their low cost and available:

https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html

https://www.ti.com/tool/PACKET-SNIFFER

You can of course also sniff with a Silicon Labs Ember / EmberZNet adapter

https://github.com/zsmartsystems/com.zsmartsystems.zigbee.sniffer

Your network sniffers does not join the network but instead passively eavesdrop on radio signals to listen in on the traffic.

closed 2020-12-01 by Adminiuga