zigpy/zigpy #547: Schneider Electric devices "invalid install code" on ZHA

axellebot, 2020-11-10:
Hey !

I'm using [Home-Assistant](https://github.com/home-assistant/core) with [ZHA integration](https://www.home-assistant.io/integrations/zha/) and I recently bought a new zigbee device from Schneider Electric ([this product ](https://www.se.com/ww/en/product/NU550818F/new-unica---wiser-blind-control-switch---white-%28p%29/) matching [Zigbee Alliance](https://zigbeealliance.org/zigbee_products/wiser-blinds-and-shutters-control-switch/))

I'm trying to install this zigbee device through `zha.permit` service with the `qr_code` or with the `ieee_source` and `install_code` tuple. According to my code review it looks like the qr_code fit [an existing pattern](https://github.com/home-assistant/core/blob/4b7829d777bd4eb701ef1260a73e1d1ccce51141/homeassistant/components/zha/core/helpers.py#L263) but it does throw an "invalid install code" message that might come from [here](https://github.com/zigpy/zigpy/blob/dev/zigpy/util.py#L209-L219).

QR code ([MAC]|[install code]) : `83AC0DFEFF5ECFD0|6CD535528BAA3BF615DB5DC0560AD6F003DD`

Adminiuga, 2020-11-14:
Is there any install code printed on the product and/or documentation and does it match the QR code?
Because `6CD535528BAA3BF615DB5DC0560AD6F003DD` does not match the CRC. If we assume the code is correct, then the CRC (last two bytes) should be `6CD535528BAA3BF615DB5DC0560AD6F03468`

I've tried flashing the `6CD535528BAA3BF615DB5DC0560AD6F0` install code using Silabs tools and got the following 
[image]

I still believe that zigpy  does calculate code correctly or at least it matches the calculations of the Silabs tool and they control the standard. I think something is wrong with the QR code. Try comparing to the actual install code which usually is printed on the device.

Or try joining with the Install Code with the last two bytes replaced with `3468`, but that would only work if the code is right and CRC is wrong. BTW, does the ieee in the QR code matches the IEEE of the device when joining?

axellebot, 2020-11-14:
[image]

According to Google Lens and I :
```
QR CODE
83AC0DFEFF5ECFD0|6CD535528BAA3BF615DB5DC0560AD6F003DD
EU-64
83AC0DFEFF5ECFD0
Install code
6CD5 3552 8BAA 3BF6 15DB 5DC0 560A D6F0 03DD
```

I was not able to make the device join my zigbee network. There is no readable info about how to begin Zigbee peering process with the device.
According to a last email with the company this device is shipped with Zigbee `2.4.x` instead of `3.0` for now. They said the product should, for now, only work with their wiser gateway but should be working with others after upgrading to `3.0` in the next years (if they do).

Adminiuga, 2020-11-14:
:shrug: I don't think that's a valid code and/or correct CRC. If it is not known how to initiate the pairing process, then can't even try joining with the standard key

Adminiuga, 2020-11-18:
Gonna close this for now. BTW, most ZB3 devices still should be able to join the network with the well known link key. If it still does not joins, maybe it isn't really trying to join.

closed 2020-11-18 by Adminiuga

axellebot, 2021-04-03:
According to Schneider Electric customer service, I received a model with bluetooth enabled but Zigbee disabled because of factory settings. I didn't manage to enable zigbee even with their instructions.

Might be helpful for future purposes : 

Here was my model :
- According to product ref "NU550818F" : [SE](https://www.se.com/ww/en/product/NU550818F/new-unica---wiser-blind-control-switch---white-%28p%29/) 
- According to type number "NH3508" : [ZigbeeAllince](https://zigbeealliance.org/zigbee_products/wiser-blinds-and-shutters-control-switch/)