zigpy/zigpy #548: ZHA - Aqara Motion sensor timeout?

jvalst, 2020-11-12:
Just a small question here as well: Someone mentioned a software based solution which was available in zha_new component for the Aqara Motion Sensor Timeouts. (https://community.home-assistant.io/t/xiaomi-motion-sensor-with-zha-vs-zha-new/181215)

Is there a way to look into a way to get the Aqara motion sensors to work with less timeout? Now the timeout for motion is something like 2 minutes.

puddly, 2020-11-20:
Unless you've done the hardware mod to reduce their reset period, the two minute timeout is what it takes to get the sensor to stay triggered continuously as long as there is motion.

If you want to change it you can either modify ZHA or patch it at runtime with this custom component https://github.com/walthowd/ha-constant-modifier (see the example config, specifically the second option).

Hedda, 2020-11-30:
Believe it refers to this Aqara / Xiaomi hardware modification to make it send motion every 5 seconds instead of every 2 minutes:

https://community.home-assistant.io/t/aqara-motion-sensor-hack-for-5-sec/147959

https://community.smartthings.com/t/making-xiaomi-motion-sensor-a-super-motion-sensor/139806

closed 2021-09-07 by github-actions[bot]