zigpy/zigpy #552: PiZiGate + TS0601

Kermit79, 2020-11-17:
Hello,
I bought a Tuya thermostatic valve (TS0601) and I would like to pair it with my Pizigate, but it doesn't work very well...
The pair is ok, but i can't read anything, and ZigPy doesn't create any entity.

I put the verbose mode, this is the log :
`
2020-11-17 22:40:21 DEBUG (SyncWorker_23) [homeassistant.components.ssdp] Scanning
2020-11-17 22:40:32 DEBUG (MainThread) [zigpy_zigate.uart] Frame received: 004d000d49c9e4bc33acfffe63fb1e800000
2020-11-17 22:40:32 DEBUG (MainThread) [zigpy_zigate.api] data received 0x4d b'c9e4bc33acfffe63fb1e8000' LQI:0
2020-11-17 22:40:32 DEBUG (MainThread) [zigpy_zigate.zigbee.application] zigate_callback_handler [0xc9e4, bc:33:ac:ff:fe:63:fb:1e, 128]
2020-11-17 22:40:32 INFO (MainThread) [zigpy.application] Device 0xc9e4 (bc:33:ac:ff:fe:63:fb:1e) joined the network
2020-11-17 22:40:32 INFO (MainThread) [zigpy.device] [0xc9e4] Requesting 'Node Descriptor'
2020-11-17 22:40:32 DEBUG (MainThread) [zigpy.util] Tries remaining: 2
2020-11-17 22:40:32 DEBUG (MainThread) [zigpy.device] [0xc9e4] Extending timeout for 0x13 request
2020-11-17 22:40:32 DEBUG (MainThread) [zigpy_zigate.zigbee.application] request (0xC9E4, 0, <ZDOCmd.Node_Desc_req: 0x0002>, 0, 0, 19, b'\x13\xe4\xc9', True, False)
2020-11-17 22:40:32 DEBUG (MainThread) [zigpy_zigate.uart] Send: 0x530 b'02c9e400000002000000000313e4c9'
2020-11-17 22:40:32 DEBUG (MainThread) [zigpy_zigate.uart] Frame to send: b'\x050\x00\x0f*\x02\xc9\xe4\x00\x00\x00\x02\x00\x00\x00\x00\x03\x13\xe4\xc9'
2020-11-17 22:40:32 DEBUG (MainThread) [zigpy_zigate.uart] Frame escaped: b'\x02\x150\x02\x10\x02\x1f*\x02\x12\xc9\xe4\x02\x10\x02\x10\x02\x10\x02\x12\x02\x10\x02\x10\x02\x10\x02\x10\x02\x13\x13\xe4\xc9'
2020-11-17 22:40:32 DEBUG (MainThread) [homeassistant.components.websocket_api.http.connection.2869733360] Sending {'id': 55, 'type': 'event', 'event': {'type': 'log_output', 'log_entry': {'name': 'zigpy.application', 'message': ['Device 0xc9e4 (bc:33:ac:ff:fe:63:fb:1e) joined the network'], 'level': 'INFO', 'source': ('/usr/local/lib/python3.8/site-packages/zigpy/application.py', 188), 'timestamp': 1605649232.6054955, 'exception': '', 'count': 1, 'first_occurred': 1605649232.6054955}}}
2020-11-17 22:40:32 DEBUG (MainThread) [homeassistant.components.websocket_api.http.connection.2869733360] Sending {'id': 55, 'type': 'event', 'event': {'type': 'device_joined', 'nwk': 0xC9E4, 'ieee': 'bc:33:ac:ff:fe:63:fb:1e'}}
2020-11-17 22:40:32 DEBUG (MainThread) [homeassistant.components.websocket_api.http.connection.2869733360] Sending {'id': 55, 'type': 'event', 'event': {'type': 'log_output', 'log_entry': {'name': 'zigpy.device', 'message': ["[0xc9e4] Requesting 'Node Descriptor'"], 'level': 'INFO', 'source': ('/usr/local/lib/python3.8/site-packages/zigpy/device.py', 84), 'timestamp': 1605649232.6085906, 'exception': '', 'count': 1, 'first_occurred': 1605649232.6085906}}}
2020-11-17 22:40:32 DEBUG (MainThread) [homeassistant.components.websocket_api.http.connection.2869733360] Sending {'id': 55, 'type': 'event', 'event': {'type': 'log_output', 'log_entry': {'name': 'zigpy.util', 'message': ['Tries remaining: 2'], 'level': 'DEBUG', 'source': ('/usr/local/lib/python3.8/site-packages/zigpy/util.py', 108), 'timestamp': 1605649232.6146495, 'exception': '', 'count': 1, 'first_occurred': 1605649232.6146495}}}
2020-11-17 22:40:32 DEBUG (MainThread) [homeassistant.components.websocket_api.http.connection.2869733360] Sending {'id': 55, 'type': 'event', 'event': {'type': 'log_output', 'log_entry': {'name': 'zigpy.device', 'message': ['[0xc9e4] Extending timeout for 0x13 request'], 'level': 'DEBUG', 'source': ('/usr/local/lib/python3.8/site-packages/zigpy/device.py', 176), 'timestamp': 1605649232.6154602, 'exception': '', 'count': 1, 'first_occurred': 1605649232.6154602}}}
2020-11-17 22:40:32 DEBUG (MainThread) [zigpy_zigate.uart] Frame received: 80000007ca00780530000000
2020-11-17 22:40:32 DEBUG (MainThread) [zigpy_zigate.api] data received 0x8000 b'007805300000' LQI:0
2020-11-17 22:40:32 DEBUG (MainThread) [zigpy_zigate.zigbee.application] zigate_callback_handler [0, 120, 1328, b'\x00\x00']
2020-11-17 22:40:32 DEBUG (MainThread) [zigpy_zigate.uart] Frame received: 8002001aa20000000013000002c9e402fffd82e4c91efb63feffac33bc808d
2020-11-17 22:40:32 DEBUG (MainThread) [zigpy_zigate.api] data received 0x8002 b'0000000013000002c9e402fffd82e4c91efb63feffac33bc80' LQI:141
2020-11-17 22:40:32 DEBUG (MainThread) [zigpy_zigate.zigbee.application] zigate_callback_handler [0, 0, 19, 0, 0, <Address address_mode=ADDRESS_MODE.NWK address=0xc9e4>, <Address address_mode=ADDRESS_MODE.NWK address=0xfffd>, b'\x82\xe4\xc9\x1e\xfbc\xfe\xff\xac3\xbc\x80']
2020-11-17 22:40:32 DEBUG (MainThread) [zigpy.zdo] [0xc9e4:zdo] ZDO request ZDOCmd.Device_annce: [0xC9E4, bc:33:ac:ff:fe:63:fb:1e, 128]
2020-11-17 22:40:32 DEBUG (MainThread) [homeassistant.components.websocket_api.http.connection.2869733360] Sending {'id': 55, 'type': 'event', 'event': {'type': 'log_output', 'log_entry': {'name': 'zigpy.zdo', 'message': ['[0xc9e4:zdo] ZDO request ZDOCmd.Device_annce: [0xC9E4, bc:33:ac:ff:fe:63:fb:1e, 128]'], 'level': 'DEBUG', 'source': ('/usr/local/lib/python3.8/site-packages/zigpy/zdo/__init__.py', 54), 'timestamp': 1605649232.8068078, 'exception': '', 'count': 1, 'first_occurred': 1605649232.8068078}}}
2020-11-17 22:40:32 DEBUG (MainThread) [zigpy_zigate.uart] Frame received: 8002001f320000008002000002c9e40200001300e4c90240800210525200002c52000090
2020-11-17 22:40:32 DEBUG (MainThread) [zigpy_zigate.api] data received 0x8002 b'0000008002000002c9e40200001300e4c90240800210525200002c520000' LQI:144
2020-11-17 22:40:32 DEBUG (MainThread) [zigpy_zigate.zigbee.application] zigate_callback_handler [0, 0, 32770, 0, 0, <Address address_mode=ADDRESS_MODE.NWK address=0xc9e4>, <Address address_mode=ADDRESS_MODE.NWK address=0x0000>, b'\x13\x00\xe4\xc9\x02@\x80\x02\x10RR\x00\x00,R\x00\x00']
2020-11-17 22:40:32 DEBUG (MainThread) [zigpy_zigate.uart] Frame received: 87010005070000040080
2020-11-17 22:40:32 DEBUG (MainThread) [zigpy_zigate.api] data received 0x8701 b'00000400' LQI:128
2020-11-17 22:40:32 DEBUG (MainThread) [zigpy_zigate.zigbee.application] zigate_callback_handler [0, 0]
2020-11-17 22:40:32 INFO (MainThread) [zigpy.device] [0xc9e4] Node Descriptor: NodeDescriptor(byte1=2, byte2=64, mac_capability_flags=128, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=0)
2020-11-17 22:40:32 INFO (MainThread) [zigpy.device] [0xc9e4] Discovering endpoints
2020-11-17 22:40:32 DEBUG (MainThread) [zigpy.util] Tries remaining: 3
2020-11-17 22:40:32 DEBUG (MainThread) [zigpy.device] [0xc9e4] Extending timeout for 0x15 request
2020-11-17 22:40:32 DEBUG (MainThread) [zigpy_zigate.zigbee.application] request (0xC9E4, 0, <ZDOCmd.Active_EP_req: 0x0005>, 0, 0, 21, b'\x15\xe4\xc9', True, False)
2020-11-17 22:40:32 DEBUG (MainThread) [zigpy_zigate.uart] Send: 0x530 b'02c9e400000005000000000315e4c9'
2020-11-17 22:40:32 DEBUG (MainThread) [zigpy_zigate.uart] Frame to send: b'\x050\x00\x0f+\x02\xc9\xe4\x00\x00\x00\x05\x00\x00\x00\x00\x03\x15\xe4\xc9'
2020-11-17 22:40:32 DEBUG (MainThread) [zigpy_zigate.uart] Frame escaped: b'\x02\x150\x02\x10\x02\x1f+\x02\x12\xc9\xe4\x02\x10\x02\x10\x02\x10\x02\x15\x02\x10\x02\x10\x02\x10\x02\x10\x02\x13\x15\xe4\xc9'
2020-11-17 22:40:32 DEBUG (MainThread) [homeassistant.components.websocket_api.http.connection.2869733360] Sending {'id': 55, 'type': 'event', 'event': {'type': 'log_output', 'log_entry': {'name': 'zigpy.device', 'message': ['[0xc9e4] Node Descriptor: NodeDescriptor(byte1=2, byte2=64, mac_capability_flags=128, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=0)'], 'level': 'INFO', 'source': ('/usr/local/lib/python3.8/site-packages/zigpy/device.py', 91), 'timestamp': 1605649232.8197913, 'exception': '', 'count': 1, 'first_occurred': 1605649232.8197913}}}
2020-11-17 22:40:32 DEBUG (MainThread) [homeassistant.components.websocket_api.http.connection.2869733360] Sending {'id': 55, 'type': 'event', 'event': {'type': 'log_output', 'log_entry': {'name': 'zigpy.device', 'message': ['[0xc9e4] Discovering endpoints'], 'level': 'INFO', 'source': ('/usr/local/lib/python3.8/site-packages/zigpy/device.py', 101), 'timestamp': 1605649232.8204286, 'exception': '', 'count': 1, 'first_occurred': 16056492 [... cut ...] e tsn=25 command_id=1>
2020-11-17 22:41:03 WARNING (MainThread) [zigpy.zcl] [0xc9e4:1:0xef00] Unknown cluster-specific command 1
2020-11-17 22:41:03 DEBUG (MainThread) [zigpy.zcl] [0xc9e4:1:0xef00] ZCL request 0x0001: b'\x02!p\x00\x00\x12\x08\x08\x16\x0f\x05\x11\x0f\x06\x0f\x0f\x07\x0f\x11\x1e\x14\x16\x00\x0f'
2020-11-17 22:41:03 DEBUG (MainThread) [zigpy.zcl] [0xc9e4:1:0xef00] No handler for cluster command 1
`
[http://pastebin.fr/73875](url)

I'll also search, I'll keep you informed.

Thanks !

Kermit79, 2020-11-17:
PS : I think zigpy-zigate is better than here, i posted overt there sorry !

Adminiuga, 2020-11-18:
It is not zigpy nor zigpy-zigate issue. Tuya uses a proprietary protocol. I can only suggest not supporting vendors who do proprietary protocols just to lock you in. 

An effort to make it work is done in https://github.com/zigpy/zha-device-handlers/pull/538

closed 2020-11-18 by Adminiuga