zigpy/zigpy #557: [RFC] Unified radio command line interface and standard network backup format

puddly, 2020-11-23:
At the moment both [bellows](https://github.com/zigpy/bellows/) and [zigpy-znp](https://github.com/zigpy/zigpy-znp/) have their own command line tools. I think it would be beneficial to replace both with a unified `zigpy` command that interfaces with new methods on `ControllerApplication`. This'll reduce reducated code within the various radio libraries and make it easier to expose features that aren't yet available in ZHA's web interface.

I will have to test this this more thoroughly but I believe the API-specific commands can actually be sent as ZDO requests to the coordinator, at least with ZNP, allowing for network scanning and channel changing to possibly be done in a radio-agnostic way.

For example:

```shell
$ zigpy [ezsp|znp|...] --device /dev/... --database ./zigbee.db [create|destroy|scan|...] [--arg1 ...]`
```

Some possible commands (most already exist as a part of `bellows`):

 - `info` (all are properties exposed by `ControllerApplication`, with the exception of `network_key`)
 - `create` (`ControllerApplication.form_network)`
 - `destroy` (may require a new method, or we can combine `create/destroy` into `reset`)
 - `backup`
 - `restore`
 - `scan` (and `scan -e`)
 - `update` (channel the channel or the network key and broadcast it to all routers)
 - `sniff` (only supported by bellows but it should be easy to support arbitrary additional commands)
 - Some minimal interface for just running the base `ControllerApplication`, permitting joins, sending commands, and so on? `IPython` is now `asyncio`-based so it may be a good base.

The `bellows` backup JSON contains enough information to migrate networks between coordinators so I think we should also come up with a common JSON backup format as well:

 - Since JSON doesn't support bytestrings, we have a few possible representations: `[222, 188, 58, 18]` is the easiest to serialize but decimal is unreadable when the common representation is hex everywhere else. Some alternatives are `"0x123ABCDE"`  and `"12:3A:BC:DE"`. ZNP uses the former, but I sort of prefer the latter, as it can be pasted into Wireshark.
 - Add a top-level `"radio"` key that identifies the generating radio library, its current version, and some hardware information. It may also be useful to hold radio-specific backup data (possibly NVRAM for Texas Instruments radios?).

I think ZiGate and possibly the Conbee II also expose enough network information to make this useful, though EZSP's inability to change the EPID more than once will be an issue. Is there a way to detect if this has never been done before on new hardware?

Any thoughts or suggestions?

Adminiuga, 2020-11-24:
Thoughts about the CLI tool being a separate package/module, just depending on zigpy???

And what is better long run?
1. having an interface allowing "radios" to "extend" CLI tool command set, by allowing to register radio lib specific commands? 
2. keeping all radio specific commands  in the tool source???

> Since JSON doesn't support bytestrings, we have a few possible representations: [222, 188, 58, 18] is the easiest to serialize but decimal is unreadable when the common representation is hex [...]

can JSON serialize/deserialize "0x123ABCDE" or "12:3A:BC:DE" without having to write custom serializers? IMO making JSON readable should not matter. You should be able just save/load it. And for human readable re-presentation there could be a CLI command

Hedda, 2020-11-30:
@doudz Do you a commonly shared unified CLI this would be usefull for the zigpy-zigate ZiGate radio library for zigpy as well?

doudz, 2020-11-30:
It could be useful to have an unified CLI because it would be easy to use, but some radio have limitations
For example, a backup of a zigate use a special format different than a backup of conbee, so can't backup zigate and restore on conbee
sniff mode isn't supported by zigate firmware (it requires a specific firmware)

puddly, 2020-11-30:
> Thoughts about the CLI tool being a separate package/module, just depending on zigpy???

Works for me.

> And what is better long run?
> 1. having an interface allowing "radios" to "extend" CLI tool command set, by allowing to register radio lib specific commands?
> 2. keeping all radio specific commands in the tool source???

I don't think having the tool directly rely on radio libraries would be very maintainable. I propose we instead have a friendly name to package name mapping (`ezsp` to `bellows`, `znp` to `zigpy_znp`, etc.) and then dynamically import custom commands from `radio_module.zigbee.cli` as decorated functions. That should keep the interface clean and allow for custom commands to be added and fixed within radio libraries without having to modify an unrelated package.

Plus, the testing infrastructure for many packages differ (e.g. ZNP's simulated adapter for integration tests) so it would not be maintainable to have a separate copy of that in `zigpy-cli`.

> can JSON serialize/deserialize "0x123ABCDE" or "12:3A:BC:DE" without having to write custom serializers? IMO making JSON readable should not matter. You should be able just save/load it. And for human readable re-presentation there could be a CLI command

No, but JSON is meant to be human readable. You're going to be doing serialization/deserialization anyways with `list(data)` and `bytes(text)` to handle bytearrays so why not use `"0x" + data.hex()` and `bytes.fromhex(text[2:])`? Or `":".join(f"{b:02X}" for b in data)` and `bytes.fromhex(text.replace(":", ""))`? Call them `serialize_bytes` and `deserialize_bytes` and drop their implementation in some `utils` submodule. I find it useful and since there's no extra code I don't see any downsides.

> For example, a backup of a zigate use a special format different than a backup of conbee, so can't backup zigate and restore on conbee

I think this is all we need for a network backup:

 - PAN ID and Extended PAN ID
 - Network update ID
 - Current channel
 - Stack version (may not be necessary)
 - Network key, key update ID, and frame counter
 - Frame counters and all children (may not be necessary)
 - Link keys for Zigbee 3.0 devices
 - More???

The radio library would then implement `ControllerApplication.network_backup` or something similar that returns a common object with this information and shuts down the network to prevent the frame counters from increasing. `ControllerApplication.network_restore` would then write back the same radio settings (frame counter would need to be incremented either by the radio or within zigpy-cli before the restore).

Adminiuga, 2020-12-01:
> nd then dynamically import custom commands from radio_module.zigbee.cli as decorated functions. That should keep the interface clean and allow for custom commands to be added and fixed within radio libraries without having to modify an unrelated package.

I concur

> No, but JSON is meant to be human readable. You're going to be doing serialization/deserialization anyways with list(data) and bytes(text) to handle bytearrays so why not use "0x" + data.hex() and [...]

Well, it was designed to send object over "human readable text" which for me is different from human readable. Unless it is pretty-printed, json is hardly more human readable than xml. Yes you can do __"0x" + data.hex() and bytes.fromhex(text[2:])__ but why? when the purpose of json was to dump your data/object into a text file and then restore it? No arguing it could be done, but why complicate the code? as long as CLI can print any network params dump in the human readable form, just keep the basic json

>Stack version (may not be necessary)

Not necessary, but good to have the source radio type and stack version just as meta information

>The radio library would then implement ControllerApplication.network_backup or something similar that returns a common object with this information and shuts down the network to prevent the frame counters from increasing.

I'm fine with the method, but IMO shutting down the network (as in destroying the network) should be optional, as for me a valid use-case is dumping the network parameters just for backup purposes

MattWestb, 2020-12-02:
@puddly Is all NCPs having the list of primary channels (Standard 11. 15. 20 and 25) stored or is it only one thing in the radio lib ?
If the NCP have it its better also backing it up and restoring it then possible, i think.

Hedda, 2020-12-02:
> I think this is all we need for a network backup:
>
> * PAN ID and Extended PAN ID
> * Network update ID
> * Current channel
> * Stack version (may not be necessary)
> * Network key, key update ID, and frame [...]

Off-topic but it would be awesome to have a ZHA migration feature to be able to backup Zigbee2MQTT and then restore to zigpy.

Currently, it can be a daunting task to migrate a lot of Zigbee devices from Zigbee2MQTT to ZHA when switching in Home Assistant.

puddly, 2020-12-02:
> No arguing it could be done, but why complicate the code?

`data.hex()` and `bytes.fromhex(text)` IMO have clearer intent than `list(data)` and `bytes(array)`. Enough bikeshedding 😆.  If you don't care about the specific method, let's go with the former (without the `"0x"`). It's easier for me to debug with `jq backup.json` and is like eight characters of extra code.

> I'm fine with the method, but IMO shutting down the network (as in destroying the network) should be optional, as for me a valid use-case is dumping the network parameters just for backup purposes

Agreed. In that case we should increment the frame counter during restore by some large value so that restoration will actually work and not require a few hundred rejected transmissions.

> Off-topic but it would be awesome to have a ZHA migration feature to be able to backup Zigbee2MQTT and then restore to zigpy.

It should be possible to extract some of the required information from a Zigbee2MQTT coordinator backup with zigpy-znp once we have that information finalized. Migrating Z-Stack Home 1.2 networks probably won't work, however, as no NVRAM backup is performed and the network is just recreated from the config file (how does that work when the frame counters are reset?) Or nicely ask Zigbee2MQTT to switch to our format once we get it working?

Hedda, 2020-12-03:
> Or nicely ask Zigbee2MQTT to switch to our format once we get it working?

+1 If you try to make it non-product-specific then it does not hurt to ask 👍

I am sure people like to try switching back and forth between zigpy and Z2M.

puddly, 2020-12-15:
Format that I'm working with for now:

```Python
{
  "metadata": {
    # Minor version bumps are backwards compatible, major are not
    "version": [1, 0],

    # Package that generated the backup
    "source": "zigpy-znp",

    # Everything in here is opaque
    "internal": {
      "version": {
        "TransportRev": 2,
        "ProductId": 0,
        "MajorRel": 2,
        "MinorRel": 6,
        "MaintRel": 3,
        "CodeRevision": 20190608,
        "BootloaderBuildType": 2,
        "BootloaderRevision": 0
      }
    }
  },
  "coordinator_ieee": "5cacaa1c004b1200",
  "pan_id": "abaf",
  "extended_pan_id": "0d1e255f21d78367",
  "nwk_update_id": 0,
  "security_level": 5,
  "channel": 15,
  "network_key": {
    "key": "a5ba5afec488f3a5c43c350900ac6ba3",
    "sequence_number": 0,
    "frame_counter": 21251  # incremented by 2500 during restore
  }
}
```

I was unable to get any of the link keys from `bellows backup` to show up in NVRAM for any Texas Instruments coordinator after joining multiple Zigbee 3.0 routers, both with and without install codes.

Bidirectional migration between the ZZH running Z-Stack 4 and a CC2531 running Z-Stack Home 1.2 works but the reverse fails if routers are still a part of the old network, since Z-Stack will refuse to form a new network with that specific PAN ID if it discovers a collision. Changing the PAN ID back and forth and incrementing the `nwk_update_id` twice is an option if low-level hacks prove to be difficult, though I'm not sure how well Xiaomi devices will deal with this change.

I think it may be possible to do most of this within `ControllerApplication.form_network()` by adding a few extra keys, like `coordinator_ieee`, `frame_counter`, etc.? That way, zigpy radio libraries won't have to explicitly support network restoration.

Adminiuga, 2020-12-15:
No aps frame counter?

For ieee and keys, any reason not to separate bytes with  colon?

Should we include nwk and logical device type too? Just in case zigpy would support being other devices doing n the future?

puddly, 2020-12-15:
> No aps frame counter?

Is it necessary? It's only 8 bits so I believe wrapping is expected by other devices and should be handled.

> For ieee and keys, any reason not to separate bytes with colon?

At the moment I'm doing this the easy way with just `.serialize()`. The endianness will have to be handled per-field:

```Python
In [1]: import zigpy_znp.types as t

In [2]: t.EUI64.convert("00:12:4b:00:1c:ce:33:85").serialize().hex()
Out[2]: '8533ce1c004b1200'

In [3]: t.NWK(0xAABB).serialize().hex()
Out[3]: 'bbaa'

In [4]: t.KeyData(range(16)).serialize().hex()
Out[4]: '000102030405060708090a0b0c0d0e0f'
```

> Should we include nwk and logical device type too? Just in case zigpy would support being other devices doing n the future?

Maybe in version 1.1 or something, I don't want to make this too complicated right now. We'd also have to include the channel mask and the alternate key. Maybe the trust center address as well?

Adminiuga, 2020-12-15:
> Is it necessary? It's only 8 bits so I believe wrapping is expected by other devices and should be handled.

I think you have a different counter in mind. Silabs KB says to restore out going frame counter of the Trust Center (`b"ZigBeeAlliance09"`) key and restore that as APS frame counter. I'm pretty sure that counter is the same size as the network key frame counter. https://github.com/zigpy/bellows/blob/90c7c31bffab6ea45c5bc0bf08fa9e1844b1dfb6/bellows/cli/backup.py#L260-L264

Adminiuga, 2020-12-15:
>In [3]: t.NWK(0xAABB).serialize().hex()
>Out[3]: 'bbaa'

Well, that's hardly human readable for an NWK address :) just save it as `str(t.NWK(0xAABB))` and it would be better. 
Endianness is going to be a PITA, and that's why may be saving just raw python types in json makes more sense, cause then you don't have to deal with endianness? 

So maybe just save everything as string and keys in `"aa:bb:cc:dd:ee..."` format?

Hedda, 2021-02-02:
@puddly Great to read that castorw plan on implementing your unified backup format into zigbee-herdsman and zigbee2mqtt!  

https://github.com/Koenkk/zigbee-herdsman/pull/303

https://github.com/Koenkk/zigbee-herdsman/issues/286

👍👍

Hedda, 2021-02-03:
> At the moment both [bellows](https://github.com/zigpy/bellows/) and [zigpy-znp](https://github.com/zigpy/zigpy-znp/) have their own command line tools. I think it would be beneficial to replace both with a unified `zigpy` command that interfaces with new methods on `ControllerApplication`. This'll reduce duplicated code within the various radio libraries and make it easier to expose features that aren't yet available in ZHA's web interface.

What do you think about the idea of also implementing "bootloader" CLI command in this unified radio command line interface?

FYI, Adminiuga has already added support for "bootloader" CLI command in bellows last year -> https://github.com/zigpy/bellows/pull/249

If it possible to add support for "bootloader" CLI command in other radio libraries like zigpy-znp, zigpy-deconz and zigpy-zigate?

Launching bootloader could be very nice for firmware update via GUI implementations in integrations like ZHA for Home Assistant.

castorw, 2021-02-07:
Just a quick update. We have been working with @puddly to get the format working and were able to create unified backup format including successful backup&restore of everything essential - including APS encryption keys stored in 2 formats within Z-Stack (https://github.com/Koenkk/zigbee-herdsman/pull/303).

We have agreed to create a repo to track the format so I guess that should be available soon.

Hedda, 2021-02-08:
> Just a quick update. We have been working with @puddly to get the format working and were able to create unified backup format including successful backup&restore of everything essential - including APS encryption keys stored in 2 formats within Z-Stack ([Koenkk/zigbee-herdsman#303](https://github.com/Koenkk/zigbee-herdsman/pull/303)).
> 
> We have agreed to create a repo to track the format so I guess that should be available soon.

That's awesome! Slightly off-topic but wondering if you also plan on looking into compatibility with non-ZStack coordinators too?

That could possibly allow people to migrate back and forth to Zigbee coordinators from different manufacturers as well.

At least both zigpy and zigbee-herdsman already supports Z-Stack (Texas Instruments), deCONZ (Dresden Elektronik), and ZiGate today, and I read a rumour that someone might also be working to add EmberZNet (Silicon Labs) support to zigbee-herdsman.

castorw, 2021-02-08:
@Hedda not off-topic at all. That's exactly the point of the common backup format - to backup data essential for network restore on any stack which provides the instrumentation to restore the network data. My original motivation was to update the Z2M's format, which only backed up portions of NV in binary format. Now we have understandable data, which can be recovered on any platforms that allows for such.

The problem we have discovered is that Z-Stack uses TCLK seed to derive link keys for APS encryption. Therefore minimising the storage. The TCLK table is usually large enough on Z-Stack coordinators, so if you have TCLK seed stored you can recover the keys to TCLK table formats. In case you are migrating without TCLK seed (eg. from different coordinator which uses randomly generated keys) you need to utilise the `APS_LINK_KEY_TABLE` and `APS_LINK_KEY_DATA_TABLE`, which is currently limited to 3 entries in most Z-Stack firmware builds - so this is something that needs to be handled by Z-Stack firmware builders.

Below is a sample of backup we got to so far.
<details>
<summary>Z2M Coordinator Backup Sample</summary>
```json
{
  "metadata": {
    "version": 1,
    "source": "zigbee2mqtt",
    "internal": {
      "zhFormat": 2
    }
  },
  "stack_specific": {
    "zstack": {
      "tclk_seed": "4973f5450493c3cd90552c5f1d388a12"
    }
  },
  "coordinator_ieee": "00124b0006104e22",
  "pan_id": 52490,
  "extended_pan_id": "00124b0009d69f77",
  "nwk_update_id": 0,
  "security_level": 5,
  "channel": 21,
  "channel_mask": [
    21
  ],
  "network_key": {
    "key": "1a0355fd7abb5dc588a34998f8ee1233",
    "sequence_number": 0,
    "frame_counter": 96022
  },
  "devices": [
    {
      "nwk_address": 6873,
      "ieee_address": "04cf8cdf3c79455f",
      "link_key": {
        "key": "9c88e969f3d3d23cd557304f2ac9cb97",
        "rx_counter": 6,
        "tx_counter": 2521
      }
    },
    {
      "nwk_address": 27485,
      "ieee_address": "00124b002226ef87",
      "link_key": {
        "key": "4422b6772c140f380dfd6f51f50e1693",
        "rx_counter": 0,
        "tx_counter": 27650
      }
    },
    {
      "nwk_address": 7842,
      "ieee_address": "00158d0002ec571d"
    },
    {
      "nwk_address": 12931,
      "ieee_address": "00158d0004506f9a"
    },
    {
      "nwk_address": 1814,
      "ieee_address": "680ae2fffeae5647",
      "link_key": {
        "key": "0e255bbbfb71c9a5d70382a1e2da807a",
        "rx_counter": 20586,
        "tx_counter": 45980
      }
    },
    {
      "nwk_address": 9999,
      "ieee_address": "0f01020304050607",
      "link_key": {
        "key": "f1f2f3f4f5f6f7f8a1a2a3a4a5a6a7a8",
        "rx_counter": 20000,
        "tx_counter": 48506
      }
    }
  ]
}
```
</details>

MattWestb, 2021-02-08:
I think one god thing that  is also off topic is deleting / resetting the NVR in the NCP so getting one clean uninitiated NCP for the Z2M that ZHA already have (for TI NCP platforms) one script that can preforming it and dont need re flashing the NCP  if having getting corrupted NVR or "reused" one without.
Its very basic but good for the user then having strange problems and dont need extra HW for doing it.

puddly, 2021-02-08:
I've setup a repository to track this format's spec. It's a WIP: https://github.com/zigpy/open-coordinator-backup/

Hedda, 2021-02-10:
>  another consideration: with the latest firmware it is possible to copy the encryption keys to other modules, and I think may be I could use it to implement a failover to another HA instance in the future :)

@Shulyaka You think this type of backup and restore could be used in a failover-implementation for Zigbee with Home Assistant?

While the Zigbee specification does not support two Zigbee coordinators active at the same time, an idea could be to utilize a Zigbee coordinator backup and restore feature to achieve a high-availability function for your Zigbee network by having two instances of Home Assistant in a so-called "active-passive cluster" setup where you have a service that performs nightly backups of your Zigbee coordinator on the active-node and those backups could then be restored to the passive-node in the "active-passive cluster" setup with its own Home Assistant instance if it has compatible Zigbee hardware.

IIRC believe read that Adminiuga mentioned somewhere that he is thinking of letting ZHA in Home Assistant act as a Zigbee Router, and if support for that was implemented in ZHA then the passive-node in the "active-passive cluster" setup might not even always have to be an offline standby node, as instead, the passive-node could then possibly be configured as a Zigbee Router to run as an online standby instead, and only if the cluster service initiates a failover it could perform a restore of the last backup to the passive-node before making that the active-node.

Sorry, a high-availability discussion is off-topic here, but FYI that idea has been discussed before in https://github.com/home-assistant/architecture/issues/152

MattWestb, 2021-02-10:
@Hedda The problem with one "hot swap" situation in Zigbee network is that only one trustcenter can manager the security in the network. And the trustcenter link keys frame counters must being intact. If changing the hardware of the coordinator (IEEE) all link keys is invalid in the network and ther counters is not valid, The same scenario is for the network key and its counters.
At least for Silabs NCP they can have the trustcenter instance on one different IEEE and forwarding all TC calls to it. Then its possible changing active coordinator and pointing the trutcenter to the TC  hardware IEEE. I think its theoretically possible having more coordinators running with one trustcenter but i dont have seeing some  have trying it out (or i have misunderstanding how its working).

Hedda, 2021-05-05:
Congratulations on initial support for "Open ZigBee Coordinator Backup Format" in zigpy-znp v0.5.0 release. That is awesome!

https://github.com/zigpy/zigpy-znp/releases/tag/v0.5.0

"**_Huge thank you to @castorw from the Z2M project for working with me to bring concurrent support for the [Open Coordinator Backup Format](https://github.com/zigpy/open-coordinator-backup/) to Z2M. This allows for bidirectional migration between any coordinators supported by zigpy-znp. More documentation is available in [TOOLS.md#Backup and restore](https://github.com/zigpy/zigpy-znp/blob/dev/TOOLS.md#backup-and-restore)._**"

Hedda, 2021-05-17:
For reference; "ZNP Adapter Manager/Backup Refactor pull request by @castorw for zigbee-herdsman has now been merged:

https://github.com/Koenkk/zigbee-herdsman/pull/303

This means that Zigbee2MQTT and IoBroker can also uses the same unified "Open ZigBee Coordinator Backup Format" as zigpy:

https://github.com/zigpy/open-coordinator-backup

Hedda, 2021-11-15:
This is actively being worked on by puddly so not stale :)

Hedda, 2022-05-16:
For reference; this is related to the work in progress by puddly who has several draft pull requests which together addresses these:

https://github.com/zigpy/zigpy/issues/842

https://github.com/zigpy/zigpy/pull/848

https://github.com/zigpy/zigpy-cli/pull/2

https://github.com/zigpy/zigpy-znp/pull/97

https://github.com/zigpy/bellows/pull/445

https://github.com/zigpy/zigpy-deconz/pull/172

https://github.com/zigpy/zigpy-zigate/pull/117

https://github.com/zigpy/zigpy-xbee/pull/123

closed 2022-07-14 by puddly