zigpy/zigpy #561: Multi Endpoint device Behavior

sarakha63, 2020-11-26:
Hello i'm a developer on a home automation solution using zigpy. We are currently working on the integration of ZGP into zigpy but that is not the point here.
I'm working on the integration of a device. A danfoss master floor controller. This device is great. But i have some issues. 
First of all this device is a dynamic multi endpoint device. In fact depending on how many thermostats you link to it you will have 1 to 15 endpoints. Theres is also an endpoint for the controller itself and an endpoint for the greenpower proxy. So to resume we have 
1 endpoint  id232 (master controller)
1 endpoint id242 (greenpower proxy)
0-15 endpoints id 1-15 (thermostats endpoints)

Each endpoints as its own model and manufacturer . there are some manufacturer attributes (i currently handle with a quirk i created) but the issue is there :  

After initilization i can confirm that the zigpydevice will have manufacturer and model setted to the one of a random endpoint and all endpoints will have the manufacturer and model setted to the same

in fact each endpoints as its basic and i can confirm its manufacturer and model by reading the basic attribute. this one is good. but the endpoints objects and device object in zigpy will have a wrong manufacturer and model and what i noticed is one is picked and applied everywhere.

This is a big issue for quirks for example because some times the zigpy object will have one model and sometimes another atough the basic info is clearly good.
in fact all endpoints  1-15 will have a basic info with model 0x8030 (in my case)
endpoint 232 will have model 0x0200

but randomly upon restart and on initialisation zigpy device object and endpoints will always have the same and sometimes 0x8030 and 0x0200. This is exactly the same with power configuration sometimes the device will be on battery because considered as a model from endpoint 1-15 and sometimes DC powered as considered as a model from endpoint 232.

How does zigpy handle the attribution of device model/manuf/etccc and endpoints/model/manuf. Does it pick the first endpoint discovered and apply it everywhere ? this will be wrong for this kind of device. Or should it do it the right way and i must look deeper in the code to see waht bug is happening.

i will also have to discuss once this model issur solved with quirks.... for now WE HAVE to create 16 quirks with signature with the optionnal presence of endpoint 1-15......... is there any other way to handle this case. Because i dont really need a quirk for this device, it respects zigbee but it has some manufacturer attributes such as output status and i need to configure report on them. And i did not manage to configure report on an attribute that exsits if it is not declared in zigpy

Thanks

puddly, 2020-11-26:
Interesting. The ZCL spec does indicate that this behavior is allowed:

> Note: Where a node supports multiple endpoints it will often be the case that many of these settings willapply to the whole node, that is, they are the same for every endpoint on the node. In such cases they can be implemented once for the node, and mapped to each endpoint.

Some thoughts: 

 - The device model information is currently loaded by the `appdb` module (https://github.com/zigpy/zigpy/blob/79cd7d9e505fff1511c6d07a2e88d7fc462bfbf7/zigpy/appdb.py#L476). Perhaps we need an `ORDER BY ieee ASC, endpoint_id ASC` to make the loading order consistent and then pull the two attributes from the first endpoint with both attributes instead of updating them one by one from each? During discovery it's also set by `device` in a slightly different manner (https://github.com/zigpy/zigpy/blob/79cd7d9e505fff1511c6d07a2e88d7fc462bfbf7/zigpy/device.py#L127). Endpoints also do support models and manufacturers but the private attribute doesn't seem to be set within zigpy and it always falls back to the device's model/manufacturer. Maybe we need to move all of this out of appdb and into individual endpoints, with the device's attributes being chosen from the first endpoint that has them? Or are there really devices whose model and manufacturer are contained in different endpoints??

 - As for quirks, maybe we may need a lower-level matching system? Right now it's based heavily on hard-coded manufacturer/model names and endpoint/cluster signature. This is enough for writing simple quirks but I think a simpler solution would be to just use listener events everywhere and let custom quirks match if a callable returns `True`, not only if the hard-coded signature matching works. This can move a lot of quirks matching code into the quirks repo and allow for unusual devices such as this one to properly be matched, though fixing the random device model info may be enough.

sarakha63, 2020-11-26:
Hello thanks for your answer. I digged earlier into zigpy and noticed somme of the things you mention

1) i guess an order by is perhaps a consistent solution to not have change randomly. but in the case of this product the endpoint of the module is 232 and when thermostat are paired to it they use 1 to 15 with their own model id (that can be different there are different types of thermostat. I guess order by ie a good solution to avoid changing this randomly but not the real solution.  but order by is needed whatever. because not only the model is the issue but also version attributes/ power configuration for battery and so on. maybe after initialisation the app using zigpy could simply call a function to tell zigpy (for this device the reference endpoint is XXXX) and zigpy will then treat it like that...

2) I noticed that model manufcaturer and so on attributes exists for endpoint but were not used and always defined by the device. This could be a great "bugfix" improvement to have the good infos (app using zigpy could reliably use this info)

3) finally yes quirks is quite rigid for this case. because as you understood from my previous explanation. this device will report 0-16 endpoints depending on how the user configured it and how much zones "thermostat" he defined. Now im completely using the product on purpose of integration. I still did not PRed the quirks because of that. That will mean creating 16 quirks for handling the case 0-15 thermostats. In fact the only real case this devices needs a quirk is for being able to config_report a manufacturer attribute.... and for that in zigpy it must be defined with its friendly name

Thanks again. I'm willing to work on optimizing zigpy once the discussion leads to development.

MattWestb, 2020-11-26:
From "ZigBee Lighting & Occupancy Device Specification Version 1.0" 25 Control bridge.

25.1 Device configuration
When the control bridge device type is implemented on an endpoint, the following configurations
The application device version field of the corresponding simple descriptor SHALL be set to 0x1.
The device class SHALL be dynamic.

So dynamic endpoints is one part of the zigbee 3 standard.

But I think no open system have implanting it then normally devices is "simple" = fixed endpoints.

Adminiuga, 2020-11-27:
If you omit model/manufacturer from quirk, then it would do the match based on endpoint content, but that wouldn't help much if the content of the other endpoints is dynamic.

As about manufacturer/model for the quirk, during the device initialization, when scanning each endpoint it would try to read model and manufacturer if currently model/manufacturer is none.  And it scans the endpoints in the order of endpoint id as presented in the active endpoint response. Once it get a reply for model) manufacturer it stops doing that, as zigpy assumes a physical device can be only from a single manufacturer and it is going to be the same manufacturer across all endpoints.

When loading the device from the db, model/manufacturer is restored from the attribute cache and order is not really defined. The last restored basic cluster for the device takes precedence.

Adminiuga, 2020-11-27:
What's the purpose  of reporting different manufacturer/model for each endpoint? 
Per ZCL it should represent the model and manufacturer of the physical device and since it the same physical device it could only be from a single manufacturer, no?

sarakha63, 2020-11-27:
Hello

In fact @Adminiuga  usually 99% of the times even if there is multiple endpoints on devices they are the same. But not  always. That's why ZCL has a basic cluster for each endpoints.
And as @puddly  said  ZCL says :

> Note: Where a node supports multiple endpoints it will often be the case that many of these settings will
> apply to the whole node, that is, they are the same for every endpoint on the node. In such cases they can be implemented once for the node, and mapped to each endpoint.

"often be the case"   ----> "in such cases they can"

 This device reports other physical devices plugged to it trough endpoints. You can plug different things on it thats why endpoints can have different model.

One endpoint (id 232 in this case) reports the device itsefl. and endpoints 1-15 reports what is plugged on it and those can have deifferent models/  there own battery level / own software and so on.  

The dynamic part of reporting 1-15 endpoints depending on what is linked on the product is only a problem for quirks as for now i have to create 16 quirks, one for each combination even tough 1-15 is the same. And i need a quirk not because the device is not standard. but only because it has one usefull attribute (output status) where i need to configure report it and i noticed that i cant configure report on an attribute that as not be defined and cleraly declared with it's frinedly name

MattWestb, 2020-11-27:
I was only finding TI naming the different device types.
[What's New in Zigbee 3.0](https://www.ti.com/lit/an/swra615a/swra615a.pdf?ts=1606402301485&ref_url=https%253A%252F%252Fwww.google.com%252F) 6.3 Zigbee Device.

Adminiuga, 2020-11-27:
Well, it n this case just make the device ZCL compliant so it doesn't need a quirk? Having a quirk is not a requirement with zigpy.

sarakha63, 2020-11-27:
> Well, it n this case just make the device ZCL compliant so it doesn't need a quirk? Having a quirk is not a requirement with zigpy.

I don't understand ?

How do i handle a manufacturer attribute in the thermostat cluster for example if zigpy doesnt handle it (as it is a manufacturer attribute) and there is no quirk to handle it ?

[image]
[image]
[image]
[image]

Here are some screens that shows that the device respects zcl with dynamic endpoints with different models. and for thermostat cluster respects ZCL but has some manufacturer attributes 0x41xx

How to configure_report on those attributes without defining those in a quirk ?

Adminiuga, 2020-11-27:
What do you mean? You either do endpoints[ep_id].basic.read_attributes([4, 5]) or configure_reporting method if you want to receive attribute reports?

sarakha63, 2020-11-27:
reading attributes is not an issue.

but configure_reporting method gives me a "unknow attribute XXX" unless i declare this manuf attribute in a quirk. i probably miss something or misunderstant the configure_reporting method. But if it is posisble to use configure_reporting with any attributes this is a godd news but how ?

Adminiuga, 2020-11-27:
For configure_reporting to succeed,  zigpy needs to know the endpoint is there, but all endpoints are only scanned once when the device joins. If device changes it's configuration, then just call device.schedule_initialize() to update the list of endpoints

sarakha63, 2020-11-27:
The issue i have is not that zigpy doesnt know the endpoint.  it is that configure_reporting refuses to execute on a manufacturer attribute 

example i want to configure_reporting  endpoint 2  cluster 0x0201 attribute 0x4110   if i don't declare 0x4110 as an attribute in a quirk.  zigpy doesnt configure_reporting on this attribute with a "unknown attribute XXXX error/warning"

Adminiuga, 2020-11-27:
If your app knows that the configuration of the device has changed then either rescan the device, or add endpoint and clusters from the controller Application. Currently, to remove the endpoints you have to pop them manually and send the event to save the updated device to Zigbee.db

sarakha63, 2020-11-27:
hello for that part no problem i do that. but i have two issues

1) the mismatch between model manufacturer trough endpoints due to the way zigpy does that considering all device and endpoints will have the same). thats what we discussed earlier. i deal with that for now but zigpy improving this by not applying model and endpoints to everything could be a good start. but this is not the big issue

2) the one we are discussing now is not about dynamic endpoints or so. its about a manufacturer specific attribute in the thermostat cluster. is there a way to configure_reporting this attribute without creating a quirk describing this attribute

sarakha63, 2020-11-27:
[image]

Adminiuga, 2020-11-27:
> the mismatch between model manufacturer trough endpoints due to the way zigpy does that considering all device and endpoints will have the same). thats what we discussed earlier. i deal with that [...]

When user adds a device to the network, it expects to see that device on ZigZag map and be able to identify that device by model and manufacturer, cause you know, physically it is still a single device and a single ZDO endpoint. But how without the sacred knowledge that endpoint 232 is the real manufacturer/model  endpoint, application should know which one to pick for the model and manufacturer, if each basic cluster reports a different one?
`endpoint.manufacturer` and `endpoint.model` attributes are representing the vendor of the device and were left at the endpoint level for compatibility purposes. Now it is the `device.model` and `device.manufacturer` which represent the device model and manufacturer. Perhaps, should remove those from the endpoints to minimize confusion. 

and secondly, in 99.9% cases the manufcaturer/model attributes are going to be the same across multiple endpoints, so reading those attributes for each endpoint was removed, because Xiaomi devices were slow and with multiple endpoint, and reading attributes for each endpoint was unproductive. 

>Point 2 is important because with that we could avoid creating multiple quirks for each combination. In fact in this precise example i need to configure report attribute 0x4110 of cluster 0x0201 on each endpoints that has this. I already do that and its working well but i did it we a quirk adding 0x4110 as an attribute to the thermostat cluster. i did not find a way to use configure_reporting by specifiyng the attribute ID for an "unreferenced attribute"... and this will solve lots of issue if it is possible ? maybe it is ?

Don't create quirk. Just have custom cluster defined with those attributes and when adding the cluster to the endpoint, specify that cluster class to `add_cluster()` method.
e.g.
```
ep = device.add_endpoint(1)
ep.status = zigpy.endpoint.Status.ZDO_INIT
ep.profile = 0x0260
ep.device_type = Thermostat
ep.add_input_cluster(0x0201, YourCustomClusterWithManufAttributesClass)
```

sarakha63, 2020-11-27:
Oh yes i will try that for the manuf attribute.

For the other point i totally agree with you. Zigpy cant guess it. I have an idea  i already do something like that give me your point of view

the app on the initialization process check if for the manufacturer and model there a maybe things to check. And what  i do is i check the list of endpoints and the list of cluster to see if it matches. and if it does i do this once i checked that im sur what device i'm on

device.model, device.manufacturer = await ep.get_model_info()  with ep the 232 endpoint

and then register it.   those particular 0.01% can be handled by app itself. but i just want to be sure that nowhere else zigpy upon a restart or else can mess this out.

in fact im not asking for zigpy to guess. just needs zigpy to let us override some things (which we can already d) but be sure to not mess things up.

For example the order by idea mentionned by @puddly could give consistancy 
i'm pretty sure this is the first device working like that but maybe not the last one
i also agree that you cant go trough each endpoints for model and manufacturer and 99.9% taking the first one is ok. but maybe a function that could be called by the app to tell "i want you to go trough all endpoints to set manuf and model correctly) ... those are only ideas to me flexible

Im going to test setting cluster with manuf attributes and bypass quirks

Adminiuga, 2020-11-27:
>This is the particular code that bothers me and forces me to create a quirk. But maybe is there another way to add attributes that will avoid quirk usage in this case

yeah, that's true. From the architecture standpoint of view `zigpy` has to know the schema for all attributes and commands for helpers to work correctly, otherwise you always can send raw requests

Adminiuga, 2020-11-27:
>For example the order by idea mentioned by @puddly could give consistancy

Better is to have DB schema upgrade process in place, then I'd rather just save model/manufacturer in the device table. This would allow quirking the device without  restoring all the attributes 1st. And also we could save the quirk class in the table and load device as quirk upfront. 
Just need to have seamless schema migration from different versions.

sarakha63, 2020-11-27:
hello @Adminiuga  sorry to bother but when i try your solution to setting directly

ep.add_input_cluster(0x0201, YourCustomClusterWithManufAttributesClass)

i keep getting this error.

File "/usr/local/lib/python3.7/dist-packages/zigpy/endpoint.py", line 101, in add_input_cluster
cluster.add_listener(listener)
TypeError: add_listener() missing 1 required positional argument: 'listener'

i checked zigpy code and i don't really understand whats missing ?

do i have to create a full class for that. or can i only set it as a thermostatcluster and adding only manufacturer atributes ?

sarakha63, 2020-11-27:
custom = CustomCluster(Thermostat)
                custom.manufacturer_attributes={
                        DANFOSS_ROOM_STATUS_CODE: ("room_status_code", types.uint16_t),
                        DANFOSS_OUTPUT_STATUS: ("output_status", types.int8s),
                        DANFOSS_ROOM_FLOOR_SENSOR_MODE: ("room_floor_sensor_mode", types.int8s),
                        DANFOSS_FLOOR_MIN_SETPOINT: ("floor_min_setpoint", types.int16s),
                        DANFOSS_FLOOR_MAX_SETPOINT: ("floor_max_setpoint", types.int16s),
                        DANFOSS_SCHEDULED_TYPE_USED: ("scheduled_type_used", types.int8s),
                }
                ep.add_input_cluster(Thermostat.cluster_id, custom)

i'm probably missing something but this doesnt seem to work as written above

sarakha63, 2020-11-30:
Hello @Adminiuga  i'm coming back about my "issue". In fact i understand that i should be able to define a cluster as a custom cluster with normal thermostat attributes   + some manufacturer attributes. But anything i tried seems to confuse Zigpy.

have you got a little example on how after initialisation an app can tell zigpy this known endpoint with this cluster is in fact  a thermostat customcluster with those attributes in addition.

I tried to make an ep.add_input_cluster   with a custom thermostat cluster definition with manufacturer attributes. but it seems to not work at all. I guess maybe its because zigpy already initiliazed the device.

Do you have any clue on how and when i can set that and being sure that it will be taken into account and persistent

Regards

PS : by the way i agree with you proposal to have the DB schema with model/manuf savec in database so it is persistent and doesnt randomly change upon restart.

Adminiuga, 2020-11-30:
> i'm probably missing something but this doesnt seem to work as written above

I was wrong. You do need to pass a cluster instance to `ep.add_input_cluster()` https://github.com/zigpy/zigpy/blob/77eb534bbb392d72b0227527af801c9275522aff/zigpy/endpoint.py#L93
try like
```
ep = device.add_endpoint(1)
thermostat_cluster = Custom(ep)
ep.add_input_cluster(thermostat_cluster.cluster_id, thermostat_cluster)
```

Also, because of all that trickery with metaclasses, I'm not positive
```
custom = CustomCluster(Thermostat)
custom.manufacturer_attributes={
DANFOSS_ROOM_STATUS_CODE: ("room_status_code", types.uint16_t),
DANFOSS_OUTPUT_STATUS: ("output_status", types.int8s),
DANFOSS_ROOM_FLOOR_SENSOR_MODE: ("room_floor_sensor_mode", types.int8s),
DANFOSS_FLOOR_MIN_SETPOINT: ("floor_min_setpoint", types.int16s),
DANFOSS_FLOOR_MAX_SETPOINT: ("floor_max_setpoint", types.int16s),
DANFOSS_SCHEDULED_TYPE_USED: ("scheduled_type_used", types.int8s),
}
```
would work. Define class for your thermstat cluster as you would do for a quirk.

Adminiuga, 2020-11-30:
> Do you have any clue on how and when i can set that and being sure that it will be taken into account and persistent

yeah, to make it persistent would need some more modifications, as zigpy would restore device with standard thermostat clusters
upon each re-start, you would need to proccess endpoints 1--15 and for each:
1. remove thermostat cluster from input_clusters
2. add instance of your custom thermostat cluster
3. copy attribute cache `_attr_cache` from deleted cluster to the instance of your custom cluster. in case you care about attribute cache.

sarakha63, 2020-11-30:
hello really appreciate your answers.

but i dont get it

i define a custom cluster class like that (pretty the same i used in the quirk)

class DanfossThermostatCluster(CustomCluster, Thermostat):
    """Danfoss Thermostat cluster."""
    cluster_id = Thermostat.cluster_id
    manufacturer_attributes = {
        DANFOSS_WINDOWS_DETECTION: ("etrv_open_windows_detection", types.enum8),
        DANFOSS_EXTERNAL_OPEN_WINDOWS_DETECTION: ("external_open_windows_detected", types.Bool),
        DANFOSS_ORIENTATION: ("orientation", types.Bool),
        DANFOSS_ROOM_STATUS_CODE: ("room_status_code", types.uint16_t),
        DANFOSS_OUTPUT_STATUS: ("output_status", types.int8s),
        DANFOSS_ROOM_FLOOR_SENSOR_MODE: ("room_floor_sensor_mode", types.int8s),
        DANFOSS_FLOOR_MIN_SETPOINT: ("floor_min_setpoint", types.int16s),
        DANFOSS_FLOOR_MAX_SETPOINT: ("floor_max_setpoint", types.int16s),
        DANFOSS_SCHEDULED_TYPE_USED: ("scheduled_type_used", types.int8s),
    }

A pretty simple Thermostat Cluster Overrided by manufacturer attributes.

But each time i do ep.add_input_cluster(Thermostat.cluster_id,DanfossThermostatCluster) 
ep is the endpoint object of course

I keep getting

File "/var/www/html/plugins/zigbee/resources/zigbeed/channels/specific.py", line 98, in init
ep.add_input_cluster(Thermostat.cluster_id,DanfossThermostatCluster)
File "/usr/local/lib/python3.7/dist-packages/zigpy/endpoint.py", line 101, in add_input_cluster
cluster.add_listener(listener)
TypeError: add_listener() missing 1 required positional argument: 'listener'

Avoiding creating a quirk for this device that can have dynamic endpoints could be very great. and handling this dynamicly on app side is a good idea. But i really dont understand what is going wrong on add_input_cluster..... 

The way you describe it looks like it is as simple as creating a custom class with manuf attributes juste like in a quirk and using add_input_cluster to add it in place of the thermostat cluster

The part for configure_reporting those manuf attributes on app side is working well and only needs this manufacturer attributes to be setted with a quirk or directly in app. AS you know i would really like to do it on app side. because as this device is dynamicly reporting 1-16 endpoints it will be a pain to handle that with quirks (meaning creating 16 quirks for a same device, for having all the possible signatures)

I'm sure i'm completely missing something dumb. but in fact every time i do something i have the error above. or no error but invalid cluster_id when working with the device.

Do you see something obvious i'm missing ?
i'm pretty ashamed because on one side we are really progressing on ZGP in zigpy with @zoic  but on the other side i'm struglling with this that looks so obvious

puddly, 2020-11-30:
I think something like this should work:

```Python
class DanfossThermostatCluster(CustomCluster, Thermostat):
    """Danfoss Thermostat cluster."""
    cluster_id = Thermostat.cluster_id  # may not be necessary
    manufacturer_attributes = {
        DANFOSS_WINDOWS_DETECTION: ("etrv_open_windows_detection", types.enum8),
        DANFOSS_EXTERNAL_OPEN_WINDOWS_DETECTION: ("external_open_windows_detected", types.Bool),
        DANFOSS_ORIENTATION: ("orientation", types.Bool),
        DANFOSS_ROOM_STATUS_CODE: ("room_status_code", types.uint16_t),
        DANFOSS_OUTPUT_STATUS: ("output_status", types.int8s),
        DANFOSS_ROOM_FLOOR_SENSOR_MODE: ("room_floor_sensor_mode", types.int8s),
        DANFOSS_FLOOR_MIN_SETPOINT: ("floor_min_setpoint", types.int16s),
        DANFOSS_FLOOR_MAX_SETPOINT: ("floor_max_setpoint", types.int16s),
        DANFOSS_SCHEDULED_TYPE_USED: ("scheduled_type_used", types.int8s),
    }

...

ep = device.add_endpoint(1)
ep.status = zigpy.endpoint.Status.ZDO_INIT
ep.profile = 0x0260
ep.device_type = Thermostat

cluster = DanfossThermostatCluster(ep, is_server=True)
ep.add_input_cluster(cluster.cluster_id, cluster)
```

sarakha63, 2020-11-30:
Oh my.......

I understand now when @Adminiuga was talking lastly about a cluster instance

In fact i never did the cluster instance..... I was missing this line each Time. 
On my case i dont need the add endpoint part. It already exists it juste need to have its thermostat cluster redefined

I will try that after dinner. thanks again you two for your explanations

sarakha63, 2020-12-01:
thank you to you two.

I was simply missing the cluster= Class(ep,is_server = True) part

I'm going to check all that needs to be done  upon restart. But i guess with that the dynamic endpoint part/quirk is no more an issue. as now i only do : if endpoint exists thant remap cluster with custom and configure-reporting on those manuf attributes.... This will avoir creating 16 quirks to map signatures...

closed 2021-09-07 by github-actions[bot]