zigpy/zigpy #563: errors with ServiceCall zha.remove

uav2608, 2020-11-30:
I have several ZigBee devices installed in my Homeassistant with ZHA. I wanted to remove several of them with service 'ZHA.remove' and got error with the below log:
Logger: homeassistant.core
Source: components/zha/api.py:889
First occurred: 4:50:02 PM (2 occurrences)
Last logged: 4:50:04 PM

Error executing service: <ServiceCall zha.remove (c:1bfdc945a315675ac97491271717c9ce): ieee=00:15:8d:00:02:b0:fa:ae>
Error executing service: <ServiceCall zha.remove (c:ea476d66a59affd227817aaeb10043f2): ieee=00:15:8d:00:02:b0:fa:ae>
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/core.py", line 1467, in catch_exceptions
    await coro_or_task
  File "/usr/src/homeassistant/homeassistant/core.py", line 1486, in _execute_service
    await handler.job.target(service_call)
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 569, in admin_handler
    await result
  File "/usr/src/homeassistant/homeassistant/components/zha/api.py", line 889, in remove
    await application_controller.remove(ieee)
  File "/usr/local/lib/python3.8/site-packages/zigpy/application.py", line 150, in remove
    await self.force_remove(dev)
  File "/usr/local/lib/python3.8/site-packages/zigpy_znp/zigbee/application.py", line 660, in force_remove
    assert leave_rsp.Status == t.ZDOStatus.SUCCESS
AssertionError

puddly, 2020-11-30:
This issue has been fixed in the development branch of https://github.com/zigpy/zigpy-znp, which I'm going to release in a few days. Here are instructions for how to install it if you want to try it out: https://github.com/zigpy/zigpy-znp#testing-dev-with-home-assistant-core

I've also submitted a PR to zigpy to prevent this issue from causing `remove` to fail if the radio library's `force_remove` fails: https://github.com/zigpy/zigpy/pull/564

uav2608, 2020-11-30:
> This issue has been fixed in the development branch of https://github.com/zigpy/zigpy-znp, which I'm going to release in a few days. Here are instructions for how to install it if you want to try it [...]

tried in the dev branch and all works well. thank you.

closed 2020-11-30 by uav2608

mrneutron42, 2021-03-17:
This isn't working for removing a replaced Zigbee Coordinator!
core-2021.3.4
Home Assistant OS 5.12
supervisor-2021.03.6
I switched from a Sonoff Zigbee Bridge (Silicon Labs EZSP) to a TI CC2531 USB stick and the Silicon Labs EZSP device will not go away!
With the ZHA.remove command, I used the argument:  ieee: '60:a4:23:ff:fe:xx:xx:xx'
the result in the log:
`2021-03-17 17:29:55 INFO (MainThread) [homeassistant.components.zha.api] Removing the coordinator (60:a4:23:ff:fe:xx:xx:xx) is not allowed`