zigpy/zigpy #573: Possible to change Zigbee channel via API without un-pair and re-pair devices?

Gamester17, 2020-12-11:
This should maybe be posted as a feature request to Home Assistant devs instead but I thought I would try asking here first.

Does the zigpy API already have the function to change Zigbee channels for your Zigbee network and devices without unjoining/removing and rejoining/joining all of your Zigbee devices? 

I ask as ZHA in Home Assistant is at least in GUI currently missing feature to change Zigbee channel from Home Assistant's UI.  I would also like to avoid having to go manually re-pair each and every device individually just to change the Zigbee channel so I think that it would be very nice to be able to have a UI feature in ZHA to change Zigbee channel for all your connected devices.

The changing of Zigbee channel is sometimes wanted due to interference from your neighbours when you can not simply change the WiFi-channels that others use. Some other products like the Philips Hue Bridge, Samsung SmartThings Hub, and Phoscon deCONZ Gateway all allow you to change the Zigbee channel from a graphical interface via their apps or web-interface. I'm also sure most of you are aware of how easy it is to change the Wi-Fi channel via the apps or web-interface of most WiFi-routers and if so you will probably agree that it would be very nice if it was just as easy to change Zigbee channel via ZHA UI.

I know some brands of devices are limited so certain Zigbee channels but that is not an issue if you do not have such devices.

Adminiuga, 2020-12-11:
Currently there's no API. Per se it is not a Zigbee API but a radio protocol API that changes the channel, so would need to be implemented at the radio level. 
Xiaomi most likely are not going to survive channel chage.

puddly, 2020-12-19:
Can it be done with a ZDO `Mgmt_NWK_Update_req` to the coordinator?

Adminiuga, 2020-12-19:
i have not tried. Would need to confirm for which radios does it work.

If you send NWK Update to a router, do you see it sending a beacon request on the new channel immediately?

Hedda, 2021-01-21:
Would be very nice if integrations like ZHA for Home Assistant could change Zigbee channel after a network has been formed.

Today you can configure the channel mask and the specific channel in HA YAML config, but only during network formation.

> would need to be implemented at the radio level.

Are there already support in any radio libraries for zigpy to change Zigbee channel and channel mask?

Don't know if their serial protocol supports it but deCONZ UI supports changing the Zigbee channel for ConBee/RaspBee.

Adminiuga, 2021-01-21:
yes, that would be awesome

MattWestb, 2021-01-21:
Channel changing and rolling network key is one part of the Zigbee 3 security concept and should working with https://github.com/zigpy/zigpy/issues/573#issuecomment-748532201 but perhaps not all radios libs have implanting it or not working the there firmware.
By the way one router that is receiving one network update with broadcast is first transmitting it as broadcast to its neighbors and then waiting before changing channel and its also sending the the channel change as unicast or broadcast to all devices and then one network update for executing it as above.

channels in HA config have more functions in the network one is the energie scans and also then pairing they is being the primary channels the devices is scanning for beacons and devices that like try to joining. ZLL and ZB3 lights is listening for beacon first on the ZLL primary channels (11. 15. 20 and 25) and then the secondary is not have fund and OK network to joining. If setting the channels to not ZLL primary ones and letting the network changing channel it can happening its choosing one no ZLL primary channels and many devices cant joining the network but its no problem in ZHA for the moment then the security function is not implanted (but requested form some fanatic users like My).

closed 2021-09-07 by github-actions[bot]