zigpy/zigpy #592: Delayed message processing (Tuya)

xonestonex, 2020-12-20:
Hi,

I have a tuya radiator thermostat and see a huge delay in message processing of this device compared to Zigbee2Mqtt. 
Setting the temperature on the device is a good example for this problem. The temperature on the device is changed in 0.5 degree steps. Each step produces at least 3 messages (configured temperature, current measured temperature, device state, …). Zigpy somehow queues the messages and processes only one event per 10 seconds.
Changing the temperature by 5° results in a delay of at least 7 minutes until homeassistant shows the right value. Zigbee2Mqtt processes the messages immediately.
I’m see this behavior with a cc2531 as well a EmberZNet (Sonoff ZBBridge) coordinator.

[homeassistant.log](https://github.com/zigpy/zigpy/files/5721010/homeassistant.log)
[zigbee2mqtt.log](https://github.com/zigpy/zigpy/files/5721013/zigbee2mqtt.log)

puddly, 2020-12-20:
Your device only seems to be sending the attribute reports every minute or so:

```Python
2020-12-20 12:49:22 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xE1D4:1:0x0201]: Attribute report 'SiterwellThermostat'[occupied_heating_setpoint] = 550
2020-12-20 12:50:14 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xE1D4:1:0x0201]: Attribute report 'SiterwellThermostat'[occupied_heating_setpoint] = 600
2020-12-20 12:51:07 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xE1D4:1:0x0201]: Attribute report 'SiterwellThermostat'[occupied_heating_setpoint] = 650
2020-12-20 12:51:59 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xE1D4:1:0x0201]: Attribute report 'SiterwellThermostat'[occupied_heating_setpoint] = 700
2020-12-20 12:52:52 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xE1D4:1:0x0201]: Attribute report 'SiterwellThermostat'[occupied_heating_setpoint] = 750
2020-12-20 12:53:44 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xE1D4:1:0x0201]: Attribute report 'SiterwellThermostat'[occupied_heating_setpoint] = 800
2020-12-20 12:54:37 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xE1D4:1:0x0201]: Attribute report 'SiterwellThermostat'[occupied_heating_setpoint] = 850
```

ZHA should have setup periodic attribute reporting (or whenever the value of attribute changes by at least 1), which should instruct the device to send out an attribute report immediately when these values change. Maybe that did not succeed. Try navigating to the ZHA device and clicking "reconfigure". The logs will contain debug messages about whether or not that worked for each attribute.

xonestonex, 2020-12-20:
This log contains 2 clicks on reconfigure.

[reconfigure.log](https://github.com/zigpy/zigpy/files/5721198/reconfigure.log)

puddly, 2020-12-20:
Looks like it's configured to report when the attribute changes by at least 25, at most once every 30 seconds and at least once every 15 minutes:

```Python
2020-12-20 16:46:40 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x62F1:1:0x0201]: attr reporting for '{'local_temp': (30, 900, 25), 'occupied_cooling_setpoint': (30, 900, 25), 'occupied_heating_setpoint': (30, 900, 25), 'unoccupied_cooling_setpoint': (30, 900, 25)}' on 'thermostat': ConfigureReportingResponseRecord(status=0)
```

Z2M appears to configure reporting with `5, 35, 0.5` instead of `30, 900, 25`, which is probably why you're seeing slow updates.

zigpy the associated radio libraries appear to be working as intended. I suggest you consolidate the logs you posted and open a new issue in the HA repo for the ZHA component, which does all the high-level logic: https://github.com/home-assistant/core/issues/new (take a look at https://www.home-assistant.io/integrations/zha/#reporting-issues).

Adminiuga, 2020-12-20:
It's tuya thermostat, it does not have thermostat clusters, it is emulated

Adminiuga, 2020-12-20:
I haven't looked at the quirk code, but probably thermostat channel in zha is not notified about the change, and that why it takes so long. 
 I'm moving this to zha device handlers, but it do t have any of those devices so can't test

xonestonex, 2020-12-20:
I've looked at the zha devices handler code and the manufacturer specific cluster function "handle_cluster_request" is only executed every 10 seconds.

Adminiuga, 2020-12-20:
That still doesn't mean the problem is with zigpy. Thermostats with native thermostat cluster update the state as expected when they receive attribute update. So the problem must be in the quirk interaction and emulation of the zha thermostat.

xonestonex, 2020-12-21:
I think I've found the reason for the delayed messages, but I'm not sure that it's really a device handler bug.

The relevant code is in zigpy/zcl/__init__.py: 
`
  def handle_message(self, hdr, args):
     
      self.debug("ZCL request 0x%04x : %s", hdr.command_id, args)     
      if hdr.frame_control.is_cluster:
            self.handle_cluster_request(hdr.tsn, hdr.command_id, args)
            self.listener_event("cluster_command", hdr.tsn, hdr.command_id, args)
            return
        self.listener_event("general_command", hdr, args)
        self.handle_cluster_general_request(hdr, args)

    def handle_cluster_request(self, tsn, command_id, args):
        self.debug("No handler for cluster command %s", command_id)

    def handle_cluster_general_request(
        self, hdr: foundation.ZCLHeader, args: List
    ) -> None:
        if hdr.command_id == foundation.Command.Report_Attributes:
            valuestr = ", ".join(
                [
                    f"{self.attributes.get(a.attrid, [a.attrid])[0]}={a.value.value}"
                    for a in args[0]
                ]
            )
            self.debug("Attribute report received: %s", valuestr)
            for attr in args[0]:
                try:
                    value = self.attributes[attr.attrid][1](attr.value.value)
                except KeyError:
                    value = attr.value.value
                except ValueError:
                    self.debug(
                        "Couldn't normalize %a attribute with %s value",
                        attr.attrid,
                        attr.value.value,
                        exc_info=True,
                    )
                    value = attr.value.value
                self._update_attribute(attr.attrid, value)

            if not hdr.frame_control.disable_default_response:
                self.create_catching_task(
                    self.general_command(
                        foundation.Command.Default_Response,
                        hdr.command_id,
                        foundation.Status.SUCCESS,
                        tsn=hdr.tsn,
                    )
                )
`

For an attribute report a confirmation is send (foundation.Command.Default_Response) in 'handle_cluster_general_request'. For a cluster request no confirmation is send. A device handler could send the 'Default_Response' but 'hdr.frame_control.disable_default_response' isn't passed to the 'handle_cluster_request' function.
I think that my TRV waits with a 10sec timeout for the confirmation until the next request is send.
The logs of Z2M show that it sends a default response for all the cluster requests.

Adminiuga, 2020-12-21:
I don't think this is it. The code you are seeing if technically for HUE motion sensor, a reply is sent back to the network, when an attribute report comes in. This is not needed for Tuya quirk.

My suggestion is to focus on the following flow:
1. When climate entity/channel sends a request to change the temperature, it sends a "write_attribute" request
2. Quirk translates that write_attribute request into Tuya command
3. Back to the original requestor, quick should return a "write attribute success" response, because in case of an "unsuccessful" attribute write, the entity does not update its state

xonestonex, 2020-12-23:
I'm changing the temperature on the thermostat and not in home assistant. The thermostat sends tuya messages that get then converted to attributes and then shown in home assistant. It looks like the thermostat waits for a confirmation before a next message is send. This is what Z2M does.
I've made the following change and it improves the situation. There are still some delays, but I think that these may be home assistant related.

```
--- zcl/__init__.py.orig	2020-12-19 19:17:08.408815773 +0000
+++ zcl/__init__.py	2020-12-23 15:40:36.805271959 +0000
@@ -192,6 +192,14 @@
         if hdr.frame_control.is_cluster:
             self.handle_cluster_request(hdr.tsn, hdr.command_id, args)
             self.listener_event("cluster_command", hdr.tsn, hdr.command_id, args)
+            if not hdr.frame_control.disable_default_response:
+                self.debug("ZCL default response 0x%04x: %d", hdr.command_id, hdr.tsn)
+                schema = foundation.COMMANDS[foundation.Command.Default_Response][0]
+                self.create_catching_task(
+                    self.reply(
+                        True, foundation.Command.Default_Response, schema, hdr.command_id, tsn=hdr.tsn
+                    )
+                )
             return
         self.listener_event("general_command", hdr, args)
         self.handle_cluster_general_request(hdr, args)
```

Look at the timestamps. Before the patch one message is processed every 10 sec:
```
2020-12-23 15:34:36 DEBUG (MainThread) [zigpy.zcl] [0x52e0:1:0xef00] ZCL request 0x0001: [Command(status=31, tsn=58, command_id=514, function=0, data=[4, 0, 0, 0, 55])]
2020-12-23 15:34:47 DEBUG (MainThread) [zigpy.zcl] [0x52e0:1:0xef00] ZCL request 0x0001: [Command(status=31, tsn=59, command_id=515, function=0, data=[4, 0, 0, 0, 225])]
2020-12-23 15:34:58 DEBUG (MainThread) [zigpy.zcl] [0x52e0:1:0xef00] ZCL request 0x0001: [Command(status=31, tsn=60, command_id=515, function=0, data=[4, 0, 0, 0, 225])]
2020-12-23 15:35:08 DEBUG (MainThread) [zigpy.zcl] [0x52e0:1:0xef00] ZCL request 0x0001: [Command(status=31, tsn=61, command_id=1028, function=0, data=[1, 2])]
2020-12-23 15:35:18 DEBUG (MainThread) [zigpy.zcl] [0x52e0:1:0xef00] ZCL request 0x0001: [Command(status=31, tsn=62, command_id=1028, function=0, data=[1, 2])]
2020-12-23 15:35:29 DEBUG (MainThread) [zigpy.zcl] [0x52e0:1:0xef00] ZCL request 0x0001: [Command(status=31, tsn=63, command_id=514, function=0, data=[4, 0, 0, 0, 60])]
2020-12-23 15:35:39 DEBUG (MainThread) [zigpy.zcl] [0x52e0:1:0xef00] ZCL request 0x0001: [Command(status=31, tsn=64, command_id=515, function=0, data=[4, 0, 0, 0, 225])]
2020-12-23 15:35:50 DEBUG (MainThread) [zigpy.zcl] [0x52e0:1:0xef00] ZCL request 0x0001: [Command(status=31, tsn=65, command_id=515, function=0, data=[4, 0, 0, 0, 225])]
2020-12-23 15:36:00 DEBUG (MainThread) [zigpy.zcl] [0x52e0:1:0xef00] ZCL request 0x0001: [Command(status=31, tsn=66, command_id=1028, function=0, data=[1, 2])]
2020-12-23 15:36:11 DEBUG (MainThread) [zigpy.zcl] [0x52e0:1:0xef00] ZCL request 0x0001: [Command(status=31, tsn=67, command_id=1028, function=0, data=[1, 2])]
2020-12-23 15:36:21 DEBUG (MainThread) [zigpy.zcl] [0x52e0:1:0xef00] ZCL request 0x0001: [Command(status=31, tsn=68, command_id=514, function=0, data=[4, 0, 0, 0, 65])]
2020-12-23 15:36:32 DEBUG (MainThread) [zigpy.zcl] [0x52e0:1:0xef00] ZCL request 0x0001: [Command(status=31, tsn=69, command_id=515, function=0, data=[4, 0, 0, 0, 225])]
2020-12-23 15:36:42 DEBUG (MainThread) [zigpy.zcl] [0x52e0:1:0xef00] ZCL request 0x0001: [Command(status=31, tsn=70, command_id=515, function=0, data=[4, 0, 0, 0, 225])]
2020-12-23 15:36:53 DEBUG (MainThread) [zigpy.zcl] [0x52e0:1:0xef00] ZCL request 0x0001: [Command(status=31, tsn=71, command_id=1028, function=0, data=[1, 2])]
2020-12-23 15:37:03 DEBUG (MainThread) [zigpy.zcl] [0x52e0:1:0xef00] ZCL request 0x0001: [Command(status=31, tsn=72, command_id=1028, function=0, data=[1, 2])]
2020-12-23 15:37:14 DEBUG (MainThread) [zigpy.zcl] [0x52e0:1:0xef00] ZCL request 0x0001: [Command(status=31, tsn=73, command_id=514, function=0, data=[4, 0, 0, 0, 70])]
2020-12-23 15:37:24 DEBUG (MainThread) [zigpy.zcl] [0x52e0:1:0xef00] ZCL request 0x0001: [Command(status=31, tsn=74, command_id=515, function=0, data=[4, 0, 0, 0, 225])]
2020-12-23 15:37:35 DEBUG (MainThread) [zigpy.zcl] [0x52e0:1:0xef00] ZCL request 0x0001: [Command(status=31, tsn=75, command_id=515, function=0, data=[4, 0, 0, 0, 225])]
2020-12-23 15:37:47 DEBUG (MainThread) [zigpy.zcl] [0x52e0:1:0xef00] ZCL request 0x0001: [Command(status=31, tsn=76, command_id=1028, function=0, data=[1, 2])]
2020-12-23 15:37:56 DEBUG (MainThread) [zigpy.zcl] [0x52e0:1:0xef00] ZCL request 0x0001: [Command(status=31, tsn=77, command_id=1028, function=0, data=[1, 2])]
2020-12-23 15:38:06 DEBUG (MainThread) [zigpy.zcl] [0x52e0:1:0xef00] ZCL request 0x0001: [Command(status=31, tsn=78, command_id=514, function=0, data=[4, 0, 0, 0, 75])]
2020-12-23 15:38:17 DEBUG (MainThread) [zigpy.zcl] [0x52e0:1:0xef00] ZCL request 0x0001: [Command(status=31, tsn=79, command_id=515, function=0, data=[4, 0, 0, 0, 225])]
2020-12-23 15:38:27 DEBUG (MainThread) [zigpy.zcl] [0x52e0:1:0xef00] ZCL request 0x0001: [Command(status=31, tsn=80, command_id=515, function=0, data=[4, 0, 0, 0, 225])]
2020-12-23 15:38:38 DEBUG (MainThread) [zigpy.zcl] [0x52e0:1:0xef00] ZCL request 0x0001: [Command(status=31, tsn=81, command_id=1028, function=0, data=[1, 2])]
[... cut ...]
2020-12-23 15:45:12 DEBUG (MainThread) [zigpy.zcl] [0x52e0:1:0xef00] ZCL request 0x0001: [Command(status=32, tsn=25, command_id=515, function=0, data=[4, 0, 0, 0, 235])]
2020-12-23 15:45:12 DEBUG (MainThread) [zigpy.zcl] [0x52e0:1:0xef00] ZCL request 0x0001: [Command(status=32, tsn=26, command_id=515, function=0, data=[4, 0, 0, 0, 235])]
```

Here the full logs:
[home assistant log without patch](https://github.com/zigpy/zigpy/files/5736073/home-assistant.log)
[home assistant log with the patch](https://github.com/zigpy/zigpy/files/5736074/home-assistant_patched.log)

closed 2021-09-07 by github-actions[bot]