zigpy/zigpy #599: Warning: Image size must be a multiple of 64 bytes

samueltardieu, 2020-12-28:
I get this warning in my logs: "2020-12-28 08:27:50 WARNING (MainThread) [zigpy.ota.validators] Image OTAImageHeader(upgrade_file_id=200208670, header_version=256, header_length=56, field_control=<FieldControl.0: 0>, manufacturer_id=4476, image_type=16643, file_version=587531825, stack_version=2, header_string='GBL GBL_tradfri_light_unified_w', image_size=203378) is invalid: Image size (203316) must be a multiple of 64 bytes
".

Does this prevent the bulbs firmware from being upgraded or is it harmless? Is this a strong requirement? Looking for example at https://github.com/Koenkk/zigbee-OTA/tree/master/images I see several firmwares (for example in the Tuya directory) which have a size not divisible by 64.

puddly, 2020-12-28:
Looks like all of IKEA's images are very broken this time.

```
Image size (200968) must be a multiple of 64 bytes: 20201228/10005777-6.1-TRADFRI-control-outlet-2.0.024.ota.ota.signed
Image size (179824) must be a multiple of 64 bytes: 20201228/10043101-3.1-TRADFRI-dimmer-2.1.024.ota.ota.signed
Image size (180248) must be a multiple of 64 bytes: 20201228/10054470-TRADFRI-shortcut-button-2.3.015.ota.ota.signed
Image size (178776) must be a multiple of 64 bytes: 20201228/10005778-10.1-TRADFRI-onoff-shortcut-control-2.2.010.ota.ota.signed
Image size (207376) must be a multiple of 64 bytes: 20201228/10040611-3.2-TRADFRI-sy5882-unified-2.3.050.ota.ota.signed
Image size (207608) must be a multiple of 64 bytes: 20201228/10047227-1.2-TRADFRI-cv-cct-unified-2.3.050.ota.ota.signed
Image size (208180) must be a multiple of 64 bytes: 20201228/10038562-2.1-TRADFRI-sy5882-bulb-ws-2.0.029.ota.ota.signed
Image size (208232) must be a multiple of 64 bytes: 20201228/191100-4.1-TRADFRI-sy5882-driver-ws-2.0.029.ota.ota.signed
Image size (203316) must be a multiple of 64 bytes: 20201228/10046695-1.1-TRADFRI-light-unified-w-2.3.050.ota.ota.signed
```

The header contains `image_size=203378`, yet the JSON says:

```JSON
{
  "fw_binary_url": "http://fw.ota.homesmart.ikea.net/global/GW1.0/01.12.034/bin/10046695-1.1-TRADFRI-light-unified-w-2.3.050.ota.ota.signed",
  "fw_file_version_LSB": 1585,
  "fw_file_version_MSB": 8965,
  "fw_filesize": 204478,
  "fw_image_type": 16643,
  "fw_manufacturer_id": 4476,
  "fw_type": 2
}
```

Interestingly, not only is the header wrong, but the actual subelement containing the firmware contains the wrong length as well, as there are indeed `204478 - 203378 = 1100` bytes concatenated onto the end of the image. The good news is that the image passes validation after the header and subelement length is fixed.

@Adminiuga  If even the Zigbee OTA header and subelement lengths are unreliable, what do you think we should do about this?

 - "Fixing" the image by updating the header and subelement lengths would probably work (since the SiLabs firmware seems to ignore these values anyways), but it would "modify" the image.
 - Treating *all* firmware images as just a header in front of unknown data would also work, but it would undo the IKEA OTA loop fix, since we wouldn't be able to parse firmware images anymore.
 - Attempt to fix the firmware images only during validation?

I really have no idea how this kind of corruption can be happening so frequently to IKEA's OTA images.

Adminiuga, 2020-12-28:
if you feed the raw file as downloaded from the servers, does the device accepts it?

puddly, 2020-12-28:
I don't think it would but I can try it out with an IKEA gateway (thanks @walthowd!).

The external metadata from the JSON contains the "true" image length and is probably what the gateway uses for building the `query_next_image` response. If we use the image size contained within the Zigbee OTA header in the `query_next_image` response, the device has no way to know that it needs to read past the end of the file to receive the full image.

Adminiuga, 2020-12-28:
What about other approach, (I don't remember if we discussed it) but just to store and send the content of the file starting with OTA header till the end of the raw file?

I need to check if I have any ikea devices on the older versions.

puddly, 2020-12-28:
This patch gets the images to validate and pass the CRC-32 checksum but I'm out of out-of-date devices to test with:

```diff
diff --git a/zigpy/ota/image.py b/zigpy/ota/image.py
index bcfc6a4..88686b7 100644
--- a/zigpy/ota/image.py
+++ b/zigpy/ota/image.py
@@ -2,9 +2,12 @@
 import typing

 import attr
+import logging

 import zigpy.types as t

+LOGGER = logging.getLogger(__name__)
+

 @attr.s(frozen=True)
 class ImageKey:
@@ -231,7 +234,24 @@ def parse_ota_image(data: bytes) -> typing.Tuple[BaseOTAImage, bytes]:
         if len(data) <= offset + size:
             raise ValueError(f"Data too short to be IKEA container: {len(data)}")

-        return parse_ota_image(data[offset : offset + size])
+        wrapped_data = data[offset : offset + size]
+        image, rest = OTAImage.deserialize(wrapped_data)
+
+        if rest:
+            LOGGER.warning(
+                "Fixing IKEA OTA image with trailing data (%s bytes)",
+                size - image.header.image_size,
+            )
+            image.header.image_size += len(rest)
+
+            # No OTA images have been seen to have any other structure
+            assert len(image.subelements) == 1
+            assert image.subelements[0].tag_id == ElementTagId.UPGRADE_IMAGE
+
+            image.subelements[0].data += rest
+            rest = b""
+
+        return image, rest

     try:
         # Hue sbl-ota images start with a Zigbee OTA header but contain no valid
```

Interestingly, the "bad" IKEA images from earlier this year are now validating properly????????????

MattWestb, 2020-12-28:
@puddly Implanting the possibility of downgrading FW thru local file as was proposed before and you can downgrading to one step older firmware and doing the automatic OTA  test.
I have don it with deCONZ (on ZLL and ZB3 devices but not back to ZLL from ZB3).

Is IKEA getting more crazy or do they having problems with the dev environment (SS5 bugs ?).

puddly, 2020-12-28:
Awesome, I wasn't aware downgrading was possible!

I have successfully downgraded my outlet from 537019939 to 537011747. Upgrading with the broken image (also 537019939) available from IKEA's servers works after it is automatically patched with the above fix.

puddly, 2020-12-28:
> Is IKEA getting more crazy or do they having problems with the dev environment (SS5 bugs ?).

I'm really not sure. It looks like this and the problem from a few months ago are the same thing. It's as if they're using some custom OTA image generator (probably with their `.signed` container format) and it apparently generates some images with 1000 or more bytes missing off the end, and then concatenates those bytes to the image (???). The OTA header's `image_size` and the firmware subelement's size are both missing exactly those bytes so it appears that the OTA code does not parse the Zigbee OTA image at all, just skips to some offset after verifying the header.

samueltardieu, 2020-12-28:
Congrats for your very fast detective work, it looks like more Ikea devices will encounter new updates thanks to you all.

MattWestb, 2020-12-28:
Its was not the first time and i hope also not the last one for our sherlock holmes ;-))

I was thinking the sign file was one normal GBL file that was signed with Simplicity Commander as silabs standard procedure (without encrypting the file) and having changing the extension for keeping it clear. 
Perhaps they is not using Simplicity Commander for generating ther OTA files at all.

Can it being they is preparing adding patching data for the devices with false data setting (like strange chartres in model ID) in the data part of the flash (should being possible with GBL files) ?

If you like one challenges its gets some very old IKEA OTA files (3.5 year old ones) you can testing if being bored one rainy day :-)) [basilfx/TRADFRI-Hacking/firmwares/ikea](https://github.com/basilfx/TRADFRI-Hacking/tree/master/firmwares/ikea). 
Also for your [control outlet 1.4.018](https://github.com/basilfx/TRADFRI-Hacking/blob/master/firmwares/ikea/otau/test/10005777-TRADFRI-control-outlet-1.4.018.ota.ota.signed).

By the way Z2M was implanting the alternative URL for IKEA firmware for 2 weeks ago (better late than ever) "inspired of ZHA" but not the local file updating :-))))

Still great work done devs !!!

commit 3bd6a4fa (2020-12-29): Single function for OTA image parsing, CachedImage simplification, and Hue SBL images (#597)

* Rewrite `CachedImage` to be a container, not an `OTAImage` subclass

* Centralize image parsing with a `parse_ota_image` function.
This allows for Hue OTA files to be specially handled and for
`FileProvider` to properly load them, along with wrapped IKEA images.

* Cache serialized image data to prevent needless serialization

* Re-add the `header` property to the `CachedImage` class

* Fix read-past-end issue with Hue OTA parser

* Add unit test to ensure `CachedImage.header` exists

* Automaticall [... cut]
closed 2020-12-29 by Adminiuga