zigpy/zigpy #600: Power

wernerhp, 2020-12-28:
Not sure if this is the right place, but here's my story...
I recently bought a Sonoff ZB Bridge, flashed with Tasmota and using with with ZHA in Home Assistant.  ZHA uses zigpy.   The problem I'm experiencing is that all my connected zigbee devices report Power as 100%.  I have many devices which I've had for over 2 years and they were down to 40% before I paired them with the new bridge.  This happens for all my Xiaomi and Sonoff sensors and switches (PIR, Door/Window, Temperature, Buttons etc).

Hedda, 2021-01-14:
Suggest maybe changing the subject/topic-description to be same as you have in https://github.com/home-assistant/core/issues/44578 as more describing?

MattWestb, 2021-01-14:
@wernerhp  Most Xiaomi sensors is not reporting battery % as Zigbee standard but battery voltages.
The devs have rewritten the converter for Xiaomi battery calculation then its was not working so good and its on the way being committed see https://github.com/zigpy/zha-device-handlers/pull/720

Adminiuga, 2021-01-21:
Fixed in the latest zha-device-handlers

closed 2021-01-21 by Adminiuga

sstefanowski, 2021-03-01:
Hello, 

I think I still have a problem with the battery-operated Xiaomi Mijia Door/Window Sensor.
Model: MCCGQ01LM
https://zigbee.blakadder.com/Xiaomi_MCCGQ01LM.html

Previously ZHA has always shown batteries at 100% when zigbee2mqtt shown ~91%.
After update to 2021.2.3 with FIX (Thank you!) I can see the battery dropped to 66%, which I wonder is this approximate value for exact 3V .... and I wonder if this is really OK?

**Potential problem:**
I can see the battery voltage reading for MCCGQ01LM is just integer "3", not a decimal like "3.???".
Is the default ~66% reading a problem related to the precision of the voltage reading from MCCGQ01LM?

Screenshots attached:
[image]
[image]

wernerhp, 2021-03-02:
Sonoff sensors
[image]

Xiaomi / Aqara sensors
Strangely the Living Room Motion Sensor reports `3.1V`.
[image]

sstefanowski, 2021-03-02:
Hi @wernerhp,
Is your Sonoff Sensor works correctly reporting 2,7V (displaying 9% of the battery)? 
If YES... then maybe the Xiaomi mV range is incorrectly calibrated? I think it's now quite narrow now...
 
https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/xiaomi/__init__.py
```
(...)
    MAX_VOLTS_MV = 3100  # 100%
    MIN_VOLTS_MV = 2820  # 0%
(...)
```
My battery readings now vary from 59% (taking into consideration range it's 2985mV) and 73%  (it's ~3024mV).

@Adminiuga 
My idea is that we could get it Xiaomi Mija mV range acuratelly calibrated by experience... 
My offer is, I could monitor and find a moment when frequency of "last seen" events will drop significantly (even if this is already reporting 0% for a long time) but I need to know precise mV reading - possible two or even three decimal digits needed.

Adminiuga, 2021-03-02:
Yes, should be changed to provide 1 decimal

wernerhp, 2021-03-03:
> Hi @wernerhp,
> Is your Sonoff Sensor still working correctly reporting 2,7V (and displaying 9% of the battery)?
> If YES... then maybe the Xiaomi mV range is incorrectly calibrated? I think it's now too narrow now...

I just moved into a new place this weekend.  The sensor seems to work reliably, but it may be too soon to tell.
[image]

I renamed the sensor from the previous **Sonoff Motion Sensor** to **Scullery Motion Sensor**.  Here are the data points for the past week.  Something is very odd with the level dropping from `100%` to `9%` 🤷  I'd say something is not reporting correctly.  I'll pull the battery and measure it with a multimeter.

[image]

[image]

[image]

[image]

[image]

wernerhp, 2021-03-03:
Something ain't right...

[image]
[image]

sstefanowski, 2021-03-03:
@wernerhp 
Is this sensor in a cold place? Batteries tend to discharge at lower voltage in cold conditions (usually something around 0.1 mV)
source: https://www.murata.com/en-eu/products/productdetail?partno=CR1632
[image]

@Adminiuga Suggestion of improvement 
I guess the different types of batteries will have different discharge curves, and power requirements (e.g. heater valves need more power to close/open the valve), therefore the voltage should not be just set on "global level" it also should be allowed to override on individual "device type level".
 
E.g. I guess Xiaomi Weather station (mentioned in a code pull-request when new voltage ranges were set to 3100-2820 (100%-0%)) is using 2xAAA batteries (am I right?), which will have probably a different curve than popular CR1632 coin batteries used in most of the small devices.
 
The "Energizer" discharge curve for 1632 (https://data.energizer.com/PDFs/cr1632.pdf) shows that it starts at ~3V (suggested 100%) and goes usable as low 2,7V (~0%), being completely dead at 2V.

Hedda, 2021-03-03:
> Is this sensor in a cold place? Batteries tend to discharge at lower voltage in cold conditions

Off-topic DIY tip is to use CR2050 batteries instead of CR2032 in devices when used in cold conditions. CR2050 has the same width dimensions as CR2032 but is thicker however it can still be used in sensors like Aqara/Xiaomi Temperature Sensor without the lid.

puddly, 2021-03-03:
@wernerhp The Sonoff sensors I've seen self-report battery percentage as well as the voltage. What you're seeing is what the device is reporting. This issue was related to the calculation done for Aqara devices, which report only their measured battery voltage.

@sstefanowski 

> E.g. I guess the Xiaomi Weather station (mentioned in a code pull-request when new voltage ranges were set to 3100-2820 (100%-0%)) is using 2xAAA batteries (am I right?), which will have probably a different curve than the popular CR1632 coin batteries used in most of the small devices.

They use CR2032s but the reported voltages are not very good for a simple battery percentage calculation. I made some graphs of the raw voltage information sent by Aqara devices here, along with a graph of reported voltage and reported temperature: https://github.com/zigpy/zha-device-handlers/issues/415. There's no single curve to accommodate every battery and device, however, without temperature calibration.

For comparison, here is how Z2M and ZHA calculate the percentage, with the 2.7V to 3.0V range in green:

[image]

sstefanowski, 2021-03-03:
@puddly This chart is great! Thanks. 

@al
I know that voltage is just an approximation of the remaining power (especially without temperature), therefore I think individual calibration of the device would be nice. 

....but if we cannot have it....

@Adminiuga 
**I still think reporting 0% at 2820mV by default is much too early and... not very useful!** 
When the sensor says 0% you are losing all the information about the remaining margin of your battery and you can't do anything about it. I think it's much better to report 10%@2700mV, even if any device is malfunctioning at this range (it may be due to the specific placement of the sensor far from the router). 

I mean, it's better to learn by experience the useful power-to-range ratio of your sensors. I suggest lowering the default low range (0%) to 2700mV (or even lower 2650mV). In such a case, every user can get it's own feeling when individual sensors need a replacement at 10% or at 5%?. 

Finally... If you have a "wide range", you have an open option of configuring a derived "corrected-battery-remaining sensor" from the original ZHA sensor (if you need it in strange cases of sensors in e.g. cold place). When the original ZHA sensor shows 0% too early (@2800mV) then you cannot do anything like that.

Adminiuga, 2021-03-03:
No. You still have voltage and you can derive the battery % all day long if you wish. at 2.85V you can barely see the led flashing on xiaomi + different xiaomi have different  offsets.

wernerhp, 2021-03-03:
> @wernerhp
> Is this sensor in a cold place? Batteries tend to discharge at lower voltage in cold conditions (usually something around 0.1 mV)

No. It's between 23°C and 29°C

wernerhp, 2021-03-03:
Weird...now it's up from 9% to 64% and still 2.7V. 
[image]

sstefanowski, 2021-03-03:
> No. You still have voltage and you can derive the battery % all day long if you wish. at 2.85V you can barely see the led flashing on xiaomi + different xiaomi have different offsets.

@Adminiuga, 
OK, I understand your point. Now, implementing a custom "corrected battery level" would need better precision. what about reporting two decimals in voltage? Is this possible?

Adminiuga, 2021-03-06:
Try 
https://github.com/home-assistant/core/pull/47520
https://github.com/zigpy/zha-device-handlers/pull/807

sstefanowski, 2021-03-08:
Thank you, you are super fast!